zigpy/zha-device-handlers #791: [Device Support Request] Updated Zemismart Zigby blind motors

davek00997744, 2021-02-27:
The new updated Zemismart zigby blind motors will pair with Zigby Home Assistant but no entities are created so control is not possible

**Describe the solution you'd like**
An update to the interface to allow control of Open, Close, Midway and Stop

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

**Additional context**
Add any other context or screenshots about the feature request here.

Daniel-Sanche, 2021-03-02:
Is [this the one?](https://zigbee.blakadder.com/Moes_AM43-0_45_40-ES-EB.html)

If so, there's an [open issue](https://github.com/zigpy/zha-device-handlers/issues/744). I'm also watching it as I have the same problem

davek00997744, 2021-03-02:
Mine is the M515EGBZTN from Zemismart/Benexmart, it has the zigbee adaptor built in.

blauret, 2021-03-03:
[This](https://github.com/zigpy/zha-device-handlers/pull/801) should be a good starting point

closed 2021-09-06 by github-actions[bot]