zigpy/zha-device-handlers #804: [BUG] Philips Hue Motion sensor outdoor SML002, On/Off not working and Occupancy acts like On/Off

yfands, 2021-03-04:
Hey Guys,

Just got a Philips Hue SML002, that's the outdoor version, motion detector.
But there's more it also reports illumination, temperature, battery and occupancy.

So I paired it with Home assistant, 2021.2.3. works like a charm.

But:

The Motion detector On/Off part does not work at all. (the one with the running man as icon).
And the Occupancy part acts like the On/Off part, detects motion and resets after 10 seconds.
So I googled my findings and indeed there are others who encountered the same problems, but no solution
and the strange thing is I cant find a submitted issue, I think clearly a bug. So hereby....

My log file while connecting:
There is also a 'polling' of sorts in the log file from my light.lamp_achter_woonkamer, dont know what it means..

[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0x511a:11:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=60 command_id=Command.Report_Attributes>
[0x511a:11:0x0000] ZCL request 0x000a: [[Attribute(attrid=1, value=<TypeValue type=uint8_t, value=66>)]]
[0x511a:11:0x0000] Attribute report received: app_version=66
[0x5719:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=True> manufacturer=4447 tsn=57 command_id=Command.Report_Attributes>
[0x5719:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=65281, value=<TypeValue type=CharacterString, value=!�(!�C!>)]]
[0x5719:1:0x0000] Attribute report received: 65281=!�(!�C!
[0x5719:1:0x0001] Voltage mV: [Min]:2820 < [RAW]:3015 < [Max]:3100, Battery Percent: 69.5
[0x1D33:1:0x0b04]: async_update
[0x810C:1:0x0b04]: async_update
[0x1d33:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=202 command_id=Command.Read_Attributes_rsp>
[0x810c:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=204 command_id=Command.Read_Attributes_rsp>
Device 0x5026 (00:17:88:01:06:48:cc:ed) joined the network
[0x5026] Requesting 'Node Descriptor'
Tries remaining: 2
[0x5026] Extending timeout for 0xce request
Device 0x5026 (00:17:88:01:06:48:cc:ed) joined the network
Skip initialization for existing device 00:17:88:01:06:48:cc:ed
Device 0x5026 (00:17:88:01:06:48:cc:ed) joined the network
Skip initialization for existing device 00:17:88:01:06:48:cc:ed
[0x5026:zdo] ZDO request ZDOCmd.Device_annce: [0x5026, 00:17:88:01:06:48:cc:ed, 128]
[0x5026] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4107, maximum_buffer_size=89, maximum_incoming_transfer_size=63, server_mask=0, maximum_outgoing_transfer_size=63, descriptor_capability_field=0)
[0x5026] Discovering endpoints
Tries remaining: 3
[0x5026] Extending timeout for 0xd0 request
[0x5026] Discovered endpoints: [2, 1]
[0x5026:2] Discovering endpoint information
Tries remaining: 3
[0x5026] Extending timeout for 0xd2 request
[0x5026:2] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=2, profile=260, device_type=263, device_version=0, input_clusters=[0, 1, 3, 1030, 1024, 1026], output_clusters=[25])
[0x5026] Extending timeout for 0xd4 request
[0x5026:2:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=212 command_id=Command.Read_Attributes_rsp>
[0x5026:2] Manufacturer: Philips
[0x5026:2] Model: SML002
[0x5026:1] Discovering endpoint information
Tries remaining: 3
[0x5026] Extending timeout for 0xd6 request
[0x5026:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=49246, device_type=2128, device_version=2, input_clusters=[0], output_clusters=[0, 3, 4, 6, 8, 768, 5])
Checking quirks for Philips SML002 (00:17:88:01:06:48:cc:ed)
Considering <class 'zhaquirks.philips.sml002.PhilipsSML002'>
Found custom device replacement for 00:17:88:01:06:48:cc:ed: <class 'zhaquirks.philips.sml002.PhilipsSML002'>
device - 0x5026:00:17:88:01:06:48:cc:ed entering async_device_initialized - is_new_join: True
device - 0x5026:00:17:88:01:06:48:cc:ed has joined the ZHA zigbee network
[0x5026](SML002): started configuration
[0x5026:ZDO](SML002): 'async_configure' stage succeeded
[0x5026] Extending timeout for 0xd8 request
[0x5026] Extending timeout for 0xda request
[0x5026] Extending timeout for 0xdc request
[0x5026:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
[0x5026:1:0x0000]: finished channel configuration
[0x5026] Extending timeout for 0xde request
[0x5026:1:0x0006]: bound 'on_off' cluster: Status.SUCCESS
[0x5026:1:0x0006]: finished channel configuration
[... 207 lines cut ...]
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0107",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0400",
        "0x0402",
        "0x0406"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "Philips",
  "model": "SML002",
  "class": "zhaquirks.philips.sml002.PhilipsSML002"
}

Regards
Frank

Adminiuga, 2021-03-05:
on_off is not a real entity, should be disabled. it is a client cluster so at best it might generate a zha_event, although I don't know if anyone have tried. I guess the idea was to bind the on/off cluster directly to a light and control it. Same as ikea motion sensors.

Occupancy does act as on/off, it is either occupied or not.

yfands, 2021-03-05:
Hello Adminiuga,

I understand, but I have also Aqara motion sensors, the 'on/off' function is defined as motion detected, (the run icon) resets after 1 minute. and also the occupancy is triggered and resets after 10 minutes (the house icon). So I expected The philips would act the same. and yes tried the zha_event and I got nothing..

So not a bug, but by design ?

MattWestb, 2021-03-05:
One IKEA motion sensor is in Zigbee one "on/off controller" that using cluster 0x0006 and sending one "on with off XX seconds" (time depends of the setting inn the device and is in the "payload" of the command) to one group of lights. ZHA is reading the device type and making the "motion".

Aqara is one occupation sensor sending attribute occupation or no occupation on one cluster (No standard but zigbee is using  0x0406 (Occupancy) for it but Xiaomi) and cant controlling lights.
Some "creative persons" have making one virtual On/Off cluster that is not working to steering light then the coordinator and host system is off line = "bad magic".

The Philips HUE have 2 endpoints one for lights with clusters one on/off =x0006 and one for HA eith cluster 0x0406 (Occupancy) so in HA you is getting one motion and one occupancy = wot clusters the sensor is having and HA is using them.
The philips hue On/Off cluster need being bonded o one group for sending the on/off command and your coordinator need being member of that group.

You shall not triggering your automation on both only one of the functions.
If you dont like have both you can hiding it in HA.

deviantintegral, 2021-09-03:
I see the same clusters with the indoor SML001 sensor.

Is there somewhere this could be documented? As is, it's a bit of a confusing experience to have a cluster that doesn't seem to activate at all. Otherwise, I think this can be closed.

jeeftor, 2022-01-21:
Me too on the outdoor sensor. Probably going to just connect it back to hue rather than diy setup

TRusselo, 2022-05-13:
i have this issue to on SML001

TRusselo, 2022-05-13:
please let me know if there are any logs or information needed to make progress

TRusselo, 2022-05-13:
> One IKEA motion sensor is in Zigbee one "on/off controller" that using cluster 0x0006 and sending one "on with off XX seconds" (time depends of the setting inn the device and is in the "payload" of [...]

I think there might be a solution here, but the broken english is breaking my brain.
i dont understand.
i cant find any documentation to help either.

TRusselo, 2022-05-13:
After randomly clicking things...
I think i fixed it or at least got it working better.

Go to Device Page in HA.
Manage Clusters
Select Cluster : OnOff 
Select Cluster Attribute : off_wait_time
Enter Value : 1
Click "SET ZIGBEE ATTRIBUTE"

now the motion sensor (on_off)  resets to OFF after 1 minute.

[image]

TRusselo, 2022-05-13:
battery still shows 100%  (its not full)  batteries are old.

this is not perfect fix. im not smart enough to fix this myself.

TRusselo, 2022-05-13:
workaround stopped working. deleted post

TheJulianJES, 2022-05-14:
The ``on_off`` sensor should be disabled:
[image]

---
Then, the occupancy binary sensor can/should be changed to "Show as: Motion":
[image]

---
If you want to change the timeout for the occupancy/motion sensor, go to "Manage Clusters", select the ``OccupancyCluster``, and select ``pir_o_to_u_delay (id: 0x0010)`` for the attribute.
Then click the "Get Zigbee attribute" button and wave in front of the sensor (to make it wake up) and see the current occupancy/motion timeout. It should be set to 0 by default.
A value equal or less than 10 is the default "10 seconds timeout" these sensors have.
You can enter another number in that field and then click "Set Zigbee attribute" and wave in front of the sensor again (make sure to leave the "Manufacturer Code Override" field empty).
If you set it to 20 for example, you have a 20 seconds timeout.
[image]

TRusselo, 2022-05-14:
> The `on_off` sensor should be disabled:
>
> Then, the occupancy binary sensor can/should be changed to "Show as: Motion":
>
> If you want to change the timeout for the occupancy/motion sensor, go to [...]

Thanks for the pretty pictures but i would prefer the code get fixed rather than a workaround.

Plus...
Occupancy does nothing, even after setting it up with your instructions.
does not change with motion, ever.  ive paired 3 motion sensors yesterday dozens of times.
just did it again using your instructions.  never changes.

On_off DID detect motion but took 5 minutes to reset. was unable to change the cluster value for it
5 min reset on this sensor, with its location, is not useful.
Illuminance reads wrong.  78 lux, if paired while face down in the carpet. 26 lux face up! never changes
battery reads wrong.  says 100%. its actually 29% according to the hue hub.
temp doesnt change

this device handler is broken.

ive moved my hue motion sensors back to the hue hub, they work better with home assistant than this
and they work better on the hue hub than the smartthings hub they were connected to before yesterday.
hue hub does local communication with home assistant too so. screw it.  progress.  they operate local now.

again,  i would prefer the code get fixed than ME making janky changes to have it work "good enough"

dmulcahey, 2022-05-14:
> Thanks for the pretty pictures but i would prefer the code get fixed rather than a workaround.
>
> Plus... Occupancy does nothing, even after setting it up with your instructions. does not change with [...]

I understand being frustrated but let’s try not to blast the folks that spend their free time trying to help you.

TRusselo, 2022-05-14:
im not frustrated.
ive fixed my issue by going to the hub.

im not good with words.  i dont know how to say the things above "nicer"
im not even sure what ISNT nice above to make you say that.

words are not my thing.  im confused now

dmulcahey, 2022-05-14:
Online stuff in all caps is interpreted as yelling… just FYI here is an article: https://www.cnn.com/2018/07/23/us/all-caps-typography-history-tweets-trnd/index.html

And the last paragraph wasn’t necessary in general. Not the end of the world but we try to stay cordial in general.

TRusselo, 2022-05-14:
> Online stuff in all caps is interpreted as yelling… just FYI here is an article: https://www.cnn.com/2018/07/23/us/all-caps-typography-history-tweets-trnd/index.html
> 
> And the last paragraph wasn’t necessary in general. Not the end of the world but we try to stay cordial in general.

are you effing serious?  it WAS the caps?
i was trying to be funny.
im out.  unsub. i cant handle people that cant hand CAPITOL LETTERS.

closed 2022-11-17 by github-actions[bot]