zigpy/zha-device-handlers #805: [Device Support Request] Xiaomi Smoke Detector

1800Zeta, 2021-03-05:
I am trying to get some additional functionality out of the smoke detector if possible please. There is already a handler which can be found at https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/xiaomi/mija/smoke.py

Would it be possible to get the additional entities presented?
https://zigbee.blakadder.com/Xiaomi_JTYJ-GD-01LM_BW.html
This lists Detector Sensitivity and Smoke Density

https://www.youtube.com/watch?v=EdezXlPoc4U
This video shows the app initiating a self test. It would be nice to be able to issue a self test, and potentially set the sensitivity.

I've put the pairing debug log below. If I start viewing the clusters within Home Assistant, it looks like the information might be there, but I am not 100% sure. Things like the smoke density, I would be interested to see if I can have a pre-warning that the alarm will go off when density is say 0-10, then when it hits 11, presumably the alarm would go off. 

Thank you in advance

`2021-03-05 15:52:26 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e41000700aa00
2021-03-05 15:52:26 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2021-03-05 15:52:26 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2021-03-05 15:52:26 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x1741000800010001
2021-03-05 15:52:26 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x17410025001e0022020000010272bc0b04010000070018700a0100204200afff219d2c00d8
2021-03-05 15:52:26 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [30, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xbc72>, 11, 260, 0, b'\x18p\n\x01\x00 B', 0, 175, 255, 33, 157, 44, 0, -40]
2021-03-05 15:52:26 DEBUG (MainThread) [zigpy.zcl] [0xbc72:11:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=112 command_id=Command.Report_Attributes>
2021-03-05 15:52:26 DEBUG (MainThread) [zigpy.zcl] [0xbc72:11:0x0000] ZCL request 0x000a: [[Attribute(attrid=1, value=<TypeValue type=uint8_t, value=66>)]]
2021-03-05 15:52:26 DEBUG (MainThread) [zigpy.zcl] [0xbc72:11:0x0000] Attribute report received: app_version=66
2021-03-05 15:52:26 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xbc72>, ep: 11, profile: 0x0104, cluster_id: 0x0000, data: b'18700a01002042'
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1c420012000b00039dd78705008d1500ffd1
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1c43000c00050002eb37ffd1
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e44000700aa00
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x1742000800010001
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1742002a0023002202fdff0002eb3700000013000c00fdeb379dd78705008d15008000afff219d2c00d2
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [35, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xfffd>, 0, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x37eb>, 0, 0, 19, b'\xfd\xeb7\x9d\xd7\x87\x05\x00\x8d\x15\x00\x80', 0, 175, 255, 33, 157, 44, 0, -46]
2021-03-05 15:52:27 INFO (MainThread) [zigpy_deconz.zigbee.application] New device joined: 0x37eb, 00:15:8d:00:05:87:d7:9d
2021-03-05 15:52:27 INFO (MainThread) [zigpy.application] Device 0x37eb (00:15:8d:00:05:87:d7:9d) joined the network
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy.zdo] [0x37eb:zdo] ZDO request ZDOCmd.Device_annce: [0x37EB, 00:15:8d:00:05:87:d7:9d, 128]
2021-03-05 15:52:27 INFO (MainThread) [zigpy.device] [0x37eb] Requesting 'Node Descriptor'
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy.util] Tries remaining: 2
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy.device] [0x37eb] Extending timeout for 0x38 request
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 56 under 57 request id, data: b'38eb37'
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x37eb>, ep: 0, profile: 0x0000, cluster_id: 0x0013, data: b'fdeb379dd78705008d150080'
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (18, 57, 0, <DeconzAddressEndpoint address_mode=2 address=0x37EB endpoint=0>, 0, <ZDOCmd.Node_Desc_req: 0x0002>, 0, b'8\xeb7', 2, 0)
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x12430019001200390002eb37000000020000030038eb370200
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x124300090002002239
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 57]
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e44000700aa00
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x1744000800010001
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1744003a003300220200000102eb3701040100001c0018000a050042116c756d692e73656e736f725f736d6f6b650100200100afffae9c2c00d2
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [51, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x37eb>, 1, 260, 0, b'\x18\x00\n\x05\x00B\x11lumi.sensor_smoke\x01\x00 \x01', 0, 175, 255, 174, 156, 44, 0, -46]
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy.application] Received frame on uninitialized device 00:15:8d:00:05:87:d7:9d (Status.NEW) for endpoint: 1
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy.device] [0x37eb] Received ZCL while uninitialized on endpoint id 1, cluster 0x0000 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=0 command_id=Command.Report_Attributes>, payload: b'\x05\x00B\x11lumi.sensor_smoke\x01\x00 \x01'
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy.device] [0x37eb] Uninitialized device command 'Command.Report_Attributes' args: [[Attribute(attrid=5, value=<TypeValue type=CharacterString, value=lumi.sensor_smoke>), Attribute(attrid=1, value=<TypeValue type=uint8_t, value=1>)]]
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy.device] [0x37eb] Found 'MijiaHoneywellSmokeDetectorSensor' quirk for 'lumi.sensor_smoke' model
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy.device] Canceling old initialize call
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for LUMI lumi.sensor_smoke (00:15:8d:00:05:87:d7:9d)
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
2021-03-05 15:52:27 DEBUG (MainThread) [zigpy.quirks.registry] Found custom device replacement for 00:15:8d:00:05:87:d7:9d: <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
2021-03-05 15:52:27 INFO (MainThread) [zigpy.device] [0x37eb] Was quickly initialized from 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor' quirk
2021-03-05 15:52:27 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] device - 0x37EB:00:15:8d:00:05:87:d7:9d entering async_device_initialized - is_new_join: True
[... cut ...] 080a01ff42280121ef0b03281e0421a80105211000062404000000000a21000008210410642000962300000000'
2021-03-05 15:52:42 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1c51000c00050002eb37ffd6
2021-03-05 15:52:45 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1c52000c00050002eb37ffd7
2021-03-05 15:52:48 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1c53000c00050002eb37ffd4
2021-03-05 15:52:51 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1c54000c00050002eb37ffd3`

1800Zeta, 2021-03-05:
Apparently this is a really good Hubitat driver:
https://raw.githubusercontent.com/markus-li/Hubitat/release/drivers/expanded/zigbee-xiaomi-smoke-detector-expanded.groovy

I can see test within this although I am not sure on Groovy and how to convert to Python/zigpy

1800Zeta, 2021-03-07:
Doing more digging, I can see what looks like the correct cluster to issue the test command to when using Manage Clusters option on the device. But when I attempt to issue the command it errors with ValueError: Wrong number of parameters for request, expected 2 argument(s).

Adminiuga, 2021-03-07:
I don't know if ui handles multiple commands parameters. You would need to use service call, that can send multiple command params

1800Zeta, 2021-03-08:
I have managed to get the test to work:
`service: zha.set_zigbee_cluster_attribute
    data:
      ieee: 00:15:8d:00:05:87:d7:9d
      endpoint_id: 1
      cluster_id: 1280
      attribute: 65521
      value: 50397184
      manufacturer: 4447`
A few moments after issuing this command it beeps, same as pressing the manual test button.

nashant, 2021-07-26:
I would very much like this too. The relevant Zigbee2MQTT code is here:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/xiaomi.js#L792-L804
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js#L5335-L5364
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/toZigbee.js#L1993-L2013

The sigpy/ZHA code is, apparently, here:
https://github.com/zigpy/zha-device-handlers/blob/9c278b8f9c42a8d8f0c1d1b8dc3b137992f19981/zhaquirks/xiaomi/__init__.py#L185-L217

I've been suggested to add a PM2.5 cluster a la:
https://github.com/zigpy/zigpy/blob/dev/zigpy/zcl/clusters/measurement.py#L332

I'm just not sure where to start

LinusT, 2022-01-23:
Checked with the a test version. The capability is not there. Since this is one of the more afordable zigbee smoke detectors the capabilities described and researched in this thread has high value. Smart muting of alarms and using using multiple wireless smoke alarms as sirens for home alarm comes to mind. 

Would prefer to keep this open. Unsure how I can assist in progressing the issue though.

1800Zeta, 2022-01-23:
I am seeing more and more people in general asking questions about this smoke alarm.  Have had a couple on my YouTube video where I unboxed and added it to Home Assistant. It would be good to get this sorted. Like @LinusT and @nashant I am unsure how to work with the files etc. I have read and read but struggling to get head around it all

closed 2022-07-29 by github-actions[bot]