zigpy/zha-device-handlers #813: [Device Support Request] Tuya/MoesHouse wall switch TS0601 _TZE200_amp6tsvy

LouisHerber, 2021-03-12:
When adding a device as the MoesHouse Zigbee 3.0 wall switch to the Zigbee network using ZHA integration (e.g. : https://aliexpress.com/item/1005002061498115.html), the device will be added, but no endpoints are added.

I've already tried myself to edit the quirk TS0601.py in Tuya folder offline, then placing it in the right folder in my HA and reboot. Not sure if this would be sufficient though, but anyway I can't get it to work - logging shows that zigpy.quirks still doesn't match the re-added device to any quirk. It's probably my programming skill lacking a bit here :-) Therefore I'd like to ask some help to get it added and recognized.

2021-03-12 12:00:57 DEBUG (MainThread) [zigpy.endpoint] [0x4697:1] Manufacturer: _TZE200_amp6tsvy
2021-03-12 12:00:57 DEBUG (MainThread) [zigpy.endpoint] [0x4697:1] Model: TS0601
2021-03-12 12:00:57 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZE200_amp6tsvy TS0601 (_mac_address_)
2021-03-12 12:00:57 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
2021-03-12 12:00:57 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {232, 230} {1}
2021-03-12 12:00:57 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
2021-03-12 12:00:57 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {232, 230} {1}
2021-03-12 12:00:57 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
2021-03-12 12:00:57 DEBUG (MainThread) [zigpy.quirks.registry] Fail because device_type mismatch on at least one endpoint
2021-03-12 12:00:57 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
2021-03-12 12:00:57 DEBUG (MainThread) [zigpy.quirks.registry] Fail because device_type mismatch on at least one endpoint
2021-03-12 12:00:57 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
2021-03-12 12:00:57 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 2} {1}
2021-03-12 12:00:57 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
2021-03-12 12:00:57 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 2} {1}
2021-03-12 12:00:57 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
2021-03-12 12:00:57 DEBUG (MainThread) [zigpy.quirks.registry] Fail because device_type mismatch on at least one endpoint
2021-03-12 12:00:57 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
2021-03-12 12:00:57 DEBUG (MainThread) [zigpy.quirks.registry] Fail because device_type mismatch on at least one endpoint
2021-03-12 12:00:57 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
2021-03-12 12:00:57 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {11, 13} {1}
2021-03-12 12:00:57 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
2021-03-12 12:00:57 DEBUG (MainThread) [zigpy.quirks.registry] Fail because device_type mismatch on at least one endpoint

The Device Signature below is for the 1-button switch version. I have invested in a lot of those so would be glad to see getting them to work :)

Device Signature:

{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_amp6tsvy",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

MattWestb, 2021-03-12:
The last line with `"class": "zigpy.device.Device"` is saying that your quirk is not loaded.
Then you getting the name of your quirk there its being loaded.

LouisHerber, 2021-03-12:
> The last line with `"class": "zigpy.device.Device"` is saying that your quirk is not loaded.
> Then you getting the name of your quirk there its being loaded.

Thanks for your reply, and I understand that. I don't know WHY it doesn't work, nor if I'm editting the right file or even if the quirk would also apply to the mentioned device to be honest... I've just tried to add the manufactorer/model of my device directly into the file mentioned here: https://github.com/zigpy/zha-device-handlers/pull/784 as it is the closest match I could find.

LouisHerber, 2021-03-12:
And some additional info. I do see in the logs that it detects an "unknown cluster", not sure if this is relevant:

2021-03-12 12:00:57 INFO (MainThread) [zigpy.endpoint] [0x4697:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10])
2021-03-12 12:00:57 WARNING (MainThread) [zigpy.zcl] Unknown cluster 61184
2021-03-12 12:00:57 DEBUG (MainThread) [zigpy.device] [0x4697] Extending timeout for 0x38 request

MattWestb, 2021-03-12:
From Z2M:
```
    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_amp6tsvy'}],
        model: 'ZTS-EU',
        vendor: 'Moes',
        description: 'Wall touch light switch (1 gang)',
        exposes: [e.switch().setAccess('state', ea.STATE_SET)],
        fromZigbee: [fz.tuya_switch_1, fz.tuya_switch_2],
        toZigbee: [tz.tuya_switch_state],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint, logger) => {
            await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        },
    },
```
Its have its own device handle so you cant only adding the signature in the quirk.
Perhaps its similar enough for getting it working but i dont think that.

You must putting all the cluster on the endpoints in the signature for getting it triggered.

LouisHerber, 2021-03-12:
Thanks for the info! But... I'm a bit lost on that.  Not sure on what to do (next) on this, any help appreciated. Would we need to edit / add to TS0601.py? Or build a new file? And what info would we need for that.

Could someone help me build that file?

vincenttan75, 2021-03-16:
hi is there possible to add _TZE200_tz32mtza? this is a 3 gang switch

vincenttan75, 2021-05-04:
> From Z2M:
>
> ```
>     {
>         fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_amp6tsvy'}],
>         model: 'ZTS-EU',
>         vendor: 'Moes',
>         description: 'Wall touch light switch [...]

I put all 3 clusters on the endpoints but still getting unknown cluster 11684 error..
`Logger: zigpy.zcl
Source: /usr/local/lib/python3.8/site-packages/zigpy/zcl/__init__.py:91
First occurred: 2:46:58 AM (1 occurrences)
Last logged: 2:46:58 AM

Unknown cluster 61184`

any clues?

sstefanowski, 2021-05-07:
Hello,
I bought 2 such devices for testing - because it's quite **unique**, Zigbee +working in "no-neutral" with "no-capacitor" needed.  
https://zigbee.blakadder.com/Moes_ZTS-EU1-W.html

...and... I was also surprised that MOES Wall switch "_TZE200_amp6tsvy" is not correctly supported.

**My config and findings:** 
- I tried and tested to use it in "no-neutral" and "no capacitor) wiring. I have just this installation at home.
- Quirk was not found so I added `("_TZE200_amp6tsvy", "TS0601")` to `ts601.py` and `TuyaSingleSwitch` class and quirk. I tested registering under the other quirks (like `ts001x.py` and `TuyaSingleNoNeutralSwitch`) but this quirk is NOT recognized correctly (clusters don't match). For, now I know it matches just the one I mentioned above.
So... First... I vote for adding this adding `("_TZE200_amp6tsvy", "TS0601")` to `ts601.py` permanently, and then we can focus on fixing.

With this config, I can see the switch is correctly registered under HA but have two issues.
**The issues are:**

1 After every switch on and off I can see in the logs 12 DEBUG entries generated by `TuyaOnOff.switch_event(..,) `method (cluster defined in `replacement`). 
OFF (channel 1, state: 0)
```
May 07 01:25:36 raspberrypi hass[1562]: 2021-05-07 01:25:36 DEBUG (MainThread) [zhaquirks.tuya] 60:a4:23:ff:fe:aa:dc:72 - Received switch event message, channel: 1, state: 0
(... repeated 12 times within 5 seconds ...)
May 07 01:25:41 raspberrypi hass[1562]: 2021-05-07 01:25:41 DEBUG (MainThread) [zhaquirks.tuya] 60:a4:23:ff:fe:aa:dc:72 - Received switch event message, channel: 1, state: 0
```
ON (channel 1, state: 1)
```
May 07 01:25:36 raspberrypi hass[1562]: 2021-05-07 01:25:36 DEBUG (MainThread) [zhaquirks.tuya] 60:a4:23:ff:fe:aa:dc:72 - Received switch event message, channel: 1, state: 0
(... repeated 12 times within 5 seconds ...)
May 07 01:25:41 raspberrypi hass[1562]: 2021-05-07 01:25:41 DEBUG (MainThread) [zhaquirks.tuya] 60:a4:23:ff:fe:aa:dc:72 - Received switch event message, channel: 1, state: 0
```
Don't know if such log entries are expected but **when switching off the bulb it blinks (one time)** after a second or two after switching. This must be something with ZHA as when not paired with ZHA the "single blink after off" doesn't happen. Hypothesis: It looks like the switch confirms some message reception from ZHA.
**Any comments?**

2. When switched from a physical button on the device (only in this case!) two additional entries are showing after the 12 lines mentioned above. They are also generated by `TuyaOnOff.switch_event(...)` but the channels are 782 and 783. The state is always 0 an2 respectively no matter if switching goes to ON or OFF.

```
(...12 entires like mentined above....)
May 07 01:25:41 raspberrypi hass[1562]: 2021-05-07 01:25:41 DEBUG (MainThread) [zhaquirks.tuya] 60:a4:23:ff:fe:aa:dc:72 - Received switch event message, channel: 1, state: 0
May 07 01:25:42 raspberrypi hass[1562]: 2021-05-07 01:25:42 DEBUG (MainThread) [zhaquirks.tuya] 60:a4:23:ff:fe:aa:dc:72 - Received switch event message, channel: 782, state: 0
May 07 01:25:43 raspberrypi hass[1562]: 2021-05-07 01:25:43 DEBUG (MainThread) [zhaquirks.tuya] 60:a4:23:ff:fe:aa:dc:72 - Received switch event message, channel: 783, state: 2
```
The side effect of this (BUG!) is that the ligh-switch toggle in HomeAssistan remains **always in ON** position no matter if the light is ON or OFF. This is probably caused by the bug in `TuyaOnOff.switch_event(...) `because it changes HA state ignoring the channel info. 
```
(...)
def switch_event(self, channel, state):
        """Switch event."""
        _LOGGER.debug(
            "%s - Received switch event message, channel: %d, state: %d",
            self.endpoint.device.ieee,
            channel,
            state,
        )
        self._update_attribute(ATTR_ON_OFF, state)
(...)
```
The last log event is always generated with the `channel: 783, state: 2` (state is > 0), which is probably interpreted in HA as "the light ON". I guess this could be corrected by checking if the channel is 1:

```
    (...)
    if channel==1:
        self._update_attribute(ATTR_ON_OFF, state)
    (...)
```
but **I need a confirmation from ZHA and HA devs that my fix proposal is OK...**

Anyway... the biggest problem is issue no 1 ("single blink after switching OFF") and I need guidance on how to go deeper in debugging.

sstefanowski, 2021-05-09:
OK, I see. I'm planning to buy more than switches (also 2-gang) so probably also need a new quirk, but... the first question:

Can you confirm a bug I'm suspecting?
Can you see additional channel 782, 783 log reports when using the physical touch-button on the device? Can you confirm the bug I described in p.2 above? 
I suspect that receiving event `channel=783 state=2` and sending it to HA leaves the HA toggle in the wrong position. I mean when you turn off the light using the physical button the toggle in HA frontend first goes to OFF (as a reaction to events with `state=0`) and... then it stays ON (I guess as a reaction on `channel=783 state=2`). As the final result, the light is really OFF but the HA shows the toggle is ON. 
Note: This happens only when using a physical touch-button on the device as switching from HA does not generate channel 782 and 783 events in my case.

So... can you confirm you can see the same bug?

AlterSb, 2021-05-10:
Just got multiple gang 'TS0601_TZE200_amp6tsvy' today and found out it was unsupported. Got an unknown cluster-specific command 1 error. I'm new to HA and still no clue how to get it work.

vincenttan75, 2021-05-10:
> Just got 2 gang 'TS0601_TZE200_g1ib5ldv' today and found out it was unsupported. Got an unknown cluster-specific command 1 error. I'm new to HA and still no clue how to get it work.

just change the model info of the ts0601.py quirk to your model. and follow the steps
https://github.com/zigpy/zha-device-handlers#testing-quirks-in-development-in-docker-based-install

vincenttan75, 2021-05-11:
yup it wlll only see 1 endpoint -- because from the signature - its a smart plug (device id = 0x0051) 
[tuya protocol](https://developer.tuya.com/en/docs/iot/tuya-zigbee-universal-docking-access-standard?id=K9ik6zvofpzql)
the tuya protocol for smart device has only 1 endpoint. 
and from the Z2M [documentation](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/moes.js) that supported the tuya switch..
the "endpoint" is "nested" under the switch state ..  so normal way of replacement endpoints on a quirk probably wont work.. without some heavy modification

vincenttan75, 2021-05-13:
> > > Just got 2 gang 'TS0601_TZE200_g1ib5ldv' today and found out it was unsupported. Got an unknown cluster-specific command 1 error. I'm new to HA and still no clue how to get it work.
> > 
> > 
> > just change the model info of the ts0601.py quirk to your model. and follow the steps
> > https://github.com/zigpy/zha-device-handlers#testing-quirks-in-development-in-docker-based-install
> 
> Thank you for advice, I added _TZE200_g1ib5ldv and it see only one endpoint. So I try to write new Quirk but still keep getting mismatch error.

https://www.digiblur.com/2021/03/zigbee2mqtt-with-sonoff-zigbee-bridge.html

i have stop messing ard the quirk for ZHA and used Z2M with sonoff ZBridge instead.. works like a charm..

vincenttan75, 2021-05-13:
currently ZHA quirk cant pickup ts0601 multi gang switches endpoints because tuya "hides" them in the switch states.. Z2M CAN..
highly recommend using Z2m instead and pair it with your fav coordinator.. i'm using sonoff zbridge/tasmota

sstefanowski, 2021-05-13:
> i have stop messing ard the quirk for ZHA and used Z2M with Sonoff ZBridge instead.. works like a charm..
Wooow, is this really possible to connect Z2M to Sonoff over Wifi? 
I like Sonoff ZBridge because it has a much better range, than my USB dongle (also it's placed in a better place than my HA host).
How I can use Z2M with Sonoff over Wifi? Should I just pull Z2M dev branch? and setup `adapter` and `port`???
Note: I'm NOT using "dockerized" version of HA because I need LIRC support (doesn't work under docker)

vincenttan75, 2021-05-14:
> Wooow, is this really possible to connect Z2M to Sonoff over Wifi?
> I like Sonoff ZBridge because it has a much better range, than my USB dongle (also it's placed in a better place than my HA host). [...]

https://www.digiblur.com/2021/03/zigbee2mqtt-with-sonoff-zigbee-bridge.html
yes - i am using it . not very sure of how stable the zb/wifi interference will cause problems.. so far its my mqtt that is disconnecting.. probably because it is hosting on the same rpi4 .. I will set up a separate broker on my pi zero - hopefully it will be better.. functional wise.. everything works.. and u will see .. the endpoints is nested elsewhere .. that is why ZHA cannot pick up

ShadowFistOkami, 2021-05-15:
#824

hdlineage, 2021-07-11:
I was able to adapt the ts0601.py single switch quirk. 
It works fine for now without major bugs.
However, I do notice that toggling the light through lovelace manually can some time be buggy. (toggling off the light would cause the light to go: off->on->off)
Which is weird but I do not observe the same behavior with automations.

alexsalvo1, 2021-09-07:
> OK, I see. I'm planning to buy more than switches (also 2-gang) so probably also need a new quirk, but... the first question:
> 
> Can you confirm a bug I'm suspecting?
> Can you see additional channel 782, 783 log reports when using the physical touch-button on the device (the bug I described in p.2 above)?
> 
> I suspect that receiving event `channel=783 state=2` and forwarding `2` to HA leaves the HA toggle in the wrong position. I mean when you turn off the light using the physical button the toggle in HA frontend first goes to OFF (as a reaction to events with `state=0`) and... then it stays ON (I guess as a reaction on `channel=783 state=2`). As the final result, the light is really OFF but the HA shows the toggle is ON.
> Note: This happens only when using a physical touch-button on the device as switching from HA does not generate channel 782 and 783 events in my case.
> 
> So... can you confirm you can see the same bug?

exactly the same behavior for me

closed 2021-11-23 by dmulcahey

JoryHogeveen, 2022-07-14:
Related to this issue I started the following topic on the community: https://community.home-assistant.io/t/user-experience-request-differernt-zigbee-tuya-based-touch-light-switches/439249
Any contributions to experience with these devices are very welcome!

PR #1166 (2021-11-23): Add quirks for Tuya multigang switches
This PR add quirks for Tuya multigang switches.

It should correct the management of the On/Off cluster for the multiswitch devices that make use of the Tuya cluster (`0xEF00`) for command/status reporting.

The `MODELS_INFO` of those devices that have been reported or an issue has been reported have been added.
For those devices whose number of gangs is not clear, they have been defined as 1-gang devices until someone report correct info.

This could fix the following issues (only on/off switches):
fixes: #813
fixes: #824
fixes: #913
fixes: #972
fixes: # [... cut]