zigpy/zha-device-handlers #819: [Device Support Request] TS0601 TRV

nickg90, 2021-03-15:
I bought 4x TS0601 Radiator TRVs the other day, 2 connected fine with all climate control options, 2 added but with no entities.
the ones that work had manufacturer _TZE200_ckud7u2l and quirk zhaquirks.tuya.valve.MoesHY368_Type1, the ones that didnt were _TZE200_cwnjrr72 with no quirk.

working:
`{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0201",
        "0x0204",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_ckud7u2l",
  "model": "TS0601",
  "class": "zhaquirks.tuya.valve.MoesHY368_Type1"
}`

Not working:
`{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_cwnjrr72",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}`

Please update ZHA to support these devices

MattWestb, 2021-03-16:
The first one is supported with #726 

Moes | HY369-ZB | _TZE200_ckud7u2l | TS0601 | ✅ Moes I | 726
-- | -- | -- | -- | -- | --

So shall not being any big problem but i dont knowing wot is working or not then luck of feedback  from users.

The second one looks being the same "smart pug" that is being used of the first one and with the same Zigbee module but with new / other firmware.

I have searching all guthub and also downloaded Z2M device.jar and i cant finding the new model ID.

Its possible trying adding the model ID bu then you need doing some diving in the docker container and modding one fill (its being overwritten with  the next HA update so its not dangerous).

But the devs like having verification that is the right device that is being added (some users have doing it wrong and it was one large mess with broken devices for other users).

So is you first (the working model) one Moes HY369-ZB from the [tuya TRV matrix](https://github.com/zigpy/zigpy/discussions/653#discussioncomment-314395) and the picture is the right ?

And then is the second one (not working) looks 100% the same ?

If not pleas provide one photo of the device and one link to the manufacture for verifying the device.

nickg90, 2021-03-16:
They were purchased at the same time from the same vendor on alibaba but little detail on the specific manufacturer / model. They are all visually and functionally identical from what I can tell and appear to be Moes HY368-ZB. They look similar to the first picture there but with a green home icon in the centre.

[image]

the first one is working completely as far as I can tell. it reports temperature, set temperature, heating/idle times, can set the temoperature, there is a on/off switch and the power level is reported. I dont imagine there will be any differences to the zigbee communication of these devices, hopefully just needs the new model ID to be added.

nickg90, 2021-03-16:
more detail on functionality of the ones which work, Screenshots below. on the last image, the heat / power buttons do not appear to have any effect. the power switch does operate the valve as expected. There do not appear to be any other modes and settings / reporting valve position as per the checklist in the link you sent.

[image]

[image]

[image]

[image]

MattWestb, 2021-03-16:
Use the first pat of this guide for editing the valve.py https://github.com/zigpy/zha-device-handlers/discussions/693 if you is knowing how to opening one command line interface in your HA docker container. If not you need starting with https://github.com/zigpy/zha-device-handlers#testing-quirks-in-development-in-docker-based-install first part until you have the command line.

In valve.py you is having at line 794:
```
        MODELS_INFO: [
            ("_TZE200_kfvq6avy", "TS0601"),
            ("_TZE200_zivfvd7h", "TS0601"),
        ],
```
Adding the new model ID after the last one like this:
```
        MODELS_INFO: [
            ("_TZE200_kfvq6avy", "TS0601"),
            ("_TZE200_zivfvd7h", "TS0601"),
            ("_TZE200_cwnjrr72"), "TS0601"),
        ],
```
Then press `CTR` + `S` and then `CTR` + `Q` for saving and quit the editor.

Restart HA and see if your TRV have getting one quirk  on the device card.

Wot i have seen is all parameters looks the same but one that is one shown on paring can being different so perhaps its not working.

nickg90, 2021-03-16:
Done, its better but not fully working. climate entity is working but no power reporting and no on/off switch. I can set temperature and the valve does trigger when it temperature drops below set temp.

[image]

MattWestb, 2021-03-16:
Great work done  !!

If you is comparing the first type and the second type is they working the same way and all functions is working for both types ?

I cant changing the function then its to deep for my knowledge but other devs can fixing not working parts.

If  you is getting both types working the same i can making on PR for getting the devices supported in the HA.

nickg90, 2021-03-16:
oooo interesting, the buttons on the climate control itself work now, these do not work on the other devices and is done using a seperate switch 
[image]

nickg90, 2021-03-16:
the only thing then which is missing is the power sensor which works on the first type reporting 100% and exists for the second but reports nothing

nickg90, 2021-03-16:
> Great work done !!
>
> If you is comparing the first type and the second type is they working the same way and all functions is working for both types ?
>
> I cant changing the function then its to deep [...]

thanks for all your help with this, I now have something usable which I did not have before! :)

MattWestb, 2021-03-16:
Sorry i was looking in the wrong section so your device is using the wrong model :-((
At line 843:
```
        MODELS_INFO: [
            ("_TZE200_ckud7u2l", "TS0601"),
            ("_TZE200_ywdxldoj", "TS0601"),
        ],
```
and adding you ID in the end:
```
        MODELS_INFO: [
            ("_TZE200_ckud7u2l", "TS0601"),
            ("_TZE200_ywdxldoj", "TS0601"),
            ("_TZE200_cwnjrr72"), "TS0601"),
        ],
```
And deleting the changes you was making around line 794.

Save, quit and restart HA.

MattWestb, 2021-03-16:
The "old" is writing Quirk:  "zhaquirks.tuya.valve.MoesHY368_Type1" on the device card info and is its working right the new one  shall doing the same.

Your new is now shown as one Siterwell GS361A and its  not having all functions of the Mose.

nickg90, 2021-03-16:
ahh ok, will try that.

I also noted a typo in your code, there is an extra bracket on the new lind

MattWestb, 2021-03-16:
More then true !!

It (bad quirks) can braking ZHA and its not starting.

I correcting it if some other user is copying it and trying to using it ;-)

nickg90, 2021-03-16:
spotted it first time but forgot to mension!
just tried it out and all up and running. all devices behaving the same now with seperate switch for on/off but buttons on the climate control not doing anything. not too much of an issue as the overall functionality is still there.

Thanks again! please do make a PR for this as im sure you will be able to say what needs changing better than I!

closed 2021-03-16 by nickg90

MattWestb, 2021-03-16:
I have putting it as pending in the tuya TRV matrix and linked to this issue so users can applying the fix and getting them working.

I doing one PR tomorrow and linking your request to it so the devs can see its working OK.

MattWestb, 2021-03-16:
PR #820 made and matrix is updated :-)

Moes | HY368-ZB | _TZE200_cwnjrr72 | TS0601 | ☑️ Moes I | 820
-- | -- | -- | -- | -- | --
Moes | HY368-ZB | _TYST11_cwnjrr72 | wnjrr72 | ☑️ Moes II | 820

☑️ = User added not merged so need manual patching and is experimental.

nickg90, 2021-03-22:
these TRVs appear to update temperatures very intermittently. sometimes it can be 5-10 minutes, other times 1-2 hours and some drop out for several hours at a time. They still can recieve temperature changes but this does not trigger them to update temperatures.

on another forum (https://github.com/Koenkk/zigbee2mqtt/issues/3821) they noted that if the calibration is updated, the valve returns the current temperature and this could be used to force an update. This was done via zigbee2MQTT by publishing "zigbee2mqtt/bedroom_moes_trv/set/local_temperature_calibration"

ive been trying to replicate this myself in nodered but cannot work out how to adapt this to an ZHA command or whether this will not be possible untill further support is added to this valve. any thoughts?

MattWestb, 2021-03-24:
The quirk is in dev branch now :-))

I have trying the calibration on Siterwell (its the first and not complete integration of tuya TRV then the devs was not having one TRV and was doing it blind) but its not working so not possible getting the temperature reporting often in them until some devs is fixing the broken function.

I dont knowing if the calibration is working on HY368-ZB but @xonestonex was  doing on great work implanting all  function hi could finding in the HY368-ZB.

If you like to see wot the quirk is doing with your TRV add `zhaquirks.tuya: debug` in your log section of advance ZHA config.

almico, 2021-04-13:
I currently have a `_TZE200_ywdxldoj` and a `_TZE200_cwnjrr72`.
This is what I see for the `_TZE200_ywdxldoj` at the moment:
[image]
The circled "Preset" should read "schedule", but it is showing "None". I will check later, as it might be that it starts with "None" and, hours later, it changes to "schedule".

This is what I see for the `_TZE200_cwnjrr72` at the moment:
[image]
As you can see, there is no "Preset". Is this expected or is it an error somewhere?

pkoretic, 2022-01-19:
@almico did you get this sorted? I see the same using latest version of official home assistant on _TZE200_cwnjrr72. There is only Heat as operation choice.

almico, 2022-01-20:
@pkoretic 
This is what I have at the moment:
[image]

`Operation` is something I didn't care too much. What I was missing was the `Preset`.

Quapcio, 2022-01-29:
I have 5 1-year old _TZE200_ckud7u2l working great. Today I received new one wich is (sadly to me) is _TZE200_cwnjrr72.

state received from this device is:

"""
hvac_modes:
  - heat
  - auto
  - 'off'
min_temp: 5
max_temp: 35
target_temp_step: 0.5
preset_modes:
  - none
  - away
  - schedule
  - manual
  - boost
  - complex
  - comfort
  - eco
current_temperature: null
temperature: 15
preset_mode: none
auto_lock: null
away_mode: null
away_preset_days: null
away_preset_temperature: null
battery_low: null
boost_time: null
child_lock: null
comfort_temperature: null
current_heating_setpoint: 15
eco_temperature: null
force: normal
linkquality: 63
local_temperature_calibration: null
max_temperature: null
min_temperature: null
position: null
system_mode: auto
update:
  state: null
update_available: null
valve_detection: null
window_detection: null
friendly_name: głowica garaż
supported_features: 17
"""

I just upgraded from 1.18 to newest 1.22, I'm going to try Edge version

Quapcio, 2022-01-29:
edge version is a little bit more helpful:

"""
hvac_modes:
  - heat
  - auto
  - 'off'
min_temp: 5
max_temp: 35
target_temp_step: 0.5
preset_modes:
  - none
  - away
  - schedule
  - manual
  - boost
  - complex
  - comfort
  - eco
current_temperature: 14
temperature: 15
preset_mode: none
auto_lock: null
away_mode: null
away_preset_days: null
away_preset_temperature: null
battery_low: null
boost_time: null
child_lock: null
comfort_temperature: null
current_heating_setpoint: 15
eco_temperature: null
force: normal
linkquality: null
local_temperature: 14
local_temperature_calibration: null
max_temperature: null
min_temperature: null
position: null
system_mode: auto
update:
  state: null
update_available: null
valve_detection: null
window_detection: null
friendly_name: głowica garaż
supported_features: 17
"""

There is a current temperature/local temperature value.
But the main missing part is (for me) "position: null"
In other TS0601 I can use:
if (state_attr('climate.glowica_2', 'position') > 0)
to trigger "need heating" ;-)

pkoretic, 2022-02-04:
@Quapcio what do you mean by edge version? I'm only aware of Edge Zigbee2MQTT version while here we are discussing ZHA integration so want to understand what you have installed.

I'm running latest Home Assistant (beta) 
core-2022.2.1

I'm also running latest
  - zha-quirks==0.0.66
  - zigpy==0.43.0

and what I get is

[image]

[image]

For tze200_cwnjrr72_ts0601_b17029fe_thermostat

I was under impression from all closed tickets / posts that in the 'latest' version there is clear support for cwnjrr72 with all options.

MattWestb, 2022-02-04:
@pkoretic Your device have getting one standard quirk in ZHA.
You shall having presets if you is looking (clicking on the thermostat "device" on the device card. From there you can changing the working mode of the TRV. If you like getting GUI entris you need installing one local quirk that is adding more functions in the GUI but you shall having all on the device cluster in the standard one.
The issue with the extender quirk that  is WIP https://github.com/zigpy/zha-device-handlers/issues/1212.

pkoretic, 2022-02-04:
@MattWestb thanks for a quick response and sorry for pinging on two different tickets. 
I'll then wait for WIP ticket to be done.

Indeed there are presets but since they don't have configuration available from HA either (e.g. schedule) I was thinking that I'm not running a proper version compared to some other screenshots shared here show more options.

RhavoX, 2022-08-25:
I guess that this extender quirk died right? Will any more work be done on this?

MattWestb, 2022-08-25:
ZAH have doing some changes that was braking all tuya TRVs local quirk but most is being fixed like my Siterwell is working great with the latest local quirk (battery need being replaced in 2 of 5 after 2.5 year and only one have being changed before).

The maintainer have demanding some additional information and we have applying it and asking if its OK but no answer so all extended quirks is on hold and the user that have making most of them is also off line after the "problem".

Hope some with the knowledge getting the PRs made merged and adding the moderate test code so user can start using them in HA without adding external code then its starts being cold now (27°C outside).

almico, 2022-09-22:
In my setup, at least one of my TRV didn't properly handle the "away" state any more 😢