zigpy/zha-device-handlers #823: [Device Support Request] Lidl / Silvercrest Door and Window Sensor (Tuya based TY0203)

Purii, 2021-03-18:
Thanks for all your work!

Product: https://www.lidl-service.com/cps/rde/SID-E00E6105-DB4B80B8/lsp/hs.xsl/product.html?id=5027327410&rdeLocaleAttr=en&title=Smart%20Window%20or%20Door%20Sensor

I'm able to pair the device, but don't receive any status updates.

**Describe the solution you'd like**
* Current state: Door open/closed
* Battery State

**Device signature**
```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4619, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ1800_ejwkn2h2",
  "model": "TY0203",
  "class": "zigpy.device.Device"
}
```

<details>
<summary>
Pairing Log
</summary>

```
Device 0x2578 (ec:1b:bd:ff:fe:70:b1:3b) joined the network
Device ec:1b:bd:ff:fe:70:b1:3b changed id (0x88b4 => 0x2578)
Checking quirks for _TZ1800_ejwkn2h2 TY0203 (ec:1b:bd:ff:fe:70:b1:3b)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
device - 0x2578:ec:1b:bd:ff:fe:70:b1:3b entering async_device_initialized - is_new_join: True
device - 0x2578:ec:1b:bd:ff:fe:70:b1:3b has joined the ZHA zigbee network
[0x2578](TY0203): started configuration
[0x2578:ZDO](TY0203): 'async_configure' stage succeeded
[0x2578] Extending timeout for 0x3c request
[... 120 lines cut ...]
[0x0c3d:1:0x0500] ZCL request 0x0000: [<ZoneStatus.Restore_reports: 32>, <bitmap8.0: 0>, 0, 0]
[0x0c3d:1:0x0500] No handler for cluster command 0
[0x0C3D:1:0x0500]: Updated alarm state: ZoneStatus.0
Device 0x0c3d (bc:33:ac:ff:fe:35:91:39) joined the network
Skip initialization for existing device bc:33:ac:ff:fe:35:91:39
[0x0c3d:zdo] ZDO request ZDOCmd.Device_annce: [0x0C3D, bc:33:ac:ff:fe:35:91:39, 128]
Device 0x0c3d (bc:33:ac:ff:fe:35:91:39) joined the network
Skip initialization for existing device bc:33:ac:ff:fe:35:91:39
[0x0c3d:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=115 command_id=Command.Report_Attributes>
[0x0c3d:1:0x0001] ZCL request 0x000a: [[Attribute(attrid=33, value=<TypeValue type=uint8_t, value=200>)]]
[0x0c3d:1:0x0001] Attribute report received: battery_percentage_remaining=200
[0x0c3d:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=116 command_id=0>
[0x0c3d:1:0x0500] ZCL request 0x0000: [<ZoneStatus.Restore_reports|Alarm_1: 33>, <bitmap8.0: 0>, 0, 0]
[0x0c3d:1:0x0500] No handler for cluster command 0
[0x0C3D:1:0x0500]: Updated alarm state: ZoneStatus.Alarm_1
[0x0c3d:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=117 command_id=0>
[0x0c3d:1:0x0500] ZCL request 0x0000: [<ZoneStatus.Restore_reports|Alarm_1: 33>, <bitmap8.0: 0>, 0, 0]
[0x0c3d:1:0x0500] No handler for cluster command 0
[0x0C3D:1:0x0500]: Updated alarm state: ZoneStatus.Alarm_1
```
</details>

Zigbee2Mqtt: https://www.zigbee2mqtt.io/devices/HG06336.html

--> Let me know if you need any further details 👍

Purii, 2021-03-19:
After repairing I don't event get any entities

closed 2021-09-22 by github-actions[bot]

bobvmierlo, 2025-05-06:
Any chance anybody wants to add the 'tamper' sensor in this library? The herdsman library (used by z2mqtt) has had this extra sensor for ages.