zigpy/zha-device-handlers #824: [Device Support Request] EU Tuya ZigBee Smart Light Switch

Kalininator, 2021-03-20:
I am able to pair the device, but dont get any entities
[Product Link](https://www.aliexpress.com/item/1005002180713176.html?spm=a2g0s.9042311.0.0.7ee34c4dDeZ57Y)

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices 
screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
Device 0xe670 (60:a4:23:ff:fe:94:be:74) joined the network
[0xe670] Requesting 'Node Descriptor'
Tries remaining: 2
[0xe670] Extending timeout for 0xec request
Device 0xe670 (60:a4:23:ff:fe:94:be:74) joined the network
Skip initialization for existing device 60:a4:23:ff:fe:94:be:74
Device 0xe670 (60:a4:23:ff:fe:94:be:74) joined the network
Skip initialization for existing device 60:a4:23:ff:fe:94:be:74
[0xe670:zdo] ZDO request ZDOCmd.Device_annce: [0xE670, 60:a4:23:ff:fe:94:be:74, 128]
[0xe670:zdo] ZDO request ZDOCmd.Node_Desc_req: [0x0000]
[0xe670:zdo] Unsupported ZDO request:ZDOCmd.Node_Desc_req
[0xba32] Delivery error for seq # 0xe8, on endpoint id 0 cluster 0x0021: message send failure
[0xba32:1:0x0019]: Failed to bind 'ota' cluster: [0xba32:0:0x0021]: Message send failure
[0xba32:1:0x0019]: finished channel configuration
[0xba32:1:0x0000]: 'async_configure' stage succeeded
[0xba32:1:0x0019]: 'async_configure' stage succeeded
[0xba32](TS0601): completed configuration
[0xba32](TS0601): stored in registry: ZhaDeviceEntry(name='_TZE200_tz32mtza TS0601', ieee='60:a4:23:ff:fe:94:be:74', last_seen=1616247432.734196)
Received frame on uninitialized device 60:a4:23:ff:fe:94:be:74 (Status.NEW) for endpoint: 1
[0xe670] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=25 command_id=Command.Read_Attributes>, payload: b'\x07\x00'
[0xe670] Uninitialized device command 'Command.Read_Attributes' args: [[7]]
Received frame on uninitialized device 60:a4:23:ff:fe:94:be:74 (Status.NEW) for endpoint: 1
[0xe670] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=25 command_id=Command.Read_Attributes>, payload: b'\x07\x00'
[0xe670] Uninitialized device command 'Command.Read_Attributes' args: [[7]]
Received frame on uninitialized device 60:a4:23:ff:fe:94:be:74 (Status.NEW) for endpoint: 1
[0xe670] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=25 command_id=Command.Read_Attributes>, payload: b'\x07\x00'
[0xe670] Uninitialized device command 'Command.Read_Attributes' args: [[7]]
Received frame on uninitialized device 60:a4:23:ff:fe:94:be:74 (Status.NEW) for endpoint: 1
[0xe670] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=25 command_id=Command.Read_Attributes>, payload: b'\x07\x00'
[0xe670] Uninitialized device command 'Command.Read_Attributes' args: [[7]]
Received frame on uninitialized device 60:a4:23:ff:fe:94:be:74 (Status.NEW) for endpoint: 1
[0xe670] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=25 command_id=Command.Read_Attributes>, payload: b'\x07\x00'
[0xe670] Uninitialized device command 'Command.Read_Attributes' args: [[7]]
Received frame on uninitialized device 60:a4:23:ff:fe:94:be:74 (Status.NEW) for endpoint: 1
[0xe670] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=25 command_id=Command.Read_Attributes>, payload: b'\x07\x00'
[0xe670] Uninitialized device command 'Command.Read_Attributes' args: [[7]]
Received frame on uninitialized device 60:a4:23:ff:fe:94:be:74 (Status.NEW) for endpoint: 1
[0xe670] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=25 command_id=Command.Read_Attributes>, payload: b'\x07\x00'
[0xe670] Uninitialized device command 'Command.Read_Attributes' args: [[7]]
[0xe670] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)
[0xe670] Discovering endpoints
Tries remaining: 3
[0xe670] Extending timeout for 0xee request
[0xe670] Discovered endpoints: [1]
[0xe670:1] Discovering endpoint information
Tries remaining: 3
[0xe670] Extending timeout for 0xf0 request
[0xe670:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10])
Unknown cluster 61184
[0xe670] Extending timeout for 0xf2 request
[0xe670:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=242 command_id=Command.Read_Attributes_rsp>
[0xe670:1] Manufacturer: _TZE200_tz32mtza
[0xe670:1] Model: TS0601
Checking quirks for _TZE200_tz32mtza TS0601 (60:a4:23:ff:fe:94:be:74)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
[... 23 lines cut ...]
[0xe670:1:0xef00] No handler for cluster command 36
[0xba32] Delivery error for seq # 0xf4, on endpoint id 0 cluster 0x0021: message send failure
[0xba32:1:0x0000]: Failed to bind 'basic' cluster: [0xba32:0:0x0021]: Message send failure
[0xba32:1:0x0000]: finished channel configuration
[0xba32] Delivery error for seq # 0xf6, on endpoint id 0 cluster 0x0021: message send failure
[0xba32:1:0x0019]: Failed to bind 'ota' cluster: [0xba32:0:0x0021]: Message send failure
[0xba32:1:0x0019]: finished channel configuration
[0xba32:1:0x0000]: 'async_configure' stage succeeded
[0xba32:1:0x0019]: 'async_configure' stage succeeded
[0xba32](TS0601): completed configuration
[0xba32](TS0601): stored in registry: ZhaDeviceEntry(name='_TZE200_tz32mtza TS0601', ieee='60:a4:23:ff:fe:94:be:74', last_seen=1616247432.734196)
[0xba32](TS0601): started initialization
[0xba32:ZDO](TS0601): 'async_initialize' stage succeeded
[0xba32:1:0x0000]: initializing channel: from_cache: False
[0xba32:1:0x0000]: finished channel configuration
[0xba32:1:0x0019]: initializing channel: from_cache: False
[0xba32:1:0x0019]: finished channel configuration
[0xba32:1:0x0000]: 'async_initialize' stage succeeded
[0xba32:1:0x0019]: 'async_initialize' stage succeeded
[0xba32](TS0601): power source: Battery or Unknown
[0xba32](TS0601): completed initialization
```

**Additional context**
Add any other context or screenshots about the feature request here.

ShadowFistOkami, 2021-05-04:
Having the same issue here. Device paired but no entities added.

After having a look at the quirks folder, there is an entry for the oisqyl4o, but no no entry for the tz32mtza, so no quirks are being applied.
Based on my (VERY) limited understanding, the solution would be to simply add the model in zha-device-handlers/zhaquirks/tuya/ts0601.py , line 35:

       MODELS_INFO: [
            ("_TZE200_oisqyl4o", "TS0601"),
            ("_TZE200_tz32mtza", "TS0601"),
        ], 

My reasoning for the above is that both these models use the same endpoints. The currently supported oisqyl4o variant seems to be the single gang version (https://zigbee.blakadder.com/Moes_ZTS-EU1-W.html), while the unsupported variant is the 3 gang version (https://zigbee.blakadder.com/Moes_ZTS-EU3-W.html). I have a couple of (untested) 2 gang versions which might/might not use one of the above model names. Will try to pair them this weekend and update with further info.

In the meantime, is there anything we can do from our end to test the above proposed change?

vincenttan75, 2021-05-04:
> Having the same issue here. Device paired but no entities added.
>
> After having a look at the quirks folder, there is an entry for the oisqyl4o, but no no entry for the tz32mtza, so no quirks are [...]

`I` tested it couple of hrs ago.. the signature dont match - quirk didnt scan in.. and there's an error on unknown cluster 11864...
humm.. still reading up and debugging.. 

`
class TuyaTripleSwitch(TuyaSwitch):
    """Tuya 3 gang touch switch device."""
    signature = {
        # "node_descriptor": "<NodeDescriptor byte1=1 byte2=64 mac_capability_flags=128 manufacturer_code=4098
        #                       maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264
        #                       maximum_outgoing_transfer_size=82 descriptor_capability_field=0>",
        # device_version=1
        # input_clusters=[0x0000, 0x0004, 0x0005, 0xef00]
        # output_clusters=[0x000a, 0x0019]
        # <SimpleDescriptor endpoint=1 profile=260 device_type=51 device_version=1 input_clusters=[0, 4, 5, 61184] output_clusters=[10, 25]>
        MODELS_INFO: [("_TZE200_tz32mtza", "TS0601")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerClusterOnOff,
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerClusterOnOff,
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            3: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerClusterOnOff,
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }
`

ShadowFistOkami, 2021-05-04:
>  `I tested it couple of hrs ago.. the signature dont match - quirk didnt scan in.. and there's an error on unknown cluster 11864...
> humm.. still reading up and debugging..
 `
Strange, it almost seems like the cluster is an inverted version of the device signature :

      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        **"0xef00"** <-- converts to 61184 in decimal.

Yet, your error shows it's complaining about cluster 11864 (0x2E58). Note the use of the same numbers. 

Again, I'm just spitballing here, but I find it a strange coincidence. Any ideas?

vincenttan75, 2021-05-04:
the unknown cluster seem to be from the zipy
Logger: zigpy.zcl
Source: /usr/local/lib/python3.8/site-packages/zigpy/zcl/__init__.py:91
First occurred: 1:25:44 AM (1 occurrences)
Last logged: 1:25:44 AM

Unknown cluster 61184

ShadowFistOkami, 2021-05-04:
> Logger: zigpy.zcl
> Source: /usr/local/lib/python3.8/site-packages/zigpy/zcl/**init**.py:91
> First occurred: 1:25:44 AM (1 occurrences)
> Last logged: 1:25:44 AM
> 
> Unknown cluster 61184
> 
> something seems missing.. will try debugging

Honestly, I have no clue regarding all this. My background is in QA, not programming. However, let me know if I can help in any way and I'll do whatever I can.

vincenttan75, 2021-05-09:
changing the model info to _TZE200_tz32mtza", "TS0601 on tuyasingleswitch.py 
got it to scan in for 1 endpoint..

ShadowFistOkami, 2021-05-09:
> changing the model info to _TZE200_tz32mtza", "TS0601 on tuyasingleswitch.py
> got it to scan in for 1 endpoint..

By one endpoint, do you mean 1 switch entity?

The only difference I can see between tuyasingleswitch.py & ts0601.py is that the Time cluster is defined as an input in the former & an output in the latter

ShadowFistOkami, 2021-05-09:
Update. Tried out the 2 gang version and it's giving the same results - no entities. To make matters worse, I can't find the variant (_TZE200_g1ib5ldv) listed anywhere, not even in z2mqtt.

Signature & logs below:
`{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_g1ib5ldv",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}`

Logs (including some entries regarding my zigbee CCT bulbs):
`
Device 0x336b (60:a4:23:ff:fe:b1:52:41) joined the network
[0x336b] Requesting 'Node Descriptor'
Tries remaining: 2
[0x336b] Extending timeout for 0xcd request
Device 0x336b (60:a4:23:ff:fe:b1:52:41) joined the network
Skip initialization for existing device 60:a4:23:ff:fe:b1:52:41
Device 0x336b (60:a4:23:ff:fe:b1:52:41) joined the network
Skip initialization for existing device 60:a4:23:ff:fe:b1:52:41
[0x336b:zdo] ZDO request ZDOCmd.Device_annce: [0x336B, 60:a4:23:ff:fe:b1:52:41, 128]
[0x336b:zdo] ZDO request ZDOCmd.Node_Desc_req: [0x0000]
[0x336b:zdo] Unsupported ZDO request:ZDOCmd.Node_Desc_req
Received frame on uninitialized device 60:a4:23:ff:fe:b1:52:41 (Status.NEW) for endpoint: 1
[0x336b] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=0 command_id=Command.Read_Attributes>, payload: b'\x07\x00'
[0x336b] Uninitialized device command 'Command.Read_Attributes' args: [[7]]
[0xaedb:1:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=172 command_id=Command.Report_Attributes>
[0xaedb:1:0x0300] ZCL request 0x000a: [[Attribute(attrid=4, value=<TypeValue type=uint16_t, value=24701>)]]
[0xaedb:1:0x0300] Attribute report received: current_y=24701
Received frame on uninitialized device 60:a4:23:ff:fe:b1:52:41 (Status.NEW) for endpoint: 1
[0x336b] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=0 command_id=Command.Read_Attributes>, payload: b'\x07\x00'
[0x336b] Uninitialized device command 'Command.Read_Attributes' args: [[7]]
Received frame on uninitialized device 60:a4:23:ff:fe:b1:52:41 (Status.NEW) for endpoint: 1
[0x336b] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=0 command_id=Command.Read_Attributes>, payload: b'\x07\x00'
[0x336b] Uninitialized device command 'Command.Read_Attributes' args: [[7]]
Received frame on uninitialized device 60:a4:23:ff:fe:b1:52:41 (Status.NEW) for endpoint: 1
[0x336b] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=0 command_id=Command.Read_Attributes>, payload: b'\x07\x00'
[0x336b] Uninitialized device command 'Command.Read_Attributes' args: [[7]]
Received frame on uninitialized device 60:a4:23:ff:fe:b1:52:41 (Status.NEW) for endpoint: 1
[0x336b] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=0 command_id=Command.Read_Attributes>, payload: b'\x07\x00'
[0x336b] Uninitialized device command 'Command.Read_Attributes' args: [[7]]
[0x336b] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)
[0x336b] Discovering endpoints
Tries remaining: 3
[0x336b] Extending timeout for 0xd0 request
Received frame on uninitialized device 60:a4:23:ff:fe:b1:52:41 (Status.NEW) for endpoint: 1
[... cut ...]
[0x336b:1:0xef00] ZCL request 0x0011: b'\x00\x06@'
[0x336b:1:0xef00] No handler for cluster command 17
[0xd723:4:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=29 command_id=Command.Report_Attributes>
[0xd723:4:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.true>)]]
[0xd723:4:0x0006] Attribute report received: on_off=1

`

vincenttan75, 2021-05-10:
you need to add the model info in the quirk on the ts0601.py 

that quirk doesnt really "worked"  but at least 1 endpoint is scanned in.. because my switch is 3 gang.. 
for 2 or 3 gang.. it should need a new quirk..

ShadowFistOkami, 2021-05-11:
> you need to add the model info in the quirk on the ts0601.py
> 
> that quirk doesnt really "worked" but at least 1 endpoint is scanned in.. because my switch is 3 gang..
> for 2 or 3 gang.. it should need a new quirk..

Long shot, but try using https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts001x.py instead of singleswitch or ts0601. It's already set up for multiple endpoints. You just need to modify line 74 (for 2 gang model) or line 146 (3 gang model) with your model name.

If the quirk doesn't scan in, you might need to change any reference to input cluster in ts001x.py from 6 to 61184

AlterSb, 2021-05-12:
I got _TZE200_g1ib5ldv and tried to write simple quirk to detect multiple endpoint and it failed. Seem like it need a new custom cluster just like 'Moes' window cover switch. I have a look into it and it's way out of my capabilities (but will try to) since I'm new with both Quirk building and Python.

ShadowFistOkami, 2021-05-13:
> I got _TZE200_g1ib5ldv and tried to write simple quirk to detect multiple endpoint and it failed. Seem like it need a new custom cluster just like 'Moes' window cover switch. I have a look into it [...]

Refer to my comment in the post just above yours. The ts001x.py quirk **_should_** work with some slight modifications. I still need to set up my system in order to load custom quirks (the proxmox installation seems a bit daunting tbh). 
If you're all set up for custom quirks, then please follow my suggestion & post your results here.

vincenttan75, 2021-05-13:
> Refer to my comment in the post just above yours. The ts001x.py quirk **_should_** work with some slight modifications. I still need to set up my system in order to load custom quirks (the proxmox [...]

no it wont work.. the signature and endpoint will not match - that was the first week of testing that i tried.. 
none of the quirks work for TZE200 series.. because they are smart plugs.. in the tuya documentation - its already listed they only have 1 endpoint.. even though they are mutiple gang switches..  ts001x device ids are 0x0100 = switches.. it wont match.

anyways.. knowing the problem cannot be solved by simply creating replacement quirk.. i have given up on ZHA and switched to Z2M . Z2M released alpha stage support for sonoff zbridge - which is what i am using .. so all well ends well.. i am purchasing more tuya switches now.. and dropping ZHA - they need a more robust quirk for ts0601

vincenttan75, 2021-05-13:
https://www.digiblur.com/2021/03/zigbee2mqtt-with-sonoff-zigbee-bridge.html

for all who wants to use ts0601 multi gang switches .. and sonoff ZB bridges

ShadowFistOkami, 2021-05-14:
Unfortunately that's not an option for me. If I switch to Z2M I'll have the same problem with my power strip & filament lamps, because they don't seem to be supported.

My only way out (unless someone builds a ZHA quirk for these switches) is to get another ZB bridge & run 2 independent zigbee networks, which defeats the point of having a large zigbee mesh.

MattWestb, 2021-11-11:
Looks like one user have getting the multi gang switches working but need more testing for getting it verified and merged in the dev branch.
https://github.com/zigpy/zha-device-handlers/issues/1038#issuecomment-962601999

PR #1166 (2021-11-23): Add quirks for Tuya multigang switches
This PR add quirks for Tuya multigang switches.

It should correct the management of the On/Off cluster for the multiswitch devices that make use of the Tuya cluster (`0xEF00`) for command/status reporting.

The `MODELS_INFO` of those devices that have been reported or an issue has been reported have been added.
For those devices whose number of gangs is not clear, they have been defined as 1-gang devices until someone report correct info.

This could fix the following issues (only on/off switches):
fixes: #813
fixes: #824
fixes: #913
fixes: #972
fixes: # [... cut]
closed 2021-11-23 by dmulcahey