zigpy/zha-device-handlers #826: [Device Support Request] add TS0011 support Tuya no neutral switch

sylvaing, 2021-03-24:
**Is your feature request related to a problem? Please describe.**
This device is no neutral tuya switch. I can include it with ZHA but, when switch is pairing with ZHA, it turns off itself.
I found same problem on github with deconz, but it seems to be ok with z2M:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3693

perhaps this device need to use one quirks and not classic zigbee device...

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_9hpxg80k",
  "model": "TS0011",
  "class": "zigpy.device.Device"
}
```

**Additional context**
Add any other context or screenshots about the feature request here.

GooG2e, 2021-03-24:
Same problem here

sACLeFR, 2021-03-28:
Same issue. I bought 2 ts0012 for testing.
Switch is working well when not appaired.
Once appaired, it shut off after 2 minutes

kev-kclub, 2021-03-30:
Same problem with TS0013.  
Once paired turns itself off after 2 minutes.

sACLeFR, 2021-03-30:
Do you know switches with 2 buttons, no neutral needed, EU format compatible that I can buy safely for my Conbee II ?

HakunaMatmata, 2021-04-03:
Same problem here with TS0011, and 2* TS0012
I saw this conversation about device handler for smartthings, is it portable for a quirck for us?
https://community.smartthings.com/t/uk-light-switches-my-experiences-2020/196035/95

Seems to be a routine check that happens evert 2min to check brand of the device (that would differ).

HakunaMatmata, 2021-04-03:
Here is a post I found about this issue and a change made to fix it :
https://www.gitmemory.com/issue/zigpy/zha-device-handlers/535/715596185

I did not test it yet.

MattWestb, 2021-04-03:
The original post: https://github.com/zigpy/zha-device-handlers/issues/535

HakunaMatmata, 2021-04-03:
> The original post: #535

Thanks, in the mean time I tested it and it does not work, as stated in the complete original post.

HakunaMatmata, 2021-04-04:
Correction, I managed to make my TS0012 work with this quirck (based on post #535 + TS0042.py) : 

`class TuyaNoNeutralDoubleSwitch(TuyaSwitch):
    """Tuya double channel switch device."""

    signature = {
        # "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, 
		# maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, 
		# maximum_outgoing_transfer_size=82, descriptor_capability_field=0)"
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=0x0100, input_clusters=[0x0000,0x0004, 0x0005,0x0006], output_clusters=[0x000a, 0x0019]))
        # SizePrefixedSimpleDescriptor(endpoint=2, profile=260, device_type=0x0100, input_clusters=[0x0004, 0x0005,0x0006,], output_clusters=[])
        MODELS_INFO: [("_TZ3000_fvh3pjaz", "TS0012")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
					OnOff.cluster_id
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
					OnOff.cluster_id
                ],
                OUTPUT_CLUSTERS: [],
            }
        }
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
					OnOff.cluster_id
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
					OnOff.cluster_id
                ],
                OUTPUT_CLUSTERS: [],
            }
        }
    }

    device_automation_triggers = {
        (SHORT_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: SHORT_PRESS},
        (LONG_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: LONG_PRESS},
        (DOUBLE_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: DOUBLE_PRESS},
        (SHORT_PRESS, BUTTON_2): {ENDPOINT_ID: 2, COMMAND: SHORT_PRESS},
        (LONG_PRESS, BUTTON_2): {ENDPOINT_ID: 2, COMMAND: LONG_PRESS},
        (DOUBLE_PRESS, BUTTON_2): {ENDPOINT_ID: 2, COMMAND: DOUBLE_PRESS},
    }`

Seems that the OUTPUT_CLUSTER "Time.cluster_id" could be responsible for the problem.

sACLeFR, 2021-04-05:
> Correction, I managed to make my TS0012 work with this quirck (based on post #535 + TS0042.py) :

Hello, Is this correction specific to ZHA ? or can it be applied to Deconz directly ?

sACLeFR, 2021-04-05:
It seems Jeedom and Deconz will be too slow to solve the issue.
So I'm trying to migrate from Jeedom to Home Assistant. Installation is done, Zigbee working ... with of course still the issue of switch off after 2 minutes. How to I test the "quirck" below ? Thank you

HakunaMatmata, 2021-04-06:
You have to install the project zha-device-handlers (explanation on main page of the project)
then create a "ts0012.py" (below) file in the "/zhaquirks/tuya/" directory (or ts0011, if 1 switch version), 
then remove directories "__pycache__/" in zhaquirks/ and "/zhaquirks/tuya/" folders

Worked for me, then problem yesterday, repaired device and now ok since then.

file ts0012.py:
`"""Tuya 2 Button Switch."""

from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, Groups, Scenes, OnOff, Ota, PowerConfiguration, Time

from zhaquirks.const import (
    BUTTON_1,
    BUTTON_2,
    COMMAND,
    DEVICE_TYPE,
    DOUBLE_PRESS,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    LONG_PRESS,
	MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SHORT_PRESS,
)
from zhaquirks.tuya import (
	TuyaSwitch
)

class TuyaNoNeutralDoubleSwitch(TuyaSwitch):
    """Tuya double channel switch device."""

    signature = {
        # "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, 
		# maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, 
		# maximum_outgoing_transfer_size=82, descriptor_capability_field=0)"
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=0x0100, input_clusters=[0x0000,0x0004, 0x0005,0x0006], output_clusters=[0x000a, 0x0019]))
        # SizePrefixedSimpleDescriptor(endpoint=2, profile=260, device_type=0x0100, input_clusters=[0x0004, 0x0005,0x0006,], output_clusters=[])
        MODELS_INFO: [("_TZ3000_fvh3pjaz", "TS0012")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
					OnOff.cluster_id
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
					OnOff.cluster_id
                ],
                OUTPUT_CLUSTERS: [],
            }
        }
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
					OnOff.cluster_id
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
					OnOff.cluster_id
                ],
                OUTPUT_CLUSTERS: [],
            }
        }
    }

    device_automation_triggers = {
        (SHORT_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: SHORT_PRESS},
        (LONG_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: LONG_PRESS},
        (DOUBLE_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: DOUBLE_PRESS},
        (SHORT_PRESS, BUTTON_2): {ENDPOINT_ID: 2, COMMAND: SHORT_PRESS},
        (LONG_PRESS, BUTTON_2): {ENDPOINT_ID: 2, COMMAND: LONG_PRESS},
        (DOUBLE_PRESS, BUTTON_2): {ENDPOINT_ID: 2, COMMAND: DOUBLE_PRESS},
    }`

kev-kclub, 2021-04-06:
Many thanks for this work, I've been struggling to get to grips with it for a few days now.
FWIW I don't think you need the "device_automation_triggers" section for these switches.

Finally got my TS0013 to work and stay on with this...
file ts0013.py
========================================
"""Tuya 3 gang no neutral light switch."""

from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, Groups, Scenes, OnOff, Ota, Time

from zhaquirks.const import (
    COMMAND,
    DEVICE_TYPE,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SKIP_CONFIGURATION,
)

from zhaquirks.tuya import (
    TuyaSwitch,
)

class TuyaTripleNoNeutralSwitch(TuyaSwitch):
    """Tuya 3 gang no neutral light switch."""
    signature = {
        #"NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, 
        #manufacturer_code=4098, maximum_buffer_size=82, 
        #maximum_incoming_transfer_size=82, server_mask=11264, 
        #maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
        MODEL: "TS0013",
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=100
            # device_version=1
            # input_clusters=[0, 4, 5, 6]
            # output_clusters=[a, 19]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id, 
                    Ota.cluster_id
                ]
            },
            # <SimpleDescriptor endpoint=2 profile=260 device_type=100
            # device_version=1
            # input_clusters=[4, 5, 6]
            # output_clusters=[]>
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id
                ],
                OUTPUT_CLUSTERS: []
            },
            # <SimpleDescriptor endpoint=3 profile=260 device_type=100
            # device_version=1
            # input_clusters=[4, 5, 6]
            # output_clusters=[]>
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id
                ],
                OUTPUT_CLUSTERS: []
            }
        }
    }

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id]
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id
                ],
                OUTPUT_CLUSTERS: []
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id
                ],
                OUTPUT_CLUSTERS: []
            }
        }
    }
===========================

And sACLeFR if you look at https://github.com/zigpy/zha-device-handlers it will tell you how to implement quirks in homeassistant.

sACLeFR, 2021-04-06:
> You have to install the project zha-device-handlers (explanation on main page of the project)
> then create a "ts0012.py" (below) file in the "/zhaquirks/tuya/" directory (or ts0011, if 1 switch [...]

Thank you, will test that this week !

sACLeFR, 2021-04-06:
> And sACLeFR if you look at https://github.com/zigpy/zha-device-handlers it will tell you how to implement quirks in homeassistant.

Thank you. I think I understood how to implement a quirk once zha-device-handlers is installed ... but the documentation don't explain (or I didn't understood) how to install zha-device-handlers itself 
Is it just the folders to copy/paste ?

dmulcahey, 2021-04-06:
If you’re using HA it’s pulled in by ZHA already

sylvaing, 2021-04-12:
This week-end had test some different quirk.
I attache a file which contain 3 Quirks for : TS0011, TS0012, TS0013.
It seems work for me for TS0011, but I will test to delete and re appair the device tonight.

Can you test with this file, before propose pull-request ?

```
"""Tuya 1,2,3 gang no neutral light switch."""

from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, Groups, Scenes, OnOff, Ota, Time

from zhaquirks.const import (
    COMMAND,
    DEVICE_TYPE,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SKIP_CONFIGURATION,
)

from zhaquirks.tuya import (
    TuyaSwitch,
)

class TuyaSingleNoNeutralSwitch(TuyaSwitch):
    
    """Tuya 1 gang no neutral light switch."""
    signature = {
        #"node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, 
        #                       maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264,
        #                       maximum_outgoing_transfer_size=82, descriptor_capability_field=0)

        MODELS_INFO: [("_TZ3000_9hpxg80k", "TS0011")],
        ENDPOINTS: {
        # <SimpleDescriptor endpoint=1 profile=260 device_type=100
        # device_version=1
        # input_clusters=[0, 4, 5, 6]
        # output_clusters=[a, 19]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            },
         },
    }

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
[... 152 lines cut ...]
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }

```

sylvaing, 2021-04-13:
I confirm, it works for me with TS0011 !

sACLeFR, 2021-04-13:
I tried to include it in Jeedom Zigbee plugin (based on zigpy) but no success.
I was hoping some magic maybe but we have there a folder resources/zigbeed/quirks/

In ZHA I didn't find where to load the file py

sylvaing, 2021-04-14:
I don't know integration of zigpy on jeedom..
For HA, what is your type of installation : supervised , OS ?

kev-kclub, 2021-04-14:
Now I'm more confused... 
Upgraded HA to 2021.4.0 and, as expected, the custom quirk was removed. BUT the TS0013 still works and does not turn itself off.  So, on the basis that if it ain't bust don't fix it, I'm leaving it without the quirk for now.

sylvaing, 2021-04-14:
@kev-kclub  ,
Yes wait in order to see,
But i'm curious if you delete your TS0013, and try to reinclude without the quirk.

sylvaing, 2021-04-15:
@sACLeFR can you test this PR ? https://github.com/zigpy/zha-device-handlers/pull/845

Say me what is your installation type: OS or supervised ?

sACLeFR, 2021-04-15:
> @sACLeFR can you test this PR ? https://github.com/zigpy/zha-device-handlers/pull/845
> 
> Say me what is your installation type: OS or supervised ?
> 

i'm on OS installation.

(I had to check because I used an installation script specific to create a vm on Proxmox)

Do you have a step by step how to activate quirks and where to save the py file ?
I admit I got lost from my readings.

It's a test vm so no problem to do activate some beta tools if needed.

Thanks.

sylvaing, 2021-04-15:
@sACLeFR 

ok step by step:

- download this file:
https://github.com/sylvaing/zha-device-handlers/blob/d2923252181b630520ede941917d75c80961829a/zhaquirks/tuya/ts001x.py
- connect to homehassistant web interface and upload file with file editor:
	- browse direcory with top left icon of direcory
	- click on icon to upload ( the arrow)

- log as root on your proxmox VM
- connect to your docker homeassistant container with: `docker exec -it homeassistant /bin/bash`
- copy ts001x.py file to the correct direcory:
`cp /config/ts001x.py /usr/local/lib/python3.8/site-packages/zhaquirks/tuya/`
- rm cache directory:
`rm -rf /usr/local/lib/python3.8/site-packages/zhaquirks/tuya/__pycache__/ /usr/local/lib/python3.8/site-packages/zhaquirks/__pycache__/`
- you can now restart home assitant from web interfaces and test your device.

sACLeFR, 2021-04-15:
> * connect to your docker homeassistant container with: `docker exec -it homeassistant /bin/bash`

I'm locked at this stage

[image]

MattWestb, 2021-04-15:
docker exec --help

Usage:  docker exec [OPTIONS] CONTAINER COMMAND [ARG...]

Run a command in a running container

Options:
  -d, --detach               Detached mode: run command in the background
      --detach-keys string   Override the key sequence for detaching a
                             container
  -e, --env list             Set environment variables
      --env-file list        Read in a file of environment variables
  -i, --interactive          Keep STDIN open even if not attached
      --privileged           Give extended privileges to the command
  -t, --tty                  Allocate a pseudo-TTY
  -u, --user string          Username or UID (format:
                             <name|uid>[:<group|gid>])
  -w, --workdir string       Working directory inside the container

sACLeFR, 2021-04-15:
I tried -i alone, -t alone
and "docker exec homeassistant /bin/bash"

sylvaing, 2021-04-17:
Ok, this step are for supervised installation, you must have os installation.
Si try to install portainer addon on ha and connect you to your container console with portainer

semajorlt, 2021-04-27:
Thanks a lot!  It's working properly!!

sylvaing, 2021-05-11:
release in 0.57 so we can close the issue.

closed 2021-05-11 by sylvaing

Smanar, 2021-07-08:
Hello, someone can explain me what do the quirk (from zigbee side)
I don't understand how hiding the ouput time cluster, can change something on the normal working mode ?

MattWestb, 2021-07-08:
I think (without knowing) that the switch dont like the time its getting from the coordinator and therefor starting doing strange things.
The ZHA users have only disabling the time cluster (by exuding the cluster in the replacement part of the quirk) so ZHA is not using  it = can not answer the time requests from the device so it not being so "confused" and continuing working as one "normal" Zigbee switch.

The quirk code: https://github.com/weddle/zha-device-handlers/commit/ccdc7ade86300cc863c2a6bde03c0d65024201cf

Smanar, 2021-07-08:
But it don't prevent the device trying to read the attribute on the gateway (or I don't know how, hiding the cluster on the lib have no impact on the device working mode, no ?)
I m trying to do same, But I do that on gateway side, I don't answer when the device ask for local time, and I m unsuccessfull.

MattWestb, 2021-07-08:
The ZHA Quirk is only preventing  ZHA using the time cluster so it can using it, the device still can reading the endpoints and cluster of the coordinator and trying using then but ZHA  is ignoring all that  have with the time cluster form this devices.

I think its on safety function then sending timed commands to the device but its one "normal zigbee device" so no tuya commands is sent (wot i have seen but you is never knowing with some manufactures).

Look on it as one LIDL weather valve that is turning it self off if not getting any more commands for safety but without the tuya DPs.

By the way gratings to @Mimiix and hope hi have not blocking all the rest nice de(F)CONNZ users there ;-))

pipiche38, 2021-07-08:
> Hello, someone can explain me what do the quirk (from zigbee side)
> I don't understand how hiding the ouput time cluster, can change something on the normal working mode ?
I'm not a zigpy specialist, but I beleive that if the Time cluster is hidden, then when the zigpy library will get a request from the device on that cluster , it will simply respond "Not Supported"

Smanar, 2021-07-08:
Ha yep right, for the moment I just ignore it, will try with a "not supported"

Smanar, 2021-07-08:
>The ZHA Quirk is only preventing ZHA using the time cluster so it can using it, the device still can reading the endpoints and cluster of the coordinator and trying using then but ZHA is ignoring all that have with the time cluster form this devices.

But I m almost sure ZHA never use this cluster, it s only used by somes devices themself, so if ZHA doesn't do "magic" on them and let the device reading the attribute, the quirk change nothing for me.

>I think its on safety function then sending timed commands to the device but its one "normal zigbee device" so no tuya commands is sent (wot i have seen but you is never knowing with some manufactures).

The problem is the orignal gateway send less request than the generic one ^^, And like you saif nothing special, only classic request

>By the way gratings to @Mimiix and hope hi have not blocking all the rest nice de(F)CONNZ users there ;-))

Lol, no don't worry, zigbee community  is  a small family

Adminiuga, 2021-07-08:
Zigpy will respond to read attribute request on the time cluster. It works fine with compliant devices, at least centralite thermostat gets the time from the server correctly.
 But because tuya is its own thing, tuya has own commands for setting the time. I'm pretty sure same issues were faced by  Moe's radiator valve quirk author.

Smanar, 2021-07-08:
>Zigpy will respond to read attribute request on the time cluster.

But even with the "quirk" ?

MattWestb, 2021-07-08:
I think the tuya TRVs is setting the time with tuya (DP) commands and different way counting the time for different versions (ala tuya).

Many tuya lights also having the time cluster and that is not needed for "normal zigbee devices" like the LIDL LED strip and is working OK with "standard setting" without quirks with both RGB and TC with normal commands.

Smanar, 2021-07-08:
>I think the tuya TRVs is setting the time with tuya (DP) commands and different way counting the time for different versions (ala tuya).

Yep, I have that for lot of TRV, and 2 different epoch used.

Smanar, 2021-07-13:
Ok so for information, playing with time cluster have worked.
But just with NO configure reporting AT ALL solve the issue, the device still working, but without the auto power off (still on bench)

MattWestb, 2021-07-13:
Its one real nasty tuya implementation. Way have tuya not implanting it with `on with time off` and using it as manufacture command and have doing the off time in minutes instead of seconds.

One more problem you was having with devices 0x017 (old Philips HUE devices) is **not** aging out neighbors so if removing one device the HUEs  is still reporting they is still in the network to the coordinator until you is resetting the HUE device (i have removed the last HUE light from my production system and putting in IKEA CWS3).

And one more tuya is the TS004F is one more or less one Zigbee dimmer switch that is sending very normal On/Off and Level commands ala Zigbee Light standard but the user is saying its one 12 scene switch that is wrong.

I think you is needing one [TYGWZ01](https://github.com/zigpy/zigpy/discussions/650) or 2 for testing plus all devices tuuya is putting out in the market.

JoryHogeveen, 2022-07-14:
Related to this issue I started the following topic on the community: https://community.home-assistant.io/t/user-experience-request-differernt-zigbee-tuya-based-touch-light-switches/439249
Any contributions to experience with these devices are very welcome!