zigpy/zha-device-handlers #829: [BUG] Eurotronic Spirit Zigbee (SPZB0001) does not reconnect after HAss restart

SantosSi, 2021-03-28:
**Describe the bug**
Controlled by a ConBee II over USB.
After restarting always neither battery nor heating functionality available. The device does not display the symbol for an active connection. Removing and reinserting the device's battery, displays the symbol, but it is still not connected in HAss, neither battery not heating.
Other devices controlled by the ConBee II reconnect fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Restart Home Assistant
2. Look at the device status

**Expected behavior**
After a restart, the device should be able to get a temperature set and reports the currently measured temperature and battery level.

**Additional context**
hvac_modes:
  - 'off'
min_temp: 7
max_temp: 35
current_temperature: null
temperature: null
hvac_action: 'off'
friendly_name: Eurotronic Spirit Zigbee (SPZB0001)
supported_features: 1

Mär 28 10:05:36 systemname docker-compose[3671]: home-assistant   | 2021-03-28 10:05:36 WARNING (MainThread) [homeassistant.components.zha.core.channels.base] [0x0DB9:1:0x0201]: async_initialize_channel_specific: all attempts have failed: [DeliveryError('[0x0db9:1:0x0201]: Message send failure'), DeliveryError('[0x0db9:1:0x0201]: Message send failure'), DeliveryError('[0x0db9:1:0x0201]: Message send failure'), DeliveryError('[0x0db9:1:0x0201]: Message send failure')]
Mär 28 10:10:50 systemname docker-compose[3671]: home-assistant   | 2021-03-28 10:10:50 WARNING (MainThread) [homeassistant.components.zha.core.channels.base] [0x0DB9:1:0x0201]: async_initialize_channel_specific: all attempts have failed: [DeliveryError('[0x0db9:1:0x0201]: Message send failure'), DeliveryError('[0x0db9:1:0x0201]: Message send failure'), DeliveryError('[0x0db9:1:0x0201]: Message send failure'), DeliveryError('[0x0db9:1:0x0201]: Message send failure')]
Mär 28 10:11:32 systemname docker-compose[3671]: home-assistant   | 2021-03-28 10:11:32 WARNING (MainThread) [homeassistant.components.zha.core.channels.base] [0x0DB9:1:0x0201]: async_initialize_channel_specific: all attempts have failed: [DeliveryError('[0x0db9:1:0x0201]: Message send failure'), DeliveryError('[0x0db9:1:0x0201]: Message send failure'), DeliveryError('[0x0db9:1:0x0201]: Message send failure'), DeliveryError('[0x0db9:1:0x0201]: Message send failure')]
Mär 28 10:12:30 systemname docker-compose[3671]: home-assistant   | 2021-03-28 10:12:30 WARNING (MainThread) [zigpy_deconz.zigbee.application] No watchdog response
nThread) [homeassistant.components.zha.core.channels.base] [0x0DB9:1:0x0201]: async_initialize_channel_specific: all attempts have failed: [DeliveryError('[0x0db9:1:0x0201]: Message send failure'), DeliveryError('[0x0db9:1:0x0201]: Message send failure'), DeliveryError('[0x0db9:1:0x0201]: Message send failure'), DeliveryError('[0x0db9:1:0x0201]: Message send failure')]
Mär 28 10:12:30 systemname docker-compose[3671]: home-assistant   | 2021-03-28 10:12:30 WARNING (MainThread) [zigpy_deconz.zigbee.application] No watchdog response

MattWestb, 2021-03-28:
I think this (i dont have the) devices dont like having the coordinator as its parent.
Do you have some routers around that it can connecting 2 and is not leaving the network if the coordinator is not online ?

SantosSi, 2021-03-28:
Can't HAss' ZHA resolve the problem? (To be honest, this is why I opened the bug report here ;-) ) Since I am not the only one to have the problem it might be worth to develop a device specific behavior or so.
Unfortunately the only other device I have is an IKEA Tradfri gateway. Unsuccessfully tried to connect to the device.

SantosSi, 2021-04-06:
Is there anything else I can do to help sort this out? A log file, an action, a test, ...?

Adminiuga, 2021-04-06:
It is not ZHA's problem. It is ConBee non Zigbee compliant behavior (not enough information based on the logs) where it does not keep the track of it's children.

Easiest way to fix/confirm, add another Zigbee router and pair the thermostat throu it.

Alternatively:
1. Update ConBees firmware
2 remove device from zha
3. Power cycle conbee
4. Factory reset thermostat -- it is important, because it has to request the network key
5. Pair it again
6. Don't restart HA -- important for the next three hours.
7. Check visualization map and confirm that thermostat has a link to the parent device

WolfRevo, 2021-09-27:
Honestly I think this is more likely an [issue](https://github.com/EUROTRONIC-Technology/Spirit-ZigBee/issues/1) of the thermostat itself. I had 8 of them until two weeks ago and I am happy I moved away since EUROTRONIC doesn't seem to work on these issues known to them since the beginning of this year.

This reconnection issue is really the most annoying part but not the only "unlucky behaviour".

closed 2022-04-02 by github-actions[bot]