zigpy/zha-device-handlers #830: Since 0.0.55 aqara motion sensors and switches do no work anymore

mbo18, 2021-03-30:
I've updated all zigpy related packages on my system and since then, aqara/xiaomi motion sensors (and aqara switches) do not report any value anymore for motion and illuminance.
To find out were is the bug, I've reverted to zha-quirks==0.0.54 and now everything works fine again.

So there is something odd in the last commits.

Adminiuga, 2021-03-30:
Standalone installs ~~arent~~ supported on per YMMV basis. 
You do need to update HA to dev tree too.
Also, reports without providing the details noile zigbee device signature, debug logs etc, are just going to be ignored.

zoic21, 2021-03-31:
Hello,
It's not use with HA but with home automation system. There is multiple report of this issue, maybe it's from this commit https://github.com/zigpy/zha-device-handlers/commit/e47fccfb821a77f9ccfe46e3f2c0c46ba92ea55e. I will make some test and try to add log on this issue.

dmulcahey, 2021-03-31:
If you aren’t using this with HA you now need to import and call the setup method from the main module to load quirks

dmulcahey, 2021-03-31:
What are you using this with... for my own curiosity 😃

mbo18, 2021-03-31:
Jeedom 😉
https://www.jeedom.com

zoic21, 2021-03-31:
> If you aren’t using this with HA you now need to import and call the setup method from the main module to load quirks
> 
> https://github.com/zigpy/zha-device-handlers/blob/2f9b0c33e08dfbddb703e44ce017e0a0a9315562/zhaquirks/__init__.py#L385

Perfect it was that I miss it.

Very good idea to add a `CUSTOM_QUIRKS_PATH` it will be very usefull to test quirks before PR.

mbo18, 2021-03-31:
Thank you! I will now close this issue

closed 2021-03-31 by mbo18

MattWestb, 2021-03-31:
@dmulcahey Look on https://jeedom.com  they have it in English 2 but no Swedish ;-(