zigpy/zha-device-handlers #831: [BUG] Tuya/MEOS Thermostat - the time gets unsynchronized making built-in schedules unusable.

sstefanowski, 2021-03-30:
The problem:
The Tuya/MEOS Thermostat support built-in heating schedules (and change temperature up to 6 times a day). 
Unfortunately, the clock inside the device is not very precise and the time gets unsynchronized after a few days. 
ZHA should take care of the device time and re-synchronize it with some default frequency (e.g. once a day).
The device should always receive a local date/time from ZHA. This also should be taken into consideration when summer/winter time changes

Similar Zigbee solutions take care of the device clock e.g. Tuya platform and Zigbee2MQT.\
The solutions for the same issue in Zigbee2MQTT:
https://github.com/Koenkk/zigbee2mqtt/issues/5916
https://github.com/Koenkk/zigbee2mqtt/issues/5620

**To Reproduce**
Steps to reproduce the behaviour:
1. Set a correct time on the Tuya/MEOS Thermostat device manually
2. Leave it for a few days
3. Check the time on the device after a few days
4. The time on the device is no longer correct

**Expected behaviour**
When checking the time on the device it should be always correct. ZHA should take care of date/time settings and re-synchronize them periodically as the other Zigee platforms do.
The device should always receive a local date/time from ZHA. This also should be taken into consideration when summer/winter time changes

**Screenshots**
N/A

**Additional context**
N/A

Adminiuga, 2021-03-30:
Tuya should have just sticked to the standards.  Nominally nthe device is a zigbee device, but really is using a proprietary protocol.

jmoransalama, 2021-03-31:
@sstefanowski What thermostat model do you have? I have a _TZE200_aoclfnxz thermostat and I'm having the same problem.

I'm not sure, but I believe that the command to set the time from ZHA to the device is [already in the code](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/__init__.py#L113), but it's only applied to some thermostatic valves and not to all tuya thermostats.

jmoransalama, 2021-03-31:
I was looking at the code and I think I've fixed it for my device. I've just added this line of code:
`set_time_offset = 1970`
in [this line](https://github.com/zigpy/zha-device-handlers/blob/2f9b0c33e08dfbddb703e44ce017e0a0a9315562/zhaquirks/tuya/electric_heating.py#L87) 

So now the code looks like this:
```
class MoesBHTThermostat(TuyaThermostatCluster):
    """Thermostat cluster for some electric heating controllers."""

    set_time_offset = 1970

    def map_attribute(self, attribute, value):
        """Map standardized attribute value to dict of manufacturer values."""
```

I waited a few minutes until the thermostat asked for a time sync to the server:
`2021-04-01 00:30:35 DEBUG (MainThread) [zhaquirks.tuya] [0xd829:1:0xef00] Got set time request (command 0x0024)`

After that, I checked the thermostat and the time was correct. I'll keep an eye on it for the next few days to see if the problem was solved.

sstefanowski, 2021-03-31:
OK. I can see the code and it looks like this handler should be avaialble. 
The problem is I cannot see any debugs with time requests in a log like "...Got set time request..."
I enabled debug in configuration.yaml yesterday 

```
logger:
  default: info
  logs:
    zhaquirks.tuya: debug
```

and NOT a single log entry with `[zhaquirks.tuya] [0xd829:1:0xef00] Got set time request (command 0x0024)` was generated.
I can see just updated of temerature readings generated by` [zhaquirks.tuya]`

but....
I can see periodic ERRORS in a log as follows:
```
Mar 31 23:52:23 raspberrypi hass[19459]: 2021-03-31 23:52:23 ERROR (MainThread) [zigpy.device] Failed to parse message (b'191c24004b') on cluster 61184, because Data is too short to contain 1 bytes
Mar 31 23:52:50 raspberrypi hass[19459]: 2021-03-31 23:52:50 ERROR (MainThread) [zigpy.device] Failed to parse message (b'195324004c') on cluster 61184, because Data is too short to contain 1 bytes
Mar 31 23:52:54 raspberrypi hass[19459]: 2021-03-31 23:52:54 ERROR (MainThread) [zigpy.device] Failed to parse message (b'1913240058') on cluster 61184, because Data is too short to contain 1 bytes
(...)
Mar 31 23:53:57 raspberrypi hass[19459]: 2021-03-31 23:53:57 ERROR (MainThread) [zigpy.device] Failed to parse message (b'19562403ea') on cluster 61184, because Data is too short to contain 1 bytes
(...)
Apr  1 00:52:49 raspberrypi hass[19459]: 2021-04-01 00:52:49 ERROR (MainThread) [zigpy.device] Failed to parse message (b'192124004e') on cluster 61184, because Data is too short to contain 1 bytes
Apr  1 00:53:21 raspberrypi hass[19459]: 2021-04-01 00:53:21 ERROR (MainThread) [zigpy.device] Failed to parse message (b'195b240050') on cluster 61184, because Data is too short to contain 1 bytes
Apr  1 00:53:26 raspberrypi hass[19459]: 2021-04-01 00:53:26 ERROR (MainThread) [zigpy.device] Failed to parse message (b'1926240062') on cluster 61184, because Data is too short to contain 1 bytes
```
and `61184`=`0xef00` (MeosManufCluster)
and all the messages listed contain static `(b'19??24????')` while `0x0024 `is `setTimeRequest `command in this cluster.

Can these `ERRORS` be related to missing time sync?

-----
BTW. My Thermostat is heater valve `_TZE200_ywdxldoj`

sstefanowski, 2021-04-01:
Update:

I've changed a line
https://github.com/zigpy/zha-device-handlers/blob/2f9b0c33e08dfbddb703e44ce017e0a0a9315562/zhaquirks/tuya/__init__.py#L135

to 

`        0x0024: ("set_time_request", (t.NoData,), True),`

and now...
 I can see in the log
`Apr 01 11:00:37 raspberrypi hass[21528]: 2021-04-01 11:00:37 DEBUG (MainThread) [zhaquirks.tuya] [0xd17c:1:0xef00] Got set time request (command 0x0024)`
and one of my thermostats has valid time. I'll recheck remaining devices after some time.

Question to @xonestonex who is the main contributor to the ZHA quirk for Moes Thermostatic valve.
What do you think. Is this a proper way of fixing this?

almico, 2021-04-12:
I too have the `_TZE200_ywdxldoj` and would really love to see this fix applied!
Please, please, please 😄

almico, 2021-04-13:
I can confirm that changing line 135 to `0x0024: ("set_time_request", (t.data16,), True),` worked here too.
Thank you!

almico, 2021-04-13:
I just received a new HY368 and its name is `_TZE200_cwnjrr72`...

MattWestb, 2021-04-13:
I was adding it in https://github.com/zigpy/zha-device-handlers/pull/820 if the user putting  all info OK it shall working out of the box.

sstefanowski, 2021-04-13:
Hi @MattWestb,
I wonder.... If you have a source code already branched maybe you could update line 135 above to fix the time synchronization and create pull-request so this could to be analyzed by main ZHA contributors?

sstefanowski, 2021-04-14:
Hi @MattWestb,

Thans for update but... 

I can see you commited:
`0x0024: ("set_time_request", (t.NoData,), True),`
while this setting generates a WARN in the log about unconsumed payload in a buffer. `t.NoData` means that it's not expecting any data but there are unconsumed bytes remaining (I guess 2 bytes) in the request.

Changing it to 
`("set_time_request", (t.data16,), True),`
makes everything working without a WARN - remaining last bytes are consumed as expected but just not "interpretted". 

Also, I don't meaning of these bytes but the previous entry using TuyaTimePayload for set_time_request was a real bug.

sstefanowski, 2021-04-14:
OK, I'll comment on Pull Request
I can justify the commit

MattWestb, 2021-04-14:
Updated PR

sstefanowski, 2021-04-14:
@MattWestb,
Stupid question, how to install this commit? Do you mean just relacing the local file with the file from your repo?

MattWestb, 2021-04-14:
Download the commit as zip file or copy only the file changed and putting it in your system so we can see its working.

sstefanowski, 2021-05-07:
This is in release 0,57 so we can close the issue...

closed 2021-05-07 by sstefanowski

dariox64, 2022-03-08:
I have the same problem as the author of the issue.

I don't know if I did everything right.
- I installed hassio-addons-development and pointed to the latest version of zha-quirks == 0.0.66

a folder appeared in hasoss with the content at the address
./config/deps/lib/python3.9/site-packages/zhaquirks/

Unfortunately, each of the 4 MOES thermostats shows a different time ;) After a correct setting on the device, the clocks live their lives again after a few hours.

TS0601
manufacturer: _TZE200_aoclfnxz
Connected by Texas Instruments CC1352/CC2652, Z-Stack 3.30+ (build 20220103) 
Quirk: zhaquirks.tuya.electric_heating.MoesBHT 

Can anyone help me?