zigpy/zha-device-handlers #833: Tuya Cover - Using wrong logic for reporting "Current Position Lift Percentage"

axellebot, 2021-04-05:
Hello !

As we noticed it in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3757, some Tuya cover model are wrongly reporting the "lift percentage".

Indeed in HomeAssistant, while my cover is fully opened it does report a `0` value as `current_position`.

<details>
  <summary>1st TS130F "Zigbee Signature" JSON </summary>

Bought <a href="https://www.aliexpress.com/item/Zigbee-Curtain-Switch-SmartThing-Tuya-Zigbee-Hub-Control-For-Electric-Curtain-Motorized-Roller-Shade/33052990003.html?spm=a2g0s.9042311.0.0.6e8c6c37RRziEq">here</a> on Aliexpress

```
{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0102"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_ltiqubue",
  "model": "TS130F",
  "class": "zigpy.device.Device"
}
```
</details>

<details>
  <summary>2nd TS130F "Zigbee Signature" JSON </summary>

Bought <a href="https://www.aliexpress.com/item/Tuya-Smart-Life-ZigBee-3-0-Curtain-Blind-Switch-for-Roller-Shutter-Electric-motor-Google-Home/4001190275372.html?spm=a2g0s.9042311.0.0.6e8c6c37RRziEq">here</a> on Aliexpress

```
{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x000a",
        "0x0102"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_8kzqqzu4",
  "model": "TS130F",
  "class": "zigpy.device.Device"
}
```
</details>

deCONZ dev apparently did a trick to reverse the percentage value and it does the job, what should be done to properly implement kind of the same trick here ?

_From comment by @axellebot in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3757#issuecomment-812889442_

axellebot, 2021-05-28:
Hey @eyoz !
It looks like I had an update because 1 of the 2 model I refered is now working 👌

I figured that you only added 1 sub-model `_TZ3000_8kzqqzu4` in #836. I will try to create a new PR to add the other sub-model `_TZ3000_ltiqubue` after testing it unless there is a way to create a quirk to contain all models ?

axellebot, 2021-05-28:
As mentionned https://github.com/zigpy/zha-device-handlers/pull/836#issuecomment-830063498 @miit86 also need other sub-model :
- _TZ3000_vd43bbfq
- _TZ3000_egq7y6pr

axellebot, 2021-05-28:
Is `MODEL` attribute instead of `MODEL_INFO` can do the job as it's done here : https://github.com/zigpy/zha-device-handlers/blob/470ccf9faa7f76e3718a619c8114305f42acacaa/zhaquirks/tuya/ts0043.py#L33 ?

MattWestb, 2021-05-28:
Try making it like this:
```
"""Device handler for loratap TS130F smart curtain switch."""
from zigpy.profiles import zha
from zigpy.quirks import CustomCluster, CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.closures import WindowCovering
from zigpy.zcl.clusters.general import Basic, Groups, OnOff, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

ATTR_CURRENT_POSITION_LIFT_PERCENTAGE = 0x0008
CMD_GO_TO_LIFT_PERCENTAGE = 0x0005

class TuyaWithBacklightOnOffCluster(CustomCluster):
    """TuyaSmartCurtainOnOffCluster: fire events corresponding to press type."""

    cluster_id = OnOff.cluster_id

    LIGHT_MODE_1 = {0x8001: 0}
    LIGHT_MODE_2 = {0x8001: 1}
    LIGHT_MODE_3 = {0x8001: 2}

    attributes = {0x8001: ("backlight_mode", t.enum8)}

class TuyaCoveringCluster(CustomCluster, WindowCovering):
    """TuyaSmartCurtainWindowCoveringCluster: Allow to setup Window covering tuya devices."""

    attributes = WindowCovering.attributes.copy()
    attributes.update({0xF000: ("tuya_moving_state", t.enum8)})
    attributes.update({0xF001: ("calibration", t.enum8)})
    attributes.update({0xF002: ("motor_reversal", t.enum8)})

    def _update_attribute(self, attrid, value):
        if attrid == ATTR_CURRENT_POSITION_LIFT_PERCENTAGE:
            # Invert the percentage value (cf https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3757)
            value = 100 - value
        super()._update_attribute(attrid, value)

    async def command(
        self, command_id, *args, manufacturer=None, expect_reply=True, tsn=None
    ):
        """Override default command to invert percent lift value."""
        if command_id == CMD_GO_TO_LIFT_PERCENTAGE:
            percent = args[0]
            # Invert the percentage value (cf https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3757)
            percent = 100 - percent
            v = (percent,)
            return await super().command(command_id, *v)
        return await super().command(
            command_id,
            *args,
            manufacturer=manufacturer,
            expect_reply=expect_reply,
            tsn=tsn
        )

class TuyaTS130F(CustomDevice):
    """Tuya smart curtain roller shutter."""

    signature = {
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=0x0202, device_version=1, input_clusters=[0, 4, 5, 6, 10, 0x0102], output_clusters=[25]))
        MODELS_INFO: [
            ("_TZ3000_8kzqqzu4", "TS130F"),
            ("_TZ3000_ltiqubue", "TS130F"),
            ("_TZ3000_vd43bbfq", "TS130F"),
            ("_TZ3000_egq7y6pr", "TS130F"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    Time.cluster_id,
                    OnOff.cluster_id,
                    WindowCovering.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    Time.cluster_id,
                    TuyaWithBacklightOnOffCluster,
                    TuyaCoveringCluster,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
        },
    }
```
I have only adding the IDs and broken them so black is happy.

If the IDs is OK its loading the quirk and you can looking if all functions is working OK.

MattWestb, 2021-05-28:
If you is not getting any quirk loading then its the device signature not the same.
```
    signature = {
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=0x0202, device_version=1, input_clusters=[0, 4, 5, 6, 10, 0x0102], output_clusters=[25]))
```
Look with your device and comparing it and if you is finding  some difference adjusting it until ZHA is triggering the quirk for your device.  
(or i have copy and past wrong device ID that also can making the quirk not loading)

If the quirk is loaded but not working then you need rewriting the quirk for your device.

closed 2021-12-01 by github-actions[bot]