zigpy/zha-device-handlers #838: [Device Support Request] CentraLite / Xfinity keypad

BusinessClaes, 2021-04-08:
**Is your feature request related to a problem? Please describe.**
I have a CentraLite 3400 and a [CentraLite 3400-G](https://zigbeealliance.org/zigbee_products/3-series-security-keypad) keypad. Both devices get discovered in ZHA but many things don't work after that.
* No zha_event (either when dialing arm/disarm code, activating proximity sensor, activating tamper switch or press&holding panic keys)
* No succeed in sending commands to the keypad (I manage to call service zha.issue_zigbee_cluster_command in node-red without error messages, but the keypad doesn't respond, I gess I am not sending the right args)
* Battery entity doesn't get any readings
* Binary sensor IAS_zone is always off (not sure what this sensor is supposed to represent)
* Temperature entity seems to work well

**Describe the solution you'd like**
Port the functionality that is [implemented in Zigbee2mqtt](https://www.zigbee2mqtt.io/devices/3400-D.html) and described in [this reddit post](https://www.reddit.com/r/homeassistant/comments/k1hg4w/physical_zigbee_alarm_keypad_integrated_into_home) so the CentraLite 3400-series and Xfinity XHK1 keypads can be used with zha.
* Get zha_events with the same messages that Zigbee2mqtt manages to capture
* Be able to send command to the keypad to start "beeping countdown to disarm on home entrance" 
* Be able to send command to the keypad to start  "beeping countdown to leave home before arming"
* Be able to send acknowledge message back to the keypad after recieving dialed arm/disarm codes
* Be able to send command to change panel status in case of arming/disarming directly in HA

**Device signature**
`{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4174, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=0, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0401",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0402",
        "0x0500",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0019",
        "0x0501"
      ]
    }
  },
  "manufacturer": "CentraLite",
  "model": "3400-G",
  "class": "zigpy.device.Device"
}`

**Additional context**
When I activate T2Z on the tasmotized Sonoff ZBBridge (changing to module 75 instead of module 0, which breaks communication between zha and zigbee radio, but activates tasmotas T2Z) tasmota starts to post beautiful mqtt messages to my broker
* Arm/disarm codes
* Battery levels
* Proximity sensor
* Tamper switch
* Panic mode (press&hold the two panic keys)

Clusters described in [zigbee alliances Compliance Document](https://api.knack.com/v1/applications/54e658034b4f44e42fb18201/download/asset/5834781858cd46a32d9ed6e7/centralite3400picsinxml.xml) matches the clusters in zha's device signature. But there are mismatches when coming down to attributes/commands. As an example, bypass_response (id: 0x0007) that appears in zha as an available command for cluster 0x0501 in not described in the Compliance document.

dmulcahey, 2021-04-10:
I am working on support for these in ZHA.

BusinessClaes, 2021-04-10:
@dmulcahey that sounds great! Please tell me if I can help in any way, testing the quirk or in any other way.
After posting the issue I continued testing and I got this working with `zha.issue_zigbee_cluster_command`
* cluster 0x0501 IAS ACE, command 0x00 ArmResponse, payload [0] --> Beep once hi-pitch (to indicate correct code)
* cluster 0x0501 IAS ACE, command 0x00 ArmResponse, payload [4] --> Beep once lo-pitch (to indicate wrong code)
* cluster 0x0501 IAS ACE, command 0x04 PanelStatusChanged, payload [0,0,0,0] --> Disarmed
* cluster 0x0501 IAS ACE, command 0x04 PanelStatusChanged, payload [1,0,0,0] --> Armed stay
* cluster 0x0501 IAS ACE, command 0x04 PanelStatusChanged, payload [2,0,0,0] --> Armed night
* cluster 0x0501 IAS ACE, command 0x04 PanelStatusChanged, payload [3,0,0,0] --> Armed away
* cluster 0x0501 IAS ACE, command 0x04 PanelStatusChanged, payload [5,X,1,0] --> Entry delay, beeping X seconds
* cluster 0x0501 IAS ACE, command 0x04 PanelStatusChanged, payload [8,0,0,0] --> Arming Stay (red flashing, no beep even with 1 in 3rd arg, works well if fired together with Entry delay beeping)
* cluster 0x0501 IAS ACE, command 0x04 PanelStatusChanged, payload [9,0,0,0] --> Arming Night (red flashing, no beep)
* cluster 0x0501 IAS ACE, command 0x04 PanelStatusChanged, payload [10,0,0,0] --> Arming Away (red flashing, no beep)

Notes
* PanelStatusChanged with payload 4 should be exit delay, but does nothing in 3400-keypad and gives error message in 3400G-panel
* PanelStatusChanged with payload [7,something,something,something] gave me once solid red lights top and bottom of the keypad, but I have not been able to reproduce it. Might be useful to indicate ongoing alarm, maybe false fire alarm that could be silenced dialing right code
* I am not completely satisfied with the keypads behaviour when Arming or Entry Delay. The keys only light up if proximity sensor detects presence and shut down after a couple of seconds (the beeping continues though). In my opinion, they should stay awake as long as the beeping goes on. Maybe there is a function to keep the keys awake?

BusinessClaes, 2021-04-10:
I will try to put here tomorrow the messages that I get om the ZbBridge when I activate Z2T

BusinessClaes, 2021-04-18:
Here comes the list of messages that I have been receiving with Z2T:
- The first group of messages are showing up now and then.
- The line 0501!07 is response when the proximity sensor detects movement.
- The group of messages 0501!00 are response to dialing (de)activation codes. Payload format 0X043Y3Y3Y3Y00 where X is (de)activation mode (0=deactivate, 1=stay, 3=away, 2=night) and YYYY is the dialed code (i dialed 6666 for all of them).
- The last line is response to press and hold panic keys.

```{"ZbReceived":{"Centralite3400G":{"Device":"0xB6D4","Name":"Centralite3400G","Manufacturer":"CentraLite","ModelId":"3400-G","Endpoint":1,"LinkQuality":81}}}
{"ZbReceived":{"Centralite3400G":{"Device":"0xB6D4","Name":"Centralite3400G","ZoneType":541,"Endpoint":1,"LinkQuality":89}}}
{"ZbReceived":{"Centralite3400G":{"Device":"0xB6D4","Name":"Centralite3400G","ConfigResponse":{"BatteryPercentage":{"Status":134,"StatusMsg":"UNSUPPORTED_ATTRIBUTE"}},"Endpoint":1,"LinkQuality":94}}}
{"ZbReceived":{"Centralite3400G":{"Device":"0xB6D4","Name":"Centralite3400G","ConfigResponse":{},"Endpoint":1,"LinkQuality":94}}}
{"ZbReceived":{"Centralite3400G":{"Device":"0xB6D4","Name":"Centralite3400G","0500<00":"000000000000","ZoneStatusChange":0,"Panic":0,"Endpoint":1,"LinkQuality":86}}}
{"ZbReceived":{"Centralite3400G":{"Device":"0xB6D4","Name":"Centralite3400G","0019!01":"004E102C0001530015","Endpoint":1,"LinkQuality":52}}}
{"ZbReceived":{"Centralite3400G":{"Device":"0xB6D4","Name":"Centralite3400G","Temperature":20.58,"Endpoint":1,"LinkQuality":47}}}
{"ZbReceived":{"Centralite3400G":{"Device":"0xB6D4","Name":"Centralite3400G","ZoneStatus":0,"Panic":0,"Endpoint":1,"LinkQuality":50}}}
{"ZbReceived":{"Centralite3400G":{"Device":"0xB6D4","Name":"Centralite3400G","0020<00":"","Endpoint":1,"LinkQuality":45}}}
{"ZbReceived":{"Centralite3400G":{"Device":"0xB6D4","Name":"Centralite3400G","BatteryVoltage":3,"BatteryPercentage":100,"Endpoint":1,"LinkQuality":47}}}

{"ZbReceived":{"Centralite3400G":{"Device":"0xB6D4","Name":"Centralite3400G","0501!07":"","Endpoint":1,"LinkQuality":79}}}

{"ZbReceived":{"Centralite3400G":{"Device":"0xB6D4","Name":"Centralite3400G","0501!00":"00043636363600","Endpoint":1,"LinkQuality":39}}}
{"ZbReceived":{"Centralite3400G":{"Device":"0xB6D4","Name":"Centralite3400G","0501!00":"01043636363600","Endpoint":1,"LinkQuality":29}}}
{"ZbReceived":{"Centralite3400G":{"Device":"0xB6D4","Name":"Centralite3400G","0501!00":"03043636363600","Endpoint":1,"LinkQuality":58}}}
{"ZbReceived":{"Centralite3400G":{"Device":"0xB6D4","Name":"Centralite3400G","0501!00":"02043636363600","Endpoint":1,"LinkQuality":34}}}

{"ZbReceived":{"Centralite3400G":{"Device":"0xB6D4","Name":"Centralite3400G","0501!04":"","Endpoint":1,"LinkQuality":29}}}
```

BusinessClaes, 2021-04-18:
And these are two examples from Tasmota console, before mqtt messages are sent (my previous post is just the mqtt messages that I log to file from node red). Plese tell me if this help you in some way with the device handler and if you want me to fetch any other messages.
```
Proximity sensor
13:07:22 ZIG: {"ZbEZSPReceived":"450000040101050101400100004564B59090FFFF0301100702"}
13:07:22 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1281,"srcaddr":"0x9090","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":31,"securityuse":0,"seqnumber":69,"fc":"0x01","manuf":"0x0000","transact":16,"cmdid":"0x07","payload":""}}
13:07:22 ZIG: ZbZCLRawReceived: {"0x9090":{"0501!07":"","Endpoint":1,"LinkQuality":31}}
13:07:22 MQT: tele/zbBridge/SENSOR = {"ZbReceived":{"0x9090":{"Device":"0x9090","0501!07":"","Endpoint":1,"LinkQuality":31}}}

Deactivate (code 5555)
13:12:47 ZIG: {"ZbEZSPReceived":"4500000401010501014001000084A0C49090FFFF0A012E000004353535350002"}
13:12:47 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1281,"srcaddr":"0x9090","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":71,"securityuse":0,"seqnumber":132,"fc":"0x01","manuf":"0x0000","transact":46,"cmdid":"0x00","payload":"00043535353500"}}
13:12:47 ZIG: ZbZCLRawReceived: {"0x9090":{"0501!00":"00043535353500","Endpoint":1,"LinkQuality":71}}
13:12:47 MQT: tele/zbBridge/SENSOR = {"ZbReceived":{"0x9090":{"Device":"0x9090","0501!00":"00043535353500","Endpoint":1,"LinkQuality":71}}}
13:12:50 ZIG: {"ZbEZSPReceived":"23000901909013F0800B006F0D0004"}
13:12:50 ZIG: {"ZbEZSPReceived":"2400909013F0800B006F0D0000030000"}
13:12:50 MQT: tele/zbBridge/RESULT = {"ZbState":{"Status":34,"IEEEAddr":"0x000D6F000B80F013","ShortAddr":"0x9090","ParentNetwork":"0x0000","Status":0,"Decision":3}}
13:12:51 ZIG: {"ZbEZSPReceived":"620013F0800B006F0D00"}
13:12:51 ZIG: {"ZbEZSPReceived":"450004000013000000000400008584BD9090FFFF0C96909013F0800B006F0D008002"}
13:12:51 MQT: tele/zbBridge/RESULT = {"ZbState":{"Status":30,"IEEEAddr":"0x000D6F000B80F013","ShortAddr":"0x9090","PowerSource":false,"ReceiveWhenIdle":false,"Security":false}}
```

BusinessClaes, 2021-04-18:
I realize now that the transaction number appears in the raw data but is not sent in the mqtt message (the transaction number that the [info page at zigbee2mqtt](https://www.zigbee2mqtt.io/devices/3400-D.html) says has to sent back to the keypad when sending `ArmResponse` messages). I have not succeeded in sending this transaction number with `zha.issue_zigbee_cluster_command` (first because I dont recieve it with mqtt and second because I don´t know how to bake it into the payload of the `ArmResponse` messages).

Hedda, 2021-04-26:
FYI, dmulcahey is working on alarm control panel support to ZHA for Home Assistant core here -> https://github.com/home-assistant/core/pull/49080

Also see zha-device-handlers discussion about Centralite 3405-L / Lowes Iris Keypad V2 which are similar keypads -> https://github.com/zigpy/zha-device-handlers/issues/87

PS: There is also a zigpy discussion about Zigbee IAS and ACE (for "Alarm Control Panel" support) here -> https://github.com/zigpy/zigpy/issues/419

nerdosity, 2021-04-27:
Please contact me for any testing, I would gladly help. Have the HK1 version, the one with the PIR.

Hedda, 2021-04-27:
> Please contact me for any testing, I would gladly help. Have the HK1 version, the one with the PIR.

dmulcahey development branch for it is found here -> https://github.com/dmulcahey/home-assistant/tree/dm/zha-ias-ace-support

Comments if testing that branch before the PR is merged into HA should go into the pull request -> home-assistant/core#49080

Hedda, 2021-04-30:
FYI; Home Assistant 2021.5 (planned for release May 5, 2021) Beta release notes: "**_ZHA now has support for alarm control panels_**".

https://rc.home-assistant.io/blog/2021/04/28/release-20215/ -> https://github.com/home-assistant/core/pull/49080

snel6424, 2021-05-06:
Any update for support on these devices using ZHA? I have an Xfinity XHK1-UE keypad that I have not been able to get working. Running HA 2021.5.0, and in ZHA I get 3 entities: temp sensor, power sensor (which doesn't work), and an IAS binary sensor just like BusinessClaes said in his original post.

dmulcahey, 2021-05-06:
> Any update for support on these devices using ZHA? I have an Xfinity XHK1-UE keypad that I have not been able to get working. Running HA 2021.5.0, and in ZHA I get 3 entities: temp sensor, power sensor (which doesn't work), and an IAS binary sensor just like BusinessClaes said in his original post.

Can I have the content of the zigbee device information dialog for this device please?

support launched in this last release

snel6424, 2021-05-06:
> 
> Can I have the content of the zigbee device information dialog for this device please?
> 

`URC4450BC0-X-R
by Universal Electronics Inc
ConBee II
Firmware: 0x20181015
Zigbee info
IEEE: 00:0d:6f:00:0f:7c:20:10
Nwk: 0x63e7
Device Type: EndDevice
LQI: 255
RSSI: -39
Last Seen: 2021-05-06T12:20:25
Power Source: Battery or Unknown`

nerdosity, 2021-05-06:
> Can I have the content of the zigbee device information dialog for this device please?
>
> support launched in this last release

Me too.
Tried with beta a couple of days ago, so I waited.
I can delete the device and re-bind it, what type of debug info do you want me to enable? 

I've got this one in the screenshot, the Technicolo TKA105:
[image]

dmulcahey, 2021-05-06:
I need the content of the dialog when you click the zigbee device information button

snel6424, 2021-05-06:
> I need the content of the dialog when you click the zigbee device information button

Please excuse my ignorance, but do you mean the 'Zigbee Device Signature'? I do not have a 'zigbee device information' button

`{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4335, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=0, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0401",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0402",
        "0x0500",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019",
        "0x0501"
      ]
    }
  },
  "manufacturer": "Universal Electronics Inc",
  "model": "URC4450BC0-X-R",
  "class": "zigpy.device.Device"
}`

dmulcahey, 2021-05-06:
Yeah that.. sorry 😣

dmulcahey, 2021-05-06:
@snel6424 @nerdosity are either of you comfortable modding HA files to try something?

snel6424, 2021-05-06:
> @snel6424 @nerdosity are either of you comfortable modding HA files to try something?

I am if you are willing to walk me through it

dmulcahey, 2021-05-06:
find this in your install: 

https://github.com/home-assistant/core/blob/ce692afead189ac63a2aa1f5b286caae8581e8e3/homeassistant/components/zha/core/registries.py#L86

we are going to add a line to see if it makes this work... make that block look like this: 

```
SINGLE_OUTPUT_CLUSTER_DEVICE_CLASS = {
    zcl.clusters.general.OnOff.cluster_id: BINARY_SENSOR,
    zcl.clusters.security.IasAce.cluster_id: ALARM
}
```

BusinessClaes, 2021-05-06:
@dmulcahey thank you so much for putting all this together. Awesome!
Here comes the device signature for Centralite 3400 and Centralite 3400-G.

Centralite 3400
```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4174, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=0, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0401",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0402",
        "0x0500",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0019",
        "0x0501"
      ]
    }
  },
  "manufacturer": "CentraLite",
  "model": "3400",
  "class": "zigpy.device.Device"
}
```

Centralite 3400-G
```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4174, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=0, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0401",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0402",
        "0x0500",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0019",
        "0x0501"
      ]
    }
  },
  "manufacturer": "CentraLite",
  "model": "3400-G",
  "class": "zigpy.device.Device"
}
```

BusinessClaes, 2021-05-06:
> find this in your install:
>
> https://github.com/home-assistant/core/blob/ce692afead189ac63a2aa1f5b286caae8581e8e3/homeassistant/components/zha/core/registries.py#L86
>
> we are going to add a line to [...]

Where do I find ”my install”? 😅
I have the `File editor` addon and there I can only browse the `config` directory. I don’t see any `install` or `registries.py` in there...

BusinessClaes, 2021-05-06:
I cannot see any `components/zha/` folder either... only `custom_componens` folder...

nerdosity, 2021-05-06:
> @snel6424 @nerdosity are either of you comfortable modding HA files to try something?

Yeah, it's a docker, so no problem.

Mine is:
```

{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4470, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=0, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0401",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0402",
        "0x0500",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019",
        "0x0501",
        "0x0b05"
      ]
    }
  },
  "manufacturer": "Technicolor",
  "model": "TKA105",
  "class": "zigpy.device.Device"
}
```

tube0013, 2021-05-06:
if you are on HASSOS or another supervised install use this guide:

https://github.com/zigpy/zha-device-handlers#testing-quirks-in-development-in-docker-based-install

snel6424, 2021-05-06:
> if you are on HASSOS or another supervised install use this guide:
> 
> https://github.com/zigpy/zha-device-handlers#testing-quirks-in-development-in-docker-based-install
> 
> then find the ZHA component here:
> 
> /usr/src/homeassistant/homeassistant/components/zha

What is the 'User' supposed to be? I have tried leaving it blank, 'root', 'homeassistant', 'hass', and 'hassio', and it always briefly pops up and say "Exec into container as default user using command /usr/src/homeassistant/homeassistant/components/zha"

kdknigga, 2021-05-06:
> find this in your install:
>
> https://github.com/home-assistant/core/blob/ce692afead189ac63a2aa1f5b286caae8581e8e3/homeassistant/components/zha/core/registries.py#L86
>
> we are going to add a line to [...]

After making this change I now have an alarm_control_panel entity for my URC4450BC0-X-R!

dmulcahey, 2021-05-06:
Huzzah! Is it functional?

kdknigga, 2021-05-06:
I'm still trying to figure out how alarm panels are supposed to work in HA.  I can arm and disarm (with 1234, a default set somewhere?) through the web interface.  Going to try with the panel shortly.

kdknigga, 2021-05-06:
Yeah, seems to work from the panel, too.

[image]

dmulcahey, 2021-05-06:
Awesome… you can set the code in the zha config panel

kdknigga, 2021-05-06:
Hmmmm, I'm not seeing that.  All I get in Global Options is "Default light transition time" and "Enable identify effect when devices join the network".

Thanks a lot for your work on this, by the way!

dmulcahey, 2021-05-06:
Ok looks like there are a few bugs… I’ll dig in later tonight

dmulcahey, 2021-05-07:
@kdknigga can you try the changes here: https://github.com/home-assistant/core/pull/50235

nerdosity, 2021-05-07:
https://github.com/home-assistant/core/pull/50235/commits/9d5841271b731af55ac1eee34fd4e87739da51d3
For me these modifications did the trick.

[image]

[image]

[image]

[image]

How can we use the ias_zone? My device lights when a hand is placed near the device itself.
Do you want some debug info?

nerdosity, 2021-05-07:
FIY:

[image]
[image]

OK, there are 4 AA batteries, that's correct.
But, at 1.5v x 4, it's impossible that voltage is correct.
Maybe that's 68%?

dmulcahey, 2021-05-07:
Battery will need a quirk per device. I haven’t touched that yet. 

I think the IasZone is tamper only. We need changes in ZHA to make that work. That said, I think we can kind of hack the IR sensor with a quirk to make that IasZone work like a motion sensor. I have to look at it some more.

nerdosity, 2021-05-07:
Tell me if you need debug.
This device (TKA105) has got two tampers I think:
- one in the back (a little button, that is kept pressed, is unpressed when the device is moved from its base)
- on on the front, a PIR. I can see it lighting up when moving my hand on front.

kdknigga, 2021-05-07:
@dmulcahey, I now see Alarm Control Panel Options in the zha config panel and setting the code there does seem to change the code accepted by the panel.  Awesome!

I did get this when restarting HA, though:

```
Logger: homeassistant.components.zha.core.channels.base
Source: components/zha/core/channels/base.py:356
Integration: Zigbee Home Automation (documentation, issues)
First occurred: 8:14:39 AM (1 occurrences)
Last logged: 8:14:39 AM

[0x2D90:1:0x0501]: received command get_panel_status
```

Seems like a "not-an-error" error?

dmulcahey, 2021-05-07:
Shouldn’t be an error… I’ll look at that later

snel6424, 2021-05-11:
> Is there a setting so that instead of arming instantly you get a set amount of seconds of pending?

This is easily done with either the built in alarm_control_panel, or other alarm systems, like Alarmo.  You set you the times of pending, arming, etc, on there, not the keypad.
Then you have 2 automations: alarm_control_panel --> Keypad and keypad --> alarm_control_panel.

The only setting the keypad itself needs is a password

Gamerayers, 2021-05-13:
Hey guys, has anybody figured out how to send the ZHA command to make this beep?  I want to either have it beep based on alarmo or based on door open/shut.
Looks like I should be able to use the zha.issue_zigbee_cluster_command and I tried, but I am not sure what I am doing wrong:
"service: zha.issue_zigbee_cluster_command
data:
  cluster_id: 0x0501
  ieee: '00:0d:6f:00:0b:a1:94:86'
  command: 0x00
  endpoint_id: 1
  args:
    - 0
    - 0"

drjjr2, 2021-05-23:
> Hey guys, has anybody figured out how to send the ZHA command to make this beep? I want to either have it beep based on alarmo or based on door open/shut.
> Looks like I should be able to use the [...]

Did you ever get this figured out? Just discovered that the keypad I was about to throw out now works with ZHA.

drjjr2, 2021-05-23:
> Hey guys, has anybody figured out how to send the ZHA command to make this beep? I want to either have it beep based on alarmo or based on door open/shut.
> Looks like I should be able to use the [...]

OK, I got this to work in a simple test script. Looks like it wants the cluster ID to be in decimal. (Your IEEE should be different):
```
service: zha.issue_zigbee_cluster_command
data:
  ieee: '00:0d:6f:00:0f:c1:e2:dc'
  cluster_id: 1281
  command: 0
  command_type: client
  cluster_type: out
  args: 0
  endpoint_id: 1

```

nerdosity, 2021-05-24:
> OK, I got this to work in a simple test script. Looks like it wants the cluster ID to be in decimal. (Your IEEE should be different):
>
> ```
> service: zha.issue_zigbee_cluster_command
> data:
>   ieee: [...]

good, this should be in the device, it's good for alarm activation timer.

Gamerayers, 2021-05-24:
Hmmm.  So I tried that, and I see in the log: 2021-05-24 07:05:41 WARNING (MainThread) [homeassistant.components.zha.core.channels.base] [0x2247:1:0x0501]: received command get_panel_status
But it doesn't beep.  I am not sure if maybe I have a different version of the 3400 or what is going on.

Below is my script.

service: zha.issue_zigbee_cluster_command
data:
  ieee: '00:0d:6f:00:0b:a1:94:86'
  cluster_id: 1281
  command: 0
  command_type: client
  cluster_type: out
  args: 0
  endpoint_id: 1

thecobra666, 2021-05-26:
> This is easily done with either the built in alarm_control_panel, or other alarm systems, like Alarmo. You set you the times of pending, arming, etc, on there, not the keypad.
> Then you have 2 [...]

Could you elaborate on this? I now have 2 alarm panels. One is the keypad and one is the built in. The built in alarm panel has a one minute delay. The keypad one activates immediately when the entered code is correct.

Gamerayers, 2021-05-26:
> Could you elaborate on this? I now have 2 alarm panels. One is the keypad and one is the built in. The built in alarm panel has a one minute delay. The keypad one activates immediately when the [...]

Create an automation for when the keypad is set for to arm away that triggers Alarmo (or the built in panel) to arm.  This will simply use the keypad as a trigger method.  You will need automation for arm away, arm home and disarm.

...Unfortunately, I have been unable to get the keypad to beep.

thecobra666, 2021-05-26:
That worked. I call the alarm service and push the corresponding state to
the keypad or built in alarm to the same state. That makes my keypad beep
when activated or disarmed with the service. I have the xfinity.

This way I have the delay and correct states on both. It does take a few
automations :)

Hedda, 2021-05-27:
> Is there a setting so that instead of arming instantly you get a set amount of seconds of pending?

Feature request for ZHA as a 10-second delay before arming is probably wanted by default on all alarm control panels as standard?

I understand that the industry standard on the commercial security alarm system is to beep for 10-seconds to warn before arming?

https://github.com/home-assistant/architecture/discussions

https://community.home-assistant.io/c/feature-requests/

thecobra666, 2021-05-27:
> Feature request for ZHA as a 10-second delay before arming is probably wanted by default on all alarm control panels as standard?
>
> I understand that the industry standard on the commercial security [...]

Done: https://community.home-assistant.io/t/zha-delay-arming/311361

Hedda, 2021-05-27:
> > Is there a setting so that instead of arming instantly you get a set amount of seconds of pending?
> 
> Feature request for ZHA as a 30-second delay before arming is probably wanted by default on all alarm control panels as standard?
> 
> I understand that the industry standard on the commercial security alarm system is to beep for 30-seconds to warn before arming?

Looks as if Home Assistant manual alarm control panel platform delay_time integer is optional and defaults to 60 seconds if set:

https://www.home-assistant.io/integrations/alarm_control_panel/

https://www.home-assistant.io/integrations/manual

https://www.home-assistant.io/integrations/alarm_control_panel.template/

Gamerayers, 2021-05-27:
After playing a bunch with this, I think it is best to have the setting of instant arm and use something with Alarmo with automation to trigger timing.
The automated beep would be nice though.  Although, I believe you could send the panel the exit beep command with the timing from Alarmo and it will work too.  But I've never been able to send my panel beeps at all.  I think because it is the 3400 and not the 3405-L or the 3400-G or Xfinity Keypad.

Gamerayers, 2021-06-02:
> good, this should be in the device, it's good for alarm activation timer.

Okay, so I've now tried this on a 3400 and a 3405-L.  My master is a tasmota Zbbridge.  I'm not sure if that is the root cause of my issue, but I can't get either keypad to beep.  I'm not entirely sure what is wrong.
I've done this as a script.

```
service: zha.issue_zigbee_cluster_command
data:
  ieee: '00:0d:6f:00:10:99:fa:f5'
  cluster_id: 1281
  command: 0
  command_type: client
  cluster_type: out
  args: 0
  endpoint_id: 1
```

And I've tried using the issue commands, but it doesn't seem to actually do anything.  Anybody who has a clue how to fix this, it would be greatly appreciated it.

BusinessClaes, 2021-06-02:
> But I've never been able to send my panel beeps at all. I think because it is the 3400

I have both a 3400 and a 3400-G, and I get both to beep...

Do you get all the other commands on [my previous post](https://github.com/zigpy/zha-device-handlers/issues/838#issuecomment-817200362) working? Have you tried lo-pitch beep (args [4] instead of [0])?
Have you tried to send the commands from Node-RED instead of with a script? I guess it shouldn´t matter, but I have only done it with Call Service node in Node-RED. I look at your code and I can´t find anything wrong, so I am curious if this is the only command that´s not working and you get all the others to work.

Another question, does anybody get the keypad to light up when sending "exit delay" command? (or in any other way)
I use "exit delay" because "entry delay" doesn´t seem to work. Anyway, in my opinion, the keypad should light up (as well ass beeping entry delay) when my front door magnet sensor opens, telling the user "here I am for you to input your code". But I am only able to make it beep, not light up (yes, it lights up when I wave my hand in front of it to input the code, but that´s not how it should be in my opinion).

BusinessClaes, 2021-06-02:
Oh, by the way :)
The keypad has to be awake (with that I mean light up) for the beep command to work.
First wave your hand in front of it and then send the command.
Otherwise no beep :)

Gamerayers, 2021-06-02:
> Oh, by the way :)
> The keypad has to be awake (with that I mean light up) for the beep command to work.
> First wave your hand in front of it and then send the command.
> Otherwise no beep :)

Ohhhhh.... Well, that's not a great thing.  I expected to be able to use the beep for when doors opened in the house. Any way to force the keypad to open.
I haven't validated that the beeps function when I wave in front, but that could definely be part of my issue.

drjjr2, 2021-06-02:
> Oh, by the way :)
> The keypad has to be awake (with that I mean light up) for the beep command to work.
> First wave your hand in front of it and then send the command.
> Otherwise no beep :)

I don't find that to be the case with my keypad. Just sent it the beep script above and it beeped and I had not been near it.

Gamerayers, 2021-06-02:
Just curious, what controller is everybody using to run zha?  In currently running on sonoff flashed with tasmota zbbridge. But maybe that's the issue I have. I do a have conbee, but my server isn't in a good location so it's not ideal for being the master.

BusinessClaes, 2021-06-02:
@drjjr2 which keypad do you have? (I have a 3400 and a 3400-G, plus an Xfinity on its way)

@Gamerayers I have the same config as you, ZBBridge with Tasmota

[13 more replies]

closed 2022-07-23 by github-actions[bot]