zigpy/zha-device-handlers #844: Ikea E1745 False Trigger [BUG]

wes1993, 2021-04-12:
**Describe the bug**
Sometimes the ikea motion trigger a false on

**To Reproduce**
Steps to reproduce the behavior:
Just the sensor turn on when should not

**Expected behavior**
Still OFF

**Screenshots**
Not Applicabile

**Additional context**
The log of one false event:
'''
2021-04-12 15:19:34 DEBUG (MainThread) [zigpy.zcl] [0xfb6b:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=2 command_id=66>
2021-04-12 15:19:34 DEBUG (MainThread) [zigpy.zcl] [0xfb6b:1:0x0006] ZCL request 0x0042: [0, 1800, 0]
2021-04-12 15:19:34 DEBUG (MainThread) [zigpy.zcl] [0xfb6b:1:0x0006] No handler for cluster command 66
'''

Log of one REAL motion event:
'''
2021-04-12 17:41:14 DEBUG (MainThread) [zigpy.zcl] [0xfb6b:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=6 command_id=66>
2021-04-12 17:41:14 DEBUG (MainThread) [zigpy.zcl] [0xfb6b:1:0x0006] ZCL request 0x0042: [0, 1800, 0]
2021-04-12 17:41:14 DEBUG (MainThread) [zigpy.zcl] [0xfb6b:1:0x0006] No handler for cluster command 66
'''

The strange is that the logs of the events seems the same, there is a way to spoof a more complete log to better debug the problem?

Best Regards
Stefano

MattWestb, 2021-04-12:
My experience of both the old and the new model (you have the new model) is that after long time and the battery is going down little but not critical they can start triggering "false alarms" and replacing both batteries is normally fixing it.
It can also that you have some very small animals that is building on home in the PIR censor (oft happen with smoke sensors with spiders).
I have seeing report of hardware failure this but its not so common so first go for new batteries (they is in parallel so both is the same state and need being changed at the same time).

wes1993, 2021-04-12:
@MattWestb 
Thanks a lot for your reply!! :)
I have changed the battery some time ago but the problem still here, when I come back to home I will change the battery again.
For now I have covered the sensor so no one pass in front of the sensor so no animals or other stuff.

Thanks again, I will update when I have changed the battery and tested

Best regards
Stefano

MattWestb, 2021-04-12:
I was potting one down and putting it in one box and it was still sending motion command and the batters was around 60%. New batteries and it was OK.

wes1993, 2021-04-12:
Ok thank a il check in the next hours if is fixed by changing the battery’s .
But the question also is why if the battery it’s 60% the sensor send this false trigger? Can we fix this?

closed 2021-10-16 by github-actions[bot]