zigpy/zha-device-handlers #848: [BUG]

wes1993, 2021-04-14:
**Describe the bug**
When i set an automation with 3 press won't work 

**To Reproduce**
Create and automatio
Press 3 times the button

The automation i have created:
```
alias: FIXPulsanteAllarme
description: ''
trigger:
  - device_id: 4e525f47a1bc1bf3e7c2205f4092d425
    domain: zha
    platform: device
    type: remote_button_triple_press
    subtype: remote_button_triple_press
condition: []
action:
  - service: notify.telegram
    data:
      message: Preme 1
mode: single
```
**Expected behavior**
Run the automation

**Screenshots**
No screen

**Additional context**
Log of ZHA
```
2021-04-14 11:40:45 DEBUG (MainThread) [zigpy.zcl] [0x69f9:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=80 command_id=Command.Report_Attributes>
2021-04-14 11:40:45 DEBUG (MainThread) [zigpy.zcl] [0x69f9:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=32768, value=<TypeValue type=uint8_t, value=3>)]]
2021-04-14 11:40:45 DEBUG (MainThread) [zigpy.zcl] [0x69f9:1:0x0006] Attribute report received: 32768=3
2021-04-14 11:40:45 DEBUG (MainThread) [zigpy.zcl] [0x69f9:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=True> manufacturer=4447 tsn=81 command_id=Command.Report_Attributes>
2021-04-14 11:40:45 DEBUG (MainThread) [zigpy.zcl] [0x69f9:1:0x0000] ValueError exception for: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=True> manufacturer=4447 tsn=81 command_id=Command.Report_Attributes> payload: b'01ff42090421a8430a219118'
2021-04-14 11:40:45 DEBUG (MainThread) [zigpy.zcl] [0x69f9:1:0x0000] new data: b'1c5f11510a01ff42080421a8430a219118'
2021-04-14 11:40:45 DEBUG (MainThread) [zigpy.zcl] [0x69f9:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=True> manufacturer=4447 tsn=81 command_id=Command.Report_Attributes>
2021-04-14 11:40:45 DEBUG (MainThread) [zigpy.zcl] [0x69f9:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=65281, value=<TypeValue type=CharacterString, value=!�C
2021-04-14 11:40:45 DEBUG (MainThread) [zigpy.zcl] [0x69f9:1:0x0000] Attribute report received: 65281=!�C
```
Best Regards
Stefano

MattWestb, 2021-04-14:
manufacturer=4447 is Xiaomi / Aqara but you is not writing wot device you is having problem with.

wes1993, 2021-04-14:
Sorry, i forget to add i have added now :-D

MattWestb, 2021-04-14:
Wot is you getting in `zha_event` then doing tripling ?

wes1993, 2021-04-14:
where i can find the zha_event? in the HA log?

MattWestb, 2021-04-14:
Developer Tools > Events and then putting `zha_event` in "Listen to events Event to subscribe" to and then "start listening" and you is getting all events ZHA is "detecting".

wes1993, 2021-04-14:
Thanks a lot!!! I'm new to ZHA, i love this function :-D
The LOG:

```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:15:8d:00:01:e7:51:cd",
        "unique_id": "00:15:8d:00:01:e7:51:cd:1:0x0006",
        "device_id": "4e525f47a1bc1bf3e7c2205f4092d425",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "attribute_updated",
        "args": {
            "attribute_id": 32768,
            "attribute_name": "Unknown",
            "value": 3
        }
    },
    "origin": "LOCAL",
    "time_fired": "2021-04-14T10:08:14.763753+00:00",
    "context": {
        "id": "3f38a4263d81cc2316ff035f1e206b2c",
        "parent_id": null,
        "user_id": null
    }
}
```

MattWestb, 2021-04-14:
Xiaomi / Aara is tricky then the is not sending "zigbee commands / events" they is sending attributes that must being converted to events that is not so easy to finding.
You can triggering QUADRUPLE_PRESS with value = 4 :-))

wes1993, 2021-04-14:
yes infact quadruple press single and double works just triple press won't work
how can i try with this template instead of device?

platform: event
event_type: ZHA Event
event_data:
  device_ieee: '‘00:15:8d:00:01:e7:51:cd’'
  endpoint_id: 1
  cluster_id: 6
  command: What can i insert here?

MattWestb, 2021-04-14:
That i dont knowing then im not having this device.
If looking on the quirk https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/xiaomi/aqara/switch_aq2.py its looks like the triple is made little different then the other so perhaps its not working.

If you getting normal events on the other functions than i thinking its one **BUG**.

@Adminiuga is normally coming on line in "our afternoon" and hi is having the knowledge of Xiaomi devices.

wes1993, 2021-04-14:
Found the problem and fixed the string thanks for your help :-D

I have created a pull request, see here:

https://github.com/zigpy/zha-device-handlers/pull/849

Thanks 
Stefano

closed 2021-04-14 by wes1993

MattWestb, 2021-04-14:
Great !!

Then the devs is getting little to checking but great that you is getting it working all the way !! 

Thanks for the PR !!!

wes1993, 2021-04-14:
No problem it's a pleasure :-D