zigpy/zha-device-handlers #851: Improve / fix support for Moes HY368 (_TZE200_cwnjrr72)

almico, 2021-04-15:
I have two Moes Hy368 (according to the packaging). One is a `_TZE200_ywdxldoj` and the other is a `_TZE200_cwnjrr72`.

While the `_TZE200_ywdxldoj` shows the following controls:
[image]

The `_TZE200_cwnjrr72` shows these:
[image]

As you can see, there is no reference to the "Preset". This makes me think that either there are some differences between the two sub-models that prevent some register from being read or that some part of the code is not reached for the `_TZE200_cwnjrr72`.

I would like to see the "Preset" for the `_TZE200_cwnjrr72` too.

This is the device signature for `_TZE200_ywdxldoj`

```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0201",
        "0x0204",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_ywdxldoj",
  "model": "TS0601",
  "class": "zhaquirks.tuya.valve.MoesHY368_Type1"
}
```

This is the device signature for `_TZE200_cwnjrr72`

```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0201",
        "0x0204",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_cwnjrr72",
  "model": "TS0601",
  "class": "zhaquirks.tuya.valve.MoesHY368_Type1"
}
```

pbvdven, 2021-04-15:
i also have a moes BHT-002 i used to have it in zigbee2mqtt and there where more entities exposed for my model like child lock and switching to programming mode instead of manual. could it be possible in ZHA as well? would be great.

these entities where exposed in zigbee2mqtt 
lock (state), deadzone_temperature, max_temperature_limit, climate (current_heating_setpoint, local_temperature, local_temperature_calibration, system_mode, running_state, preset, sensor), linkquality

{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.Router: 1>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0201",
        "0x0204",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_aoclfnxz",
  "model": "TS0601",
  "class": "zhaquirks.tuya.electric_heating.MoesBHT"

almico, 2021-04-18:
Ok. I found out that this issue has to be fixed on Home Assistant ZHA component and not here. Sorry 😄

closed 2021-04-18 by almico