zigpy/zha-device-handlers #852: [Device Support Request] Zemismart tubular roller blind - M2805EGBZTN

mikkelzzz3, 2021-04-15:
**Is your feature request related to a problem? Please describe.**
I have bought two of these: https://pl.aliexpress.com/item/1005001994026867.html?spm=a2g0s.9042311.0.0.72a84c4dsBfami 

I can add them to home assistant with the ZHA integration without issues. Once added, there are no entities:
[image]

**Describe the solution you'd like**
For ZHA to recognise this as a type "cover". I manually tried specifying in configurations.yaml that it was a cover, but that did not change anything. 
I would like this device to be controller as other smart curtains. up / down / 50% down.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

**Device signature**
`{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_5sbebbzs",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}`

**Log from pairing**
`[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
Device 0xf97f (84:71:27:ff:fe:23:6e:69) joined the network
[0xf97f:zdo] ZDO request ZDOCmd.Device_annce: [0xF97F, 84:71:27:ff:fe:23:6e:69, 128]
[0xf97f] Requesting 'Node Descriptor'
Tries remaining: 2
[0xf97f] Extending timeout for 0x25 request
[0xf97f] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)
[0xf97f] Discovering endpoints
Tries remaining: 3
[0xf97f] Extending timeout for 0x27 request
[0xf97f] Discovered endpoints: [1]
[0xf97f:1] Discovering endpoint information
Tries remaining: 3
[0xf97f] Extending timeout for 0x29 request
[0xf97f:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10])
Unknown cluster 61184
[0xf97f] Extending timeout for 0x2b request
[0xf97f:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=43 command_id=Command.Read_Attributes_rsp>
[0xf97f:1] Manufacturer: _TZE200_5sbebbzs
[0xf97f:1] Model: TS0601
Checking quirks for _TZE200_5sbebbzs TS0601 (84:71:27:ff:fe:23:6e:69)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
device - 0xF97F:84:71:27:ff:fe:23:6e:69 entering async_device_initialized - is_new_join: True
device - 0xF97F:84:71:27:ff:fe:23:6e:69 has joined the ZHA zigbee network
[0xF97F](TS0601): started configuration
[0xF97F:ZDO](TS0601): 'async_configure' stage succeeded
[0xf97f] Extending timeout for 0x2d request
[0xf97f] Extending timeout for 0x2f request
[0xF97F:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
[0xF97F:1:0x0000]: finished channel configuration
[0xF97F:1:0x0019]: bound 'ota' cluster: Status.SUCCESS
[0xF97F:1:0x0019]: finished channel configuration
[0xF97F:1:0x0000]: 'async_configure' stage succeeded
[0xF97F:1:0x0019]: 'async_configure' stage succeeded
[0xF97F](TS0601): completed configuration
[0xF97F](TS0601): stored in registry: ZhaDeviceEntry(name='_TZE200_5sbebbzs TS0601', ieee='84:71:27:ff:fe:23:6e:69', last_seen=1618495439.3135388)
[0xF97F](TS0601): started initialization
[0xF97F:ZDO](TS0601): 'async_initialize' stage succeeded
[0xF97F:1:0x0000]: initializing channel: from_cache: False
[0xF97F:1:0x0000]: finished channel configuration
[0xF97F:1:0x0019]: initializing channel: from_cache: False
[0xF97F:1:0x0019]: finished channel configuration
[0xF97F:1:0x0000]: 'async_initialize' stage succeeded
[0xF97F:1:0x0019]: 'async_initialize' stage succeeded
[0xF97F](TS0601): power source: Battery or Unknown
[0xF97F](TS0601): completed initialization
[0xf97f:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=3 command_id=Command.Read_Attributes>
[0xf97f:1:0x000a] ZCL request 0x0000: [[7]]
[0xf97f:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=4 command_id=36>
[0xf97f:1:0xef00] Unknown cluster-specific command 36
[0xf97f:1:0xef00] ZCL request 0x0024: b'\x00\t'
[0xf97f:1:0xef00] No handler for cluster command 36`

**Additional context**
I'm happy to help in any way I can.

hirsts, 2021-05-28:
I too have Zemismart Zigbee Tubular and had the same issue using ZHA. I have tried using Zigbee2Mqtt which work but is generally unstable with my 15+ Philips Hue Sensors and 10 Zemismart blinds. 

ZHA was great with the Philips Hue sensors and so adding support for the Zemismart blinds would be great. I'll then be able to decom Z2M and move back to ZHA.

teunaldo, 2021-07-27:
Hi @hirsts , I have Zemismart AM15 blind rollers. 
Can you tell me how you made this working with Zigbee2Mqtt?

MattWestb, 2021-11-15:
ZHA have adding this model and shall working OK after last fixes is being added.
https://github.com/zigpy/zha-device-handlers/pull/976

MattWestb, 2021-11-15:
Support have being added in ZHA and shall working OK after the last fixes.
https://github.com/zigpy/zha-device-handlers/pull/1102

closed 2022-05-21 by github-actions[bot]