zigpy/zha-device-handlers #853: [Device Support Request] Develco Frient WISZB-120 Entry Sensor Pro

Veldkornet, 2021-04-16:
**Is your feature request related to a problem? Please describe.**
I am able to pair the device, the temperature sensor works, but the battery sensor does not. The binory_sensor does not work. The contact is always reported as "open".
Upon investigation, found errors in logs.

**Describe the solution you'd like**
Add quirks for the Frient devices as needed

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=0, manufacturer_code=4117, maximum_buffer_size=80, maximum_incoming_transfer_size=80, server_mask=0, maximum_outgoing_transfer_size=80, descriptor_capability_field=0, *allocate_address=False, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 49353,
      "device_type": "0x0001",
      "in_clusters": [
        "0x0003",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    },
    "35": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x000f",
        "0x0020",
        "0x0500"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "38": {
      "profile_id": 260,
      "device_type": "0x0302",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0402"
      ],
      "out_clusters": [
        "0x0003"
      ]
    }
  },
  "manufacturer": "frient A/S",
  "model": "WISZB-120",
  "class": "zigpy.device.Device"
}
```

**Additional context**
ZHA Issue: https://github.com/home-assistant/core/issues/49127
```
2021-04-16 17:42:15 DEBUG (MainThread) [zigpy.zcl] [0x7562:35:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=50 command_id=0>
2021-04-16 17:42:15 DEBUG (MainThread) [zigpy.zcl] [0x7562:35:0x0500] ZCL request 0x0000: [<ZoneStatus.Restore_reports|Supervision_reports|Tamper|Alarm_1: 53>, <bitmap8.0: 0>, 0, 65535]
2021-04-16 17:42:15 DEBUG (MainThread) [zigpy.zcl] [0x7562:35:0x0500] No handler for cluster command 0
...
2021-04-16 17:46:29 DEBUG (MainThread) [zigpy.zcl] [0x7562:35:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=51 command_id=0>
2021-04-16 17:46:29 DEBUG (MainThread) [zigpy.zcl] [0x7562:35:0x0500] ZCL request 0x0000: [<ZoneStatus.Restore_reports|Supervision_reports|Tamper|Alarm_1: 53>, <bitmap8.0: 0>, 0, 65535]
2021-04-16 17:46:29 DEBUG (MainThread) [zigpy.zcl] [0x7562:35:0x0500] No handler for cluster command 0
...
2021-04-16 17:48:04 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2021-04-16 17:48:04 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2021-04-16 17:48:04 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2021-04-16 17:48:04 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [32, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 42>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x7562>, 35, 260, 1280, b'\t4\x005\x00\x00\x00\xff\xff', 0, 175, 215, 162, 27, 121, 0, -67]
2021-04-16 17:48:04 DEBUG (MainThread) [zigpy.zcl] [0x7562:35:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=52 command_id=0>
2021-04-16 17:48:04 DEBUG (MainThread) [zigpy.zcl] [0x7562:35:0x0500] ZCL request 0x0000: [<ZoneStatus.Restore_reports|Supervision_reports|Tamper|Alarm_1: 53>, <bitmap8.0: 0>, 0, 65535]
2021-04-16 17:48:04 DEBUG (MainThread) [zigpy.zcl] [0x7562:35:0x0500] No handler for cluster command 0
2021-04-16 17:48:04 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x7562>, ep: 35, profile: 0x0104, cluster_id: 0x0500, data: b'09340035000000ffff'
2021-04-16 17:48:04 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2021-04-16 17:48:04 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [29, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x7562>, 35, 260, 1280, b'\t4\x005\x00\x00', 0, 175, 215, 162, 27, 121, 0, -67]
2021-04-16 17:48:04 DEBUG (MainThread) [zigpy.zcl] [0x7562:35:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=52 command_id=0>
2021-04-16 17:48:04 ERROR (MainThread) [zigpy.device] Failed to parse message (b'093400350000') on cluster 1280, because Data is too short to contain 1 bytes
2021-04-16 17:48:04 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x7562>, ep: 35, profile: 0x0104, cluster_id: 0x0500, data: b'093400350000'
2021-04-16 17:48:11 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2021-04-16 17:48:11 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2021-04-16 17:48:11 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2021-04-16 17:48:11 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [32, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 42>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x7562>, 35, 260, 1280, b'\t5\x005\x00\x00\x00\xff\xff', 0, 175, 223, 162, 27, 121, 0, -66]
2021-04-16 17:48:11 DEBUG (MainThread) [zigpy.zcl] [0x7562:35:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=53 command_id=0>
2021-04-16 17:48:12 DEBUG (MainThread) [zigpy.zcl] [0x7562:35:0x0500] ZCL request 0x0000: [<ZoneStatus.Restore_reports|Supervision_reports|Tamper|Alarm_1: 53>, <bitmap8.0: 0>, 0, 65535]
2021-04-16 17:48:12 DEBUG (MainThread) [zigpy.zcl] [0x7562:35:0x0500] No handler for cluster command 0
2021-04-16 17:48:12 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x7562>, ep: 35, profile: 0x0104, cluster_id: 0x0500, data: b'09350035000000ffff'
2021-04-16 17:48:12 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2021-04-16 17:48:12 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [29, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x7562>, 35, 260, 1280, b'\t5\x005\x00\x00', 0, 175, 223, 162, 27, 121, 0, -66]
2021-04-16 17:48:12 DEBUG (MainThread) [zigpy.zcl] [0x7562:35:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=53 command_id=0>
2021-04-16 17:48:12 ERROR (MainThread) [zigpy.device] Failed to parse message (b'093500350000') on cluster 1280, because Data is too short to contain 1 bytes
2021-04-16 17:48:12 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x7562>, ep: 35, profile: 0x0104, cluster_id: 0x0500, data: b'093500350000'
```

Veldkornet, 2021-05-07:
I see that support for this was supposed to be added via #317 but it's clearly not working.

Hedda, 2021-05-19:
Wondering if the root cause is the same as that described by dmulcahey in https://github.com/zigpy/zha-device-handlers/issues/605 issue? That is, a a limitation in [ZHA integration](https://www.home-assistant.io/integrations/zha/) for Home Assistant which currently only support 1 sensor per cluster, and these type of devices requiring 2 or more sensor per cluster?

Adminiuga, 2021-09-03:
regarding the data too short. Per ZCL specs, server-to-client cluster command 0x0000 should contain:

| Bits | 16 | 8 | 8 | 16 |
| --- | --- | --- | --- | -- | 
| Field Name | Zone Status | Extended Status | Zone ID | Delay |
| Data | 0x0035 | 0x00 | missing | missing |

You device sends the following information: 
09 -- ZCL Frame control: Server-to-Client, Cluster specific command
34 -- TSN
00 -- Command ID --Cluster command ZoneStatus Change Notification
3500 -- 0x0035 -- Zone status
00 -- extended status

So the device does not send enough data and is not compliant with ZCL v7, as per specs zoneid and delay fields should be included with zone status change notification,
It would need a quirk and IIRC there were some other IAS sensors in quirks with the similar problem

093400350000

closed 2022-03-09 by github-actions[bot]