zigpy/zha-device-handlers #857: [Device Support Request] Develco Frient SPLZB-131 Smart Plug Mini

Veldkornet, 2021-04-17:
**Is your feature request related to a problem? Please describe.**
I am able to pair the device, but it adds two binary sensors, of which one doesn’t work.  It also adds electrical measurement and smartenergy metering but according to zigbee2mqtt it should have at least switch (state), power, current, voltage, energy.

**Describe the solution you'd like**
Add quirks for the Frient devices as needed. There is apparently [ZigBee2MQTT support](https://www.zigbee2mqtt.io/devices/SPLZB-131.html).

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```

{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4117, maximum_buffer_size=72, maximum_incoming_transfer_size=1560, server_mask=0, maximum_outgoing_transfer_size=1560, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.Router: 1>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 49353,
      "device_type": "0x0001",
      "in_clusters": [
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009",
        "0x0702",
        "0x0b04"
      ],
      "out_clusters": [
        "0x0000",
        "0x0003",
        "0x000a",
        "0x0019",
        "0x0406"
      ]
    }
  },
  "manufacturer": "frient A/S",
  "model": "SPLZB-131",
  "class": "zigpy.device.Device"
}
```

**Additional context**
ZHA Issue: https://github.com/home-assistant/core/issues/49127

Hedda, 2021-05-19:
https://www.develcoproducts.com/products/smart-plugs/smart-plug-mini/

https://www.develcoproducts.com/media/1995/splzb-13x-technical-manual-smart-plug-mini.pdf

According to its manual, it uses Zigbee SE Metering Cluster id 0x0702 so might be issue with same ZHA limitations as in ->  https://github.com/zigpy/zha-device-handlers/issues/605

Also, not sure zigpy which ZHA depends on support DLMS/COSEM and Zigbee SEP (Zigbee Smart Energy Profile) specification?

https://github.com/zigpy/zigpy

https://zigbeealliance.org/wp-content/uploads/2019/11/docs-07-5356-19-0zse-zigbee-smart-energy-profile-specification.pdf

https://community.smartthings.com/t/dlms-and-zigbee-smart-energy-profile-sep-1-1/143411

https://e2e.ti.com/support/wireless_connectivity/zigbee_6lowpan_802-15-4_mac/f/158/t/661416?CC2538-Using-smart-energy-devices-in-z-stack-3-0

Hedda, 2021-05-19:
Develco also has several other products which also looks to use the same Zigbee Smart Energy specifications:

https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Develco

Examples:

- **Develco EMI Norwegian HAN**

https://github.com/zigpy/zha-device-handlers/issues/267

https://www.develcoproducts.com/products/meter-interfaces/emi-norwegian-han/

https://github.com/zigpy/zha-device-handlers/issues/857

https://www.develcoproducts.com/products/meter-interfaces/emi-norwegian-han/

- **Develco Prosumer Meter** (SMMZB-310101)

https://www.develcoproducts.com/products/meter-interfaces/prosumer-meter/

https://zigbeealliance.org/zigbee_products/prosumer-meter/

https://zigbeealliance.org/no/zigbee_Produkter/prosumer-meter/

Hedda, 2021-10-06:
FYI, it looks like support for more clusters attributes is now being worked on for [ZHA integration](https://www.home-assistant.io/integrations/zha) in Home Assistant core, see:

https://github.com/home-assistant/core/pull/56909

and

https://github.com/home-assistant/core/pull/56666

Also, note the last comment from Adminiuga in PR for ZHA in Home Assistant core -> https://github.com/home-assistant/core/pull/56909

"**_Support for other attributes could be extended later, as I couldn't find too many devices other than Sinopé RM3250ZB Electrical Load Controller – 50A – Zigbee https://www.amazon.com/dp/B07CGJ6XH2/ reporting anything interesting on the electrical measurement cluster._**"

So maybe now you could try to get this working with the latest Home Assistant beta or wait for the next upcoming release:

https://www.home-assistant.io/faq/release/

PS: Again, there is much more discussion about the current ZHA limitation of only supporting 1 sensor per cluster here -> https://github.com/zigpy/zha-device-handlers/issues/605

Adminiuga, 2021-10-06:
@Hedda stop spamming and posting same message bro every old thread. This is your last warning.