zigpy/zha-device-handlers #858: [Device Support Request] Philips Hue Wall Switch Module

aleex1848, 2021-04-19:
**Describe the solution you'd like**
There should be a zha_event triggered when the switch is pushed.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be ``
`Device 0x2ab8 (00:17:88:01:0b:00:12:18) joined the network
[0x2ab8:zdo] ZDO request ZDOCmd.Device_annce: [0x2AB8, 00:17:88:01:0b:00:12:18, 128]
[0x2ab8] Requesting 'Node Descriptor'
Tries remaining: 2
[0x2ab8] Extending timeout for 0x52 request
[0x2ab8] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4107, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)
[0x2ab8] Discovering endpoints
Tries remaining: 3
[0x2ab8] Extending timeout for 0x53 request
[0x2ab8] Discovered endpoints: [1]
[0x2ab8:1] Discovering endpoint information
Tries remaining: 3
[0x2ab8] Extending timeout for 0x54 request
[0x2ab8:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=2080, device_version=1, input_clusters=[0, 1, 3, 64512], output_clusters=[3, 4, 6, 8, 25])
[0x2ab8] Extending timeout for 0x55 request
light.osram_classic_b40_tw_lightify_e287a900_level_light_color_on_off: polling current state
[0x5a9a:3:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=86 command_id=Command.Read_Attributes_rsp>
[0x5a9a:3:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=87 command_id=Command.Read_Attributes_rsp>
[0x5A9A:3:0x0008]: received attribute: 0 update with value: 26
[0x5a9a:3:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=88 command_id=Command.Read_Attributes_rsp>
[0x2ab8:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=85 command_id=Command.Read_Attributes_rsp>
[0x2ab8:1] Manufacturer: Signify Netherlands B.V.
[0x2ab8:1] Model: RDM001
Checking quirks for Signify Netherlands B.V. RDM001 (00:17:88:01:0b:00:12:18)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
device - 0x2AB8:00:17:88:01:0b:00:12:18 entering async_device_initialized - is_new_join: True
device - 0x2AB8:00:17:88:01:0b:00:12:18 has joined the ZHA zigbee network
[0x2AB8](RDM001): started configuration
[0x2AB8:ZDO](RDM001): 'async_configure' stage succeeded
[0x2ab8] Extending timeout for 0x59 request
[0x2ab8] Extending timeout for 0x5a request
[0x2ab8] Extending timeout for 0x5b request
[0x2AB8:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
[0x2AB8:1:0x0000]: finished channel configuration
[0x2AB8:1:0x0001]: bound 'power' cluster: Status.SUCCESS
[0x2ab8] Extending timeout for 0x5c request
[0x2ab8] Extending timeout for 0x5d request
[0x2ab8:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=92 command_id=Command.Configure_Reporting_rsp>
[0x2AB8:1:0x0001]: reporting 'battery_voltage' attr on 'power' cluster: 3600/10800/1: Result: '[[ConfigureReportingResponseRecord(status=0)]]'
[0x2ab8] Extending timeout for 0x5e request
[0x2AB8:1:0xfc00]: bound 'manufacturer_specific' cluster: Status.SUCCESS
[0x2AB8:1:0xfc00]: finished channel configuration
[0x2ab8] Extending timeout for 0x5f request
[0x2ab8:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=94 command_id=Command.Configure_Reporting_rsp>
[0x2AB8:1:0x0001]: reporting 'battery_percentage_remaining' attr on 'power' cluster: 3600/10800/1: Result: '[[ConfigureReportingResponseRecord(status=0)]]'
[0x2AB8:1:0x0001]: finished channel configuration
[0x2ab8] Extending timeout for 0x60 request
[0x2AB8:1:0x0008]: bound 'level' cluster: Status.SUCCESS
[0x2AB8:1:0x0008]: finished channel configuration
[0x2AB8:1:0x0006]: bound 'on_off' cluster: Status.SUCCESS
[0x2AB8:1:0x0006]: finished channel configuration
[0x2AB8:1:0x0019]: bound 'ota' cluster: Status.SUCCESS
[0x2AB8:1:0x0019]: finished channel configuration
[0x2AB8:1:0x0000]: 'async_configure' stage succeeded
[0x2AB8:1:0x0001]: 'async_configure' stage succeeded
[0x2AB8:1:0xfc00]: 'async_configure' stage succeeded
[0x2AB8:1:0x0008]: 'async_configure' stage succeeded
[0x2AB8:1:0x0006]: 'async_configure' stage succeeded
[0x2AB8:1:0x0019]: 'async_configure' stage succeeded
[0x2AB8](RDM001): completed configuration
[0x2AB8](RDM001): stored in registry: ZhaDeviceEntry(name='Signify Netherlands B.V. RDM001', ieee='00:17:88:01:0b:00:12:18', last_seen=1618816060.5545013)
[0x2ab8] Extending timeout for 0x61 request
[0x2ab8:1:0x0003] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=97 command_id=Command.Default_Response>
[0x2AB8:1:0x0003]: executed 'trigger_effect' command with args: '(2, 0)' kwargs: '{}' result: [64, <Status.UNSUP_CLUSTER_COMMAND: 129>]
[0x2AB8](RDM001): started initialization
[0x2AB8:ZDO](RDM001): 'async_initialize' stage succeeded
[0x2AB8:1:0x0000]: initializing channel: from_cache: False
[0x2AB8:1:0x0000]: finished channel configuration
[0x2AB8:1:0x0001]: initializing channel: from_cache: False
[0x2ab8] Extending timeout for 0x62 request
[0x2AB8:1:0xfc00]: initializing channel: from_cache: False
[0x2AB8:1:0xfc00]: finished channel configuration
[0x2AB8:1:0x0008]: initializing channel: from_cache: False
[0x2AB8:1:0x0008]: finished channel configuration
[0x2AB8:1:0x0006]: initializing channel: from_cache: False
[0x2AB8:1:0x0006]: finished channel configuration
[0x2AB8:1:0x0019]: initializing channel: from_cache: False
[0x2AB8:1:0x0019]: finished channel configuration
[0x2ab8:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=98 command_id=Command.Read_Attributes_rsp>
[0x2ab8] Extending timeout for 0x63 request
[0x2ab8:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=99 command_id=Command.Read_Attributes_rsp>
[0x2AB8:1:0x0001]: finished channel configuration
[0x2AB8:1:0x0000]: 'async_initialize' stage succeeded
[0x2AB8:1:0x0001]: 'async_initialize' stage succeeded
[0x2AB8:1:0xfc00]: 'async_initialize' stage succeeded
[0x2AB8:1:0x0008]: 'async_initialize' stage succeeded
[0x2AB8:1:0x0006]: 'async_initialize' stage succeeded
[0x2AB8:1:0x0019]: 'async_initialize' stage succeeded
[0x2AB8](RDM001): power source: Battery or Unknown
[0x2AB8](RDM001): completed initialization`

Hedda, 2021-04-20:
Can you check and verify if that Philips Hue Wall Switch Module is a ZGP (Zigbee Green Power) device or not?

I am guessing that it is and if so then you should know that ZGP (Zigbee Green Power) is not yet supported by zigpy/ZHA, see:

https://github.com/zigpy/zigpy/issues/341

Developers are however working on it in https://github.com/zigpy/zigpy/pull/519 and https://github.com/zigpy/zigpy/pull/656

aleex1848, 2021-04-20:
> Can you check and verify if that Philips Hue Wall Switch Module is a ZGP (Zigbee Green Power) device or not?
> 
> I am guessing that it is and if so then you should know that ZGP is not yet supported by zigpy (or ZHA), see: [zigpy/zigpy#341](https://github.com/zigpy/zigpy/issues/341)
> 
> Developers are however working on it in [zigpy/zigpy#519](https://github.com/zigpy/zigpy/pull/519) and [zigpy/zigpy#656](https://github.com/zigpy/zigpy/pull/656)

I'm sorry, but i've no clue how to check if this device is ZGP. Never heard about that before.
On the main product page there is no info about ZGP https://www.philips-hue.com/de-de/p/hue-philips-hue-wall-switch-modul/8719514318045#specifications

Hedda, 2021-04-20:
> On the main product page there is no info about ZGP https://www.philips-hue.com/de-de/p/hue-philips-hue-wall-switch-modul/8719514318045#specifications

Ah, that specific device (Zigbee ID "RDM001") is always connected to mains-power and that means that it would never be a ZGP.

https://zigbee.blakadder.com/Philips_RDM001.html

https://zigbeealliance.org/zigbee_products/philips-hue-wall-switch-module/

ZGP are low-energy devices that normally are battery-operated by small coin-cell battery or use energy harvesting technology.

So you can ignore my comments and links about ZGP in regards to that device.

Adminiuga, 2021-04-20:
Do you get anything in the logs when you press buttons? Post the logs when you turn the switch on/off.

aleex1848, 2021-04-21:
> Do you get anything in the logs when you press buttons? Post the logs when you turn the switch on/off.

I activated debug logging like this:
https://www.home-assistant.io/integrations/zha/#debug-logging

But unfortunately there is nothing logged when i push the button.

Adminiuga, 2021-04-21:
Well, if device doesn't send anything, then the quirk unlikely to help. 
Can you try the following:
1. Unbind remote from the coordinator: devices -> your remote -> Manage zigbee clusters -> device binding -> pick your coordinator (what radio are you using) -> unbind. After you press unbind, wake up the remove by pressing any button on it
2. check what groups your coordinator is member of?
3. Bind the remote the one of the groups that coordinator is member of

post the separate logs of unbinding and binding. 
Alternatively you could put the lights you'd like to control with that remote into a new group and bind remote to that group and see if it control the lights

jonas18z, 2021-05-05:
Yes, this would be greate if wall switch module is supported

dieneuser, 2021-05-13:
@Adminiuga: Perhaps the following links will help. Both deconz and zigbee2mqtt support the switch:
deconz: [https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4566]
zigbee2mqtt: [https://github.com/Koenkk/zigbee-herdsman-converters/issues/2393]

@Hedda: It's not a ZGP device.

Adminiuga, 2021-05-13:
There's nothing specific in those other implementations. 
Out of the box zha should originate the events.

dieneuser, 2021-05-14:
@Adminiuga "

> Nothing specific

is probably not it. 
I got a wall switch and can register it with ZHA, but only see a sensor with battery status but i cannot see any ZHA event when the switch is pressed. But i can see much of "unknown cluster-specific commands" in the log:
```
2021-05-14 16:38:44 DEBUG (MainThread) [zigpy.zcl] [0xb6ad:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=45 command_id=Command.Report_Attributes>
2021-05-14 16:38:44 DEBUG (MainThread) [zigpy.zcl] [0xb6ad:1:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=27316>)]]
2021-05-14 16:38:44 DEBUG (MainThread) [zigpy.zcl] [0xb6ad:1:0x0400] Attribute report received: measured_value=27316
2021-05-14 16:38:52 DEBUG (MainThread) [zigpy.zcl] [0x2e33:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=106 command_id=Command.Report_Attributes>
2021-05-14 16:38:52 DEBUG (MainThread) [zigpy.zcl] [0x2e33:1:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=39043>)]]
2021-05-14 16:38:52 DEBUG (MainThread) [zigpy.zcl] [0x2e33:1:0x0400] Attribute report received: measured_value=39043
2021-05-14 16:38:55 DEBUG (MainThread) [zigpy.zcl] [0x655a:1:0xfc00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=True is_reply=True disable_default_response=True> manufacturer=4107 tsn=19 command_id=0>
2021-05-14 16:38:55 WARNING (MainThread) [zigpy.zcl] [0x655a:1:0xfc00] Unknown cluster-specific command 0
2021-05-14 16:38:55 DEBUG (MainThread) [zigpy.zcl] [0x655a:1:0xfc00] ZCL request 0x0000: b'\x01\x00\x000\x00!\x00\x00'
2021-05-14 16:38:55 DEBUG (MainThread) [zigpy.zcl] [0x655a:1:0xfc00] No handler for cluster command 0
2021-05-14 16:38:55 DEBUG (MainThread) [zigpy.zcl] [0x655a:1:0xfc00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=True is_reply=True disable_default_response=True> manufacturer=4107 tsn=20 command_id=0>
2021-05-14 16:38:55 WARNING (MainThread) [zigpy.zcl] [0x655a:1:0xfc00] Unknown cluster-specific command 0
2021-05-14 16:38:55 DEBUG (MainThread) [zigpy.zcl] [0x655a:1:0xfc00] ZCL request 0x0000: b'\x01\x00\x000\x02!\x01\x00'
2021-05-14 16:38:55 DEBUG (MainThread) [zigpy.zcl] [0x655a:1:0xfc00] No handler for cluster command 0
2021-05-14 16:38:55 DEBUG (MainThread) [zigpy.zcl] [0x655a:1:0xfc00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=True is_reply=True disable_default_response=True> manufacturer=4107 tsn=21 command_id=0>
2021-05-14 16:38:55 WARNING (MainThread) [zigpy.zcl] [0x655a:1:0xfc00] Unknown cluster-specific command 0
2021-05-14 16:38:55 DEBUG (MainThread) [zigpy.zcl] [0x655a:1:0xfc00] ZCL request 0x0000: b'\x01\x00\x000\x00!\x00\x00'
2021-05-14 16:38:55 DEBUG (MainThread) [zigpy.zcl] [0x655a:1:0xfc00] No handler for cluster command 0
2021-05-14 16:38:55 DEBUG (MainThread) [zigpy.zcl] [0x655a:1:0xfc00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=True is_reply=True disable_default_response=True> manufacturer=4107 tsn=22 command_id=0>
2021-05-14 16:38:55 WARNING (MainThread) [zigpy.zcl] [0x655a:1:0xfc00] Unknown cluster-specific command 0
2021-05-14 16:38:55 DEBUG (MainThread) [zigpy.zcl] [0x655a:1:0xfc00] ZCL request 0x0000: b'\x01\x00\x000\x02!\x01\x00'
2021-05-14 16:38:55 DEBUG (MainThread) [zigpy.zcl] [0x655a:1:0xfc00] No handler for cluster command 0
2021-05-14 16:38:56 DEBUG (MainThread) [zigpy.zcl] [0x655a:1:0xfc00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=True is_reply=True disable_default_response=True> manufacturer=4107 tsn=23 command_id=0>
2021-05-14 16:38:56 WARNING (MainThread) [zigpy.zcl] [0x655a:1:0xfc00] Unknown cluster-specific command 0
2021-05-14 16:38:56 DEBUG (MainThread) [zigpy.zcl] [0x655a:1:0xfc00] ZCL request 0x0000: b'\x01\x00\x000\x00!\x00\x00'
2021-05-14 16:38:56 DEBUG (MainThread) [zigpy.zcl] [0x655a:1:0xfc00] No handler for cluster command 0
2021-05-14 16:38:56 DEBUG (MainThread) [zigpy.zcl] [0xb4d4:1:0x0019] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=162 command_id=1>
2021-05-14 16:38:56 DEBUG (MainThread) [zigpy.zcl] [0xb4d4:1:0x0019] ZCL req
```

aleex1848, 2021-06-06:
I just updated from core-2021.4.3 to core-2021.6.2 but unfortunately nothing changed. There are still no zha_events when pushing the hue wall switch.

jonas18z, 2021-08-12:
@dieneuser I have exactly the same problem as you I can ad but only find the battery level. I have a hue dimmer which I can find all different buttons. Anyone figure ut how to use this? 

I have also updated the firmware on my conbeeII

dieneuser, 2021-08-13:
@jonas18z ZHA does not (yet) support the switch. As a workaround, I use the Deconz integration in parallel - also for existing Hue Tap switch modules. There, the wall switch is already supported in the beta phoscon web app.

jonas18z, 2021-08-13:
Ok, let’s hope that it will be supported. It’s only ZHA that misses the support now. I don’t think you can run conbee II at the same time in ZHA and deconz?

htodep, 2021-08-14:
How/where can we request ZHA support? Really need this device ;)

dieneuser, 2021-08-14:
> I don’t think you can run conbee II at the same time in ZHA and deconz?
That's right. I use a second Conbee II with deconz integration.

jonas18z, 2021-08-21:
Is there some guide how to add new devices. I will try to add it I was thinking to just modify the hue dimmer switch file to make this wall switch to work. But I don’t find anything here

MattWestb, 2021-08-21:
https://github.com/zigpy/zha-device-handlers#how-to-contribute

jonas18z, 2021-09-02:
Thanks, Installed 2021.9 and did a new search for device and the device is found and added via ZHA but now next problem is that it only founds two triggers "Device Offline" and "power battery level changes" For example my dimmer RWL021 founds all trigger on,off, dim up, dim down etc with RDM001. So maybe there is some more editing needed in the Python file but I can't find what?

MarvinSchenkel, 2021-09-23:
> Thanks, Installed 2021.9 and did a new search for device and the device is found and added via ZHA but now next problem is that it only founds two triggers "Device Offline" and "power battery level changes" For example my dimmer RWL021 founds all trigger on,off, dim up, dim down etc with RDM001. So maybe there is some more editing needed in the Python file but I can't find what?
> 
> EDIT: I don't se any zha_event when pushing a button.

I have the exact same issue. Has anyone found the reason for this?

jonas18z, 2021-09-27:
I have the same issue if some know what could be wrong or what I could try to make it work.

florianw12, 2021-11-12:
Hello I think I have the same issue but I am not sure. I am using a raspberry pi 4 with homeassistant in a docker container and the conbee 2 usb on the raspberry with deconz also in a container. I have managed to connect the hue wall switch and I get events in home assistant but I am not able to use them for anything. Wall switch also shows just as battary in devices
[image]

MattWestb, 2021-11-12:
In ZHA is normal switches (or if battery powered remotes) only getting one battery.
All things one switch is doing event and you is getting them by development, events, listening to events and putting in `zha_event` and clicking start listening to events and you is getting all in the log window.

One remote dont have any state like one light that can being on or off or dimmed and only sending events.

Also if Philips have making it good it shall having one switch configure cluster for changing the function of the wall switch but i cant saying if its having it then no one have posting one device signature.

MattWestb, 2021-11-13:
If some like implanting the switch configuration i think i have found the attribute that need being added in one quirk for getting it working.
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/3338039cb8a498f90f5000f0b118ded7ef963a84/general.xml#L211-L229

wigster, 2021-11-14:
> and you is getting them by development, events, listening to events and putting in `zha_event` and clicking start listening to events and you is getting all in the log window.

So does this mean that XHA supports the module or not yet? I was about to purchase a few but am hesitating.

water-escape, 2021-11-23:
Same here. Using ZHA. Did anyone get this to work via ZHA so it is useable in automations.

JoachimCanu, 2021-12-01:
Hello,

I see that this subject is not moving forward. the same as the Aqara MCCGQ14LM.
The only solution is to learn to contribute by integrating new products.
I have read the README.md : zigpy/zha-device-handlers

Is there a tutorial ? (if possible in FR, but English will do just as well)

tube0013, 2021-12-08:
Worked with @dmulcahey and got a rough quirk working for this device. device automation triggers are only for the 1st input.

still some thought to be done on how best to implement this in the code base, but this is usable as a custom zha quirk just drop the py file in the custom quirk folder as defined in your configuration.yaml under zha. <https://www.home-assistant.io/integrations/zha/#custom_quirks_path>

to enable the second input - go to the devices configuration page > Manage Clusters.

pick the Philips Basic cluster - pick the mode attribute, put 2 in the value and set attirbute (toggle switch to wake the device so it gets the setting.

[image]

[rdm001.py.zip](https://github.com/zigpy/zha-device-handlers/files/7673371/rdm001.py.zip)

dieneuser, 2021-12-11:
@tube0013 

Thank you for the quirk file.
But I can't manage to configure the switch to double rocker/Dual Push Button. No mode value (0,1,2,3) is taken over in the cluster manage window. When reading out the value is always enum8.undefined_0x00 or None. The switch remains in the default configuration as Single Rocker - but I really need the Dual Push Button mode (value=3).

here all possible value for the mode:
mode value="0" type="Single Rocker"
mode value="1" type="Single Push Button"
mode value="2" type="Dual Rocker"
mode value="3" type="Dual Push Button"

tube0013, 2021-12-11:
Was the switch awake when the attribute to change the mode is written? May want to enable debug logging and watch to be sure the attribute is received. I confirmed changing to mode 2 worked but did not try any others.

dieneuser, 2021-12-11:
Oh, thanks for pointing out the debug level. 
```
2021-12-11 16:39:59 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x432F](RDM001): set: None for attr: 52 to cluster: 0 for ept: 1 - res: [2, <Status.MALFORMED_COMMAND: 128>]
2021-12-11 16:39:59 DEBUG (MainThread) [homeassistant.components.zha.api] Set attribute for: cluster_id: [0] cluster_type: [in] endpoint_id: [1] attribute: [52] value: [None] manufacturer: [2] response: [[2, <Status.MALFORMED_COMMAND: 128>]]
```
I wrote the value in the wrong line in the dialog (manufacturer code), so the mode value is None. 

Now it worked. I use value 3 for Dual Push Button mode and it works perfect. 👍

htodep, 2022-01-05:
@tube0013 Thank you so much! it worked like a charm.
Is this a permanent fix or does it need some tweaking(works fine though)?

aleex1848, 2022-01-09:
> Worked with @dmulcahey and got a rough quirk working for this device. device automation triggers are only for the 1st input.
>
> still some thought to be done on how best to implement this in the code [...]

After many months  i'm finaly able to use the switch module. Thank you so much!!

WernerRH21, 2022-01-09:
@tube0013 
I am excited to find this solution, but unfortunately don't know exactly what to do. Could you please explain your sentence "just drop the py file in the custom quirk folder as defined in your configuration.yaml under zha" a littlebit more? I don't have a zha section in my configuration.yaml
Thanks in advance

tube0013, 2022-01-09:
you add the zha: section like this:

```
zha:
  custom_quirks_path: zha_custom_quirks
```

then in your config folder create a folder named zha_custom_quirks and within that drop the .py file that's in the zip I posted:

[image]

WernerRH21, 2022-01-09:
Thx for your quick reply.
Have the zip file smoothly overlooked :) Now the section entry is set and the file is in place and I see the PhilipsBasicCluster. Sorry for my maybe silly questions. But what do I have to do now? I come from ioBroker and are quite new to HA. Thanks for your patience.

WernerRH21, 2022-01-09:
This is all I need, or shall they also have an entity?
[image]

WernerRH21, 2022-01-09:
I created an automation to toggle a hue lamp when "Turn on" button pressed, but no success. Same automation with an Aqara Switch runs as expected.

htodep, 2022-01-09:
@WernerRH21 what kind of switch are you using? Don't forget to change the cluster settings as mentioned above if you have a double switch or something else. Check the logs when you press the switch and see what triggers com up. For automation use remote_button_short_press as type and turn_on as subtype.

WernerRH21, 2022-01-09:
A single push button switch and set the cluster settings like below and push the switch when I set zigbee attribute. "Set Zigbee attributes" blinks green.
[image]
Where can I check the log for this device? In the logbook is no entry when I select Hue Wall Switch entity.

htodep, 2022-01-10:
https://www.home-assistant.io/integrations/zha/#debug-logging

WernerRH21, 2022-01-10:
Hi @htodep 
Cool, thank you. I added the wall switch again and set the cluster setting. Should be fine now?
[image]
Where do I find for automation remote_button_short_press as type and turn_on as subtype. 
[image]

WernerRH21, 2022-01-10:
Amazing. I got it. Thank you so much guys. How much you can enjoy a simple smart light switch.

htodep, 2022-01-10:
Nice! I know the feeling 😎 How did you make it work at the end?

WernerRH21, 2022-01-10:
The debug-logging was important to see that the switch works. Then just "Turn on button" as trigger and bäähm, the light toggles. 
[image]

RaveGun, 2022-01-13:
Thank you!

I also followed the examples and hints and made it work on my desk.
Now I just have to drill and mount everything.

PS. Is there a possibility to connect the Switch directly to the Bulb so that an automation is not needed?

Thanks!

Matthiasfranck, 2022-01-30:
Thnx for this fix.

Can someone tell me with which version(s) of home assistant this is compatible? I am currently using `core-2021.11.5`.

ark-, 2022-03-16:
I have got the custom quirks working with my wall module. What's preventing this being merged into the repo and released to the next release of HA?

tube0013, 2022-03-16:
I'll try and get this into a proper PR so it can be put into a release.

tube0013, 2022-03-16:
#1434

m4dmin, 2022-03-24:
I'm trying to configure my Hue wall Switch Module as Dual Rocker, but I can't get it to work.

I've followed all the steps mentiones in this thread:

1. Added the Custom Quirk file

> you add the zha: section like this:
> 
> ```
> zha:
>   custom_quirks_path: zha_custom_quirks
> ```
> 
> then in your config folder create a folder named zha_custom_quirks and within that drop the .py file that's in the zip I posted:
> 
> [image]

2. Enabled the Second Input Mode via Manage Cluster: 

> After many months i'm finaly able to use the switch module. Thank you so much!!

3. Verfied the correct setting via Debug Log:

`2022-03-24 13:40:56 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x4A45](RDM001): set: 2 for attr: 52 to cluster: 0 for ept: 1 - res: [[WriteAttributesStatusRecord(status=<Status.SUCCESS: 0>)]]`
`2022-03-24 13:40:56 DEBUG (MainThread) [homeassistant.components.zha.api] Set attribute for: cluster_id: [0] cluster_type: [in] endpoint_id: [1] attribute: [52] value: [2] manufacturer: [None] response: [[[WriteAttributesStatusRecord(status=<Status.SUCCESS: 0>)]]]`

4. Created two automations, one for the first and one for the second switch.

But in the end I can only use the first switch.

- Am I missing something?
- How can I adress the second switch within an automation?

evpaassen, 2022-03-24:
> How can I adress the second switch within an automation?

@m4dmin I'm using an *Event* trigger to listen to a `zha_event` which I captured using the developer tools:

[image]

When you go into Developer tools > Events you can enter the `zha_event` type and listen to those events. If you flip your switch, this will show you the `device_ieee`.

m4dmin, 2022-03-24:
Thanks @evpaassen. That was the missing piece :)

RaveGun, 2022-04-06:
Hi, After the update to `Home Assistant 2022.4.0` the ZHA is not starting because of an error from the `rdm001.py`.
This is what is in the log recorded:

```
Error setting up entry ConBee II, s/n: DE2225070 - dresden elektronik ingenieurtechnik GmbH for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 335, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 99, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 529, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1029, in load_module
  File "<frozen importlib._bootstrap_external>", line 854, in load_module
  File "<frozen importlib._bootstrap>", line 274, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 711, in _load
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/zha_custom_quirks/rdm001.py", line 51, in <module>
    class PhilipsBasicCluster(CustomCluster, Basic):
  File "/usr/local/lib/python3.9/site-packages/zigpy/zcl/__init__.py", line 91, in __init_subclass__
    raise TypeError(
TypeError: `manufacturer_attributes` is deprecated. Copy the parent class's `attributes` dictionary and update it with your manufacturer-specific `attributes`. Make sure to specify that it is manufacturer-specific through the  appropriate constructor or tuple!
```

I do not understand the issue and how to modify the python file to make it work.

Any hints?

Thank you.

BrendanxP, 2022-04-06:
> Hi, After the update to `Home Assistant 2022.4.0` the ZHA is not starting because of an error from the `rdm001.py`. This is what is in the log recorded:
>
> ```
> Error setting up entry ConBee II, s/n: [...]

Exactly the same issue here. I deleted the ZHA quirks from my configuration.yaml file and everything works now!

tube0013, 2022-04-06:
the quirk was added to the most recent quirks release and is in the new HA release. if you installed it to custom_quirks, deleted the rdm001.py from the custom_quirks folder in your config.

RaveGun, 2022-04-06:
Oh, that was an easy fix.
Delete I can do :)
Thank you a lot.

maddingamer, 2022-04-08:
> > How can I adress the second switch within an automation?
> When you go into Developer tools > Events you can enter the `zha_event` type and listen to those events. If you flip your switch, this will show you the `device_ieee`.

When I flip switch 2 it does not trigger an zha_event. How do you make switch 2 work? 🤔 (Using zha on HA 2022.4)

deridderlab, 2022-04-23:
Thanks for sharing! Managed to get both the left_press and right_press enables through setting the Cluster Attributes. 

However.... 

Anyone has the issue that an actual button press issues both a "left_press" and "left_press_release" event? And in fact, the actual button release dues the same (also a "left_press" and "left_press_release")? Not a huge issue, but unexpected behavior...

niwi-hh, 2022-04-24:
I wonder if using the walls witch with ZHA can cause the battery to run low much faster than normally. Two of my four wall switches are out of power after using them about a month with ZHA. A third more close the my zigbee bridge is still fine as well as the fourth switch still controlled by a hue bridge. The batteries should last about 5 years. I bought the switches some months ago.

[13 more replies]

closed 2025-08-24 by TheJulianJES