zigpy/zha-device-handlers #860: Please add support for Avatto Tuya 1 button remote

miklosbagi, 2021-04-20:
**Is your feature request related to a problem? Please describe.**
TS0041 by _TZ3000_fkp5zyho seems to register itself as a switch in HA instead of a button.

**Describe the solution you'd like**
I'd like to request the quirk to be added to be added to support this device.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0006"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_fkp5zyho",
  "model": "TS0041",
  "class": "zigpy.device.Device"
}
```

**Additional context**

Pairing:
```
Sending 'zdo_leave_req' failed: 
[0x1E30:1:0x0b04]: async_update
[0x682E:1:0x0b04]: async_update
[0xB3AC:1:0x0b04]: async_update
[0x9B2A:1:0x0b04]: async_update
[0xC42D:1:0x0b04]: async_update
[0xc42d:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=239 command_id=Command.Read_Attributes_rsp>
[0xb3ac:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=237 command_id=Command.Read_Attributes_rsp>
[0x682e:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=236 command_id=Command.Read_Attributes_rsp>
[0x9b2a:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=238 command_id=Command.Read_Attributes_rsp>
[0x1e30:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=235 command_id=Command.Read_Attributes_rsp>
[0x8926:1:0x0b04]: async_update
[0x8926:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=240 command_id=Command.Read_Attributes_rsp>
Device 0x3c4b (60:a4:23:ff:fe:45:fe:d2) joined the network
[0x3c4b:zdo] ZDO request ZDOCmd.Device_annce: [0x3C4B, 60:a4:23:ff:fe:45:fe:d2, 128]
[0x3c4b] Requesting 'Node Descriptor'
Tries remaining: 2
[0x3c4b] Extending timeout for 0xf1 request
[0x3c4b] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)
[0x3c4b] Discovering endpoints
Tries remaining: 3
[0x3c4b] Extending timeout for 0xf2 request
[0x3c4b] Discovered endpoints: [1]
[0x3c4b:1] Discovering endpoint information
Tries remaining: 3
[0x3c4b] Extending timeout for 0xf3 request
[0x3c4b:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=0, device_version=1, input_clusters=[0, 1, 6], output_clusters=[25, 10])
[0x3c4b] Extending timeout for 0xf4 request
[0x3c4b:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=244 command_id=Command.Read_Attributes_rsp>
[0x3c4b:1] Manufacturer: _TZ3000_fkp5zyho
[0x3c4b:1] Model: TS0041
Checking quirks for _TZ3000_fkp5zyho TS0041 (60:a4:23:ff:fe:45:fe:d2)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
[... 74 lines cut ...]
[0x682E:1:0x0b04]: async_update
[0xB3AC:1:0x0b04]: async_update
[0x9B2A:1:0x0b04]: async_update
[0xC42D:1:0x0b04]: async_update
[0xc42d:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=3 command_id=Command.Read_Attributes_rsp>
[0x8926:1:0x0b04]: async_update
[0xb3ac:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=1 command_id=Command.Read_Attributes_rsp>
...
```

When I push the button, this seems to get into the HA logs:
```
Logger: zigpy.zcl
Source: /usr/local/lib/python3.8/site-packages/zigpy/zcl/__init__.py:110 
First occurred: 9:29:50 AM (2 occurrences) 
Last logged: 9:29:51 AM

[0x0e8c:1:0x0006] Unknown cluster-specific command 253
```

**The two buttons variant works fine**

Home Assistant OS 2021.4.3 / ZHA + ZZH Stick

A fix as part of #595 / #611 seems to have addressed some of these.

dr-matter, 2021-06-23:
I have sth like this

it show battery and swith

{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0401",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0006",
        "0x000a"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_3oitjgaa",
  "model": "TS0041",
  "class": "zigpy.device.Device"
}

miklosbagi, 2021-06-23:
Cool - well I opened this a while back and asked the channel if this is the right place / level of info to get something included in future releases and I've got a "yep" back.
Nonetheless the level of activity here makes me wonder: is there a better place to raise this?

Gershy13, 2021-08-12:
> Cool - well I opened this a while back and asked the channel if this is the right place / level of info to get something included in future releases and I've got a "yep" back.
> Nonetheless the level of activity here makes me wonder: is there a better place to raise this?

I have a similar switch but branded lonsonho. 
Any luck fixing it?

TS0041
TZ3000_fkp5zyho

MattWestb, 2021-08-12:
The commit https://github.com/zigpy/zha-device-handlers/pull/957 is making all TS0041 devices working (until tuya is changing the clusters) its merged but no new ZHAQuirks is released so you must installing the ts0041.py as one local quirk and it shall working.

Gershy13, 2021-08-12:
> The commit #957 is making all TS0041 devices working (until tuya is changing the clusters) its merged but no new ZHAQuirks is released so you must installing the ts0041.py as one local quirk and it shall working.

Where would i place that file in my home assistant instal?

MattWestb, 2021-08-13:
Adding local quirk is described in https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274 and its only copy the updated `ts0041.py` in your configured quirk folder and  restarting HA after adding the configuration.

Dont forgetting deleting the local quirk then its being updated in the future then  its being added in the HA  or its overriding the package in ZHA.

You can see if the quirk is loaded on the device card information it shall showing the name of the loaded quirk in  the last line.

closed 2022-02-16 by github-actions[bot]