zigpy/zha-device-handlers #861: [Device Support Request] Legrand Wireless Double  Switch

Oxydiums, 2021-04-21:
**Is your feature request related to a problem? Please describe.**
Hi everyone, today i paired a legrand double switch to my home assistant but the only entity is the battery sensor and actually it doesn't work properly. I have no switch entities 

**Device signature**
{
  "node_descriptor": "NodeDescriptor(byte1=18, byte2=64, mac_capability_flags=128, manufacturer_code=4129, maximum_buffer_size=89, maximum_incoming_transfer_size=63, server_mask=10752, maximum_outgoing_transfer_size=63, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=True)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0104",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x000f",
        "0x0020",
        "0xfc01"
      ],
      "out_clusters": [
        "0x0000",
        "0x0003",
        "0x0006",
        "0x0008",
        "0x0019",
        "0xfc01"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0104",
      "in_clusters": [
        "0x0003",
        "0x000f"
      ],
      "out_clusters": [
        "0x0003",
        "0x0006",
        "0x0008"
      ]
    }
  },
  "manufacturer": " Legrand",
  "model": " Double gangs remote switch",
  "class": "zigpy.device.Device"
}

**Additional context**
In first place i paired the switch to Legrand Gateway to make sure the version is updated, and it will, version seem like to be version 8.
I'm actually not at home and can't have the log details when paired with HA.

After some research i found some topics, like the clusters list from Legrand website :
NLD - Double OnOff dim. remote:
https://developer.legrand.com/documentation/product-cluster-list/

I don't know if a new quirk is required or if i made a mistake in the pairing process.
Could you guys help me ?
Thanks a lot for your help.

Oxydiums, 2021-04-21:
I found some logs i don't know if this is what you need, tell me and i will provide you wht is needed ;)

2021-04-21 08:14:55 INFO (MainThread) [zigpy.application] Device 0x9462 (00:04:74:00:00:82:66:2b) joined the network
2021-04-21 08:14:55 INFO (MainThread) [zigpy.device] [0x9462] Requesting 'Node Descriptor'
2021-04-21 08:14:55 DEBUG (MainThread) [zigpy.util] Tries remaining: 2
2021-04-21 08:14:55 DEBUG (MainThread) [zigpy.device] [0x9462] Extending timeout for 0xc0 request
2021-04-21 08:14:55 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0x9462, 0, <ZDOCmd.Node_Desc_req: 0x0002>, 0, 0, 192, b'\xc0b\x94', True, False)
2021-04-21 08:14:55 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x0530 b'029462000000020000000003c06294'
2021-04-21 08:14:55 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x0f\xf9\x02\x94b\x00\x00\x00\x02\x00\x00\x00\x00\x03\xc0b\x94'
2021-04-21 08:14:55 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x02\x1f\xf9\x02\x12\x94b\x02\x10\x02\x10\x02\x10\x02\x12\x02\x10\x02\x10\x02\x10\x02\x10\x02\x13\xc0b\x94'
2021-04-21 08:14:55 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 800000079e002c0530000000
2021-04-21 08:14:55 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'002c05300000' LQI:0
2021-04-21 08:14:55 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 44, 1328, b'\x00\x00']
2021-04-21 08:14:55 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8002001a700000000013000002946202fffd0062942b6682000074040080c6
2021-04-21 08:14:55 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'0000000013000002946202fffd0062942b6682000074040080' LQI:198
2021-04-21 08:14:55 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0, 19, 0, 0, <Address address_mode=ADDRESS_MODE.NWK address=0x9462>, <Address address_mode=ADDRESS_MODE.NWK address=0xfffd>, b'\x00b\x94+f\x82\x00\x00t\x04\x00\x80']
2021-04-21 08:14:55 DEBUG (MainThread) [zigpy.zdo] [0x9462:zdo] ZDO request ZDOCmd.Device_annce: [0x9462, 00:04:74:00:00:82:66:2b, 128]
2021-04-21 08:14:55 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 87010005870000040000
2021-04-21 08:14:55 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8701 b'00000400' LQI:0
2021-04-21 08:14:55 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0]
2021-04-21 08:14:56 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80110008c00094620080022d00
2021-04-21 08:14:56 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8011 b'0094620080022d' LQI:0
2021-04-21 08:14:56 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0x9462, 0, 32770, 45]
2021-04-21 08:14:56 DEBUG (MainThread) [zigpy_zigate.zigbee.application] ACK Data received 45 0
2021-04-21 08:15:04 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 801100085e0094620100003000
2021-04-21 08:15:04 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8011 b'00946201000030' LQI:0
2021-04-21 08:15:04 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0x9462, 1, 0, 48]
2021-04-21 08:15:04 DEBUG (MainThread) [zigpy_zigate.zigbee.application] ACK Data received 48 0
2021-04-21 08:15:04 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8002001cb6000104fc01010102946202000015211002092b6682000074040001b7
2021-04-21 08:15:04 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'000104fc01010102946202000015211002092b6682000074040001' LQI:183
2021-04-21 08:15:04 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 260, 64513, 1, 1, <Address address_mode=ADDRESS_MODE.NWK address=0x9462>, <Address address_mode=ADDRESS_MODE.NWK address=0x0000>, b'\x15!\x10\x02\t+f\x82\x00\x00t\x04\x00\x01']
2021-04-21 08:15:04 DEBUG (MainThread) [zigpy.application] Received frame on uninitialized device 00:04:74:00:00:82:66:2b (Status.NEW) for endpoint: 1
2021-04-21 08:15:04 DEBUG (MainThread) [zigpy.device] [0x9462] Received ZCL while uninitialized on endpoint id 1, cluster 0xfc01 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=True is_reply=False disable_default_response=True> manufacturer=4129 tsn=2 command_id=9>, payload: b'+f\x82\x00\x00t\x04\x00\x01'
2021-04-21 08:15:04 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8002001cb4000104fc01010102946202000015211003092b6682000074040002b7
2021-04-21 08:15:04 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'000104fc01010102946202000015211003092b6682000074040002' LQI:183
2021-04-21 08:15:04 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 260, 64513, 1, 1, <Address address_mode=ADDRESS_MODE.NWK address=0x9462>, <Address address_mode=ADDRESS_MODE.NWK address=0x0000>, b'\x15!\x10\x03\t+f\x82\x00\x00t\x04\x00\x02']
2021-04-21 08:15:04 DEBUG (MainThread) [zigpy.application] Received frame on uninitialized device 00:04:74:00:00:82:66:2b (Status.NEW) for endpoint: 1
2021-04-21 08:15:04 DEBUG (MainThread) [zigpy.device] [0x9462] Received ZCL while uninitialized on endpoint id 1, cluster 0xfc01 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=True is_reply=False disable_default_response=True> manufacturer=4129 tsn=3 command_id=9>, payload: b'+f\x82\x00\x00t\x04\x00\x02'
2021-04-21 08:15:23 DEBUG (MainThread) [zigpy.util] Tries remaining: 1
[... cut ...]
2021-04-21 08:15:30 DEBUG (MainThread) [zigpy.zcl] [0x9462:1:0x0001] ZCL request 0x0001: [[ReadAttributeRecord(attrid=49, status=<Status.UNSUPPORTED_ATTRIBUTE: 134>), ReadAttributeRecord(attrid=51, status=<Status.UNSUPPORTED_ATTRIBUTE: 134>)]]
2021-04-21 08:15:30 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x9462]( Double gangs remote switch): power source: Battery or Unknown
2021-04-21 08:15:30 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x9462]( Double gangs remote switch): completed initialization

TheJulianJES, 2021-04-27:
If you listen for ``zha_event`` in the Home Assistant developers section and press some button on the remote, do you get some output in there?

pipiche38, 2021-04-28:
be aware that if you are on the latest Legrand firmware for the Remote Switch, then you have only 3 actions possibles:

- Toggle (when pressing the up or down button)
- Dim+
- Dim-
the On/Off are not sent anymore

Oxydiums, 2021-04-28:
Hi thanks for your answer, yes actually i have found some data frome the zha_event :

{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:04:74:00:00:82:66:2b",
        "unique_id": "00:04:74:00:00:82:66:2b:2:0x0006",
        "device_id": "4b288b9543396ee4110c0b5c9f0e7d3f",
        "endpoint_id": 2,
        "cluster_id": 6,
        "command": "toggle",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2021-04-28T18:05:57.883688+00:00",
    "context": {
        "id": "1242c6fd8f68e7f9a351b8329672eac6",
        "parent_id": null,
        "user_id": null
    }
}

Maybe i could use this directly in node red to command some lights ?

And your right @pipiche38 i only have toggle command ;)

TheJulianJES, 2021-04-28:
A quirk with device automation triggers could be created if ``zha_event``s get sent for each action.
Are there any differences (args, cluster_id, ...) when pressing the different buttons?
(Would be nice to get an ``zha_event`` for pressing each button normally (and a long press if that sends another event).)

Oxydiums, 2021-04-29:
here it is !

Bottom Left button long press release:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:04:74:00:00:82:66:2b",
        "unique_id": "00:04:74:00:00:82:66:2b:1:0x0008",
        "device_id": "4b288b9543396ee4110c0b5c9f0e7d3f",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "stop",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2021-04-29T06:02:35.428325+00:00",
    "context": {
        "id": "494f1c88ceec9e45f5baf79506f68fab",
        "parent_id": null,
        "user_id": null
    }
}
Bottom Left button long press:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:04:74:00:00:82:66:2b",
        "unique_id": "00:04:74:00:00:82:66:2b:1:0x0008",
        "device_id": "4b288b9543396ee4110c0b5c9f0e7d3f",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "move",
        "args": [
            1,
            100
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2021-04-29T06:02:33.062736+00:00",
    "context": {
        "id": "bfc2dd298deb87e93902bee025ff68a4",
        "parent_id": null,
        "user_id": null
    }
}
Up Left button long press release:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:04:74:00:00:82:66:2b",
        "unique_id": "00:04:74:00:00:82:66:2b:1:0x0008",
        "device_id": "4b288b9543396ee4110c0b5c9f0e7d3f",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "stop",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2021-04-29T06:02:30.025777+00:00",
    "context": {
        "id": "ceb2149cfd25c3cf4dce4af1863d8885",
        "parent_id": null,
        "user_id": null
    }
}
Up Left button long press :
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:04:74:00:00:82:66:2b",
[... 170 lines cut ...]
        "unique_id": "00:04:74:00:00:82:66:2b:2:0x0006",
        "device_id": "4b288b9543396ee4110c0b5c9f0e7d3f",
        "endpoint_id": 2,
        "cluster_id": 6,
        "command": "toggle",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2021-04-29T06:08:00.295993+00:00",
    "context": {
        "id": "f3370a4100740f9183942a1b9d5c3239",
        "parent_id": null,
        "user_id": null
    }
}

Seems like we have no difference  between up and right press in short press, but we have : 
short press  >  cluster_id = 6   >  left endpoint_id =1  >  right endpoint_id = 2 (command = toggle)

long press up  >  cluster_id =  8  >  left endpoint_id =1 and arg = 0  >  right endpoint_id = 2 and arg = 0 (command = move)
long press down >  cluster_id =  8  >  left endpoint_id =1 and arg = 1  >  right endpoint_id = 2 and arg = 1  (command = move)

and same filters for release command = stop

I made some node red stuff and it works fine, filtering in device_ieee and clusters.

closed 2021-11-02 by github-actions[bot]