zigpy/zha-device-handlers #862: [Device Support Request] Neo Tuya Temperature, Humidity and Illumination Sensor TS0201 _TZ3000_qaaysllp

blakadder, 2021-04-21:
https://zigbee.blakadder.com/Neo_NAS-TH02B.html device in question

**Is your feature request related to a problem? Please describe.**
ZHA recognises the illumination sensor and battery. Temperature and Humidity don't have an obvious endpoint.

**Describe the solution you'd like**
Add a quirk for temperature and humidity sensors.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
Device 0x8c8d (84:2e:14:ff:fe:e5:4b:50) left the network
[0x2d91:2:0x0702] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=159 command_id=Command.Report_Attributes>
[0x2d91:2:0x0702] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint48_t, value=1>), Attribute(attrid=1024, value=<TypeValue type=int24s, value=0>), Attribute(attrid=512, value=<TypeValue type=bitmap8, value=bitmap8.0>), Attribute(attrid=256, value=<TypeValue type=uint48_t, value=281474976710655>), Attribute(attrid=258, value=<TypeValue type=uint48_t, value=281474976710655>)]]
[0x2d91:2:0x0702] Attribute report received: current_summ_delivered=1, instantaneous_demand=0, status=0, current_tier1_summ_delivered=281474976710655, current_tier2_summ_delivered=281474976710655
Device 0x8c8d (84:2e:14:ff:fe:e5:4b:50) joined the network
[0x8c8d] Requesting 'Node Descriptor'
Tries remaining: 2
[0x8c8d] Extending timeout for 0x0f request
Device 0x8c8d (84:2e:14:ff:fe:e5:4b:50) joined the network
Skip initialization for existing device 84:2e:14:ff:fe:e5:4b:50
[0x8c8d:zdo] ZDO request ZDOCmd.Device_annce: [0x8C8D, 84:2e:14:ff:fe:e5:4b:50, 128]
[0x8c8d] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)
[0x8c8d] Discovering endpoints
Tries remaining: 3
[0x8c8d] Extending timeout for 0x11 request
[0x8c8d] Discovered endpoints: [1]
[0x8c8d:1] Discovering endpoint information
Tries remaining: 3
[0x8c8d] Extending timeout for 0x13 request
Received frame on uninitialized device 84:2e:14:ff:fe:e5:4b:50 endpoint 1 for cluster: 1024
[0x8c8d:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=262, device_version=1, input_clusters=[0, 1, 1024, 57346], output_clusters=[25, 10])
Unknown cluster 57346
[0x8c8d] Extending timeout for 0x15 request
[0x8c8d:zdo] ZDO request ZDOCmd.Node_Desc_req: [0x0000]
[0x8c8d:zdo] Unsupported ZDO request:ZDOCmd.Node_Desc_req
Received frame on uninitialized device 84:2e:14:ff:fe:e5:4b:50 endpoint 1 for cluster: 1024
[0x8c8d:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=21 command_id=Command.Read_Attributes_rsp>
[0x8c8d:1] Manufacturer: _TZ3000_qaaysllp
[0x8c8d:1] Model: TS0201
Checking quirks for _TZ3000_qaaysllp TS0201 (84:2e:14:ff:fe:e5:4b:50)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
device - 0x8C8D:84:2e:14:ff:fe:e5:4b:50 entering async_device_initialized - is_new_join: True
device - 0x8C8D:84:2e:14:ff:fe:e5:4b:50 has joined the ZHA zigbee network
[0x8C8D](TS0201): started configuration
[0x8C8D:ZDO](TS0201): 'async_configure' stage succeeded
[0x8c8d] Extending timeout for 0x17 request
[0x8c8d] Extending timeout for 0x19 request
[0x8c8d] Extending timeout for 0x1b request
[0x8c8d:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=55 command_id=Command.Report_Attributes>
[0x8c8d:1:0x0400] ZCL request 0x000a: [[Attribute(attrid=61441, value=<TypeValue type=uint16_t, value=50>)]]
[0x8c8d:1:0x0400] Attribute report received: 61441=50
[0x8C8D:1:0x0400]: bound 'illuminance' cluster: Status.SUCCESS
[0x8c8d] Extending timeout for 0x1e request
[0x8C8D:1:0x0001]: bound 'power' cluster: Status.SUCCESS
[0x8c8d] Extending timeout for 0x20 request
[... 56 lines cut ...]
[0x8c8d:1:0xe002] Attribute report received: 53254=2
[0x8c8d:1:0xe002] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=61 command_id=Command.Report_Attributes>
[0x8c8d:1:0xe002] ZCL request 0x000a: [[Attribute(attrid=53263, value=<TypeValue type=enum8, value=enum8.undefined_0x02>)]]
[0x8c8d:1:0xe002] Attribute report received: 53263=2
[0x8c8d:1:0xe002] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=62 command_id=Command.Report_Attributes>
[0x8c8d:1:0xe002] ZCL request 0x000a: [[Attribute(attrid=53264, value=<TypeValue type=uint8_t, value=0>)]]
[0x8c8d:1:0xe002] Attribute report received: 53264=0
[0x2fa3:39:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=110 command_id=Command.Report_Attributes>
[0x2fa3:39:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=5657>)]]
[0x2fa3:39:0x0400] Attribute report received: measured_value=5657
[0x8c8d:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=65 command_id=Command.Report_Attributes>
[0x8c8d:1:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=15799>)]]
[0x8c8d:1:0x0400] Attribute report received: measured_value=15799
[0x8c8d:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=66 command_id=Command.Read_Attributes>
[0x8c8d:1:0x000a] ZCL request 0x0000: [[7]]
[0x8c8d:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=67 command_id=Command.Report_Attributes>
[0x8c8d:1:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=15799>)]]
[0x8c8d:1:0x0400] Attribute report received: measured_value=15799
[0x8c8d:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=69 command_id=Command.Report_Attributes>
[0x8c8d:1:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=15799>)]]
[0x8c8d:1:0x0400] Attribute report received: measured_value=15799
```

**Additional context**
Add any other context or screenshots about the feature request here.

Vebryn, 2021-05-18:
Same here, according to https://github.com/zigpy/zha-device-handlers/blob/dev/README.md, quirk is based on a Node Descriptor and a Simple Descriptor :

```
[0x8c8d:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=262, device_version=1, input_clusters=[0, 1, 1024, 57346], output_clusters=[25, 10])
``` 

Each clusters must be associated, at Home Assistant startup, values from identified clusters are displayed :

```
2021-05-18 23:06:23 DEBUG (MainThread) [zigpy.appdb] [0x920d:1:0x0000] Attribute id: 4 value: _TZ3000_qaaysllp
2021-05-18 23:06:23 DEBUG (MainThread) [zigpy.appdb] [0x920d:1:0x0000] Attribute id: 5 value: TS0201
2021-05-18 23:06:23 DEBUG (MainThread) [zigpy.appdb] [0x920d:1:0x0000] Attribute id: 4 value: _TZ3000_qaaysllp
2021-05-18 23:06:23 DEBUG (MainThread) [zigpy.appdb] [0x920d:1:0x0000] Attribute id: 5 value: TS0201
2021-05-18 23:06:23 DEBUG (MainThread) [zigpy.appdb] [0x920d:1:0x0001] Attribute id: 32 value: 30
2021-05-18 23:06:23 DEBUG (MainThread) [zigpy.appdb] [0x920d:1:0xe002] Attribute id: 53258 value: 40
2021-05-18 23:06:23 DEBUG (MainThread) [zigpy.appdb] [0x920d:1:0xe002] Attribute id: 53259 value: 10
2021-05-18 23:06:23 DEBUG (MainThread) [zigpy.appdb] [0x920d:1:0xe002] Attribute id: 53261 value: 90
2021-05-18 23:06:23 DEBUG (MainThread) [zigpy.appdb] [0x920d:1:0xe002] Attribute id: 53262 value: 20
2021-05-18 23:06:23 DEBUG (MainThread) [zigpy.appdb] [0x920d:1:0xe002] Attribute id: 53254 value: 2
2021-05-18 23:06:23 DEBUG (MainThread) [zigpy.appdb] [0x920d:1:0xe002] Attribute id: 53264 value: 0
2021-05-18 23:06:23 DEBUG (MainThread) [zigpy.appdb] [0x920d:1:0x0001] Attribute id: 33 value: 120
2021-05-18 23:06:23 DEBUG (MainThread) [zigpy.appdb] [0x920d:1:0xe002] Attribute id: 53263 value: 2
2021-05-18 23:06:23 DEBUG (MainThread) [zigpy.appdb] [0x920d:1:0x0400] Attribute id: 0 value: 11762
```

0x0400 is input_cluster 1024
11762 = 15 lux (no idea how value is converted)

```
1024 -> IlluminanceMeasurement.cluster_id
```

I think 57346 cluster include both temperature and humidity values. Quirk readme contains a lot of other usefull infos.

NB : Sensor is on blakadder list : https://zigbee.blakadder.com/Neo_NAS-TH02B.html

Vebryn, 2021-05-19:
According to https://zigbeealliance.org/wp-content/uploads/2019/12/07-5123-06-zigbee-cluster-library-specification.pdf

Illuminance calculation (p281) :

```
10,000 x log10Illuminance + 1
``` 

Clusters related to tempand humidity (p278) :

```
0x0402 Temperature Measurement 
0x0405 Relative Humidity Measurement
```

Vebryn, 2021-05-19:
21.3°C and 56.3% :

```
2021-05-19 21:06:00 DEBUG (MainThread) [zigpy.appdb] [0x920d:1:0xe002] Attribute id: 53258 value: 40
2021-05-19 21:06:00 DEBUG (MainThread) [zigpy.appdb] [0x920d:1:0xe002] Attribute id: 53259 value: 10
2021-05-19 21:06:00 DEBUG (MainThread) [zigpy.appdb] [0x920d:1:0xe002] Attribute id: 53261 value: 90
2021-05-19 21:06:00 DEBUG (MainThread) [zigpy.appdb] [0x920d:1:0xe002] Attribute id: 53262 value: 20
2021-05-19 21:06:00 DEBUG (MainThread) [zigpy.appdb] [0x920d:1:0xe002] Attribute id: 53254 value: 2
2021-05-19 21:06:00 DEBUG (MainThread) [zigpy.appdb] [0x920d:1:0xe002] Attribute id: 53264 value: 0
2021-05-19 21:06:00 DEBUG (MainThread) [zigpy.appdb] [0x920d:1:0xe002] Attribute id: 53263 value: 2
```

I will update debug according to new values.

Vebryn, 2021-05-24:
Tuya cluster is 0xEF00 according to https://github.com/Koenkk/zigbee-herdsman/blob/master/src/zcl/definition/cluster.ts
What is 0xe002 cluster ? Does Tuya change its cluster id ? What is 0x55aa header into https://developer.tuya.com/en/docs/iot/tuya-zigbee-module-uart-communication-protocol?id=K9ear5khsqoty#title-2-Frame%20format%20description ?

So many questions ! I'm wondering if it will be easier to debug with a zigbee gateway and a sniffer.

blakadder, 2021-05-24:
It could simply be NEO's special formatting, some manufacturers tend to bend the Tuya spec and tack on their own

Vebryn, 2021-05-24:
Think you're right, hope sensor will periodically send its data.

Using Zigbee2MQTT, just found meaning of some messages :

Humidity alarm (>90%, not sure about this threshold)
```
0x08 0x3e 0x0a 0x0f 0xd0 0x30 0x01
```

Humidity alarm back to normal
```
0x08 0x4b 0x0a 0x0f 0xd0 0x30 0x02
```

Back to normal alarm is also send after pairing. Following message may indicate a Temperature alarm (4th byte 0x06 instead of 0x0f) :

```
0x08 0x69 0x0a 0x06 0xd0 0x30 0x02
```

marciogranzotto, 2021-06-07:
I'm also interested on getting the temperature and humidity of this sensor. Please let me know if I can provide any additional data

marciogranzotto, 2021-06-11:
What I find strange is that the temperature and humidity show up great on the SmartLife app, but the illuminance always shows at 0 there.
I wonder if the Tuya app doesn't change the device to temp/humidity mode or something...

Vebryn, 2021-06-11:
According to https://www.aliexpress.com/item/1005002434961203.html?spm=a2g0s.9042311.0.0.15874c4dZYVRAj

```
Note: when pair with the app, some function will be not working, please don't worry, it is because the gateway needs to be updated, tuya will send the update firmware on the end of May, please don't open dispute. And if you want to quick install the update firmware, please contact us and tell us your zigbee gateway ID, we will send to Tuya and you will receive the firmware sooner.
```

This is 0 illuminance issue.

What kind of Zigbee dongle do you have ? You can use it to sniff Tuya bridge data. I think Tuya bridge send a NEO specific data to query temp and humidity.

marciogranzotto, 2021-06-12:
@Vebryn I'm using a CC2531 with ZHA, but I have a Zemismart Wireless Zigbee hub that I'm not using.
I can try to sniff it, but I have no idea on how to do it

wdesponts, 2021-06-15:
Hey everyone, I have this sensor working at 50%, hopefully we can help each other resolve the remainder. I have been able to get all the data except the Temp and Humidity values.  Those are part of a new cluster type that needs to be defined...more on that later.

To get the rest working try the following:

1. Edit the Neo.js file to add the new device
- Navigate to `/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices` 
- Edit `neo.js`
- Add the following below the NEW-AB02B0 Device
`{
        fingerprint: [{modelID: 'TS0201', manufacturerName: '_TZ3000_qaaysllp'}],
        zigbeeModel: ['0yu2xgi'],
        model: 'NAS-TH01',
        vendor: 'Neo',
        description: 'Temperature, humidity & illuminance display',
        fromZigbee: [fz.battery, fz.temperature, fz.humidity, fz.illuminance],
        toZigbee: [],
        exposes: [e.battery(), e.illuminance(), e.illuminance_lux().withUnit('lx'), e.humidity(), e.temperature()],
																	},
		},`

2. Remove the device if already added to your system
3. Restart zigbee2mqtt service
4. Pair the Neo SAS-TH01 device to Zigbee2Mqtt

You'll have access to the battery, illuminance and illuminance_lux data  

Now the for the new cluster work, I'll need to differ to others.  I was able to find the right file thanks to @Vebryn 's input but I'm at a loss as to editing that file.   For that I'll need some help.

Vebryn, 2021-06-15:
> 
> 
> @Vebryn I'm using a CC2531 with ZHA, but I have a Zemismart Wireless Zigbee hub that I'm not using.
> I can try to sniff it, but I have no idea on how to do it

Great, there is some [tutorials ](https://community.oh-lalabs.com/t/guide-build-a-zigbee-cc2531-sniffer-how-to-use-it/469)to flash CC2531 dongle and use Wireshark to sniff Zigbee data.

1. flash cc2531
2. try sniffing data
3. unpair neo device
4. start sniffing and pair neo device

We will see queries from tuya bridge and responses. It will help implementing neo device handler.

marciogranzotto, 2021-06-15:
[image]
look at this guy, happly reporting temperature on `0x0402`!

I'm trying to decrypt the logs, but I don't have any experience with it. Here's the logs of adding the device to my Zermismart hub and it reporting back temperature and humidity:
[adding_and_getting_values.zip](https://github.com/zigpy/zha-device-handlers/files/6658869/adding_and_getting_values.zip)

Vebryn, 2021-06-16:
Great job ! Hopelessly trace is encrypted, can you provide related keys ?

Best regards.

MattWestb, 2021-06-16:
If pairing one device you is having the network key in the sniff only look for transport key short after the device have joining the network and adding it in the key table for Zigbee in wireshark.

Vebryn, 2021-06-16:
oh, I simply add 5a6967426565416c6c69616e63653039 key from [this link](https://community.oh-lalabs.com/t/guide-build-a-zigbee-cc2531-sniffer-how-to-use-it/469)

MattWestb, 2021-06-16:
Yes that is working if not using Zigbee 3 and not updating the network key and using the well known one  :-))

Dont being surprised is some neighbor is starting controlling your device !!

Vebryn, 2021-06-16:
Here is a focus on Report Attributes, first 8 packets appears after device inclusion (Tuya and non Tuya gateway) :

[image]

Vebryn, 2021-06-16:
Device doen't announce temp and hum capabilities,

[image]

Does Zigpy only consider reports from previously announced capabilities ?

Tuya gateway also force reading on default clusters :

[image]

Are we able to force reading using Zigpy ?

MattWestb, 2021-06-16:
Add the missing cluster in the quirk that is for :-))

Vebryn, 2021-06-16:
As I remember, I already try forcing temp and hum like @wdesponts 
I will check again and put my quirk here

Best regards.

MattWestb, 2021-06-16:
tuya devices is normally using "tuya cluster" but they is not always being presented on many TRVs so the quirk is adding it so the system is knowing it can / shall being used.

I dont knowing how to do it but in the end it shall being added in the replacement part of the quirk.

pipiche38, 2021-06-16:
TS0201 do not use the 0xef00 Tuya cluster, and indeed TS0201 use the standard cluster to provide Temperature and Humidity

MattWestb, 2021-06-16:
Indeed but i was only pointing in the direction for finding how to adding missing clusters in on quirk ;-))

marciogranzotto, 2021-06-16:
So, I tried to create a quirk just adding the clusters, but it didn't work.
Then I realized that temperature and humidity are on endpoint 2, not 1!
[image]

So I tried:
```python
class TemperatureHumidtyIlluminanceSensor(CustomDevice):
    """Neo Tuya Temperature, Humidity and Illumination Sensor."""

    signature = {
        #  <SimpleDescriptor endpoint=1, profile=260, device_type=262
        #  device_version=1
        #  input_clusters=[0, 1, 1024, 57346]
        #  output_clusters=[25, 10]>
        MODELS_INFO: [("_TZ3000_qaaysllp", "TS0201")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: DeviceType.LIGHT_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    IlluminanceMeasurement.cluster_id,
                    0xE002,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    IlluminanceMeasurement.cluster_id,
                    0xE002,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
            2: {
                INPUT_CLUSTERS: [
                    TemperatureMeasurement.cluster_id,
                    RelativeHumidity.cluster_id,
                ]
            },
        },
    }
```

However, I get this error on Home Assitant:
```
2021-06-16 13:12:06 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry TI CC2531 USB CDC, s/n: __0X00124B001CD4B395 - Texas Instruments for zha
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/config_entries.py", line 293, in async_setup
result = await component.async_setup_entry(hass, self) # type: ignore
File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 102, in async_setup_entry
await zha_gateway.async_initialize()
File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 157, in async_initialize
self.application_controller = await app_controller_cls.new(
File "/usr/local/lib/python3.8/site-packages/zigpy/application.py", line 64, in new
await app._load_db()
File "/usr/local/lib/python3.8/site-packages/zigpy/application.py", line 56, in _load_db
await self._dblistener.load()
File "/usr/local/lib/python3.8/site-packages/zigpy/appdb.py", line 473, in load
device = zigpy.quirks.get_device(device)
File "/usr/local/lib/python3.8/site-packages/zigpy/quirks/__init__.py", line 42, in get_device
return _DEVICE_REGISTRY.get_device(device)
File "/usr/local/lib/python3.8/site-packages/zigpy/quirks/registry.py", line 148, in get_device
device = candidate(device._application, device.ieee, device.nwk, device)
File "/usr/local/lib/python3.8/site-packages/zigpy/quirks/__init__.py", line 96, in __init__
self.add_endpoint(endpoint_id, replace_device=replaces)
File "/usr/local/lib/python3.8/site-packages/zigpy/quirks/__init__.py", line 111, in add_endpoint
ep = custom_ep_type(self, endpoint_id, replacement_data, replace_device)
File "/usr/local/lib/python3.8/site-packages/zigpy/quirks/__init__.py", line 126, in __init__
set_device_attr(SIG_EP_PROFILE)
File "/usr/local/lib/python3.8/site-packages/zigpy/quirks/__init__.py", line 124, in set_device_attr
setattr(self, attr, getattr(replace_device[endpoint_id], attr))
File "/usr/local/lib/python3.8/site-packages/zigpy/device.py", line 359, in __getitem__
return self.endpoints[key]
KeyError: 2
```

looks like I can't add a replacement for endpoint 2 if it's not present on the signature. And I think I can't just add it to the signature since the device doesn't report that endpoint on discovery...

MattWestb, 2021-06-16:
I think its also possible making one virtual end point in one quirk but how to do that i think only @Adminiuga  is knowing.

Adminiuga, 2021-06-16:
Yeah, i don't think quirk could add non existing endpoints. Tuyas are such abomination.

You could try to define a custom device and override ` handle_message()` method and if src_ep == 2 replace it with 1 and call original handle_message.

Then you could add humidity and temp clusters to the 1st endpoints in replacement

marciogranzotto, 2021-06-16:
Hah! Success!
[image]

I added the new endpoint in the init function:
```python
class TemperatureHumidtyIlluminanceSensor(CustomDevice):
    """Neo Tuya Temperature, Humidity and Illumination Sensor."""

    def __init__(self, *args, **kwargs):
        """Init."""
        super().__init__(*args, **kwargs)
        # Add missing endpoint 2
        sensor_endpoint = self.add_endpoint(2)
        sensor_endpoint.status = zigpy.endpoint.Status.ZDO_INIT
        sensor_endpoint.profile_id = zha.PROFILE_ID
        sensor_endpoint.device_type = DeviceType.TEMPERATURE_SENSOR
        sensor_endpoint.add_input_cluster(
            TemperatureMeasurement.cluster_id, TemperatureMeasurement(endpoint=sensor_endpoint, is_server=True)
        )
        sensor_endpoint.add_input_cluster(RelativeHumidity.cluster_id, RelativeHumidity(endpoint=sensor_endpoint, is_server=True))

    signature = {
        #  <SimpleDescriptor endpoint=1, profile=260, device_type=262
        #  device_version=1
        #  input_clusters=[0, 1, 1024, 57346]
        #  output_clusters=[25, 10]>
        MODELS_INFO: [("_TZ3000_qaaysllp", "TS0201")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: DeviceType.LIGHT_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    IlluminanceMeasurement.cluster_id,
                    0xE002,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    IlluminanceMeasurement.cluster_id,
                    0xE002,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
        },
    }
```

Opening a PR 🎉

albalaing, 2021-06-17:
Thank you very much for your excellent work. I have added it in my quirk directory and the temperature data is not updated

[image]

Mikk36, 2021-06-17:
I get the same too, temperature and humidity stay at "Unknown"
Also, when joining the device into the network:
```
Device 0x32dc (b4:e3:f9:ff:fe:0c:4b:34) joined the network
Device b4:e3:f9:ff:fe:0c:4b:34 changed id (0x474f => 0x32dc)
[0x32dc:zdo] ZDO request ZDOCmd.Device_annce: [0x32DC, b4:e3:f9:ff:fe:0c:4b:34, 128]
device - 0x32DC:b4:e3:f9:ff:fe:0c:4b:34 entering async_device_initialized - is_new_join: True
device - 0x32DC:b4:e3:f9:ff:fe:0c:4b:34 has joined the ZHA zigbee network
[0x32DC](TS0201): started configuration
[0x32DC:ZDO](TS0201): 'async_configure' stage succeeded
[0x32dc] Extending timeout for 0x1f request
[0x32DC:1:0x0000]: finished channel configuration
[0x32dc] Extending timeout for 0x20 request
[0x32DC:1:0x0019]: finished channel configuration
[0x32dc] Extending timeout for 0x21 request
[0x32DC:1:0x0400]: bound 'illuminance' cluster: Status.SUCCESS
[0x32dc] Extending timeout for 0x22 request
[0x32DC:1:0x0001]: bound 'power' cluster: Status.SUCCESS
[0x32dc] Extending timeout for 0x23 request
[0x32DC:2:0x0402]: bound 'temperature' cluster: Status.SUCCESS
[0x32dc] Extending timeout for 0x24 request
[0x32dc:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=44 command_id=Command.Report_Attributes>
[0x32dc:1:0x0400] ZCL request 0x000a: [[Attribute(attrid=61441, value=<TypeValue type=uint16_t, value=50>)]]
[0x32dc:1:0x0400] Attribute report received: 61441=50
[0x32dc:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=34 command_id=Command.Default_Response>
[0x32DC:1:0x0400]: reporting 'measured_value' attr on 'illuminance' cluster: 30/900/1: Result: '[6, <Status.UNSUP_GENERAL_COMMAND: 130>]'
[0x32dc] Extending timeout for 0x25 request
Sending 'zdo_leave_req' failed: 
[0x32DC:1:0x0001]: failed to set reporting for 'battery_voltage' attr on 'power' cluster: 
[0x32dc] Extending timeout for 0x26 request
[0x32DC:2:0x0402]: failed to set reporting for 'measured_value' attr on 'temperature' cluster: 
[0x32DC:2:0x0402]: finished channel configuration
[0x32DC:2:0x0402]: 'async_configure' stage succeeded
[0x32DC:2:0x0405]: 'async_configure' stage failed: 
[0x32DC:1:0x0001]: failed to set reporting for 'battery_percentage_remaining' attr on 'power' cluster: Request failed after 5 attempts: <Status.MAC_TRANSACTION_EXPIRED: 240>
[0x32DC:1:0x0001]: finished channel configuration
[0x32DC:1:0x0400]: 'async_configure' stage failed: 'uint8_t' object is not subscriptable
[0x32DC:1:0x0000]: 'async_configure' stage succeeded
[0x32DC:1:0x0001]: 'async_configure' stage succeeded
[0x32DC:1:0x0019]: 'async_configure' stage succeeded
[0x32DC](TS0201): completed configuration
[0x32DC](TS0201): stored in registry: ZhaDeviceEntry(name='_TZ3000_qaaysllp TS0201', ieee='b4:e3:f9:ff:fe:0c:4b:34', last_seen=1623942267.298252)
[0x32DC](TS0201): started initialization
[0x32DC:ZDO](TS0201): 'async_initialize' stage succeeded
[0x32DC:1:0x0400]: initializing channel: from_cache: False
[0x32dc] Extending timeout for 0x27 request
[0x32DC:1:0x0000]: initializing channel: from_cache: False
[0x32DC:1:0x0000]: finished channel configuration
[0x32DC:1:0x0001]: initializing channel: from_cache: False
[0x32dc] Extending timeout for 0x28 request
[0x32DC:1:0x0019]: initializing channel: from_cache: False
[0x32DC:1:0x0019]: finished channel configuration
[0x32DC:2:0x0402]: initializing channel: from_cache: False
[0x32dc] Extending timeout for 0x29 request
[0x32DC:1:0x0400]: failed to get attributes '['measured_value']' on 'illuminance' cluster: Request failed after 5 attempts: <Status.MAC_TRANSACTION_EXPIRED: 240>
[0x32DC:1:0x0400]: finished channel configuration
[0x32DC:2:0x0405]: initializing channel: from_cache: False
[0x32dc] Extending timeout for 0x2a request
[0x32DC:1:0x0001]: failed to get attributes '['battery_voltage', 'battery_percentage_remaining']' on 'power' cluster: Request failed after 5 attempts: <Status.MAC_TRANSACTION_EXPIRED: 240>
[0x32dc] Extending timeout for 0x2b request
```

MattWestb, 2021-06-17:
`<Status.MAC_TRANSACTION_EXPIRED: 240>` = the device is not communicating so very likely it have falling in sleep.

Trye repairing it and find some way keeping it awake by pressing one button (if it have) so its not sleeping and can talking with the coordinator.

Mikk36, 2021-06-17:
Got that part nailed, but still no temp/humi measurements. Reconfiguring succeeds on binds, but fails on measurements.

marciogranzotto, 2021-06-18:
> Thank you very much for your excellent work. I have added it in my quirk directory and the temperature data is not updated

Hmm.. I tested it differently, using `hassio-addons-development` according with the docs:
https://github.com/zigpy/zha-device-handlers#testing-new-releases

Maybe try using that instead and see if it works then.
My config:
```yaml
pypi:
  - >-
    git+https://github.com/marciogranzotto/zha-device-handlers.git@dev#zha-quirks
apk: []
```

Don't forget to restart home assistant before testing

MattWestb, 2021-06-18:
And dont forget deleting the `/config/deps/*` after updating HA and like using the "new quirk" in the release or the local is overriding its and cam making problem later :-))

marciogranzotto, 2021-06-18:
ok, so I removed one of my sensors and temperature and humidity stopped working...
when I was testing, in some point I updated zigpy to 0.34.0 and later came back to 0.33.0
I though this was the problem, since zigpy seems to have refatored the setup process. But I couldn't get it working again...

I will try to fix it and update the PR. My guess is that it has something to do with the endpoint 2 creation while not adding a node descriptor (but I'm not what I'm talking about here)

albalaing, 2021-06-19:
These are the data of my installation:
```
System Health
Version				core-2021.6.5
Installation Type		Home Assistant Core
Development			false
Supervisor			false
Docker				false
Virtual Environment		true
Python Version			3.8.5
Operating System Family		Linux
Operating System Version	5.4.0-72-generic
CPU Architecture		x86_64
Timezone			Europe/Madrid
```

marciogranzotto, 2021-06-20:
Update:
I tried several things and could not get the temperature and humidity to report again.
I even added logs to `handle_message` but looks like the device is simply not reporting the endpoint 2 clusters.

Then I removed one of my Neo devices from my zha network, paired with the Zemismart hub on the SmartLife app, removed with "wipe all settings" options, and re-added it to zha.
And there it goes, it starts happily reporting the temperature and humidity again. 🤦‍♂️ 
Then I can see the `handle_message` logs showing the state reports.

So... I have no idea...
I'm trying to look deeper at the Wireshark logs that I uploaded here, but I can't find anything that could cause this behavior.

Any ideas?

Vebryn, 2021-06-21:
In addition to Report Attributes, there is also Read Attributes. Read Attributes are related to temperature, humidity and power. There is no illumination.

Power Read Attributes complies with Zigbee standard. They issue Read Attributes Response with corresponding attributes :

[image]
[image]

Temperature and humidity Read Attributes get no answer from device.

Vebryn, 2021-06-21:
Illumination reporting start before 0xe002 cluster init. Temperature reporting start after 0xe002 cluster init.

Default Response are issued with SUCCESS from Tuya agateway. What are Zigpy answers ?

[image]

MattWestb, 2021-06-21:
I think sequence 17-23 is tuya commands / attribute set on the cluster 0xe002 but i cant see if its on EP 1 or 2.
Is the sequence 16 and 24 sent to EP 1 ??

The power cluster looks being pity Zigbee standard and can report and being red OK then tuya have doing little more magic on the other cluster / attribute.

If getting data from cluster 0xe002 ZHA must sending default response or the device is stop sending attribute reports to the coordinator. Zigbee standard commands (ZCL) is the zigbee stack doing that but cluster 0xe002 is not part of ZCL so it must being done "manual" in Zigpy.

Vebryn, 2021-06-21:
> 
> 
> I think sequence 17-23 is tuya commands / attribute set on the cluster 0xe002 but i cant see if its on EP 1 or 2.
> Is the sequence 16 and 24 sent to EP 1 ??
> 

I updated pic with Src and Dst EP.

ali1234, 2021-06-26:
I wrote some discoveries about this device here: https://github.com/Koenkk/zigbee2mqtt/issues/7741#issuecomment-869028778

TL;DR 0xe002 is the alarm system and you can get it to do stuff by bringing temperature above 40 or below 10, or by bringing humidity above 90 or below 20. It will emit a message whenever crossing those thresholds. The thresholds are shown in the initial pairing messages:
```
2021-05-18 23:06:23 DEBUG (MainThread) [zigpy.appdb] [0x920d:1:0xe002] Attribute id: 53258 value: 40
2021-05-18 23:06:23 DEBUG (MainThread) [zigpy.appdb] [0x920d:1:0xe002] Attribute id: 53259 value: 10
2021-05-18 23:06:23 DEBUG (MainThread) [zigpy.appdb] [0x920d:1:0xe002] Attribute id: 53261 value: 90
2021-05-18 23:06:23 DEBUG (MainThread) [zigpy.appdb] [0x920d:1:0xe002] Attribute id: 53262 value: 20
```

Also in that thread, another user reports that their device reports the second endpoint normally. Mine does not however. It appears that there are different firmware revisions of this device with the same identifiers?

marciogranzotto, 2021-06-26:
Thanks @ali1234 !

So updating @Vebryn table:

| atribute | value | description |
|  --- | --- | --- |
| 0xd00a | 40 | temperature alarm ceiling  |
| 0xd00b | 10 | temperature alarm floor |
| 0xd00d | 90 | humidity alarm ceiling |
| 0xd00e | 20 | humidity alarm floor |
| 0xd006 | 0x01 | start of temperature alarm |
| 0xd006 | 0x02 | end of temperature alarm |
| 0xd00f | 0x01 | start of humidity alarm |
| 0xd00f | 0x02 | end of humidity alarm |
| 0xd010 | 0x00 | ? |

ali1234, 2021-06-26:
It sends 0x00 when the reading is below the alarm minimum and 0x01 when it is above the maximum.

Vebryn, 2021-06-26:
Remaining endpoint 2 mystery !

marciogranzotto, 2021-06-26:
Right, here's your data decoded:
## Humidity (0xd00f)

humidity reading to go above 90%
```
8		0x08	frame control: profile wide
16		0x10	sequence number
10		0x0a	command: report attr
15		0x0f	attr = 0xd00f
208		0xd0
48		0x30	data type: 8-bit enum
1		0x01	max alarm triggered
```

humidity went back below 90%
```
8		0x08	frame control: profile wide
29		0x1d	sequence number
10		0x0a	command: report attr
15		0x0f	attr = 0xd00f
208		0xd0
48		0x30	data type: 8-bit enum
2		0x02	alarm off
```

humidity down to 19%
```
8		0x08	frame control: profile wide
42		0x2a	sequence number
10		0x0a	command: report attr
15		0x0f	attr = 0xd00f
208		0xd0
48		0x30	data type: 8-bit enum
0		0x00	min alarm triggered
```

when it went back up to 22%
```
8		0x08	frame control: profile wide
52		0x34	sequence number
10		0x0a	command: report attr
15		0x0f	attr = 0xd00f
208		0xd0
48		0x30	data type: 8-bit enum
2		0x02	alarm off
```
## Temperature (0xd00e):

temperature went above 40
```
8		0x08	frame control: profile wide
97		0x61	sequence number
10		0x0a	command: report attr
6		0x06	attr = 0xd006
208		0xd0
48		0x30	data type: 8-bit enum
1		0x01	max alarm triggered
```

temperature dropped to 39 degrees
```
8		0x08	frame control: profile wide
65		0x41	sequence number
10		0x0a	command: report attr
6		0x06	attr = 0xd006
208		0xd0
48		0x30	data type: 8-bit enum
2		0x02	alarm off
```

So it looks like the enum is:
```
0x02 alarm off
0x01 max alarm
0x00 min alarm
```

marciogranzotto, 2021-06-27:
Ok, I just pushed a new commit to my fork. Can you guys test if it starts reporting temp/humidity for you?
Seems to be working for my sensors.

ali1234, 2021-06-27:
With the quirk it shows as supporting temp and humidity in the UI, but there is no data. Here is the pairing log:

```
Device 0x6eea (b4:e3:f9:ff:fe:07:a4:b6) joined the network
[0x6eea:zdo] ZDO request ZDOCmd.Device_annce: [0x6EEA, b4:e3:f9:ff:fe:07:a4:b6, 128]
[0x6eea] Requesting 'Node Descriptor'
Tries remaining: 2
[0x6eea] Extending timeout for 0x06 request
[0x6eea] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)
[0x6eea] Discovering endpoints
Tries remaining: 3
[0x6eea] Extending timeout for 0x07 request
[0x6eea] Discovered endpoints: [1]
[0x6eea:1] Discovering endpoint information
Tries remaining: 3
[0x6eea] Extending timeout for 0x08 request
[0x6eea:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=262, device_version=1, input_clusters=[0, 1, 1024, 57346], output_clusters=[25, 10])
Unknown cluster 57346
[0x6eea] Extending timeout for 0x09 request
[0x6eea:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=9 command_id=Command.Read_Attributes_rsp>
[0x6eea:1] Manufacturer: _TZ3000_qaaysllp
[0x6eea:1] Model: TS0201
Checking quirks for _TZ3000_qaaysllp TS0201 (b4:e3:f9:ff:fe:07:a4:b6)
Considering <class 'zhaquirks.tuya.neo_TS0201.TemperatureHumidtyIlluminanceSensor'>
Found custom device replacement for b4:e3:f9:ff:fe:07:a4:b6: <class 'zhaquirks.tuya.neo_TS0201.TemperatureHumidtyIlluminanceSensor'>
Unknown cluster 57346
[0x6eea:2] Forcing discovery information for Neo custom endpoint
device - 0x6EEA:b4:e3:f9:ff:fe:07:a4:b6 entering async_device_initialized - is_new_join: True
device - 0x6EEA:b4:e3:f9:ff:fe:07:a4:b6 has joined the ZHA zigbee network
[0x6EEA](TS0201): started configuration
[0x6EEA:ZDO](TS0201): 'async_configure' stage succeeded
[0x6eea] Extending timeout for 0x0a request
[0x6eea] Extending timeout for 0x0b request
[0x6EEA:1:0x0000]: finished channel configuration
[0x6EEA:1:0x0019]: finished channel configuration
[0x6eea] Extending timeout for 0x0c request
[0x6EEA:1:0x0400]: bound 'illuminance' cluster: Status.SUCCESS
[0x6eea] Extending timeout for 0x0d request
[0x6EEA:1:0x0001]: bound 'power' cluster: Status.SUCCESS
[0x6eea] Extending timeout for 0x0e request
[0x6EEA:2:0x0402]: bound 'temperature' cluster: Status.SUCCESS
[0x6eea] Extending timeout for 0x0f request
[0x6eea:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=17 command_id=Command.Report_Attributes>
[0x6eea:1:0x0400] ZCL request 0x000a: [[Attribute(attrid=61441, value=<TypeValue type=uint16_t, value=50>)]]
[0x6eea:1:0x0400] Attribute report received: 61441=50
[0x6eea:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=13 command_id=Command.Default_Response>
[0x6EEA:1:0x0400]: reporting 'measured_value' attr on 'illuminance' cluster: 30/900/1: Result: '[6, <Status.UNSUP_GENERAL_COMMAND: 130>]'
[0x6eea] Extending timeout for 0x10 request
[0x6eea:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=14 command_id=Command.Default_Response>
[0x6EEA:1:0x0001]: reporting 'battery_voltage' attr on 'power' cluster: 3600/10800/1: Result: '[6, <Status.UNSUP_GENERAL_COMMAND: 130>]'
[0x6EEA:1:0x0400]: 'async_configure' stage failed: 'uint8_t' object is not subscriptable
[0x6EEA:1:0x0001]: 'async_configure' stage failed: 'uint8_t' object is not subscriptable
[0x6EEA:1:0x0000]: 'async_configure' stage succeeded
[0x6EEA:1:0x0019]: 'async_configure' stage succeeded
[0x6eea:1:0xe002] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=18 command_id=Command.Report_Attributes>
[0x6eea:1:0xe002] ZCL request 0x000a: [[Attribute(attrid=53258, value=<TypeValue type=int16s, value=40>)]]
[0x6eea:1:0xe002] Attribute report received: 53258=40
[0x6EEA:2:0x0405]: bound 'humidity' cluster: Status.SUCCESS
[0x6eea] Extending timeout for 0x11 request
[0x6eea:1:0xe002] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=19 command_id=Command.Report_Attributes>
[0x6eea:1:0xe002] ZCL request 0x000a: [[Attribute(attrid=53259, value=<TypeValue type=int16s, value=10>)]]
[0x6eea:1:0xe002] Attribute report received: 53259=10
[... cut ...]
[0x6eea:1:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=0>)]]
[0x6eea:1:0x0400] Attribute report received: measured_value=0
[0x6eea:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=40 command_id=Command.Report_Attributes>
[0x6eea:1:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=0>)]]
[0x6eea:1:0x0400] Attribute report received: measured_value=0
```

marciogranzotto, 2021-06-27:
I can't figure this out... if I remove the batteries from the sensor, then try to pair it directly to ZHA, it doesn't report anything on endpoint 2.
But if I first add it to Smart Life with tuya's hub, then remove it and add it to ZHA, it sends temperature and humidity data.

I think next week I'll reflash my CC2531 and try to sniff the paring process again... We must be missing something there...

ali1234, 2021-07-17:
I have made a pcap of Tuya pairing and reporting. I can see that, after pairing and fetching the endpoint information, the coordinator sends three Read Attributes commands to the device. After this the device starts periodically reporting temperature, humidity and luminance. The luminance value does not display properly in the Tuya app (always shows 0), but I can see in the packet dump that the true value is being reported.

I can't see any other communications from the coordinator that look like they could enable temp and humidity.

I don't know if there are any security implications of sharing the full dump, so here are just the Read Attributes commands decoded:

```
No.     Time           Source                Destination           Protocol Length Info
     76 11.851521      0x0000                0x9241                ZigBee HA 50     ZCL: Read Attributes, Seq: 14

ZigBee Application Support Layer Data, Dst Endpt: 255, Src Endpt: 1
    Frame Control Field: Data (0x00)
        .... ..00 = Frame Type: Data (0x0)
        .... 00.. = Delivery Mode: Unicast (0x0)
        ..0. .... = Security: False
        .0.. .... = Acknowledgement Request: False
        0... .... = Extended Header: False
    Destination Endpoint: 255
    Cluster: Relative Humidity Measurement (0x0405)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 203
ZigBee Cluster Library Frame, Command: Read Attributes, Seq: 14
    Frame Control Field: Profile-wide (0x10)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .0.. = Manufacturer Specific: False
        .... 0... = Direction: Client to Server
        ...1 .... = Disable Default Response: True
    Sequence Number: 14
    Command: Read Attributes (0x00)
    Attribute: Measured Value (0x0000)

No.     Time           Source                Destination           Protocol Length Info
     80 11.859822      0x0000                0x9241                ZigBee HA 50     ZCL: Read Attributes, Seq: 15

ZigBee Application Support Layer Data, Dst Endpt: 255, Src Endpt: 1
    Frame Control Field: Data (0x00)
        .... ..00 = Frame Type: Data (0x0)
        .... 00.. = Delivery Mode: Unicast (0x0)
        ..0. .... = Security: False
        .0.. .... = Acknowledgement Request: False
        0... .... = Extended Header: False
    Destination Endpoint: 255
    Cluster: Temperature Measurement (0x0402)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 204
ZigBee Cluster Library Frame, Command: Read Attributes, Seq: 15
    Frame Control Field: Profile-wide (0x10)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .0.. = Manufacturer Specific: False
        .... 0... = Direction: Client to Server
        ...1 .... = Disable Default Response: True
    Sequence Number: 15
    Command: Read Attributes (0x00)
    Attribute: Measured Value (0x0000)

No.     Time           Source                Destination           Protocol Length Info
     84 11.868702      0x0000                0x9241                ZigBee HA 52     ZCL: Read Attributes, Seq: 16

ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x40)
        .... ..00 = Frame Type: Data (0x0)
        .... 00.. = Delivery Mode: Unicast (0x0)
        ..0. .... = Security: False
        .1.. .... = Acknowledgement Request: True
        0... .... = Extended Header: False
    Destination Endpoint: 1
    Cluster: Power Configuration (0x0001)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 205
ZigBee Cluster Library Frame, Command: Read Attributes, Seq: 16
    Frame Control Field: Profile-wide (0x10)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .0.. = Manufacturer Specific: False
        .... 0... = Direction: Client to Server
        ...1 .... = Disable Default Response: True
    Sequence Number: 16
    Command: Read Attributes (0x00)
    Attribute: Battery Voltage (0x0020)
    Attribute: Battery Percentage Remaining (0x0021)
```

MattWestb, 2021-07-17:
One sniffing with joining one device (with "normal zigbee keys") is also containing the current network key being exposed. 
The network key is also needed for encrypting the traffic if not doing one paring its need needs being handled over to the sniffing  "reader".

If you is doing one sniff of one production network it can being good changing the network key after the the testing is over or  you can living  with it if your neighbors is not good sniffers and knowing where you have exposing you network key.

I dont knowing if tuya can changing the network key by deleting and adding the ZBGW one more time in  the app (i have "converting" my 2 ZBGW to ZHA GWs).

Pedder007, 2021-07-18:
All in all it seems to be, that Tuya is raising the hurdles to use their deveices with other GWs than their own ones. If they continue and others might follow, that might lead to the end, that only real specialists will be able to capture respective data into open source solutions in the future. Sad ... :-(

Vebryn, 2021-08-05:
> I can't figure this out... if I remove the batteries from the sensor, then try to pair it directly to ZHA, it doesn't report anything on endpoint 2.
> But if I first add it to Smart Life with tuya's hub, then remove it and add it to ZHA, it sends temperature and humidity data.
> 
> I think next week I'll reflash my CC2531 and try to sniff the pairing process again... We must be missing something there...
> 
> Update: I'm waiting for a new CC2652R to arrive so I don't have to sacrifice my CC2531 and recreate all my network later. Should be here in 1 to 2 weeks, then I will give another go

Also ordered a CC2652R from slae.sh. I flashed it using Flash Programmer 2 and precompiled sniffer fw. Firmware disable bootloader. Flash with caution !

Pedder007, 2021-08-05:
got a Tuya gateway today and had the hope that first pairing the device to it might convince it later to send temp and humidity also to deconz. but it didn't work ... 😢 
waiting now further on here ...

marciogranzotto, 2021-08-06:
So I finally received my CC2652R and start using that on ZHA, now I'm using my older CC2531 to sniff the packets.
I got basically the same result as @ali1234 
What I think is the catch, is that for the first temperature/humidity request, the Tuya hub requests it over the 255 endpoint, not 2!

[image]

Then the sensor seems to respond on endpoint 2, and later all other requests are on endpoint 2:
[image]

I'm attaching my wireshark logs if you guys want to take a look.
[adding_first_time.zip](https://github.com/zigpy/zha-device-handlers/files/6947521/adding_first_time.zip)

axute, 2021-08-06:
"255" sounds like a broadcast

Adminiuga, 2021-08-06:
Yes, 255 is a broadcast endpoint. Send a request to it during cluster configuration?

marciogranzotto, 2021-08-07:
> Yes, 255 is a broadcast endpoint. Send a request to it during cluster configuration?

That's what I'm thinking. Just need to find a way to do that in the code.
Also I noticed my sensor connected to ZHA sometimes gets disconnected and starts blinking, but the one on the Tuya hub stays connected all the time.
I'm going to run a long wireshark log to see if there's any weird behavior that is missing on ZHA

marciogranzotto, 2021-08-12:
Ok, I managed to send the 255 requests, but that doesn't seem to fix the issue...

Another difference that I found, is that the sensor requests the hub's Node Descriptor, which then the hub responds to.
Here's the tuya's hub response:
[image]

And here's the ZHA response:
[image]

I tried setting `manufacturer_id_override` to `0x1002` but I think that overrides the device's manufacturer id, not the hub's id.
Is there a way to change the one from the actual coordinator?

Adminiuga, 2021-08-12:
btw, if you have an EZSP based radio, it could be possible to change the Manufacturer Code, but it would require modifications to `bellows` package. There's EZSP command to update it
[image]

[49 more replies]

closed 2023-09-09 by github-actions[bot]