zigpy/zha-device-handlers #863: [Device Support Request] IKEA Remote N2 Styrbar

hpsteff, 2021-04-22:
I would like to sue the new Ikea Remote Control n2 Styrbar with ZHA under Homeassistant. Curently i am only able to pair it but there is only the battery status. Buttons are not available for automations. The older IKEA 5 button remote is working fine. 

I would like to use the IKEA Styrbar N2 remote control via ZHA  in home assistant. 

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

> `{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0820",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x1000",
        "0xfc57"
      ],
      "out_clusters": [
        "0x0003",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "Remote Control N2",
  "class": "zigpy.device.Device"
}`

**Additional context**
Add any other context or screenshots about the feature request here.
[https://zigbee.blakadder.com/Ikea_E2002.html](url)

MattWestb, 2021-04-22:
Shi... !!

Im waiting for IKEA to opening in Vienna for baying it but but its in hard lock down for one months so cant getting it then its in the store.

I think the official name name is E2000/E2001 what i remember from FCC papers.

The basic Z3L function shall working then its on ZB3 device with binding to one group and controlling light but for getting the "other HA" events working its need one quirk for extracting the events from it.

Perhaps @TheJulianJES have getting on and can helping.

TheJulianJES, 2021-04-22:
Please listen to ``zha_event`` in the Home Assistant Developer section.
Then press each button on the remote (normally, double-click, long-press) and label which button press responds to which ``zha_event``.

hpsteff, 2021-04-22:
i have pressed all buttons and get some zha event output. It seems like the up and down have normal press on - off  and long press and stop events. unfortunately the left and right buttons are not giving a clear event? 
attached the output from zha events. i had pressed all different buttons multiple times. I hope the attached will give you the required info.

`{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "84:2e:14:ff:fe:8f:e2:a4",
        "unique_id": "84:2e:14:ff:fe:8f:e2:a4:1:0x0006",
        "device_id": "246f8386dd8892687eb5087806c109ac",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "on",
        "args": []
    },
[Ikea Remote N2 zha events.txt](https://github.com/zigpy/zha-device-handlers/files/6361396/Ikea.Remote.N2.zha.events.txt)

{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "84:2e:14:ff:fe:8f:e2:a4",
        "unique_id": "84:2e:14:ff:fe:8f:e2:a4:1:0x0006",
        "device_id": "246f8386dd8892687eb5087806c109ac",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "off",
        "args": []
    },

Ereignis 1231 ausgelöst 23:30:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "84:2e:14:ff:fe:8f:e2:a4",
        "unique_id": "84:2e:14:ff:fe:8f:e2:a4:1:0x0008",
        "device_id": "246f8386dd8892687eb5087806c109ac",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "stop",
        "args": []
    },

Ereignis 1230 ausgelöst 23:30:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "84:2e:14:ff:fe:8f:e2:a4",
        "unique_id": "84:2e:14:ff:fe:8f:e2:a4:1:0x0008",
        "device_id": "246f8386dd8892687eb5087806c109ac",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "move_with_on_off",
        "args": [
            0,
            83
        ]
 
[Ikea Remote N2 zha events.txt](https://github.com/zigpy/zha-device-handlers/files/6361359/Ikea.Remote.N2.zha.events.txt)

TheJulianJES, 2021-04-27:
I've now created an initial quirk: https://github.com/zigpy/zha-device-handlers/pull/867
I still need more information though:

1. It would be nice if I could get a list where I can clearly see what actions produce(s) which event(s). (The left/right click events *should* also produce some event, although I couldn't see which one it is. (I also need the events for holding each button (press + release)).
2. Does the battery entity in Home Assistant only display half the value? (If there's a new full battery in it -> does it report 50 % or 100 %?

hpsteff, 2021-04-28:
the battery entity only displays 50% with full two AAA batteries.

hpsteff, 2021-04-28:
i have captured the events from zha_event in separate text files for the button up and down. unfortunately zha_event log for the buttons right and left do not produce any events. 
[button down long press.txt](https://github.com/zigpy/zha-device-handlers/files/6389377/button.down.long.press.txt)
[button down short press.txt](https://github.com/zigpy/zha-device-handlers/files/6389378/button.down.short.press.txt)
[button up long press.txt](https://github.com/zigpy/zha-device-handlers/files/6389379/button.up.long.press.txt)
[button up short press.txt](https://github.com/zigpy/zha-device-handlers/files/6389380/button.up.short.press.txt)

any idea how to capture the interaction with the left and right buttons?

hpsteff, 2021-04-28:
i tried to reconfigure the remote and logged the output after pressing the left and right buttons. i dont know if this output contains any meaning ful data to work with? any other idea what to do to log the right output events?
[reconfigure left button long press.txt](https://github.com/zigpy/zha-device-handlers/files/6389491/reconfigure.left.button.long.press.txt)
[reconfigure left button short press.txt](https://github.com/zigpy/zha-device-handlers/files/6389492/reconfigure.left.button.short.press.txt)
[reconfigure - right button long press.txt](https://github.com/zigpy/zha-device-handlers/files/6389493/reconfigure.-.right.button.long.press.txt)
[reconfigure - right button short press.txt](https://github.com/zigpy/zha-device-handlers/files/6389494/reconfigure.-.right.button.short.press.txt)

[image]

hpsteff, 2021-04-28:
the commands sent by this remote are also discussed here:

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4572#issuecomment-804398474

TheJulianJES, 2021-04-28:
I've pushed an update to this PR: https://github.com/zigpy/zha-device-handlers/pull/867
Please download this file: https://raw.githubusercontent.com/TheJulianJES/zha-device-handlers/ikea_styrbar/zhaquirks/ikea/fourbtnremote.py and install it according to these instructions: https://github.com/zigpy/zha-device-handlers#testing-quirks-in-development-in-docker-based-install

Further instructions (untested):
1. Download this updated quirk file: https://raw.githubusercontent.com/TheJulianJES/zha-device-handlers/ikea_styrbar/zhaquirks/ikea/fourbtnremote.py and place it into a temporary folder in your Home Assistant config folder. (We'll place the file here: ``/config/tempzha/fourbtnremote.py``)
2. Read this short passage of this tutorial: https://github.com/zigpy/zha-device-handlers#testing-quirks-in-development-in-docker-based-install
3. Basically, if using Home Assistant OS: install the Portainer addon, possibly unhide the homeassistant container, get a terminal into the homeassistant container, then:
``cd /usr/local/lib/python3.8/site-packages/zhaquirks/ikea`` and do
``cp -a /config/tempzha/fourbtnremote.py .``
(Basically copy the fourbtnremote.py file into the ikea directory of the zhaquirks library).
4. Also remove the ``__pycache__`` folders: (I don't think all of them need to be removed (if any since it's a new quirk) but it shouldn't hurt. You can also try restarting Home Assistant without removing any ``__pycache__folder`` -- it will probably also work in this case)
``rm -rf __pycache__/`` and ``rm -rf ./__pycache__/`` and ``rm -rf ././__pycache__/``
should be the same as: (please double check)
``rm -rf /usr/local/lib/python3.8/site-packages/zhaquirks/ikea/__pycache__/``
``rm -rf /usr/local/lib/python3.8/site-packages/zhaquirks/__pycache__/``
``rm -rf /usr/local/lib/python3.8/site-packages/__pycache__/``
5. Then restart Home Assistant

When using Home Assistant Core, you can also mostly follow the above instructions. There shouldn't be a need for the Portainer addon though. You just need to find where the``...python3.8/site-packages/zhaquirks`` folder is and copy the ``fourbtnremote.py`` file into the ikea folder there. (Also delete the ``__pycache__`` folders and restart Home Assistant.)

NOTE: Following these instructions will only provide a temporary fix. The file will probably get removed when updating Home Assistant (until the fix is actually included there). Restarting Home Assistant/Home Assistant OS should be fine though.

TheJulianJES, 2021-04-28:
I can't guarantee this works but this quirk (if installed) should fix the following issues:
- doubles the battery percentage reported
- provides basic device automation triggers in the Home Assistant automations editor GUI
- adds the scenes cluster (id: 0x0005) to ``replacement`` (IKEA seems to send messages there but doesn't report the device having this cluster in the initial interview)
- if the above "scenes cluster fix" works, perhaps the device automation triggers (copied from the five button remote) should also work

hpsteff, 2021-04-29:
Hi Julian,
great! i managed to install the quirk and re-paired the remotes, the battery level is correct now and the automation triggers are here. Great! i will try to set up some automations.

[image]

[image]

TheJulianJES, 2021-04-29:
Nice!
Please try and see if the left/right arrow buttons work now. If not, please listen for ``zha_event`` again and press the left/right arrow buttons. I've implemented a change that could make these show up now.

TheJulianJES, 2021-04-29:
Although I doubt it helps, can you try to re-pair the device (with the quirk installed)?

hpsteff, 2021-04-29:
quirk is installed, tried to repair, no change ...

MattWestb, 2021-04-29:
Also try doing one "reconfigure this device" from the device card so the binding is done OK.

I going and baying 2 "family" sets next week then IKEA is opening agen :-))

hpsteff, 2021-04-29:
ok, let see - for now i am happy with the two buttons on the remote - thats 100 % more than before...
 lets wait until you can get hands on a N2 remote yourself.

MattWestb, 2021-05-03:
Looks being on good candidate for looking if IKEA have protecting there firmware / bootloader or not and if its possible dumping it :-))
[image]

[image]
looks nearly perfect only the S3 is little near the antenna but the trace back to the module is going around the batteries and is not disturbing the RF but in the casing its one metal for serial connecting the batteries and the batteries can being little bad 2 but not on large thing.

I think first installing the the Quirk and testing the scene buttons is generating in ZHA so it can being finished.

I think Silabs must have opening on new production lane for there EFR32MG21 modules for delivering all IKEA is needing in there products :-))

One thing then opening the battery compartment its being very well  sealed and is not going up by itself then truing the lock in open position and cant getting some tool for helping it.
Only waiting little and the spring is pushing it out and no need using force (the same with both i  have baying).

MattWestb, 2021-05-03:
Its using little different commands with scene buttons as 5B.
```
Right short cluster 5 commmand 7 data 00010d00 (Mfr 0x117c) (33)

Right hold cluster 5 commmand 9 data 0000 (Mfr 0x117c) (34)
+ cluster 6 commmand 1 (on) (no Mfr) (35)
+ cluster 5 commmand 7 data 02000000 (Mfr 0x117c) (36)
+ cluster 5 commmand 8 data 000d00 (Mfr 0x117c) (37) (same as 5B)

Right relese cluster 5 commmand 9 data a549 (Mfr 0x117c) (38)

Left short cluster 5 commmand 7 data 01010d00 (Mfr 0x117c) (39)

Left hold cluster 5 commmand 9 data 0000 (Mfr 0x117c) (40)
+ cluster 6 commmand 1 (on) (no Mfr) (41)
+ cluster 5 commmand 7 data 02000000 (Mfr 0x117c) (42)
+ cluster 5 commmand 8 data 010d00 (Mfr 0x117c) (43) (same as 5B)

Left relese cluster 5 commmand 9 data e52d (Mfr 0x117c) (44)
```
The last (numbers) is the sequence numbers in wireshark.

Command 9 = is being used for resetting / syncing the light in the bounded groupe with one normal on and one "using first scene" (command 7 with data 02000000 its the same that the 5 button is sending then long pressing the toggle for resetting).

So command 7 with data for right (00010d00) or left (01010d00) for short presseing.
Command 8 with data for right (000d00) or left (010d00) hold.
Command 9 with short random data (0xXXXX) that is not zero (0x0000) is relese after hold.

Spammed HA log is attached with the same squenc as above.
[IKEAN21.txt](https://github.com/zigpy/zha-device-handlers/files/6415933/IKEAN21.txt)

We need sending default response to this as for the shortcut bottom and SYMFONISK for not spamming the network. Its looks like the stack is not sending default response to unicast light commands then is normally being sent with millticast (group) commands that is not being acked.

hpsteff, 2021-05-08:
i had updated the Home Assistant Core to the latest version. I had to go through the procedure to copy the quirk again using the steps described above. https://github.com/zigpy/zha-device-handlers/issues/863#issuecomment-828685833
would it be easier to enable custom quirks in the configuration_yaml and just copy it into a folder?

enable_quirks boolean (optional, default: true)
Enable quirks mode for devices where manufacturers didn’t follow specs.

custom_quirks_path string (optional)
Full path to a directory containing custom quirk modules that will take precedence over any built-in quirks matching a device.

This way the quirk would survive updates to Home assistant until the quirk gets finalized?

dmulcahey, 2021-05-08:
That’s how custom_quirks_path works…

TheJulianJES, 2021-05-08:
> would it be easier to enable custom quirks in the configuration_yaml and just copy it into a folder?

Yes. I didn't mention this originally, as this change was only completely enabled in the latest Home Assistant release (2021.5).

> This way the quirk would survive updates to Home assistant until the quirk gets finalized?

Yes.

MattWestb, 2021-05-09:
Was also putting the custom_quirks_path in test and production system and now both E2001 is showing the right battery level :-))

MattWestb, 2021-05-09:
One question shall not both the 5 and 4 button controllers being one Non-color scene controller [Device ID: 0x0830] that is having scene cluster and ZHA is knowing that ?
Its not helping so much then it can receiving commands and is sending all commands as manufacture commands that is "extended" normal commands (Z3L sene cluster commands is 0x0 - 0x6 and 0x40 - 0x42).

hpsteff, 2021-05-21:
any idea how one could get the left and right buttons to work?

MattWestb, 2021-05-25:
I was looking on the new E27 CWS3 that i was putting in the production network and it have updating its firmware  and having all the default scenes and was working with the old 5 button remote so i was mowing the Styrbar to the production network and binding it to the group "kitchen table" and i have the  default scenes working and the old Philips HUE is going in the black boy for not baseball things then its not deleting neighbor device that is offline / left / deleted from the network and the dimmer switch is going the same way then its bottoms is starting  being very slippy.

I hope **puddly** is getting all the missing parts in the ZCL v7 working so we can adding scenes is our beloved devices and getting use if great futures.

Obiak, 2021-06-15:
Is there anything us "not so technical" users can do to drive this forward? 
The latest revision of the quirk does show the arrow buttons in the automations menu but doesn't send anything on the "zha_event".

MattWestb, 2021-07-15:
You can testing this version:
```
"""Device handler for IKEA of Sweden TRADFRI remote control."""
from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    Identify,
    LevelControl,
    OnOff,
    Ota,
    PollControl,
    PowerConfiguration,
    Scenes,
)
from zigpy.zcl.clusters.lightlink import LightLink

from zhaquirks import DoublingPowerConfigurationCluster
from zhaquirks.const import (
    ARGS,
    CLUSTER_ID,
    COMMAND,
    COMMAND_HOLD,
    COMMAND_MOVE,
    COMMAND_MOVE_ON_OFF,
    COMMAND_OFF,
    COMMAND_ON,
    COMMAND_PRESS,
    COMMAND_STOP,
    DEVICE_TYPE,
    DIM_DOWN,
    DIM_UP,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    LEFT,
    LONG_PRESS,
    LONG_RELEASE,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    RIGHT,
    SHORT_PRESS,
    TURN_OFF,
    TURN_ON,
)
from zhaquirks.ikea import IKEA, ScenesCluster

IKEA_CLUSTER_ID = 0xFC57  # decimal = 64599

class IkeaTradfriRemote(CustomDevice):
    """Custom device representing IKEA of Sweden TRADFRI remote control."""

    signature = {
        # <SimpleDescriptor endpoint=1 profile=260 device_type=2080
        # device_version=1
        # input_clusters=[0, 1, 3, 32, 4096, 64599]
        # output_clusters=[3, 6, 8, 25, 4096]>
        MODELS_INFO: [(IKEA, "Remote Control N2")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.NON_COLOR_CONTROLLER,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    PollControl.cluster_id,
                    LightLink.cluster_id,
                    IKEA_CLUSTER_ID,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Ota.cluster_id,
                    LightLink.cluster_id,
                ],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.NON_COLOR_CONTROLLER,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    DoublingPowerConfigurationCluster,
                    Identify.cluster_id,
                    PollControl.cluster_id,
                    LightLink.cluster_id,
                    IKEA_CLUSTER_ID,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    ScenesCluster,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Ota.cluster_id,
                    LightLink.cluster_id,
                ],
            }
        }
    }

    device_automation_triggers = {
        (SHORT_PRESS, TURN_ON): {COMMAND: COMMAND_ON, CLUSTER_ID: 6, ENDPOINT_ID: 1},
        (LONG_PRESS, DIM_UP): {
            COMMAND: COMMAND_MOVE_ON_OFF,
            CLUSTER_ID: 8,
            ENDPOINT_ID: 1,
            ARGS: [0, 83],
        },
        (LONG_RELEASE, DIM_UP): {COMMAND: COMMAND_STOP, CLUSTER_ID: 8, ENDPOINT_ID: 1},
        (SHORT_PRESS, TURN_OFF): {COMMAND: COMMAND_OFF, CLUSTER_ID: 6, ENDPOINT_ID: 1},
        (LONG_PRESS, DIM_DOWN): {
            COMMAND: COMMAND_MOVE,
            CLUSTER_ID: 8,
            ENDPOINT_ID: 1,
            ARGS: [1, 83],
        },
        (LONG_RELEASE, DIM_DOWN): {
            COMMAND: COMMAND_STOP,
            CLUSTER_ID: 8,
            ENDPOINT_ID: 1,
        },
        (SHORT_PRESS, LEFT): {
            COMMAND: COMMAND_PRESS,
            CLUSTER_ID: 5,
            ENDPOINT_ID: 1,
            ARGS: [257, 13, 0],
},
        (LONG_PRESS, LEFT): {
            COMMAND: COMMAND_HOLD,
            CLUSTER_ID: 5,
            ENDPOINT_ID: 1,
            ARGS: [3329, 0],
        },
        (SHORT_PRESS, RIGHT): {
            COMMAND: COMMAND_PRESS,
            CLUSTER_ID: 5,
            ENDPOINT_ID: 1,
            ARGS: [256, 13, 0],
        },
        (LONG_PRESS, RIGHT): {
            COMMAND: COMMAND_HOLD,
            CLUSTER_ID: 5,
            ENDPOINT_ID: 1,
            ARGS: [3328, 0],
        },
    }
```
Its have device trigger working for scene buttons right and left short and hold but it is sending more release on the scene cluster after long press on right and left button.
And its still spamming the network then configuring binding and reporting.

MattWestb, 2021-07-17:
@puddly Is it possible making one device automation for release that is only triggering then command 9 is fired but not then the data  is 0x0000 (command 9 with data 0x0000 is also used then doing the long press so it shall not being fired the release device automation) ?

The sniffed data and finding in https://github.com/zigpy/zha-device-handlers/issues/863#issuecomment-831356687 with sequences for the reset IKEA is using for the bonded group.

Also if that is working is it possible doing one release from left / right long press as is being done in the IKEA Open/Close button automation ?

Alos one more question:
IKEA is normally having its manufacture cluster `IKEA_CLUSTER_ID = 0xFC7C  # decimal = 64636` and this device is having it as `WWAH_CLUSTER_ID = 0xFC57 # decimal = 64599`.
It can  being on typo from IKEA or they have getting one more for the 3 gen devices but is it better having the "normal" IKEA name for it or one new so its not being colliding (perhaps `IKEA_CLUSTER3_ID = 0xFC57 # decimal = 64599`) ?

Then you have getting all ZCL v7 things in place i think we need putting the commands IKEA is using in the IKEA_Cluster for adding relative scenes to IKEA lights but its only 2 known command that is being used for that.

Thanks in advance !

PR #867 (2021-08-25): IKEA Styrbar remote quirk
Fixes #863, fixes #975

Adds device automation triggers and doubles the power percentage.
Left/right click are apparently working now.

Like the IKEA shortcut button, this remote seems to generate multiple ``zha_event``s.
(At least for @MattWestb on a Silabs(?) coordinator. AFAIK, some people with a TI coordinator didn't have an issue with multiple events on the shortcut button.)
The "spamming" issue needs to be looked at later (for all newer IKEA remotes), but this isn't caused by the quirk.
(Reference: https://github.com/zigpy/zha-device-handlers/issues/967)

For no [... cut]
closed 2021-08-25 by dmulcahey