zigpy/zha-device-handlers #864: [BUG]

jfisbein, 2021-04-24:
**Describe the bug**
After adding TERNCY-SD01 device to home assistant,it does not fire any zha event when pushed/rotated.

**To Reproduce**
Steps to reproduce the behavior:
1. Add TERNCY-SD01 devide to home assistant using ZHA integration (v 2021.4.6)
2. Device is recognised by home assistant _correctly_
3. Go to DEveloper Toosl -> Events
4. in "Listen to events" fill `zha_event` and click "start listening"
5. Click on the Terncy SD01 butoon an rotate it.
6. No event is shown.

**Expected behavior**
Expected events to get fired when pushing an rotating the Terncy SD01

**Screenshots**
Device _correctly_ configured
[image]

**Additional context**
I tried to remove the device and adding it several times.
I also updated the firmware through OTA.
I has not worked with the stock firmware *nor* the updated firmware.

ZHA Pairing log
[zigpy.log](https://github.com/zigpy/zha-device-handlers/files/6369497/zigpy.log)
[blakadder info](https://zigbee.blakadder.com/Terncy_TERNCY-SD01.html)

kukachik, 2021-05-21:
Which stick do you use? I have this problem on ti cc2538 and on cc2530. But terncy work perfectly on itead erf32 stick.
In my case, terncy dont work in z2m too.

MattWestb, 2021-05-21:
In the log its one zigpy_znp coordinator.

kukachik, 2021-05-21:
You can try low level serial log, maybe it will be not empty.

jfisbein, 2021-05-21:
I use a [ZZH](https://electrolama.com/projects/zig-a-zig-ah/) zigbee stick.

zeon-ua, 2021-06-17:
I do have exactly the same behavior. I use Conbee2 dongle which works perfectly with my other ZigBee devices

bbrandal, 2021-07-17:
Same behaviour here with ZHA/Conbee2. Terncy paired and can read battery level, but no events firing. Would be happy to help debug if anyone can give me some directions :)

labactani, 2022-01-14:
i am having the same result with Conbee stick. tried with ZHA >> i can only ready battery level and with Z2M where for i get null value or NA%...
Conbee is with the latest firmware as of 13 Jan 2022 - anyone made progress on this?

closed 2022-07-20 by github-actions[bot]

martin3000, 2023-01-22:
My terncy dial know was working until Dec 2022. Then it stopped sending events. I was using EZSP in december, in january I switched to ZNP.

MattWestb, 2023-01-22:
@martin3000 Did you repairing the devices then merger to ZNP ?
If not all devices is trying reporting to the old EZSP IEEE address and the ZNP is having one other so its not working.
You can try re adding some controllers and see if its working or not.

martin3000, 2023-01-22:
@MattWestb hi matt, I removed it and repaired it, reconfigured it, nothing helps.

MattWestb, 2023-01-22:
@martin3000 Its looks like ZHA its having problem adding the device in zigbee.db.
Try deleting the device and the restarting HA so the DB is being closed and new opened OK.
Then (Z)HA is stable try adding the device and see if its being joined OK and also written in the DB.

If its being OK in the DB it shall staying in ZHA after one more restart of HA and if not its being not OK in the GUI (normally unknown device).

martin3000, 2023-01-22:
@MattWestb Thanks Matt for looking into it.
But the device is there, see screenshot.
[image]

martin3000, 2023-02-06:
I have the same problem now with a osram lightify 4x wireless button: the button was working for 2 years, than the events came double. I reset the button and paired it again, now there are no events at all.

martin3000, 2023-02-06:
Is it possible to activate zigpy/zha/znp debugging to check where the events flow?

Drizzt321, 2023-05-02:
I have the same problem with this device, got to me in the last month from AliExpress, never had one of these. Using ZHA. I can get it to pair, but no zha_events or anything else afterwards.

Here's the log file of when the device joined. 

[terncy-sd01 join.log](https://github.com/zigpy/zha-device-handlers/files/11369471/terncy-sd01.join.log)

martin3000, 2023-05-02:
Last year it was working, but then suddenly (with a new HA version) it failed.

Drizzt321, 2023-05-02:
@martin3000 And nobody seems to have any idea how to get it to work? What the change that caused it to break? 

I'm happy to do any other logging, test custom quirks, etc. This would be the perfect device for my needs, mute/unmute/volume control for each rooms Snapcast powered speakers.

tube0013, 2023-05-02:
I added a SD-01 to my network (it was previously paired when I bought them a year or 2 ago) and it is working as expected. I'm using an ezsp coordinator - what are you all using? I can try another one with a test network on a cc2652 coordinator later tonight.

Drizzt321, 2023-05-02:
I'm using one of your older [cc2652 ethernet](https://github.com/tube0013/tube_gateways/tree/main/models/retired/tubeszb-cc2652-eth) coordinators. Works great for everything else (except those new Aqara Opple, even though they seem exactly the same as ones I already have that are working fine), just this one no zha_events, nothing except the Identify and Battery attributes.

Not sure this is zigpy related, or HA, but I notice that [blakadder](https://zigbee.blakadder.com/Terncy_TERNCY-SD01.html) has an OTA update file there. Tried to update following the instructions, but getting this error stacktrace. This is after changing the conf and uploading the file to the right dir, via File Editor add-on, and doing a HA restart. 

```
 Home Assistant 2023.4.6 
Supervisor 2023.04.1 
Operating System 10.1 
Frontend 20230411.1 - latest 
```

```
2023-05-02 09:50:13.302 ERROR (MainThread) [homeassistant.core] Error executing service: <ServiceCall zha.set_zigbee_cluster_attribute (c:01GZENXE8P6GJSG3E79NE8NEKD): ieee=00:0d:6f:00:16:dc:fb:2e, endpoint_id=1, cluster_id=25, attribute=2, cluster_type=out, value=49502>
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 317, in request
    return await req.result
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/core.py", line 1838, in catch_exceptions
    await coro_or_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1857, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 874, in admin_handler
    await result
  File "/usr/src/homeassistant/homeassistant/components/zha/websocket_api.py", line 1275, in set_zigbee_cluster_attributes
    response = await zha_device.write_zigbee_attribute(
  File "/usr/src/homeassistant/homeassistant/components/zha/core/device.py", line 666, in write_zigbee_attribute
    response = await cluster.write_attributes(
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 569, in write_attributes
    return await self.write_attributes_raw(attrs, manufacturer)
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 575, in write_attributes_raw
    result = await self._write_attributes(attrs, manufacturer=manufacturer)
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 326, in request
    return await self._endpoint.request(
  File "/usr/local/lib/python3.10/site-packages/zigpy/endpoint.py", line 237, in request
    return await self.device.request(
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 316, in request
    async with asyncio_timeout(timeout):
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError
```

tube0013, 2023-05-02:
I'm pretty sure that is just due not being able to reach the device, you have to make sure it's away but pressing the button when you do that. 

I am the one who contributed that FW, I sniffed it from a packet capture of the terncy hub doing it. So both of my SD-01's are on the newer fw.  I will test the other one on cc2652. 

Your aqara opple issue should be a separate issue, but you may need to update the coordinator fw.

Drizzt321, 2023-05-02:
Ah! Not able to reach the device, that makes perfect sense. Just did the button push, got the red light, hit `Call Service`, button push and got the red light again, and no error in the Core logs yet, so I presume it's working. I'll let it alone for 20-30 minutes and see if there's any output anywhere. 

So I did perform an update recently, was moving stuff around, unplugging stuff, changing it around. Had it plugged into a PoE mUSB to power it from my switch. For some reason when I plugged it back in, it wasn't getting or grabbing an IP, or something weird. So I pulled it off, used a FTDI and ESPHomeFlasher and got it into flash mode and flashed whatever the latest from your github is. Unfortunately, there's no version on the web page. I do see `zBSL` is set to Off though.

martin3000, 2023-05-02:
I'm using a CC2652RB ZNP stick, but before I was using a EZSP stick.

tube0013, 2023-05-02:
email me or open an issue on my repo getting off topic here...

for the sd-01 fw update if you enable debug logging for zigpy you should see the update progressing

ilarrain, 2023-05-02:
I was the one who contributed the original quirk. I did the original testing with an ezsp coordinator, where it works well. Later, I've switched to a Texas Instruments one, and it doesn't. Probably due to differences in coordinator implementation.
I think it has to do with the cluster command parsing done in https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/terncy/__init__.py#L131
I assumed some int types that might be wrong, tried to fix them, but it's beyond my limited python/ZigBee debugging/sniffing skills.

Drizzt321, 2023-05-03:
@ilarrain thanks for chiming in, you're probably ahead of me simply because I've never done any sort of ZigBee/quirks stuff directly, just been very fortunate and grateful at all the other folks who have done all the hard work of getting it all more or less (usually more!) working well with HA/FOSS software.

martin3000, 2023-05-03:
I modified zhaquirks/terncy/--init--.py:
```
   def handle_cluster_request(
        self,
        hdr: foundation.ZCLHeader,
        args: List[Any],
        *,
        dst_addressing: Optional[
            Union[t.Addressing.Group, t.Addressing.IEEE, t.Addressing.NWK]
        ] = None,
    ):
        _LOGGER.debug(
            "Terncy - handle_cluster_request tsn: [%s] command id: %s - args: [%s]",
            hdr.tsn,
            hdr.command_id,
            args,
        )

```
But in the log there is nothing. So it seems that handle_cluster_request is not called.

martin3000, 2023-05-03:
Going to the device information page and looking at the clusters, I see that the TerncyRawCluster does not have any attributes.

jfisbein, 2023-05-06:
I tried multiple times to upload the firmware but I think it's not working. should the firmware version on the device page change?
I always see `Firmware: 0x0000001a`.

martin3000, 2023-05-06:
> I tried multiple times to upload the firmware but I think it's not working. should the firmware version on the device page change? I always see `Firmware: 0x0000001a`.

[image]

MattWestb, 2024-07-30:
`[0x3D63:1:0x0020]` = pull control cluster.
Its one Zigbee 3 future that good end devices is using for getting pulling working OK with its parent.  
But the device dont like getting / setting 135 (i think its out of range then its have to so with slow pull intervals its **must** being greater then that). Try reading the other  attribute on the pull control cluster and see how its looking.
And i think then using EZSP Silabs stack is handing it better then the device requesting it and also if is have one other router as its parent and not the coordinator that.
Also posting one updated device signature so can see what its shall having for capacity (very likely false Zigbee 3 end  device).

martin3000, 2024-07-30:
The checkin_interval is 3600 and cannot be changed. But ok II think this is not a problem.

But the TerncyRawCluster is empty, there are no attributes. But I knows the commands "click_event" and "motion_event". This is strange. I think this should be attributes and not commands.

MattWestb, 2024-07-30:
3600 sound good (I think its 1/10 or 1/4 seconds) from my experience sniffing IKEA controllers.
I think its failing init and the device is using its default but then its not continuing init other functions like the customer cluster.

Is reconfigure from the device card working ? 
And is it doing all OK no not stopping for not OK settings ?

martin3000, 2024-07-30:
battery_voltage cannot be read, only battery_percentage_remaining. So battery_voltage is yellow. And reconfiguration of PollCluster also fails. TerncyRawCluster is not reconfigured.