zigpy/zha-device-handlers #866: tyst11_wmcdj3aq_mcdj3aq detected as light bulb instead of cover (zemismart tuya smart cover 38mm like TS0601)

almooj, 2021-04-26:
hi, 
my 2 first zemismart TS0601 covers are now detected and working correctly in HA but my last one is still detected as light bulb...

i don't really know how to help to make some convert quirk but here is the zigbee signature

{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=0, maximum_buffer_size=82, maximum_incoming_transfer_size=255, server_mask=11264, maximum_outgoing_transfer_size=255, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.Router: 1>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TYST11_wmcdj3aq",
  "model": "mcdj3aq",
  "class": "zigpy.device.Device"
}

hope someone could figure out how to make this work in HA without mqtt

thx a lot for this

almooj, 2021-05-19:
@blauret it seems to be also another ZM25TQ variant, maybe this could help https://github.com/iquix/Smartthings/blob/eb41229bad69bb591f185801355ef27cc2614cf8/devicetypes/iquix/tuya-window-shade.src/tuya-window-shade.groovy

MattWestb, 2021-11-15:
One user have adding it 2 mouths ago so if all is right dome it shall working now.
https://github.com/zigpy/zha-device-handlers/pull/1035

almooj, 2021-11-18:
hi,
in fact , ive made a pull request to add this device as quirk and fix the issue [here](https://github.com/zigpy/zha-device-handlers/pull/1035)

thx

closed 2021-11-18 by almooj

MattWestb, 2021-11-18:
Sorry for not looking so well on the name of the PR that it was you :-((

Great its working and the issue is closed.

rus-ik, 2022-01-18:
> hi, in fact , ive made a pull request to add this device as quirk and fix the issue [here](https://github.com/zigpy/zha-device-handlers/pull/1035)
> 
> thx

hi, can you please add all of them? :)
https://github.com/iquix/Smartthings/blob/eb41229bad69bb591f185801355ef27cc2614cf8/devicetypes/iquix/tuya-window-shade.src/tuya-window-shade.groovy#L27-L42