zigpy/zha-device-handlers #868: [Device Support Request] Develco Frient MOSZB-140 Motion Sensor Pro

Veldkornet, 2021-04-28:
**Is your feature request related to a problem? Please describe.**
I am able to pair the device, the temperature sensor works, but the battery sensor does not. Unclear on the multiple binary_sensor's, they seem to always be on.

**Describe the solution you'd like**
Add quirks for the Frient devices as needed. There is apparently [ZigBee2MQTT support](https://www.zigbee2mqtt.io/devices/MOSZB-140.html).

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=0, manufacturer_code=4117, maximum_buffer_size=80, maximum_incoming_transfer_size=80, server_mask=0, maximum_outgoing_transfer_size=80, descriptor_capability_field=0, *allocate_address=False, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 49353,
      "device_type": "0x0001",
      "in_clusters": [
        "0x0003",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    },
    "34": {
      "profile_id": 260,
      "device_type": "0x0107",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0406"
      ],
      "out_clusters": []
    },
    "35": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x000f",
        "0x0020",
        "0x0500"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "38": {
      "profile_id": 260,
      "device_type": "0x0302",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0402"
      ],
      "out_clusters": [
        "0x0003"
      ]
    },
    "39": {
      "profile_id": 260,
      "device_type": "0x0106",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0400"
      ],
      "out_clusters": []
    },
    "40": {
      "profile_id": 260,
      "device_type": "0x0107",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0406"
      ],
      "out_clusters": []
    },
    "41": {
      "profile_id": 260,
      "device_type": "0x0107",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0406"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "frient A/S",
  "model": "MOSZB-140",
  "class": "zigpy.device.Device"
}
```

**Additional context**
ZHA Issue: https://github.com/home-assistant/core/issues/49127

Hedda, 2021-05-19:
Wondering if the root cause is the same as that described by dmulcahey in https://github.com/zigpy/zha-device-handlers/issues/605 issue? That is, a a limitation in ZHA integration for Home Assistant which currently only support 1 sensor per cluster, and these type of devices requiring 2 or more sensor per cluster?

Veldkornet, 2021-11-15:
It's not resolved, but I've switched to using zigbee2mqtt where it does work.

rbnmk, 2021-12-08:
Bought a Develco Frient Motion Sensor Pro today (MOSZB-140) and having the issues described here with latest HA+ZHA. Could the issue be resolved in the next release or is there anything I can provide to get it to work with ZHA?

LeoWinterDE, 2022-01-16:
Support here is still missing.

alandrdre, 2022-03-04:
Can anyone please help add support for this device which is a top amazon seller in the UK.

Dawnes, 2022-06-19:
Bump. The above issue is still there. Please assist, as this device is now sold in a lot of places.

mwlodarc38, 2022-10-02:
hello, i have this motion sensor too and is not work well with my haos. pairing is difficult ant battery status not reconize and tamper switch don't work. how do this motion sensor work?

MrEditor97, 2023-01-04:
I have just received one of these, so will look into it to see if there is potential for correcting the device in ZHA.

PR #2122 (2023-01-23): Quirk for Develco Motion Sensor Pro
Enables Develco specific cluster handlers for Motion Sensor Pro

Fixes #868
commit bf636085 (2023-01-23): Quirk for Develco Motion Sensor Pro (#2122)

Fixes #868
closed 2023-01-23 by dmulcahey