zigpy/zha-device-handlers #876: [Device Support Request] Aqara vibration tilt sensor

thecobra666, 2021-05-04:
**Is your feature request related to a problem? Please describe.**
The Aqara vibration sensor has more sensors then the one vibration sensor that is now available in ZHA.

**Describe the solution you'd like**
Have all the sensors mentioned in ZHA.

dmulcahey, 2021-05-11:
Everything else is sent as `zha_event`

closed 2021-05-11 by dmulcahey

thecobra666, 2021-05-11:
> Everything else is sent as `zha_event`

I know, but why? Now I have to manually created the sensors based on input fields.

dmulcahey, 2021-05-11:
Remotes don’t get sensors per HA architecture

martin3000, 2021-07-31:
you can convert it to a sensor:

```
- trigger:
  - platform: event
    event_type: zha_event
    event_data:
      device_ieee: "00:11:22:33:44:55:66:77"
      endpoint_id: 1
      command: "vibration_strength"
  sensor:
  - name: vibration
    state: "{{ trigger.event.data.args[0]['strength'] }}"
    unit_of_measurement: "%"

```