zigpy/zha-device-handlers #878: [BUG] Hue Dimmer Switch (EU) RWL021

RefineryX, 2021-05-06:
I have a Phillips Hue Dimmer Switch (EU) RWL021 that stopped working a few versions back in Home Assistant (2021.2). I have tried removing the device and re-adding however it does not work. The error I receive in my logs is``Unknown cluster-specific command 0``.

Upon deleting the device and re-adding, I have pasted the below log. Upon clicking the buttons on the device I get the same error throughout ``Unknown cluster-specific command 0``.

I am not yet sure if this is a bug or support request. Any help would be appreciated.

```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0x44aa:1:0x0019] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=208 command_id=1>
[0x44aa:1:0x0019] ZCL request 0x0001: [0, 4137, 4364, 419955472, None]
[0x44aa:1:0x0019] OTA query_next_image handler for 'ZigBee/CCT CCT Light': field_control=0, manufacture_id=4137, image_type=4364, current_file_version=419955472, hardware_version=None
[0x44aa:1:0x0019] No OTA image is available
[0xBF0B:1:0x0500]: Updated alarm state: 0
[0x44aa:1:0x0019] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=106 command_id=1>
[0x44aa:1:0x0019] ZCL request 0x0001: [0, 4137, 4364, 419955472, None]
[0x44aa:1:0x0019] OTA query_next_image handler for 'ZigBee/CCT CCT Light': field_control=0, manufacture_id=4137, image_type=4364, current_file_version=419955472, hardware_version=None
[0x44aa:1:0x0019] No OTA image is available
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0x44aa:1:0x0019] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=4 command_id=1>
[0x44aa:1:0x0019] ZCL request 0x0001: [0, 4137, 4364, 419955472, None]
[0x44aa:1:0x0019] OTA query_next_image handler for 'ZigBee/CCT CCT Light': field_control=0, manufacture_id=4137, image_type=4364, current_file_version=419955472, hardware_version=None
[0x44aa:1:0x0019] No OTA image is available
Device 0x1a5b (00:17:88:01:10:48:1e:50) joined the network
[0x1a5b:zdo] ZDO request ZDOCmd.Device_annce: [0x1A5B, 00:17:88:01:10:48:1e:50, 128]
[0x1a5b] Requesting 'Node Descriptor'
Tries remaining: 2
[0x1a5b] Extending timeout for 0xad request
[0x1a5b] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4107, maximum_buffer_size=71, maximum_incoming_transfer_size=45, server_mask=0, maximum_outgoing_transfer_size=45, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)
[0x1a5b] Discovering endpoints
Tries remaining: 3
[0x1a5b] Extending timeout for 0xaf request
[0x1a5b] Discovered endpoints: [1, 2]
[0x1a5b:1] Discovering endpoint information
Tries remaining: 3
[0x1a5b] Extending timeout for 0xb1 request
[0x1a5b:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=49246, device_type=2080, device_version=2, input_clusters=[0], output_clusters=[0, 3, 4, 6, 8])
[0x1a5b] Extending timeout for 0xb3 request
[0x1a5b:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=179 command_id=Command.Read_Attributes_rsp>
[0x1a5b:1] Manufacturer: Philips
[0x1a5b:1] Model: RWL021
[0x1a5b:2] Discovering endpoint information
Tries remaining: 3
[0x1a5b] Extending timeout for 0xb5 request
[0x1a5b:2] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=2, profile=260, device_type=12, device_version=0, input_clusters=[0, 1, 3, 15, 64512], output_clusters=[25])
Checking quirks for Philips RWL021 (00:17:88:01:10:48:1e:50)
Considering <class 'zhaquirks.philips.rwlfirstgen.PhilipsRWLFirstGen'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1, 2}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1, 2}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1, 2}
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because endpoint list mismatch: {1} {1, 2}
device - 0x1A5B:00:17:88:01:10:48:1e:50 entering async_device_initialized - is_new_join: True
device - 0x1A5B:00:17:88:01:10:48:1e:50 has joined the ZHA zigbee network
[0x1A5B](RWL021): started configuration
[0x1A5B:ZDO](RWL021): 'async_configure' stage succeeded
[0x1A5B:1:0x0000]: finished channel configuration
[0x1a5b] Extending timeout for 0xb7 request
[0x1a5b] Extending timeout for 0xb9 request
[0x1a5b] Extending timeout for 0xbb request
[0x1A5B:1:0x0008]: bound 'level' cluster: Status.SUCCESS
[... 112 lines cut ...]
[0x1a5b:2:0xfc00] No handler for cluster command 0
[0x1a5b:2:0xfc00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=True is_reply=True disable_default_response=True> manufacturer=4107 tsn=14 command_id=0>
[0x1a5b:2:0xfc00] Unknown cluster-specific command 0
[0x1a5b:2:0xfc00] ZCL request 0x0000: b'\x01\x00\x000\x02!\x01\x00'
[0x1a5b:2:0xfc00] No handler for cluster command 0
[0x6767:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=210 command_id=Command.Report_Attributes>
[0x6767:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=65281, value=<TypeValue type=CharacterString, value=!�!�C!?>)]]
[0x6767:1:0x0000] Attribute report received: 65281=!�!�C!?
[0x6767:1:0x0001] Voltage mV: [Min]:2820 < [RAW]:3005 < [Max]:3100, Battery Percent: 66.0
[0x44aa:1:0x0019] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=158 command_id=1>
[0x44aa:1:0x0019] ZCL request 0x0001: [0, 4137, 4364, 419955472, None]
[0x44aa:1:0x0019] OTA query_next_image handler for 'ZigBee/CCT CCT Light': field_control=0, manufacture_id=4137, image_type=4364, current_file_version=419955472, hardware_version=None
[0x44aa:1:0x0019] No OTA image is available
[0x44aa:1:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=154 command_id=Command.Report_Attributes>
[0x44aa:1:0x0300] ZCL request 0x000a: [[Attribute(attrid=4, value=<TypeValue type=uint16_t, value=24701>)]]
[0x44aa:1:0x0300] Attribute report received: current_y=24701
[0x44aa:1:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=203 command_id=Command.Report_Attributes>
[0x44aa:1:0x0300] ZCL request 0x000a: [[Attribute(attrid=3, value=<TypeValue type=uint16_t, value=24939>)]]
[0x44aa:1:0x0300] Attribute report received: current_x=24939
light.bedroom_right_bedside: polling current state
[0xe10f:11:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=210 command_id=Command.Read_Attributes_rsp>
[0xe10f:11:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=212 command_id=Command.Read_Attributes_rsp>
[0xE10F:11:0x0008]: received attribute: 0 update with value: 13
[0xe10f:11:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=214 command_id=Command.Read_Attributes_rsp>
```

Adminiuga, 2021-05-07:
I don't beleive there's quirk for this device.
Listen to `zha_events` in the developer tab and check if you get any zha-events

RefineryX, 2021-05-07:
That is the strange thing. I am getting event for this. 

```
Event 3 fired 9:18:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:17:88:01:10:48:1e:50",
        "unique_id": "00:17:88:01:10:48:1e:50:1:0x0006",
        "device_id": "5547f20a22a98168e8a39a56f178d249",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "off_with_effect",
        "args": [
            0,
            0
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2021-05-07T08:18:01.741043+00:00",
    "context": {
        "id": "e516693c81c605ab4d32f11e7248ec80",
        "parent_id": null,
        "user_id": null
    }
}
Event 2 fired 9:18:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:17:88:01:10:48:1e:50",
        "unique_id": "00:17:88:01:10:48:1e:50:1:0x0008",
        "device_id": "5547f20a22a98168e8a39a56f178d249",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "step",
        "args": [
            1,
            30,
            9
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2021-05-07T08:18:01.134317+00:00",
    "context": {
        "id": "c446ddcfb49f048c1f3874f8027a334b",
        "parent_id": null,
        "user_id": null
    }
}
Event 1 fired 9:18:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:17:88:01:10:48:1e:50",
        "unique_id": "00:17:88:01:10:48:1e:50:1:0x0008",
        "device_id": "5547f20a22a98168e8a39a56f178d249",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "step",
        "args": [
            0,
            30,
            9
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2021-05-07T08:18:00.447014+00:00",
    "context": {
        "id": "87bbfb21fd23b2669c678eb916c0eed1",
        "parent_id": null,
        "user_id": null
    }
}
Event 0 fired 9:17:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:17:88:01:10:48:1e:50",
        "unique_id": "00:17:88:01:10:48:1e:50:1:0x0006",
        "device_id": "5547f20a22a98168e8a39a56f178d249",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "on",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2021-05-07T08:17:59.364684+00:00",
    "context": {
        "id": "32d8cbed7c76dcef854e70e1660a97da",
        "parent_id": null,
        "user_id": null
    }
}
```

But while getting events, the buttons are not working and I also see the errors on the logs.
``[0x1a5b:2:0xfc00] Unknown cluster-specific command 0``

dmulcahey, 2021-05-11:
what do you mean the buttons aren't working? pressing them is what generates the events...

dmulcahey, 2021-05-11:
Did you use the events as triggers in the automation?

RefineryX, 2021-05-11:
Yes. I have tried creating the manual automation and also tried the blueprints ([here](https://community.home-assistant.io/t/zha-philips-hue-dimmer-switch-individual-buttons-with-long-presses/261012)).

Both do not respond and reports the error on the logs.
``[0x1a5b:2:0xfc00] Unknown cluster-specific command 0``

I have just checked ``zha_event` again and pressed all four buttons individually and this looks like it responds find. However, even through I have an automation setup the lights do not switch on and the error gets reported in the log ``[0x1a5b:2:0xfc00] Unknown cluster-specific command 0``. 

It was working fine for about a year and noticed it stopped working a few versions of HA ago.

**NEW ZHA_EVENT LOG**

```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:17:88:01:10:48:1e:50",
        "unique_id": "00:17:88:01:10:48:1e:50:1:0x0006",
        "device_id": "5547f20a22a98168e8a39a56f178d249",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "off_with_effect",
        "args": [
            0,
            0
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2021-05-11T11:54:08.781415+00:00",
    "context": {
        "id": "53b9760c45943fa13c745360e331e042",
        "parent_id": null,
        "user_id": null
    }
}
Event 2 fired 12:54:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:17:88:01:10:48:1e:50",
        "unique_id": "00:17:88:01:10:48:1e:50:1:0x0008",
        "device_id": "5547f20a22a98168e8a39a56f178d249",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "step",
        "args": [
            1,
            30,
            9
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2021-05-11T11:54:07.930273+00:00",
    "context": {
        "id": "b9e209769c48a1a76ee1a2689922fabf",
        "parent_id": null,
        "user_id": null
    }
}
Event 1 fired 12:54:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:17:88:01:10:48:1e:50",
        "unique_id": "00:17:88:01:10:48:1e:50:1:0x0008",
        "device_id": "5547f20a22a98168e8a39a56f178d249",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "step",
        "args": [
            0,
            30,
            9
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2021-05-11T11:54:07.184881+00:00",
    "context": {
        "id": "f2896f882467c6c8c4a4a273e462bc3c",
        "parent_id": null,
        "user_id": null
    }
}
Event 0 fired 12:54:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:17:88:01:10:48:1e:50",
        "unique_id": "00:17:88:01:10:48:1e:50:1:0x0006",
        "device_id": "5547f20a22a98168e8a39a56f178d249",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "on",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2021-05-11T11:54:05.875573+00:00",
    "context": {
        "id": "8f1d5452c3620220204df88c4d0442e1",
        "parent_id": null,
        "user_id": null
    }
}
```

MattWestb, 2021-05-11:
If i remember right is HUE dimmer switched "hybrids"  so its sending "normal" light commands like On / Off, Dim up and down. Plus its sending HA "clicks" on one HA cluster (0xfc00 ?) that can being used as HA vents. Then ZHA is using the ZLL commands for sending command direct to bonded light groups and also "translating" this commands to evenest in ZHA. 
So i think the error is because ZHA is getting package that it have no use for and is "manufacture specific" and not understand (but its dont need that because its already have getting the commands from the light group).

TheJulianJES, 2021-05-12:
Yeah. All Hue remotes are currently supported with ZHA. (Even the new RWL022 dimmer.)
The RWL021 uses this quirk here: https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/philips/rwlfirstgen.py

Can you send the signature of your dimmer? (Can be seen in the ZHA device UI -> Show signature).
Also, try re-pairing (probably won't help in this case but always worth a try).

TheJulianJES, 2021-05-12:
Oh well, the (previously named) RWL020 quirk also applies for RWL021 dimmers. (I guess the firmware never got updated or something like that.)

RefineryX, 2021-05-12:
Thanks. 

I tried repairing a few times over the course of a month but unfortunately this does not work.

I am attaching the device signature too.

```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4107, maximum_buffer_size=71, maximum_incoming_transfer_size=45, server_mask=0, maximum_outgoing_transfer_size=45, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 49246,
      "device_type": "0x0820",
      "in_clusters": [
        "0x0000"
      ],
      "out_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0008"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x000c",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x000f",
        "0xfc00"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "Philips",
  "model": "RWL021",
  "class": "zigpy.device.Device"
}
```

TheJulianJES, 2021-05-12:
Yeah, I'll get a fix ready.

RefineryX, 2021-05-12:
@TheJulianJES  You are the BEST! Much appreciated - thank you!

TheJulianJES, 2021-05-12:
Download this file: https://raw.githubusercontent.com/TheJulianJES/zha-device-handlers/rwl021_fix/zhaquirks/philips/rwlfirstgen.py
(Make sure you're on Home Assistant 2021.5.x)

Add https://www.home-assistant.io/integrations/zha/#custom_quirks_path to your ZHA config, create the folder, create a philips folder in the custom quirks folder, then drop the file in there, restart Home Assistant and hope for the best.

I think something like this should work:
```
zha:
  custom_quirks_path: /config/custom_quirks
```
Then create the custom_quirks folder in your config folder. (Then the philips folder in that one, then drop the downloaded file in the philips folder).

MattWestb, 2021-05-12:
@TheJulianJES Is it possible updating the firmware of HUE device in ZHA  ?
I was updating all my HUE device in deCONZ and they have link to most known devices in there wiki.

TheJulianJES, 2021-05-12:
Yeah, all firmware is in the zigbee-OTA repo. You can also update dimmers that way.

closed 2021-06-30 by Adminiuga

Homergy, 2022-06-25:
It seems that I still have the exact same issue with 2 old Hue dimmers Model: 324131137411
LOGS:
Event 175 fired 10:01 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:17:88:01:03:e4:dd:c1",
        "unique_id": "00:17:88:01:03:e4:dd:c1:1:0x0006",
        "device_id": "ab82eaeb54a49351f8df4fc6ca23dfed",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "on",
        "args": [],
        "params": {}
    },
    "origin": "LOCAL",
    "time_fired": "2022-06-25T20:01:26.119088+00:00",
    "context": {
        "id": "01G6E78WK79G8AX3FN6JACWWD7",
        "parent_id": null,
        "user_id": null
    }
}
Event 174 fired 10:01 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:17:88:01:03:e4:dd:c1",
        "unique_id": "00:17:88:01:03:e4:dd:c1:1:0x0008",
        "device_id": "ab82eaeb54a49351f8df4fc6ca23dfed",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "step",
        "args": [
            0,
            30,
            9
        ],
        "params": {
            "step_mode": 0,
            "step_size": 30,
            "transition_time": 9,
            "options_mask": null,
            "options_override": null
        }
    },
    "origin": "LOCAL",
    "time_fired": "2022-06-25T20:01:25.536097+00:00",
    "context": {
        "id": "01G6E78W106CA4PDN8JBXQ4YC8",
        "parent_id": null,
        "user_id": null
    }
}
Event 173 fired 10:01 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:17:88:01:03:e4:dd:c1",
        "unique_id": "00:17:88:01:03:e4:dd:c1:1:0x0008",
        "device_id": "ab82eaeb54a49351f8df4fc6ca23dfed",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "step",
        "args": [
            1,
            30,
            9
        ],
        "params": {
            "step_mode": 1,
            "step_size": 30,
            "transition_time": 9,
            "options_mask": null,
            "options_override": null
        }
    },
    "origin": "LOCAL",
    "time_fired": "2022-06-25T20:01:24.896536+00:00",
    "context": {
        "id": "01G6E78VD0KS6ARXP2A6FWYPJA",
        "parent_id": null,
        "user_id": null
    }
}
Event 172 fired 10:01 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:17:88:01:03:e4:dd:c1",
        "unique_id": "00:17:88:01:03:e4:dd:c1:1:0x0006",
        "device_id": "ab82eaeb54a49351f8df4fc6ca23dfed",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "off_with_effect",
        "args": [
            0,
            0
        ],
        "params": {
            "effect_id": 0,
            "effect_variant": 0
        }
    },
    "origin": "LOCAL",
    "time_fired": "2022-06-25T20:01:23.887488+00:00",
    "context": {
        "id": "01G6E78TDF14Z5EQ8TMXZW0EDE",
        "parent_id": null,
        "user_id": null
    }
}

I tried reconfiguring via ZHA but get the error each time
[image]

I tried to download your ZHAquirck but that download isn't available anymore.

Any suggestions?

TheJulianJES, 2022-06-26:
@Homergy Can you create a new issue (https://github.com/zigpy/zha-device-handlers/issues/new/choose) and tag me there? (Please fill out device signature/diagnostic information and so on.)

The fix for this issue is included in all somewhat recent Home Assistant versions. The reconfiguring not working is likely because the dimmer is sleeping (and it needs to be woken up before reconfiguring).