zigpy/zha-device-handlers #882: [Device Support Request] lumi.switch.n0agl1

mbw2001, 2021-05-10:
**Is your feature request related to a problem? Please describe.**
I have added the lumi.switch.n0agl1 to home assistant, it is a relay but identified as a lightbulb. As I am using it for a fan for the bathroom that is misleading. Secondly toggle from lovelace, doesn't work. PowerOff works, but PowerOn doesn't. I assume that is related to being identified as a lightbulb, but I don't know for sure.
Sending zha_cluster_command with:
```
ieee: 'XXX'
endpoint_id: 1
cluster_id: 6
cluster_type: in
command: YYY
command_type: server
```
The the command being 1 to turn on and 0 to turn off works perfectly, but doesn't integrate well with the UI.

**Describe the solution you'd like**
Providing a correct device type, and fixing the poweroff and power toggle functionality.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4447, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.Router: 1>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009",
        "0x000a",
        "0x0702",
        "0x0b04",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "21": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": []
    },
    "31": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": []
    },
    "41": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0012"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.switch.n0agl1",
  "class": "zigpy.device.Device"
}
```

**Additional context**
I have failed in actually making the quirk myself, but I got my HomeAssistant setup with the custom quirks folder, so I will happily test, and provide feedback and/or more information if needed.

puddly, 2021-05-13:
The device type is `ON_OFF_LIGHT = 0x0019` so that's right. It also has the following input clusters:

```Python
 0
 zigpy.zcl.clusters.general.DeviceTemperature,
 zigpy.zcl.clusters.general.Identify,
 zigpy.zcl.clusters.general.Groups,
 zigpy.zcl.clusters.general.Scenes,
 zigpy.zcl.clusters.general.OnOff,
 zigpy.zcl.clusters.general.Alarms,
 zigpy.zcl.clusters.general.Time,
 zigpy.zcl.clusters.smartenergy.Metering,
 zigpy.zcl.clusters.homeautomation.ElectricalMeasurement,
 64704
```

Can you enable https://www.home-assistant.io/integrations/zha/#debug-logging and post the section of your `home-assistant.log` file that contains you turning it both on and off a few times, noting which command worked? If it turns on it seems like it should turn off. Both should be done through the `OnOff` cluster.

mbw2001, 2021-05-17:
Hi,

Thanks for the reply and sorry for the slow feedback!

Is there anything I can grep for in the logs, because enabling debug logs and tailing my logs is like watching the Matrix.

Pressing the switch in Lovelace shows:
```
2021-05-17 10:45:36 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=light, service=turn_on, service_data=entity_id=light.toilet_udsugning>
2021-05-17 10:45:36 DEBUG (MainThread) [homeassistant.components.zha.entity] light.toilet_udsugning: turned on: {}
```
I don't know if more relevant information can be found later in the logs as well? But I cannot see anything.

But the interface just toggles back to off and the switch doesn't activate.

If i turn it on using zigbee cluster command and use the interface to turn it off I get:
```
2021-05-17 10:52:31 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xFCDF:1:0x0006]: executed 'off' command with args: '()' kwargs: '{}' result: [0, <Status.SUCCESS: 0>]
2021-05-17 10:52:31 DEBUG (MainThread) [homeassistant.components.zha.entity] light.toilet_udsugning: turned off: [0, <Status.SUCCESS: 0>]
2021-05-17 10:52:31 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=light.toilet_udsugning, old_state=<state light.toilet_udsugning=on; supported_color_modes=['onoff'], color_mode=onoff, off_brightness=None, friendly_name=Toilet Udsugning, supported_features=8 @ 2021-05-17T10:52:14.573951+02:00>, new_state=<state light.toilet_udsugning=off; supported_color_modes=['onoff'], off_brightness=None, friendly_name=Toilet Udsugning, supported_features=8 @ 2021-05-17T10:52:31.457486+02:00>>
2021-05-17 10:52:31 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2021-05-17 10:52:31 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [30, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xfcdf>, 1, 260, 6, b'\x08.\n\x00\x00\x10\x00', 0, 175, 223, 155, 56, 137, 0, -66]
2021-05-17 10:52:31 DEBUG (MainThread) [zigpy.zcl] [0xfcdf:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=46 command_id=Command.Report_Attributes>
2021-05-17 10:52:31 DEBUG (MainThread) [zigpy.zcl] [0xfcdf:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>)]]
2021-05-17 10:52:31 DEBUG (MainThread) [zigpy.zcl] [0xfcdf:1:0x0006] Attribute report received: on_off=0
2021-05-17 10:52:31 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 46 under 159 request id, data: b'182e0b0a00'
2021-05-17 10:52:31 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xfcdf>, ep: 1, profile: 0x0104, cluster_id: 0x0006, data: b'082e0a00001000'
2021-05-17 10:52:31 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 159, 0, <DeconzAddressEndpoint address_mode=2 address=0xFCDF endpoint=1>, 260, 6, 1, b'\x18.\x0b\n\x00', 2, 0)
2021-05-17 10:52:31 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 159]
2021-05-17 10:52:31 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2021-05-17 10:52:31 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2021-05-17 10:52:31 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 159: 00
2021-05-17 10:52:31 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x9f 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xfcdf endpoint=1>, status: 0x00
```

Again, the device is a relay, so while it could technically be used to control a light, it can also be used for a million other things. So is there any way to force it to be seen as a normal switch?

GingerAdonis, 2022-02-21:
> I added the exact same relay a couple of months ago, which worked perfectly, so I assume something has changed in ZigPy/ZHA maybe? The device signature for the successfully added, and the non-successful are different (see attached files). Any chance this could be resolved?

This might be unrelated to your issue, but I'll give it a go anyways:
I've noticed that Conbee II firmwares `26700700` and `26720700` also break new `lumi.switch.n0agl1` entities when used with ZHA. Downgrading to firmware `26680700` fixed it for me. After adding the entities it's safe to upgrade to newer firmware.

aeklo, 2022-03-26:
> 

I have the Conbee 1, but this applies to this one as well. Downgraded to 26390500, and the device added successfully.
Thanks!

GingerAdonis, 2022-07-28:
This is probably still an issue on zigpy's end as this issue also (still) happens on a SONOFF ZigBee 3.0 USB Dongle Plus.

2dom, 2022-08-29:
Indeed still an issue on SONOFF ZigBee 3.0 - 
[image]

w-marco, 2022-09-03:
Can confirm, just paired this exact module and only the two entities from above show up.
no switch entity. weirdly, already paired modules still have the switch entity and work fine.

2dom, 2022-09-05:
Should we also put this in the Sonoff firmware issue tracker?

puddly, 2022-09-05:
Until there's conclusive evidence that this is a firmware problem, I don't think that's necessary.

If you are able to reliably join and use the device with one firmware version and not the other, please upload detailed debug logs (https://www.home-assistant.io/integrations/zha/#debug-logging) of both cases.

GingerAdonis, 2022-09-05:
> Should we also put this in the Sonoff firmware issue tracker?

Don't think so as it also occurs using other sticks. Other solutions like Zigbee2mqtt aren't affected by this problem.

Anymn, 2022-09-23:
Any update on this? I just added this device to HA and did get identical results as @2dom.

Anymn, 2022-09-23:
> This might be unrelated to your issue, but I'll give it a go anyways: I've noticed that Conbee II firmwares `26700700` and `26720700` also break new `lumi.switch.n0agl1` entities when used with ZHA. [...]

Do you have any kind of instructions how to downgrade the Conbee II?

Anymn, 2022-09-23:
> > 
> 
> I have the Conbee 1, but this applies to this one as well. Downgraded to 26390500, and the device added successfully. Thanks!

Downgrading indeed works, thank you!

loclamor, 2022-10-07:
Thank you ! As of 7 Oct. 2022 downgradinding to 26680700 works as a charm (lastest firmware 26780700 does not work) : after remove device from zha and add it again, all entities show up !

dontinelli, 2022-10-23:
I can confirm both the issue with the latest firmware as well as the workaround with downgrade to firmware 26680700.
One additional remark: under the latest firmware of Conbee II there were no updates on the energy consumption and the power. With the downgrade these both sensors again are updated. I therefore do not intend to update the firmware again until also this issue is solved.

KaherdinTristan, 2022-10-29:
I too have the issue with HA, ZHA integration, Conbee 2 stick, and AQARA T1; Not showing the light switch in HA. 
I was on the latest Conbee 2 firmware; "deCONZ_ConBeeII_0x26780700.bin.GCF" when this occured.

2dom, 2022-11-20:
Hm - any love for this issue. I would love to complete the cabling in my annex....

AndersWestman, 2022-12-14:
Downgrading the firmware on the ConbeeII will put me between a rock and a hard place I'm afraid, as I need a newer firmware (minimum 266b0700 I believe) to get my lumi.plug.maeu01 with fw.0020 to work...
See [https://github.com/zigpy/zha-device-handlers/issues/1952](https://github.com/zigpy/zha-device-handlers/issues/1952)

I had my lumi.switch.n0agl1 fully working until I upgraded my ConbeeII fw from 26580700 to 26780700 which solved one issue but created another one...

GingerAdonis, 2022-12-14:
> Downgrading the firmware on the ConbeeII will put me between a rock and a hard place I'm afraid, as I need a newer firmware (minimum 266b0700 I believe) to get my lumi.plug.maeu01 with fw.0020 to work... See #1952

The downgrade is just required while pairing the affected devices, afterwards you can restore your original firmware.

AndersWestman, 2022-12-14:
> The downgrade is just required while pairing the affected devices, afterwards you can restore your original firmware.

Ok, then I don't understand why my lumi.switch.n0agl1 stopped working.
Well, it partly worked.
The switch on/off still worked, but I did not got any updated measurements and assumed it was a result after the fw update.
Mabey that's not the case?
I tried to re-pair it, but are after that missing most entities as all other above...

2dom, 2023-01-18:
OK ... here is how to fix this by hand:
- Add your Lumi Switch as usual via ZHA integration
- Check device info for MAC address
- install SQLite Web from Addons and add /config/zigbee.db in configuration
- Find the table called "in_clusters_vX" where X is the highes number available
- execute query (insert your in_cluser tabe nache and MAC address below):
   INSERT INTO "in_clusters_v11" (ieee, endpoint_id, cluster)
   VALUES('54:ef:44:10:00:15:a2:16','1','6')
- restart home assistant
- reconfigure device
- light switch should appear

TheJulianJES, 2023-02-04:
From what I've read here, it seems like the device **doesn't work** properly **when** it receives Xiaomi manufacturer codes during pairing. The "standard" mode (OppleMode = 0) seems to work correctly with ZHA though.

@2dom Can you provide the device signature after you made the modifications? (Also, what coordinator and firmware version are you running?)

400HPMustang, 2023-04-11:
> From what I've read here, it seems like the device **doesn't work** properly **when** it receives Xiaomi manufacturer codes during pairing. The "standard" mode (OppleMode = 0) seems to work [...]

Not who you were asking BUT I arrived here with the same issue. I'm using a SkyConnect. Here is the device signature after I ran the SQL Query in the other comment.

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4447, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009",
        "0x000a",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019",
        "0xffff"
      ]
    },
    "21": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": []
    },
    "31": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": []
    },
    "41": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0012"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.switch.n0agl1",
  "class": "zigpy.device.Device"
```

I can control the light, but the device temperature is incorrect and I have no power monitoring available. Not sure if it's possible to fix that or not? You folks are a lot smarter than I am. 

[image]

Gilles2senior, 2023-06-12:
Just installed one of those Xiaomi Aquara SSM-U01 which looks like a great device based on size and features.
Unfortunately, similar to others, beyond switch command (as light) only device temperature appears with wrong value. Even though the power monitoring entities seem to be listed, value remains at 0.
[image]

MarkuBu, 2023-08-02:
> OK ... here is how to fix this by hand:
> 
>     * Back-up your HA
> 
>     * Add your Lumi Switch as usual via ZHA integration
> 
>     * Check device info for MAC address
> 
>     * install SQLite Web from Addons and add /config/zigbee.db in configuration
> 
>     * Find the table called "in_clusters_vX" where X is the highest number available
> 
>     * execute query (insert your in_cluser table name and MAC address below):
>       INSERT INTO "in_clusters_v11" (ieee, endpoint_id, cluster)
>       VALUES('54:ef:44:10:00:15:a2:16','1','6')
> 
>     * restart home assistant
> 
>     * reconfigure device
> 
>     * light switch should appear

Does this also work with MariaDB?

I run into the same issue. Is there still no solution for this? I use ZHA and Skyconnect, so the Conbee workaround doesn't help me

TheJulianJES, 2023-08-29:
- https://github.com/zigpy/zha-device-handlers/pull/2513

will change a few things for HA Core 2023.9.0.
Please try the Home Assistant beta tomorrow with this relay and report any issues you might still have.

With this new quirk version, it's recommended to have an up-to-date coordinator. You might need to (remove and) re-pair the device. Power usage *should* work.
There may be further changes required for the quirk (adding `OppleMode` attribute and setting that to 1), but we'll see how far these changes bring us.

TheJulianJES, 2023-09-16:
Can anyone confirm if the quirk introduced in HA 2023.9.0 improves things for this device?
Are there any issues left?

jsk-sct, 2023-09-25:
I tried to re-pair it and didn't get the new type switch (it remained as a light) but I had an old conbee stick (gen1) so I changed that to skyconnect and then it showed up as a switch in HA. Had some trouble with it not registering active power but after a few re-pairnings it now shows that too.
So it seems to work fine but haven't tested completely yet.

jsk-sct, 2023-10-07:
Okey. There is something strange going on with the Active power sensor. When I first paired the switch and everything seems to work fine. It responds on both very high and low values when the washing machine is going. But after a few days it stops responding and it just shows 0,0 W. The switch-function works fine, it's just the Active power that doesn't change (probably the other power meter sensors too but they are harder too see if they are malfunctioning). I didn't have this problem before 2023.9 so it seems as it has to do something with the quirk.
I haven't done much troubleshooting before regarding these kind of problems but if someone can show me how to get the right logs I'll gladly help out.

400HPMustang, 2023-10-24:
Not sure if it's a ZHA issue or an issue with these quirks but I have two of these relays changing to "unknown" and then going "unavailable" and the only way to recover them is to restart the ZHA integration. None of my other Zigbee accessories are having this issue. Not sure what to provide that would help diagnose that issue.

Gilles2senior, 2023-10-26:
To contribute to this discussion, I have experienced no disconnection of this Aqara relay on ZHA since it has been installed on my side. The only concern was the lack of reporting of consumption data, which is now solved with latest update. I would buy another one with no regret when I need to add one relay in my set-up.

vukisz, 2023-10-26:
> To contribute to this discussion, I have experienced no disconnection of this Aqara relay on ZHA since it has been installed on my side. The only concern was the lack of reporting of consumption [...]

Thanks. Do you have rocker switches, or "reset" switches
What are your actuator? Sky connect?
You are talking about T1 with neutral, right?

Gilles2senior, 2023-10-26:
> Thanks. Do you have rocker switches, or "reset" switches What are your actuator? Sky connect? You are talking about T1 with neutral, right?

I use ZHA, with a Sonoff type E coordinator. The switch is connected through a couple of Aetec extenders. It is a T1 with Neutral.
With the latest update, the device temperature is still not reporting correctly, which is not critical for me, but active power is reported.

vukisz, 2023-10-26:
Okay, I'll have a leap of faith with this setup: Sky connect (so ZHA) and reset switches.

eduarcusa, 2023-10-31:
> Okey. There is something strange going on with the Active power sensor. When I first paired the switch and everything seems to work fine. It responds on both very high and low values when the [...]

I have 11 aqara relays T1 with this signature "zhaquirks.xiaomi.aqara.switch_t1.SwitchT1Alt1" and this is happening to me too. But it's weird because only happens in the last three relays I bought and installed. The others don't have that problem with the active power.

eduarcusa, 2023-11-03:
> I have 11 aqara relays T1 with this signature "zhaquirks.xiaomi.aqara.switch_t1.SwitchT1Alt1" and this is happening to me too. But it's weird because only happens in the last three relays I bought [...]

I'm going to explain what I found. I have 11 aqara relays T1 with neutral. There are 8 of them that work perfectly the other 3 of them after few days the Active Power stops working and stays at 0.0. Repairing those again make Active power work again.
I found that the working ones I added with core 2023.9.1 or 2023.9.2. The ones that are not working I added with core 2023.10.2. Something between those versions happened.

Few days ago I upgraded to 2023.10.5 and something happened. The entities of the 8 aqara relay working perfect changed the language from english to spanish: switch to interruptor, active power to potencia, etc. 
Can you guess what happened with my other 3? They didn't change at all, and the problems with the active power still remain. 

So yesterday I tried to re-add those 3 and it was a nightmare. Maybe I re-added each one 20-30 times before all the entities were in spanish and the control was a switch instead of a lightbulb. I tried a lot of combinatios, remove, reboot, add. remove, add and press de botton continously while it was configuring, remove, reboot, wait 5 minutos and add. Sometimes it detected at unknown and unknown manufacturer, detected at lumi but the control was a ligthbulg. At the end and after 20-30 times 
 in each one it detected perfectly. In a few days I'm going to see if active power still works.

eduarcusa, 2023-11-07:
After a few days testing I had the same problems with those 3 aqara T1. The active power stops working after 1 or 2 days :(

eduarcusa, 2023-11-14:
Update: 

I added the last [quirk update](handlers/blob/dev/zhaquirks/xiaomi/aqara/switch_t1.py) from @TheJulianJES  and after remove and re-add the device instead of SwitchT1Alt1 now detects SwitchT1Alt2. I did the same with another device that doesn't have the problem with active power and also changed from SwitchT1Alt1 to SwitchT1Alt2.

I'll test if it fixes the problem.

400HPMustang, 2023-11-14:
> Update:
>
> I added the last [quirk update](handlers/blob/dev/zhaquirks/xiaomi/aqara/switch_t1.py) from @TheJulianJES and after remove and re-add the device instead of SwitchT1Alt1 now detects [...]

I tried to find that quirk you linked to and it doesn't work...

eduarcusa, 2023-11-14:
> I tried to find that quirk you linked to and it doesn't work...

Sorry, I updated the link.

eduarcusa, 2023-11-20:
> Update:
> 
> I added the last quirk update "https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/xiaomi/aqara/switch_t1.py" from @TheJulianJES and after remove and re-add the device instead of SwitchT1Alt1 now detects SwitchT1Alt2. I did the same with another device that doesn't have the problem with active power and also changed from SwitchT1Alt1 to SwitchT1Alt2.
> 
> I'll test if it fixes the problem.

Like I said before I have 3 aqara switches with problems with Active Power sensor. After a day always stopped working. I updated the last quirk from @TheJulianJES on two of the switches and they changed from SwitchT1Alt1 to SwitchT1Alt2, I didn't updated the third one. 

The results after a week are that the ones I updated worked perfectly and I didn't have the problem with the Active Power anymore, the third one that I didn't update has the same problem. So thank you @TheJulianJES , it seems that you were right and there are different signatures.

jsk-sct, 2023-12-09:
> Like I said before I have 3 aqara switches with problems with Active Power sensor. After a day always stopped working. I updated the last quirk from @TheJulianJES on two of the switches and they [...]

OK. I tried repairing the device after upgrading to 2023.12. I got Quirk: zhaquirks.xiaomi.aqara.switch_t1.SwitchT1Alt3 under the device but, like before, the power meter stops working after a day or 2.
So I'd like to try thq quirk Alt2 that seems to be working for @eduarcusa. My newbie question is if there's any way I could do that in the visual interface or do I have to handle it like a custom quirk and put it in a folder and edit configuration.yaml?
Much appreciated if someone could guide me through this

eduarcusa, 2023-12-11:
> OK. I tried repairing the device after upgrading to 2023.12. I got Quirk: zhaquirks.xiaomi.aqara.switch_t1.SwitchT1Alt3 under the device but, like before, the power meter stops working after a day [...]

Bad news. I had a problem with my zigbee network and I had to re-add some devices. Even though my devices are detected with zhaquirks.xiaomi.aqara.switch_t1.SwitchT1Alt2 I have again the same problem with the Active Power in some devices but others work perfect. I had a lot of problems trying to re-add the devices and I had to restart home assistant, reload zigbee coordinator, the ones that work perfect are the ones that I added after a restart. Perhaps is a problem with the coordinator adding new devices? Are you using Sonoff Dongle-E like me?

The quirk a tried manually is the same that the quirk when you upgrade 2023.12. In fact before upgrading to 2023.12 I deleted the custom quirk.

jsk-sct, 2023-12-12:
> Bad news. I had a problem with my zigbee network and I had to re-add some devices. Even though my devices are detected with zhaquirks.xiaomi.aqara.switch_t1.SwitchT1Alt2 I have again the same [...]

Crap.. 🙁
I'll try to repair them after a restart and see if that helps but I don't have much faith in that.

I have a SkyConnect but it seems to be the same chip in both of them (EFR32MG21) so it might have to do with that but who knows at this point 🤷

akarabach, 2024-02-06:
For me alt1 works better than alt2 (see my screens). For some reasons the same relays (I have 8) can have alt1 or alt2 (I have 1 alt1 and 7 alt2). I also have T2 (two channels) and it works perfectly fine. Should we have a look on quirk for T2 and implement it the same way ?
[image]
[image]

dfanica, 2024-03-18:
After reading through this thread I decided to look into my relays. I’ve been having issues with one after a power cut and re-added it a few months ago. This is the one that now works for a while and then stops reporting power consumption. The other one works perfectly.

The one that works is on an older firmware since I added the other one to Aqara hub and ungraded it. The one that works is also on a quick alt2 while the one that doesn’t work is on alt1.

Here’s the information on both:

The one that works:
[image]
[image]

The one that doesn’t work (or works for a while then stops):
[image]
[image]

I should also mention that the extra sensors appear the moment it stops working. In the beginning the sensors section looks the same as the one’s that works. Not sure if that’s important.

eduarcusa, 2024-09-20:
The last time I tried it didn't work at all. I'm gonna try again and I'll let you know.

Gilles2senior, 2024-09-20:
> There hasn't been any activity on this issue recently. Due to the high number of incoming GitHub notifications, we have to clean some of the old issues, as many of them have already been resolved [...]

There has been no progress on this point, and the device, which is still on sales, is still not supported properly on ZHA. This is unfortunate as there aren't that many alternative devices with same quality and features.

akarabach, 2024-09-20:
Waiting for Aqara Matter Thread switches, probably it's the only way to have everything in HA without such kind of hacks

homerojleal, 2024-11-06:
Also have more then a dozen of these switches with same issues on the sensors (power) except one of them (seems like to be the first one I've paired).

Not working - Firmware 0x0000001e.
[image]

Working - Firmware 0x0000001e
[image]

eduarcusa, 2024-11-06:
I tried again to re-pair them and after a few days it happend again, power meter stopped working. And like @homerojleal said only the first ones I added before X update are still working. And they are the same model because i bougth them together. It looks like there was an upgrade that broke this.

dfanica, 2024-11-24:
The problem seems to be with the Quirk: zhaquirks.xiaomi.aqara.switch_t1.SwitchT1Alt1

I had one that was working for over a year now that was on SwitchT1Alt2… I have now re-added it and it was added with SwitchT1Alt1. It has now stopped reporting power after 2 days, same as the other one.

Anyone knows how to target the SwitchT1Alt2 quirk?

dfanica, 2024-11-24:
I found a solution that I’m testing atm. Adding the relay via another repeater device, maybe better positioned, made the pairing process flawless. Before it used to take me hours to successfully pair it while with this method it’s almost instant.

The entities available are slightly less, screenshot below.

I will update this thread if the power consumption still works after a few days or, if a long time has passed without an update, assume it worked. 😂

[image]

[image]

[image]

dfanica, 2024-11-25:
Well, I didn’t have to wait long at all… it seems that the issue persists. I will try removing and adding one a few times to see if SwitchT1Alt2 quirk is still possible to get.

dfanica, 2024-11-27:
This might not be a solution for everyone but I have a Xiaomi Gateway 3 and I was able to pair the relay and it supplies power consumption via https://github.com/AlexxIT/XiaomiGateway3