zigpy/zha-device-handlers #892: Please add support for 2CH DC5-32v relay-module gateway (eWeLink ZB-SW02)

miklosbagi, 2021-05-21:
**Is your feature request related to a problem? Please describe.**
This device gets setup in Home Assistant as 2 switches, however, this has a traditional no/nc/comm relay for controlling blinds/shades/curtains up/down.

**Describe the solution you'd like**
I'd like this device to be recognized as two entities, each with an up-down 🔼 🔽 option as Comm-NC and comm-NO switchings are made.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=0, maximum_buffer_size=80, maximum_incoming_transfer_size=160, server_mask=0, maximum_outgoing_transfer_size=160, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.Router: 1>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": [
        "0x0000"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": [
        "0x0000"
      ]
    }
  },
  "manufacturer": "eWeLink",
  "model": "ZB-SW02",
  "class": "zigpy.device.Device"
}
```

**Additional context**

It seems that ZHA recognizes 2 light switches, and both act as reset buttons (turn back off within about a second):
- light.ewelink_zb_sw02_f82e6b22_on_off
- light.ewelink_zb_sw02_f82e6b22_on_off

quadrupole1988, 2021-06-25:
How did you get  ZHA to even discover the device? 

I have home assistant running ZHA and when I try to add this relay, it doesn't find it. Help please?

miklosbagi, 2021-06-25:
That is because you haven't put the device into pairing mode. It takes a few trial & error button mashing to get there.

quadrupole1988, 2021-06-25:
I realized I didn't download install the repo, but once, I did, everything worked great.

@miklosbagi Thanks!

closed 2021-12-29 by github-actions[bot]