zigpy/zha-device-handlers #893: [BUG] IKEA Tradfri E1810 does not send zha events

zhulik, 2021-05-21:
**Describe the bug**
A couple of months ago I migrated my network from a ConBee2 coordinator to an Elelabs USB coordinator. All devices work fine except the Tradfri switch. It correctly pairs, reports it's battery state, but does not produce any `zha_event`s when I press buttons. Device triggers also don't work.

In the same time if I install HASS on another computer and build a new network using the old ConBee2 coordinator - the switch works as expected.

On both setups the devices is handled by `zhaquirks.ikea.fivebtnremotezha.IkeaTradfriRemote`.

**Expected behavior**
A `zha_event` is sent every time I press a button on the remote.

**Screenshots**

**Additional context**
```json
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0820",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x1000",
        "0xfc7c"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "TRADFRI remote control",
  "class": "zhaquirks.ikea.fivebtnremotezha.IkeaTradfriRemote"
}
```
Please give me some hints on what logs I need to provide. I have relatively big ZigBee network and if I enable debug logging the logs are pretty noisy and it's hard to find anything in there. Can I somehow get logs related to only one device in the network?

MattWestb, 2021-05-21:
Try doing on "reconfigure device" from the device card and waking up the sleeping remote.
For my all IKEA controllers devices is not reporting OK if not doing it after pairing them or doing on new binding.
Also last seen and battery is not working OK until getting the reporting fixed.

I have just testing `zha_event` and i is getting:
```
Listen to events
 
Listening to
zha_event
 
Event 3 fired 6:19 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:0d:6f:ff:fe:51:43:5c",
        "unique_id": "00:0d:6f:ff:fe:51:43:5c:1:0x0006",
        "device_id": "90e0b4e29f34a30c2b1e6bd8df29cbfe",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "toggle",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2021-05-21T16:19:13.637537+00:00",
    "context": {
        "id": "37c1a8b04715cb1b1623a4366b4e1ecc",
        "parent_id": null,
        "user_id": null
    }
}
Event 2 fired 6:19 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:0d:6f:ff:fe:51:43:5c",
        "unique_id": "00:0d:6f:ff:fe:51:43:5c:1:0x0005",
        "device_id": "90e0b4e29f34a30c2b1e6bd8df29cbfe",
        "endpoint_id": 1,
        "cluster_id": 5,
        "command": "press",
        "args": [
            257,
            13,
            0
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2021-05-21T16:19:11.785674+00:00",
    "context": {
        "id": "9a2e2bfcf6bed312a8e4e86a128a129e",
        "parent_id": null,
        "user_id": null
    }
}
Event 1 fired 6:19 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:0d:6f:ff:fe:51:43:5c",
        "unique_id": "00:0d:6f:ff:fe:51:43:5c:1:0x0006",
        "device_id": "90e0b4e29f34a30c2b1e6bd8df29cbfe",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "toggle",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2021-05-21T16:19:10.467228+00:00",
    "context": {
        "id": "9f9de38f08b1371e46dba89bbc315b86",
        "parent_id": null,
        "user_id": null
    }
}
Event 0 fired 6:13 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:0d:6f:ff:fe:51:43:5c",
        "unique_id": "00:0d:6f:ff:fe:51:43:5c:1:0x0006",
        "device_id": "90e0b4e29f34a30c2b1e6bd8df29cbfe",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "toggle",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2021-05-21T16:13:55.598294+00:00",
    "context": {
        "id": "48d27c6b7fe4d7cfa4bee5c705167e9e",
        "parent_id": null,
        "user_id": null
    }
}
````

zhulik, 2021-05-21:
Thanks for a hint, but it didn't help. The reconfiguration worked fine, I got green check marks everywhere in the report, but still no "zha_events" in the dev tab

MattWestb, 2021-05-21:
Strange :((

Do you have one old quirk installed that is not being updated ?
Or some other component that is not being updated (quirk handling was updated for some mounts ago).

Look if the `<your config directory>\deps` is empty or the files there is overriding the installed HA components.

MattWestb, 2021-05-21:
Is you using `zha_even` or `zha_event` in the events tab  ?

zhulik, 2021-05-21:
I'm listening to `zha_event`. My setup is up to date, the installed version of zhaquirks is 0.0.57, which is the latest one.

MattWestb, 2021-05-21:
OK then i dont have more ideas but i hope some other have it ;-)

Adminiuga, 2021-05-22:
enable debug logging and when reconfiguring device, make sure all the "bind" commands complete successfully (i.e. keep remote awake while reconfiguring).
Unless something has changed in the firmware, this remote used to workl.

If you pair it back to ConBee and enable debug logging. can you post the "incoming messages" reported by zigpy_deconz?

MattWestb, 2021-05-22:
The cluster layout look current and i think its on the latest firmware but worth to checking then the latest is 2.3.014 for it.

zhulik, 2021-05-22:
@zhulik thanks for a hint. I recorded reconfiguration logs using this logger config:
```yaml
logger:
  default: warn
  logs:
    homeassistant.components.zha: debug
    bellows.ezsp: debug
    bellows.uart: debug
    zigpy.zdo: debug
    zigpy.application: debug
```

I removed timestamps from logs to make them more readable:
```
(MainThread) [homeassistant.components.zha.api] Reconfiguring node with ieee_address: 58:8e:81:ff:fe:2c:00:a6
(MainThread) [homeassistant.components.zha.core.device] [0x2055](TRADFRI remote control): started configuration
(MainThread) [homeassistant.components.zha.core.channels.base] [0x2055:ZDO](TRADFRI remote control): 'async_configure' stage succeeded
(MainThread) [homeassistant.components.zha.core.channels.base] [0x2055:1:0x0000]: finished channel configuration
(MainThread) [bellows.ezsp.protocol] Send command setExtendedTimeout: (58:8e:81:ff:fe:2c:00:a6, True)
(bellows.thread_0) [bellows.uart] Sending: b'37a021a92a2ab3b2756ab5a4240d935a9a7e'
(bellows.thread_0) [bellows.uart] Data frame: b'74a0a1a92a2ab4837e'
(bellows.thread_0) [bellows.uart] Sending: b'8070787e'
(MainThread) [bellows.ezsp.protocol] Application frame 126 (setExtendedTimeout) received: b''
(MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x2055, EmberApsFrame(profileId=0, clusterId=33, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=246), 247, b'\xf6\xa6\x00,\xfe\xff\x81\x8eX\x01\x01\x00\x03"\xf2\xa5\xfe\xff\xcc\xcc\xcc\x01')
(bellows.thread_0) [bellows.uart] Sending: b'40a121a9602a15e779944a04aa5592099d4e275d7d3ad8912dfdea9d767d5df067a6eacddd4d7d5d5a3924191ea58d04cc7e'
(bellows.thread_0) [bellows.uart] Data frame: b'05a1a1a9602a15bca07a7e'
(bellows.thread_0) [bellows.uart] Sending: b'8160597e'
(MainThread) [bellows.ezsp.protocol] Application frame 52 (sendUnicast) received: b'000e'
(MainThread) [bellows.ezsp.protocol] Send command setExtendedTimeout: (58:8e:81:ff:fe:2c:00:a6, True)
(bellows.thread_0) [bellows.uart] Sending: b'51a621a92a2ab3b2756ab5a4240d93308b7e'
(bellows.thread_0) [bellows.uart] Data frame: b'16a6a1a92a2aad5e7e'
(bellows.thread_0) [bellows.uart] Sending: b'82503a7e'
(MainThread) [bellows.ezsp.protocol] Application frame 126 (setExtendedTimeout) received: b''
(MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x2055, EmberApsFrame(profileId=260, clusterId=4096, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=248), 249, b'\x01\xf8A\x00')
(bellows.thread_0) [bellows.uart] Sending: b'62a721a9602a15e779904b25ba5493099d4e275314ca6673bcc63ebb7e'
(bellows.thread_0) [bellows.uart] Data frame: b'27a7a1a9602a15bde5b07e'
(bellows.thread_0) [bellows.uart] Sending: b'83401b7e'
(MainThread) [bellows.ezsp.protocol] Application frame 52 (sendUnicast) received: b'000f'
(MainThread) [bellows.ezsp.protocol] Send command setExtendedTimeout: (58:8e:81:ff:fe:2c:00:a6, True)
(bellows.thread_0) [bellows.uart] Sending: b'73a421a92a2ab3b2756ab5a4240d9316847e'
(bellows.thread_0) [bellows.uart] Data frame: b'30a4a1a92a2a51347e'
(bellows.thread_0) [bellows.uart] Sending: b'8430fc7e'
(MainThread) [bellows.ezsp.protocol] Application frame 126 (setExtendedTimeout) received: b''
(MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x2055, EmberApsFrame(profileId=0, clusterId=33, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=250), 251, b'\xfa\xa6\x00,\xfe\xff\x81\x8eX\x01 \x00\x03"\xf2\xa5\xfe\xff\xcc\xcc\xcc\x01')
(bellows.thread_0) [bellows.uart] Sending: b'04a521a9602a15e779944a04aa5592099d4e275116d89d2dfdea9d767d5df067a6cbcddd4d7d5d5a3924191ea58da3557e'
(bellows.thread_0) [bellows.uart] Data frame: b'41a5a1a9602a15a26bfe7e'
(bellows.thread_0) [bellows.uart] Sending: b'8520dd7e'
(MainThread) [bellows.ezsp.protocol] Application frame 52 (sendUnicast) received: b'0010'
(bellows.thread_0) [bellows.uart] Data frame: b'51a5b1a9112a15b65894a524ab1593499cb5d77314459874f8cf62adfc7e7ea97e'
(bellows.thread_0) [bellows.uart] Sending: b'8610be7e'
(MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00040100ef010140010000fbf0d8f98bffff050901240000'
(bellows.thread_0) [bellows.uart] Data frame: b'61a5b1a96b2a164fa6904b23aa546d499d4e27e5edce676a6e7e'
(bellows.thread_0) [bellows.uart] Sending: b'87009f7e'
(MainThread) [bellows.ezsp.protocol] Application frame 63 (messageSentHandler) received: b'03fdff0401060001ff000100004e000000'
(bellows.thread_0) [bellows.uart] Data frame: b'71a5b1a96b2a164fa6904b23aa546d499d4e27e4edce672cac7e'
(bellows.thread_0) [bellows.uart] Sending: b'8070787e'
(MainThread) [bellows.ezsp.protocol] Application frame 63 (messageSentHandler) received: b'03fdff0401060001ff000100004f000000'
(bellows.thread_0) [bellows.uart] Data frame: b'01a5b1a9112a15b259b5ca25aa1593499c1fe366b8ee9874ff3063abb57e'
(bellows.thread_0) [bellows.uart] Sending: b'8160597e'
(MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'000000218000004001000051c4cd5520ffff02f600'
(bellows.thread_0) [bellows.uart] Data frame: b'11a5b1a96b2a15e779944a04aa5592099d4e27a51ace6739727e'
[... cut ...]
(MainThread) [homeassistant.components.zha.core.channels.base] [0x2055:1:0x0005]: 'async_configure' stage succeeded
(MainThread) [homeassistant.components.zha.core.device] [0x2055](TRADFRI remote control): completed configuration
(MainThread) [homeassistant.components.zha.core.device] [0x2055](TRADFRI remote control): stored in registry: ZhaDeviceEntry(name='IKEA of Sweden TRADFRI remote control', ieee='58:8e:81:ff:fe:2c:00:a6', last_seen=1621676427.9437542)
```

Screenshot of the details table
[image]

MattWestb, 2021-05-22:
looks 110% as my then doing reconfigure.

Is your remote still bonded to the coordinator (default after joining the network) or to one Zigbee group and is the group having the coordinator in it ?

zhulik, 2021-05-22:
I didn't configure any groups, I simply add the device to my network.

MattWestb, 2021-05-22:
Then it shall sending all commands to group 0x0000 = the coordinator.
I trying if its doing that then being new joined to the network.

zhulik, 2021-05-22:
@Adminiuga paring logs from the computer with the ConBee coordinatior. Logger config:
```yaml
logger:
  default: warn
  logs:
    homeassistant.components.zha: debug
    zigpy_deconz: debug
    zigpy.zdo: debug
    zigpy.application: debug
```

Logs:
```
[zigpy.zdo] [0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, ep: 0, profile: 0x0000, cluster_id: 0x0036, data: b'033c00'
[zigpy_deconz.uart] Frame received: 0x0e18000700a600
[zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
[zigpy_deconz.api] Command Command.aps_data_confirm (0,)
[zigpy_deconz.uart] Send: 0x04180007000000
[zigpy_deconz.uart] Frame received: 0x04180012000b00220401fcff000000000000
[zigpy_deconz.api] APS data confirm response for request with id 4: 00
[zigpy_deconz.api] Request id: 0x04 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.GROUP address=0xfffc endpoint=None>, status: 0x00
[zigpy_deconz.api] Command Command.write_parameter (2, <NetworkParameter.permit_join: 33>, b'<')
[zigpy_deconz.uart] Send: 0x0b190009000200213c
[zigpy_deconz.uart] Frame received: 0x0b19000800010021
[zigpy_deconz.api] Write parameter permit_join: SUCCESS
[zigpy_deconz.uart] Frame received: 0x1c1a0012000b0003a6002cfeff818e58fff0
[zigpy_deconz.uart] Frame received: 0x1c1b000c00050002e4fffff0
[zigpy_deconz.uart] Frame received: 0x1c1c000c00050002e4fffff0
[zigpy_deconz.uart] Frame received: 0x1c1d000c00050002e4fffff0
[zigpy_deconz.uart] Frame received: 0x1c1e000c00050002e4fffff0
[zigpy_deconz.uart] Frame received: 0x1c1f000c00050002e4fffff0
[zigpy_deconz.uart] Frame received: 0x1c20000c00050002e4fffff0
[zigpy_deconz.uart] Frame received: 0x1c21000c00050002e4fffff0
[zigpy_deconz.uart] Frame received: 0x1c22000c00050002e4fffff0
[zigpy_deconz.uart] Frame received: 0x1c23000c00050002e4fffff0
[zigpy_deconz.uart] Frame received: 0x1c240012000b0003a6002cfeff818e58fff0
[zigpy_deconz.uart] Frame received: 0x1c25000c00050002e4fffff0
[zigpy_deconz.uart] Frame received: 0x0e26000700aa00
[zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
[zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
[zigpy_deconz.uart] Send: 0x171a000800010001
[zigpy_deconz.uart] Frame received: 0x171a002a0023002202fdff0002e4ff00000013000c0081e4ffa6002cfeff818e588000afff2787be00f0
[zigpy_deconz.api] APS data indication response: [35, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xfffd>, 0, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xffe4>, 0, 0, 19, b'\x81\xe4\xff\xa6\x00,\xfe\xff\x81\x8eX\x80', 0, 175, 255, 39, 135, 190, 0, -16]
zigpy_deconz.zigbee.application] New device joined: 0xffe4, 58:8e:81:ff:fe:2c:00:a6
zigpy.application] Device 0xffe4 (58:8e:81:ff:fe:2c:00:a6) joined the network
[zigpy.zdo] [0xffe4:zdo] ZDO request ZDOCmd.Device_annce: [0xFFE4, 58:8e:81:ff:fe:2c:00:a6, 128]
zigpy.device] [0xffe4] Requesting 'Node Descriptor'
[zigpy.util] Tries remaining: 2
[zigpy.device] [0xffe4] Extending timeout for 0x05 request
[zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 5 under 6 request id, data: b'05e4ff'
[zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xffe4>, ep: 0, profile: 0x0000, cluster_id: 0x0013, data: b'81e4ffa6002cfeff818e5880'
[zigpy_deconz.api] Command Command.aps_data_request (18, 6, 0, <DeconzAddressEndpoint address_mode=2 address=0xFFE4 endpoint=0>, 0, <ZDOCmd.Node_Desc_req: 0x0002>, 0, b'\x05\xe4\xff', 2, 0)
[zigpy_deconz.uart] Send: 0x121b0019001200060002e4ff000000020000030005e4ff0200
[zigpy_deconz.uart] Frame received: 0x1c1b000c00050002e4fffff0
[zigpy_deconz.uart] Frame received: 0x121b00090002002206
[zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 6]
[zigpy_deconz.uart] Frame received: 0x1c1c000c00050002e4fffff0
[zigpy_deconz.uart] Frame received: 0x0e1d000700a600
[zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
[zigpy_deconz.uart] Frame received: 0x1c1e000c00050002e4fffff0
[zigpy_deconz.api] Command Command.aps_data_confirm (0,)
[zigpy_deconz.uart] Send: 0x041c0007000000
[zigpy_deconz.uart] Frame received: 0x041c0013000c00220602e4ff00000000000000
[zigpy_deconz.api] APS data confirm response for request with id 6: 00
[zigpy_deconz.api] Request id: 0x06 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xffe4 endpoint=0>, status: 0x00
[zigpy_deconz.uart] Frame received: 0x0e1d000700aa00
[zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
[zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
[zigpy_deconz.uart] Send: 0x171d000800010001
[... 481 lines cut ...]
[zigpy_deconz.uart] Send: 0x04560007000000
[zigpy_deconz.uart] Frame received: 0x1c57000c00050002e4fffff0
[zigpy_deconz.uart] Frame received: 0x04560013000c00222c02e4ff01010000000000
[zigpy_deconz.api] APS data confirm response for request with id 44: 00
[... cut ...]
[homeassistant.components.zha.core.channels.base] [0xFFE4:1:0x0006]: 'async_initialize' stage succeeded
[homeassistant.components.zha.core.channels.base] [0xFFE4:1:0x0019]: 'async_initialize' stage succeeded
[homeassistant.components.zha.core.channels.base] [0xFFE4:1:0x0005]: 'async_initialize' stage succeeded
[homeassistant.components.zha.core.device] [0xFFE4](TRADFRI remote control): power source: Battery or Unknown
[homeassistant.components.zha.core.device] [0xFFE4](TRADFRI remote control): completed initialization
```

MattWestb, 2021-05-22:
I was deleting the remote and waiting 2 minutes and repairing it and its sending event nice in HA.

I can see in your log that its setting up binding to the group 0 that is normal but do your system having group 0 cofigurated ?
```
Group Information
Name: Default Lightlink Group
Group Id: 0x0000
```
If not ZHA is not listening to it and cant getting the commands from the remote.

I think our code master (Adminiuga) is coming online little later (US time zone).

zhulik, 2021-05-22:
Yes, the group 0 does exist and besides the tradfri remote I have a bunch of xiaomi 2- and 1-button remotes, they work perfectly

zhulik, 2021-05-22:
Ok, I figured out the problem. When I was migrating the network to the new coordinator, I removed the integration, but did not remove zha.storage and zigbee.db. For some reason information about the old coordinator was left there and ZHA for some reason thought that it keep working with the old coordinator, however I saw 2 coordinators on the 'Devices' list.

FIX:
- I manually removed all occurrences of the old coordinator and it's ieee from zha.storage
- I manually replaced all occurrences of old device's ieee in zigbee.db with the new device's ieee.

It was a clear guess, but it worked out.

After these manipulations the network started as nothing had changed and the remote started sending `zha_event`s as it should.

@Adminiuga maybe you can explain in two words why this might happen? Maybe it makes sense to update zha documentation and describe the process of migrating networks from one coordinator to another? WDYT?

Adminiuga, 2021-05-22:
While remote is on ConBee, can you show debug for receive messages when you press buttons? Interested to see in the address mode of those messages.
Conbee receives all group messages without doing anything special, but ezsp based ones have to subscribe to the group addresses

Adminiuga, 2021-05-22:
I'd need to test it. Ezsp radio have to be subscribed to multicast messages, if there were two coordinator with 0x0000 then maybe it used the wrong one to do those operations

zhulik, 2021-05-22:
There were 2 coordinators on the list, but only one of them(the new one) was actually plugged in.

MattWestb, 2021-08-03:
@zhulik ZHA shall have getting one fix for only using the active device with short address 0x0000 (coordinator) if have getting 2 or more of then in the DB and also possible deleting the inactive one in HA service that was not possible before.

I think its no problems for you then you have doing one "hard cleaning" if the DB and storage files so only one info :-))

closed 2022-02-06 by github-actions[bot]