zigpy/zha-device-handlers #897: [Device Support Request] Tuya zigbee remote TS004F

dsellers1, 2021-05-25:
**Is your feature request related to a problem? Please describe.**
Tuya-based 4-gang scene switch. Similiar to this device [https://zigbee.blakadder.com/Earda_ESW-0ZAA-EU.html](https://zigbee.blakadder.com/Earda_ESW-0ZAA-EU.html). Linked device has Zigbee ID: "TS0044"; "_TZ3000_vp6clf9d"; mine has Zigbee ID: "TS0044"; "_TZ3000_xabckq1v". Although the TS0044 seems similar, mine reports as a TS004F.

Similar support requests:
#525 TS0044 Tuya 4-button scene switch
#679 Tuya zigbee remote TS0044 - Different manufacturer
#699 Scene Switch 4 Gang TS0044 doesn't update

[image]
In Node-RED, I see zha_events for bottom right & top right buttons but not the left buttons.

Button press
bottom right: command = "step", payload.event.args = [1, 51, 10]
top right: command = "step", payload.event.args = [0, 51, 10]
bottom left: no response
top left: no response

Button hold 5 seconds
bottom right: command = "move", payload.event.args = [1,51] -- with matching "stop" commands for release
top right: command = "move", payload.event.args = [0,51]  -- with matching "stop" commands for release
bottom left: no response
top left: no response

Button double press 
bottom right: no response
top right: no response
bottom left: no response
top left: no response

Documentation says it supports double click, but I didn't see anything in the log. Only something like this but I'm not sure if it is relevant:
2021-05-24 21:19:04 DEBUG (MainThread) [bellows.ezsp.protocol] Send command nop: ()
2021-05-24 21:19:04 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 5 (nop) received: b''

Interestingly, there is an entity for the device: switch.tz3000_xabckq1v_ts004f_11766bfe_on_off
Top left button turns it on; bottom left button turns it off. Pressing the same button again does not affect the switch since it is already on/off. This isn't really useful (IMO) for a scene selector switch.

**Describe the solution you'd like**
Would like the left buttons function like the right buttons. Double press capability would be nice.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0104",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0006",
        "0x1000"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x000a",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "_TZ3000_xabckq1v",
  "model": "TS004F",
  "class": "zigpy.device.Device"
[... 172 lines cut ...]
[0xa525:1:0x0001]: 'async_initialize' stage succeeded
[0xa525:1:0x0008]: 'async_initialize' stage succeeded
[0xa525:1:0x0006]: 'async_initialize' stage succeeded
[0xa525:1:0x0019]: 'async_initialize' stage succeeded
[0xa525:1:0x0005]: 'async_initialize' stage succeeded
[0xa525](TS004F): power source: Battery or Unknown
[0xa525](TS004F): completed initialization
[0xa525:1:0x0006]: attempting to update onoff state - from cache: True
[0x0a19:2:0x0019] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=195 command_id=1>
[0x0a19:2:0x0019] ZCL request 0x0001: [0, 4107, 269, 1107315168, None]
[0x0a19:2:0x0019] OTA query_next_image handler for 'Philips SML001': field_control=0, manufacture_id=4107, image_type=269, current_file_version=1107315168, hardware_version=None
[0x0a19:2:0x0019] No OTA image is available
[0x0a19:2:0x0406] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=196 command_id=Command.Report_Attributes>
[0x0a19:2:0x0406] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=bitmap8, value=bitmap8.0>)]]
[0x0a19:2:0x0406] Attribute report received: occupancy=0
[0x0a19:2:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=197 command_id=Command.Report_Attributes>
[0x0a19:2:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=11727>)]]
[0x0a19:2:0x0400] Attribute report received: measured_value=11727
[0x369d:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=150 command_id=Command.Report_Attributes>
[0x369d:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.true>)]]
[0x369d:1:0x0006] Attribute report received: on_off=1
[0x7ef1:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=124 command_id=Command.Report_Attributes>
[0x7ef1:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>)]]
[0x7ef1:1:0x0006] Attribute report received: on_off=0
```

Hedda, 2021-05-25:
Reference discussion here https://community.home-assistant.io/t/help-adding-a-variation-of-tuya-ts0044-ts004f-to-zha/310579

dsellers1, 2021-05-25:
> 
> 
> Reference discussion here https://community.home-assistant.io/t/help-adding-a-variation-of-tuya-ts0044-ts004f-to-zha/310579
Looks like the same as me! I tried doing something similar but failed miserably. I followed the instructions at https://github.com/zigpy/zha-device-handlers#testing-new-releases. Copied the TS0044.py into the /deps/ folder where it is supposed to go. I hoped it would be a simple matter of changing the MODELS_INFO to correspond but nope. I broke HA. I believe I corrupted the zigbee.db. I had to delete it to get things kinda working and then restore from a snapshot. I may give it another shot with the info from the link but I'm going to make a complete backup before trying.

rkakrik, 2021-06-23:
Same problem

mhouston100, 2021-08-10:
After spending at least an hour or more running through the various threads regarding this 4-Gang switch, specifically the TS004F. I thought I would just put a concise rundown on how to get this working and the outcomes / limitations.

**Firstly, the limitations:**

This will not be configured as a 12 button switch. The only actions that are available once correctly configured are:

- First Button Pressed
- Second Button Pressed
- Third Button Pressed
- Fourth Button Pressed
- Third Button Continuosly Pressed
- Fourth Button Continuosly Pressed
- Fourth Button Released After Long Pressed (I dont see this mentioned elsewhere but it shows as an event on mine.

**Configuration**

Before proceeding, delete any instances of this device in your ZHA integration and ensure that your gateway is on the latest firmware version.

1.  Create a folder on your HA instance **config** folder called **custom_zha_quirks** for instance

    `\\homeassistant\config\custom_zha_quirks`

2. Create a file in this new folder called **ts004f.py** and copy the latest content of the file:

    [TS004F custom quirk](https://gist.github.com/koying/7ffb29e2db56649b8cb7fd5702a2ff46#file-ts004f-py)

3. In your **config\configuration.yaml** add the following lines:

    `zha:`
        `custom_quirks_path: /config/custom_zha_quirks/`

4. Restart your HA instance

5. In the ZHA integration, add a new device, this will start the searching process

6. On the device, hold the BOTTOM LEFT button for 10+ seconds, until all four lights begin flashing

7. Wait for several seconds until ZHA descovers the device

**Outcome**

One point of confusion seems to be that no entities show up. This is expected. This device will add EVENTS not entities.

So you may not see any change from previous in the ZHA discovered device.

However, when you are creating automations etc, you will see the new events available:

[image]

As far as I know, at this point, this is the end result that we can expect.

pmiguelcruz, 2021-08-10:
Thanks for that @mhouston100 !

PR #969 (2021-08-25): Adding support for tuya TS004F dimmer switches
Adding TS004F that is one no color controller or perhaps one no color scene controller and **not** one HA scene controller that the other TS004X is.

Fixes #914  
Fixes #897 
Fixes #917 
Plus [HA Community thread TS004X](https://community.home-assistant.io/t/help-adding-a-variation-of-tuya-ts0044-ts004f-to-zha/310579).

@koying have doing most coding in blind but the users have problem getting events from the on/off cluster that is little strange (i think its was problem in the local installation but its not verified) but the events from light lev [... cut]
closed 2021-08-25 by dmulcahey