zigpy/zha-device-handlers #900: [Device Support Request] Tuya Smart Air Box TS0601 _TZE200_8ygsuhe1

Lirrec, 2021-05-28:
This seems to be a relatively new device, containing temperature, relative humidity, CO2, VOC and formaldehyde sensors.

I will spend some time today trying to implement a pull request, but I'm not experienced with this, so might not work out.

**Additional context**
This seems to be already actively worked on in some other zigbee integrations, so cluster and attribute ids + conversions are already figured out:
zigbee-herdsman-converter: https://github.com/Koenkk/zigbee-herdsman-converters/pull/2612
zigbee2mqtt: https://github.com/Koenkk/zigbee2mqtt/discussions/7299

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0x0000](ConBee II): started initialization
[0x0000:ZDO](ConBee II): 'async_initialize' stage succeeded
[0x0000](ConBee II): power source: Mains
[0x0000](ConBee II): completed initialization
Device 0x43f7 (60:a4:23:ff:fe:d2:0f:e8) joined the network
[0x43f7:zdo] ZDO request ZDOCmd.Device_annce: [0x43F7, 60:a4:23:ff:fe:d2:0f:e8, 142]
[0x43f7] Requesting 'Node Descriptor'
Tries remaining: 2
[0x43f7] Extending timeout for 0x35 request
[0x43f7] Node Descriptor: NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)
[0x43f7] Discovering endpoints
Tries remaining: 3
[0x43f7] Discovered endpoints: [1]
[0x43f7:1] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device 60:a4:23:ff:fe:d2:0f:e8 endpoint 1 for cluster: 61184
...
[0x9CCF:1:0x0b04]: async_update
Received frame on uninitialized device 60:a4:23:ff:fe:d2:0f:e8 endpoint 1 for cluster: 61184
...
Tries remaining: 2
[0x43f7:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10])
Unknown cluster 61184
[0x43f7:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=61 command_id=Command.Read_Attributes_rsp>
[0x43f7:1] Manufacturer: _TZE200_8ygsuhe1
[0x43f7:1] Model: TS0601
Checking quirks for _TZE200_8ygsuhe1 TS0601 (60:a4:23:ff:fe:d2:0f:e8)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
device - 0x43F7:60:a4:23:ff:fe:d2:0f:e8 entering async_device_initialized - is_new_join: True
device - 0x43F7:60:a4:23:ff:fe:d2:0f:e8 has joined the ZHA zigbee network
[0x43F7](TS0601): started configuration
[0x43F7:ZDO](TS0601): 'async_configure' stage succeeded
[0x43F7:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
[0x43F7:1:0x0000]: finished channel configuration
[0x43F7:1:0x0019]: bound 'ota' cluster: Status.SUCCESS
[0x43F7:1:0x0019]: finished channel configuration
[0x43F7:1:0x0000]: 'async_configure' stage succeeded
[0x43F7:1:0x0019]: 'async_configure' stage succeeded
[0x43F7](TS0601): completed configuration
[0x43F7](TS0601): stored in registry: ZhaDeviceEntry(name='_TZE200_8ygsuhe1 TS0601', ieee='60:a4:23:ff:fe:d2:0f:e8', last_seen=1622197068.2178242)
[0x43F7](TS0601): started initialization
[0x43F7:ZDO](TS0601): 'async_initialize' stage succeeded
[0x43F7:1:0x0000]: initializing channel: from_cache: False
[0x43F7:1:0x0000]: finished channel configuration
[0x43F7:1:0x0019]: initializing channel: from_cache: False
[0x43F7:1:0x0019]: finished channel configuration
[0x43F7:1:0x0000]: 'async_initialize' stage succeeded
[0x43F7:1:0x0019]: 'async_initialize' stage succeeded
[0x43F7](TS0601): power source: Mains
[0x43F7](TS0601): completed initialization

[0x43f7:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=2 command_id=1>
[0x43f7:1:0xef00] Unknown cluster-specific command 1
[0x43f7:1:0xef00] ZCL request 0x0001: b'\x00\x06\x02\x02\x00\x04\x00\x00\x00\x00'
[0x43f7:1:0xef00] No handler for cluster command 1
... repeats several times per second

[0x9ccf:21:0x000c] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=250 command_id=Command.Report_Attributes>
[0x9ccf:21:0x000c] ZCL request 0x000a: [[Attribute(attrid=85, value=<TypeValue type=Single, value=156.73599243164062>)]]
[0x9ccf:21:0x000c] Attribute report received: present_value=156.73599243164062
[0x43f7:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=13 command_id=1>
[0x43f7:1:0xef00] Unknown cluster-specific command 1
[0x43f7:1:0xef00] ZCL request 0x0001: b'\x00\x06\x16\x02\x00\x04\x00\x00\x00\x00'
[0x43f7:1:0xef00] No handler for cluster command 1
...
[0x43f7:1:0xef00] Unknown cluster-specific command 36
[0x43f7:1:0xef00] ZCL request 0x0024: b'\x00\x05'
[0x43f7:1:0xef00] No handler for cluster command 36
[0x43f7:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=17 command_id=1>
[0x43f7:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=19 command_id=Command.Report_Attributes>
[0x43f7:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=1, value=<TypeValue type=uint8_t, value=68>), Attribute(attrid=65506, value=<TypeValue type=uint8_t, value=31>), Attribute(attrid=65508, value=<TypeValue type=uint8_t, value=1>)]]
[0x43f7:1:0x0000] Attribute report received: app_version=68, 65506=31, 65508=1
[0x0b45:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=28 command_id=Command.Report_Attributes>
[0x0b45:1:0x0405] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=5923>)]]
[0x0b45:1:0x0405] Attribute report received: measured_value=5923
```

Lirrec, 2021-05-28:
I have started on a custom integration, file attached.
This can currently receive the attributes, temperature and humidity will show up in home-assistant.
This is using new clusters only supported in zigpy 0.33.0:
The specific concentration measurement clusters used are:
CarbonDioxideConcentration, FormaldehydeConcentration, EthyleneConcentration
**Warning/TODO:** VOC is currently exposed as EthyleneConcentration, which is plain wrong. AnalogValue could be a workaround, I'm not familiar enough with zha on how this should be handled.

Also, ZHA would probably need an additional patch to handle the various Concentration* Clusters just like temperature and humidity for them to show up in the UI.
[ts0601_smart_air_box.txt](https://github.com/zigpy/zha-device-handlers/files/6561878/ts0601_smart_air_box.txt)

dougmaitelli, 2021-09-25:
I see this got merged already. Any idea if this is already part of the 2021.9 Home-Assistant core?

MattWestb, 2021-09-25:
If looking in the commits is the line https://github.com/zigpy/zha-device-handlers/blob/1c2b7ece00f494d16e38bb41093a525b1ed64f42/zhaquirks/tuya/air/ts0601_air_quality.py#L35 was added on Jun 27, 2021 so it shall being the the HA 2021.7 and all later HA.
If ZHA is loading the quirk you is see it on the device card one line with the name of the loaded quirk.

dougmaitelli, 2021-09-25:
> If looking in the commits is the line
>
> https://github.com/zigpy/zha-device-handlers/blob/1c2b7ece00f494d16e38bb41093a525b1ed64f42/zhaquirks/tuya/air/ts0601_air_quality.py#L35
>
> was added on Jun 27, [...]

I think the problem is that my device is a different hardware revision with different ID and that is why it is not working.
If I provide the necessary information would someone be able to help add that to the code too?

Thanks,

Nixellion, 2021-12-02:
For me it's only showing VOC value, only one sensors, though if I click on "Reconfigure" it shows as reporting other attributed. What gives?

MattWestb, 2021-12-02:
Can you posting the device signature from the device card ?

Nixellion, 2021-12-02:
@MattWestb Sorry to bother, I've edited my commend just as you replied. I updated Hass from 11.3 to 11.5 and either a restart or an update fixed it, it's now showing 4 or 5 sensors as it should.

MattWestb, 2021-12-02:
Great its working for you.
It being 3 different devices IDs added and i was hoping it was one of them and its look that then its working :-))

Duehok, 2022-01-05:
Does not work: the device is paired, but has no entities.

device signature:

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_yvx5lh6k",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

MattWestb, 2022-01-05:
@Duehok Your device is new and need being added in the quirk at:
https://github.com/zigpy/zha-device-handlers/blob/d65fd8d4b359b52c689e28542810d8bca678190c/zhaquirks/tuya/air/ts0601_air_quality.py#L83-L85

Duehok, 2022-01-05:
Is that guide still up to date?
[link to guide in readme.md](https://github.com/zigpy/zha-device-handlers#testing-quirks-in-development-in-docker-based-install)

MattWestb, 2022-01-05:
Partly but its easyest using local quirks https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274 and no need hacking in the HA container.

cloudbr34k84, 2022-03-16:
Hi all i just got this and i think its a new device. How can i get this to work?

`{`
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0402",
        "0x0405",
        "0x040d",
        "0x042b",
        "0x042e",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_dwcarsat",
  "model": "TS0601",
  "class": "zhaquirks.tuya.air.ts0601_air_quality.TuyaCO2SensorGPP"
}
``

closed 2022-09-19 by github-actions[bot]