zigpy/zha-device-handlers #901: [Device Support Request] Develco Keypad KEPZB-110

EverythingSmartHome, 2021-05-28:
Hello, I was hoping to get support added for a Develco Keypad KEPZB-110. This keypad also features RFID support, proximity and light sensor as well as tamper switch. As far as I know it is standard Zigbee 3.0 (although I could be wrong).

So far in Home Assistant 2021.5 I can add it, and it indeed shows up as an alarm, although ias_zone doesn't work, either does battery sensor or on_off switch (not sure what that is?):

[image]

The alarm component works and the state etc works properly, you can set the alarm via the keypad or via HA and all functions correctly, except I have not found a way to change the default code from 1234.

**Device signature:**
`{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4117, maximum_buffer_size=82, maximum_incoming_transfer_size=1500, server_mask=11264, maximum_outgoing_transfer_size=1500, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 49353,
      "device_type": "0x0001",
      "in_clusters": [
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    },
    "44": {
      "profile_id": 260,
      "device_type": "0x0401",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0500"
      ],
      "out_clusters": [
        "0x0003",
        "0x000a",
        "0x0019",
        "0x0501"
      ]
    }
  },
  "manufacturer": "Develco Products A/S",
  "model": "KEPZB-110",
  "class": "zigpy.device.Device"
}`

**Additional context**
I also captured some events from zha_events listener of arming and disarming in case that was helpful:
`{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:15:bc:00:43:00:05:52",
        "unique_id": "00:15:bc:00:43:00:05:52:44:0x0501",
        "device_id": "75f001581c1c89353997bdf30c9c956e",
        "endpoint_id": 44,
        "cluster_id": 1281,
        "command": "arm",
        "args": {
            "arm_mode": 0,
            "arm_mode_description": "Disarm",
            "code": "1234",
            "zone_id": 0
        }
    },
    "origin": "LOCAL",
    "time_fired": "2021-05-28T16:15:46.496075+00:00",
    "context": {
        "id": "3d4100541614489dc4faa9a259eb1b40",
        "parent_id": null,
        "user_id": null
    }
}

{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:15:bc:00:43:00:05:52",
        "unique_id": "00:15:bc:00:43:00:05:52:44:0x0501",
        "device_id": "75f001581c1c89353997bdf30c9c956e",
        "endpoint_id": 44,
        "cluster_id": 1281,
        "command": "arm",
        "args": {
            "arm_mode": 3,
            "arm_mode_description": "Arm_All_Zones",
            "code": "1905",
            "zone_id": 0
        }
    },
    "origin": "LOCAL",
    "time_fired": "2021-05-28T16:15:32.102086+00:00",
    "context": {
        "id": "27e8fba46f028ee0c28bdb50fa07e26a",
        "parent_id": null,
        "user_id": null
    }
}
`

If I can provide anything else please let me know. There is also a technical [manual here ](https://www.develcoproducts.com/media/2032/kepzb-110-technical-manual-keypad.pdf)which has the Zigbee commands and cluster info.

EverythingSmartHome, 2021-05-28:
Just to add, I figured out how to change the code, didn't realise/notice there was a new option in the ZHA configuration menu to change it since the last release, good job!

Do you think it's possible to get the other stuff working?

SanderLam, 2021-12-01:
Hi there, I have the same device and after a night of breaking my head trying to find the basic instructions, I am in the same state as you are.

I can see the events coming in, can lock and unlock the alarm, but all the other stuff is not working.

I might be willing to make the custom code for this device myself, but I've never done that and wouldn't know where to start testing etc.

closed 2022-06-06 by github-actions[bot]

Jessendelft, 2022-09-29:
@SanderLam : On the Integrations page (Devices & Services), locate the ZHA integration and press 'Configure'. There is an option here called 'Alarm Control Panel Options'. I assume this is the place @EverythingSmartHome meant in his reply.
I have the same Keypad in my cabin, and will try it out the next time I'm there.