zigpy/zha-device-handlers #905: [Device Support Request] RGBgenie ZB-5028 remote

ehn, 2021-05-30:
I'm trying to pair a [RGBgenie ZB-5028](https://rgbgenie.com/?product=rgbgenie-rgbw-remote-zigbee) remote with ZHA. (This device was previously known – and is sometimes still referred to – as ZB-5008.)

It times out and fails after a while. This is the log:

```plain text
Device 0xcbd8 (00:0d:6f:xx:xx:xx:xx:xx) joined the network
Device 00:0d:6f:xx:xx:xx:xx:xx changed id (0x98fb => 0xcbd8)
[0xcbd8:zdo] ZDO request ZDOCmd.Device_annce: [0xCBD8, 00:0d:6f:xx:xx:xx:xx:xx, 128]
[0xcbd8] Requesting 'Node Descriptor'
Tries remaining: 2
[0xcbd8] Extending timeout for 0xc1 request
[0xcbd8] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4644, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)
[0xcbd8] Discovering endpoints
Tries remaining: 3
[0xcbd8] Extending timeout for 0xc2 request
Received frame on uninitialized device 00:0d:6f:xx:xx:xx:xx:xx (Status.NEW) for endpoint: 1
[0xcbd8] Received ZCL while uninitialized on endpoint id 1, cluster 0x0001 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=22 command_id=Command.Report_Attributes>, payload: b'!\x00 d'
[0xcbd8] Uninitialized device command 'Command.Report_Attributes' args: [[Attribute(attrid=33, value=<TypeValue type=uint8_t, value=100>)]]
Tries remaining: 2
[0xcbd8] Extending timeout for 0xc3 request
Tries remaining: 1
[0xcbd8] Extending timeout for 0xc4 request
[0xcbd8] Failed to discover active endpoints
```

For reference, the remote is [supported by Zigbee2MQTT](https://www.zigbee2mqtt.io/devices/ZB-5028.html) and there is [an issue filed for support in deConz](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2916).

ehn, 2021-11-28:
AFAIK the issue remains.

MattWestb, 2021-11-28:
Sorry its not helping but the device dont like replay to request from the coordinator so its not getting the information from the device how its working (endpoints and cluster).

I have seen one user having the same problem with one motion sensor that was only sending attribute reports but not replaying on requests.

You can trying pressing one key on the remote every 2 seconds then its trying paring with your system and see if its start replaying to the commend sent to it.

What coordinator hardware is you running your Zigbee network ?

ehn, 2021-11-29:
That remote is currently on a different continent, but I'll give your suggestion a try as soon as I can. Thanks!

closed 2022-06-04 by github-actions[bot]