zigpy/zha-device-handlers #909: [BUG] zhaquirks.xiaomi.aqara.light_aqcn2 no color_temp, reconfiguration fails

FrancYescO, 2021-06-03:
it's not possible to set color_temp attribute to these device (i have 2 of these that where working on Z2M), reconfiguration fails

[image]

```
{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=0, maximum_outgoing_transfer_size=100, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.Router: 1>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0102",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019",
        "0x0406"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.light.aqcn02",
  "class": "zhaquirks.xiaomi.aqara.light_aqcn2.LightAqcn02"
}
```

FrancYescO, 2021-06-03:
don't ask me how, but actually reconfiguration still fail, but color temp slider showed up in HA.

closed 2021-06-03 by FrancYescO

MattWestb, 2021-06-03:
Thats failing is the setting up attribute reporting.
Its not the same as setting the CT in ZHA its only that the device is not supporting attribute reporting of it.
Philips HUE device is not using attribute reporting and ZHA need pulling the state of the device with reading the attribute from the device (pulling) that is not Zigbee standard.