zigpy/zha-device-handlers #913: [Device Support Request] _TZE200_g1ib5ldv TS0601 (2 gang Moes no neutral wired switch) ZTS-EUB

khrazik, 2021-06-08:
**Is your feature request related to a problem? Please describe.**
Device pairs successfully, but no entities are created. I tried to make a custom quirk based off similar tuya switches, but I failed. Interestingly, device seems to report only one endpoint but there are two gang switches (and physically, both work).

**Describe the solution you'd like**
I would like a quirk that handles this device and exposes control over both gangs via ZHA.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

`2021-06-08 23:04:37 INFO (MainThread) [zigpy.application] Device 0x63cc (60:a4:23:ff:fe:d5:d4:4a) joined the network
2021-06-08 23:04:37 DEBUG (MainThread) [zigpy.application] Skip initialization for existing device 60:a4:23:ff:fe:d5:d4:4a
2021-06-08 23:04:37 DEBUG (MainThread) [zigpy.zdo] [0x63cc:zdo] ZDO request ZDOCmd.Device_annce: [0x63CC, 60:a4:23:ff:fe:d5:d4:4a, 128]
2021-06-08 23:04:39 DEBUG (MainThread) [zigpy.util] Tries remaining: 1
2021-06-08 23:04:39 DEBUG (MainThread) [zigpy.device] [0x63cc] Extending timeout for 0x47 request
2021-06-08 23:04:40 DEBUG (MainThread) [zigpy.application] Received frame on uninitialized device 60:a4:23:ff:fe:d5:d4:4a (Status.NEW) for endpoint: 1
2021-06-08 23:04:40 DEBUG (MainThread) [zigpy.device] [0x63cc] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=0 command_id=Command.Read_Attributes>, payload: b'\x07\x00'
2021-06-08 23:04:40 DEBUG (MainThread) [zigpy.device] [0x63cc] Uninitialized device command 'Command.Read_Attributes' args: [[7]]
2021-06-08 23:04:41 INFO (MainThread) [zigpy.device] [0x63cc] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)
2021-06-08 23:04:41 INFO (MainThread) [zigpy.device] [0x63cc] Discovering endpoints
2021-06-08 23:04:41 DEBUG (MainThread) [zigpy.util] Tries remaining: 3
2021-06-08 23:04:41 DEBUG (MainThread) [zigpy.device] [0x63cc] Extending timeout for 0x48 request
2021-06-08 23:04:41 INFO (MainThread) [zigpy.device] [0x63cc] Discovered endpoints: [1]
2021-06-08 23:04:41 INFO (MainThread) [zigpy.endpoint] [0x63cc:1] Discovering endpoint information
2021-06-08 23:04:41 DEBUG (MainThread) [zigpy.device] [0x63cc] Extending timeout for 0x49 request
2021-06-08 23:04:41 DEBUG (MainThread) [zigpy.application] Received frame on uninitialized device 60:a4:23:ff:fe:d5:d4:4a endpoint 1 for cluster: 10
2021-06-08 23:04:41 INFO (MainThread) [zigpy.endpoint] [0x63cc:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10])
`
**Additional context**
Device signature from the Device Info screen:
`{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_g1ib5ldv",
  "model": "TS0601",
  "class": "zigpy.device.Device"`

mrbob1234, 2021-06-15:
I need some help with this too! 
I think these may become popular considering they're both neutral and no neutral and zigbee!

mylesewen, 2021-10-06:
I've got the 3 gang ZTS-EUB.  It is detected as a single gang switch within the light domain rather than switch domain.  I've overridden this in configuration.yaml (see 'MODIFYING THE DEVICE TYPE' section of https://www.home-assistant.io/integrations/zha/) so that it now appears as a switch device, but it still only works as a one gang switch.

PR #1166 (2021-11-23): Add quirks for Tuya multigang switches
This PR add quirks for Tuya multigang switches.

It should correct the management of the On/Off cluster for the multiswitch devices that make use of the Tuya cluster (`0xEF00`) for command/status reporting.

The `MODELS_INFO` of those devices that have been reported or an issue has been reported have been added.
For those devices whose number of gangs is not clear, they have been defined as 1-gang devices until someone report correct info.

This could fix the following issues (only on/off switches):
fixes: #813
fixes: #824
fixes: #913
fixes: #972
fixes: # [... cut]
closed 2021-11-23 by dmulcahey