zigpy/zha-device-handlers #914: TS004F by _TZ3000_xabckq1v

wstrater, 2021-06-08:
I tried adding a new device, `TS004F by _TZ3000_xabckq1v`. It is a 4 button scene controller using the Zigbee network. I have a Sonoff Zigbee Bridge flashed with Tasmota zbbridge 9.4.0. I am able to add the device but it it came in with two entities, `sensor.tz3000_xabckq1v_ts004f_power` and `switch.tz3000_xabckq1v_ts004f_on_off`. Other have commented that I should only see the power entity.

When I try to add automations, I don't see the all 4 buttons and multiple press events for each button. I only see `on_off turned off`, `on_off turned on` and `power battery level changes`.

I would like to create automations for all 12 events, short, double and long. I believe this is similar to the [Model ESW-4ZAA-CN manufactured by Earda](https://zigbee.blakadder.com/Earda_ESW-4ZAA-CN.html) on BlakAdder website. (I am aware of ZHA not being listed but I still tried and thus I am requesting.)

I have attached the pairing logs, device signature and home-assistant logs.
[TS004F_device_signature.txt](https://github.com/zigpy/zha-device-handlers/files/6619778/TS004F_device_signature.txt)
[TS004F_home-assistant.log.gz](https://github.com/zigpy/zha-device-handlers/files/6619779/TS004F_home-assistant.log.gz)
[TS004F_zigbee_add_device.log](https://github.com/zigpy/zha-device-handlers/files/6619781/TS004F_zigbee_add_device.log)
[image]

Thanks for all the work!
Wes.

jph1ll1ps, 2021-06-15:
i have the same 4 button scene switch.... and it seems that this manufacture tags for the device are not recognised in the quirks DB - i have the same switch from a different manufacturer and it works... and shows up with the 4 buttons and indeed can be detected in zha_event and that means i can recognise a button pressed. then automation or node-red works... so we need the zha database for quirks i believe to recognise this manufacturer.

wstrater, 2021-06-15:
I think there are a couple of versions, TS004F and TS0044. They appear to be the same externally. Don't know what the difference is on the inside.
I have read some people had success by trying to add it over and over until you only see the battery. I tried that many times and never had success.

Wes.

Draxter, 2021-06-18:
I think one of these, the 4-button variant: [Aliexpress link](https://www.aliexpress.com/item/1005002326756203.html?spm=a2g0s.9042311.0.0.50414c4dWdwMPH)
My variant is the [TS004F](https://community.hubitat.com/uploads/default/original/3X/4/4/44b60eb716b45f9c74f52c5d3a5a1c5251ad0d43.jpeg). And the zigbee signature can be found here: https://pastebin.com/4PcFLUb2

I am using mine with the ZHA integration in Home Assistant. From there I have extracted the zha_event data for buttons 2 and 4 (buttons 1 and 3 do not produce any zha_event data but switch an entity switch in home assistant). Here is the data: https://pastebin.com/sh6VYUqj

wstrater, 2021-06-18:
You got a lot farther than I have ever been able to. I have a similar device signature to you but not able to get Home Assistant to respond to any buttons.

I tried listening to `zha_event` in Home Assistant Developer Tools and NodeRED and don't see any events for the device. I do see events for my Aqara Opple so I know the listener is working.

I only have two states for the registered device, `sensor.tz3000_xabckq1v_ts004f_power` and `switch.tz3000_xabckq1v_ts004f_55add5fe_on_off`. The power is 100 and I have not been able to get the on_off state to change using any of the buttons.

How did you get it to work? Did it register automatically or did you have to do manual configuration?

I deleted it again today and re-added it but no luck.

Thanks, Wes.

Draxter, 2021-06-18:
@wstrater I suppose it may be because of different zigbee coordinators. I'm using the Dresden Electronik ConBee II. by default buttons 1 and 3 show up in HA as an on/off switch and buttons 2/4 I can automate with a zha_event event trigger.

wstrater, 2021-06-18:
@Draxter I went with the Sonoff Zigbee Bridge running Tasmota because I am planning on running my home automation stack in the basement and wanted a coordinator in the main part of the house near the devices. I have since added a couple of repeaters behind night lights. Also the back of my Pi is pretty crowded with keyboard, mouse, monitor and USB 3.0 bridge for running off of a SSD.

jph1ll1ps, 2021-06-21:
so this is the 2nd of this switch type i bought...the first i bought is working - and seems the only difference (reason it does not work) is the manufacturer ID is not recognised in the QUIRK?

here is the configuration of the one that works...
[image]
and i have this integrated with Node-red code using the zha_event - but of course it needs to be recognized first...

this is the one that does not work - because its not recognised ..... the 4 switches do not show up:
[image]

klona, 2021-08-11:
Hi, Same here
[image]

MattWestb, 2021-08-11:
@klona Your device dont have any quirk loaded.
I have making on PR for adding it but it not merged yet but  you can installing as local quirk and getting it working.
#969

PR #969 (2021-08-25): Adding support for tuya TS004F dimmer switches
Adding TS004F that is one no color controller or perhaps one no color scene controller and **not** one HA scene controller that the other TS004X is.

Fixes #914  
Fixes #897 
Fixes #917 
Plus [HA Community thread TS004X](https://community.home-assistant.io/t/help-adding-a-variation-of-tuya-ts0044-ts004f-to-zha/310579).

@koying have doing most coding in blind but the users have problem getting events from the on/off cluster that is little strange (i think its was problem in the local installation but its not verified) but the events from light lev [... cut]
closed 2021-08-25 by dmulcahey