zigpy/zha-device-handlers #917: [Device Support Request] Manufacturer: _TZ3000_xabckq1v Model: TS004F

jph1ll1ps, 2021-06-10:
This is another manufacturer for the Tuya 4 button scene switch.

**Describe the solution you'd like**
Need to add this manufactuer and model to the quirks db
Believe it is that same as Quirk: zhaquirks.tuya.ts0044.TuyaSmartRemote0044

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0104",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0006",
        "0x1000"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x000a",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "_TZ3000_xabckq1v",
  "model": "TS004F",
  "class": "zigpy.device.Device"
}
**Additional context**
Have the same switch - different manufactuer uses this signature
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0006",
        "0x000a"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    },
    "4": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_vp6clf9d",
  "model": "TS0044",
  "class": "zhaquirks.tuya.ts0044.TuyaSmartRemote0044"
}

Draxter, 2021-06-18:
I've added some extra info about this device in #914

MattWestb, 2021-07-13:
@koying from https://community.home-assistant.io/t/help-adding-a-variation-of-tuya-ts0044-ts004f-to-zha/310579/40 its as you  have saying one dimmer switch and not one scene  switch.

Its better not using `MODELS_INFO: [("_TZ3000_xabckq1v", "TS004F")],` for the TS004X switches then they looks being the same and with 2 variants of cluster configs (perhaps not the 4F) so its better / easier using `MODEL: "TS004F",`  then the quirk is being loaded of all this devices and dont need the model info to match only the clusters.

I dont have the switch but like getting one bunch of it (not the scene one) and its looks very clean Zigbee with binding and reopening and only one end point.

Perhaps need disabling some cluster that is not used but its one later story.

Is it some problems with double events (commands) being sent like the other TS004X switches or is it being normal ?

MattWestb, 2021-07-13:
[0xbb77:1:0x0006] ZCL request 0x0000: []
[0xbb77:1:0x0006] No handler for cluster command 0 
From ZCL CLuster 0x0006 and command 0x0000 is `Off`

[0xbb77:1:0x0006] ZCL request 0x0001: []
[0xbb77:1:0x0006] No handler for cluster command 
From ZCL CLuster 0x0006 and command 0x0001 is `On`

So must triggering the automation for button 1 and 2 on the `On` and `Off` command.

For the moment is no more sent from button 1 and 2 but i like cooking one dual dimmer firmware for it so is having like one IKEA 2 button remote on the left side and one the right side buttons (on - dimmer up / off - dimmer down x 2) or perhaps on 4 dimmer firmware (4 x toggle on/off and dimming up / down).

MattWestb, 2021-07-13:
From the not released IKEA 4 button remote:

`(SHORT_PRESS, TURN_ON): {COMMAND: COMMAND_ON, CLUSTER_ID: 6, ENDPOINT_ID: 1},`
`(SHORT_PRESS, TURN_OFF): {COMMAND: COMMAND_OFF, CLUSTER_ID: 6, ENDPOINT_ID: 1},`

You is the code worrier im only hardware nerd ;-))

MattWestb, 2021-07-13:
The open/close is having logic for doing stop commands for both hold up and down if you like implanting the long release for both  button 3 and 4.
https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/ikea/opencloseremote.py

MattWestb, 2021-07-13:
The Z2M device  handle:
```
    {
        fingerprint: [{modelID: 'TS004F', manufacturerName: '_TZ3000_xabckq1v'}],
        model: 'TS004F',
        vendor: 'TuYa',
        description: 'Wireless switch with 4 buttons',
        exposes: [e.battery(), e.action(
            ['on', 'off', 'brightness_move_up', 'brightness_step_up', 'brightness_step_down', 'brightness_move_down', 'brightness_stop'])],
        fromZigbee: [fz.battery, fz.command_on, fz.command_off, fz.command_step, fz.command_move, fz.command_stop],
        toZigbee: [],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            try {
                await reporting.batteryPercentageRemaining(endpoint);
            } catch (error) {/* Fails for some*/}
        },
    },
```
They is using the normal light commands for getting the events from the device = On/Off, Level up and down.

koying, 2021-07-14:
Thanks. I don't have the device, either, just trying to help ;)
Thanks for the tip regarding the on/off cluster. Will ask to try that on the thread.

jph1ll1ps, 2021-07-15:
Hi all

SO this is just another China vendor making the same switch - but like i said earlier i have the same switch from different vendor - working 100% - just that its model is recognised... and its exposed to HA for automation and i can monitor in Node-Red too.

Now.. what to do with 4 of these new switches that don't get recognised....
"manufacturer": "_TZ3000_xabckq1v",
"model": "TS004F",
"class": "zigpy.device.Device"

but the one that works:
"manufacturer": "_TZ3000_vp6clf9d",
"model": "TS0044",
"class": "zhaquirks.tuya.ts0044.TuyaSmartRemote0044"

and they are identical devices.. 4 scenes, single, double or long-press for each of the 4 buttons.

Seems either have to change the quirk to support the new manufacturer "_TZ3000_xabckq1v" then i would expect this works? Is there something i can do locally .. don't assume i can change the manufacturer attribute - its in the device... 

this post https://github.com/zigpy/zha-device-handlers/issues/914#issuecomment-864746403 i have shared how the two devices show-up in the dashboard.... and clearly one works - the other does not.

can we get the quirk updated? is that the right thing? is there something i can do to make this work?

MattWestb, 2021-07-15:
@jph1ll1ps The TS0044 is one 4 button scene controller with 12 scene = is sending "clicks" that ZHA is triggering on.
The TS004F is one Dimmer switch that is sending On, Off, Dimm up and down = its working out of  the box in ZHA for searing lights if binding is working OK and its only possible getting HA triggering then the light commands so  On, Off, Step up, Step down, Move Up, Move down and Stop move (perhaps up and down).
If the seller have saying its one 12 scene switch  hi have sheeting you and you shall demanding getting one redound or the right product (TS0044).

@koying Wot i can see is ZHA  is receiving the on and off commands OK:
```
2021-07-14 09:55:56 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5247:1:0x0006]: received 'on' command with [] args on cluster_id '6' tsn '49'
2021-07-14 09:56:41 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5247:1:0x0006]: received 'off' command with [] args on cluster_id '6' tsn '50'
```
If i comparing with my IKEA E2002 (4 button remote) is receiving the same and is making zha_event:
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "84:2e:14:ff:fe:5f:89:0e",
        "unique_id": "84:2e:14:ff:fe:5f:89:0e:1:0x0006",
        "device_id": "14c00f9f7f350057a29c75268ff7ad88",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "on",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2021-07-15T08:13:33.009592+00:00",
    "context": {
        "id": "6e8741046b94fdab1e28ec1b5f6aacf6",
        "parent_id": null,
        "user_id": null
    }
}

{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "84:2e:14:ff:fe:5f:89:0e",
        "unique_id": "84:2e:14:ff:fe:5f:89:0e:1:0x0006",
        "device_id": "14c00f9f7f350057a29c75268ff7ad88",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "off",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2021-07-15T08:14:08.721114+00:00",
    "context": {
        "id": "9ae9067cae3ceeb47953862e591e875f",
        "parent_id": null,
        "user_id": null
    }
}
```
The question is have the user deleting the `__pycache__` or  is that making the spooky thing ??
Im very confused of that not getting and events from the OnOff cluster :-(((

The IKEA 4 button quirk (with not working right and left buttons):
```
"""Device handler for IKEA of Sweden TRADFRI remote control."""
from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    Identify,
    LevelControl,
    OnOff,
    Ota,
    PollControl,
    PowerConfiguration,
    Scenes,
)
from zigpy.zcl.clusters.lightlink import LightLink
[... 113 lines cut ...]
            COMMAND: COMMAND_PRESS,
            CLUSTER_ID: 5,
            ENDPOINT_ID: 1,
            ARGS: [257, 13, 0],
        },
        (LONG_PRESS, LEFT): {
            COMMAND: COMMAND_HOLD,
            CLUSTER_ID: 5,
            ENDPOINT_ID: 1,
            ARGS: [3329, 0],
        },
        (SHORT_PRESS, RIGHT): {
            COMMAND: COMMAND_PRESS,
            CLUSTER_ID: 5,
            ENDPOINT_ID: 1,
            ARGS: [256, 13, 0],
        },
        (LONG_PRESS, RIGHT): {
            COMMAND: COMMAND_HOLD,
            CLUSTER_ID: 5,
            ENDPOINT_ID: 1,
            ARGS: [3328, 0],
        },
    }
```

MattWestb, 2021-07-15:
I was deleting the quirk for the IKEA 4 button remote and ZHA is still getting  the on and off events so perhaps is the quirk "stealing" the events and zha is not firing the event then  its not OK in the new quirk.

Im being  very confused :-((

jph1ll1ps, 2021-07-15:
Thanks @MattWestb  - so my TS0044 works great - and i can capture the ZHA events - but the TS004f is not showing any events in the ZHA_Events tracking.. and hence i can't trigger any activities.

To me, both look identical in packaging and layout etc... so i expected that do that same things. 1 press, double press, long press... nothing is tracking through the ZHA events.

MattWestb, 2021-07-15:
You must waiting until **koying** have getting the TS004F quirk working then you is getting the dimmer switch events as zha_event.

Outside is the same but tuya have 2 or more hardware inside and one with very nice hardware (EFR32MG21 module) and all looks the same and can running the same firmware (44 or 4F) so if i was having both i wold flashing the firmware i like for getting the function i like (after dumping it first !!!).

MattWestb, 2021-07-15:
@koying You is using  `DEVICE_TYPE: zha.DeviceType.REMOTE_CONTROL,` but is it exposing the right cluster for the "switch" ? (its one pure HA cluster and have nothing to do with light controllers)

IKEA E2001/E2002 is using `DEVICE_TYPE: zha.DeviceType.NON_COLOR_CONTROLLER,` that is having that is supporting OnOff, and Level control and so on.

ZHA is getting the on and off commands so the problem is in the quirk is not forwarding the commands back to ZHA OK.
I cant finding the the matrix wot ZHA is using for cluster for different cluster profiles, but  i can being worth testing.

koying, 2021-07-15:
I had the same idea, but apparently both are "remotes" as far as ZHA is concerned: https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/core/registries.py#L37

From the thread, early on, someone reported that the OnOff was actually translated as an HA switch, which might explains where the events are eaten.
Why my quirk would create a switch and not yours is a bit of a mystery, though

MattWestb, 2021-07-15:
In Zigbee lights is one switch is one "relay" that turning the power on or off like one "real switch" or like the IKEA Control outlet that is one large relay inside and no way controlling locally (if not adding on controller in the same device) .
You is getting one light or one flash in HA.
 
Zignee light is using "controllers" for devices that is controlling lights like dimmer switches and scene switches is named "controller" and they is normally battery devices and is only sending commands to lights. And you is normally only getting one battery in HA and the commands is fired and events from the device and they cant receiving commands only sending.

The light spec for zigbee lights: 
[docs-15-0014-05-0plo-Lighting-OccupancyDevice-Specification-V1.0.pdf](https://github.com/zigpy/zha-device-handlers/files/6825086/docs-15-0014-05-0plo-Lighting-OccupancyDevice-Specification-V1.0.pdf)

If you using "old HA profiles" its other cluster being used and also one ZOL (Zigbee light and occupancy) and and on HA cluster can being different if the profile is ZHA or ZLO (for making all more confusing).

koying, 2021-07-15:
@jph1ll1ps  As the member on the thread who was testing the quirk gave up, could you test if the triggers are working,?

@MattWestb What you said goes over my head, but it's confirmed the OnOff translates as a switch in HA.
Is there a way to prevent this and use events instead?
Do you think changing the device type would do it?

MattWestb, 2021-07-15:
The `DEVICE_TYPE` looks being OK then i was putting yours it the E2002 quirks and its still making zha_events and no switches on the device card.

MattWestb, 2021-07-15:
Was looking little in the logs and its looks very identical from the input from the devices:
```
TS004F HA log:
2021-07-14 09:56:41 DEBUG (MainThread) [zigpy.zcl] [0x5247:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=50 command_id=0>
2021-07-14 09:56:41 DEBUG (MainThread) [zigpy.zcl] [0x5247:1:0x0006] ZCL request 0x0000: []
2021-07-14 09:56:41 DEBUG (MainThread) [zigpy.zcl] [0x5247:1:0x0006] No handler for cluster command 0
2021-07-14 09:56:41 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5247:1:0x0006]: received 'off' command with [] args on cluster_id '6' tsn '50'

E2002 HA log:
2021-07-15 20:27:33 DEBUG (MainThread) [zigpy.zcl] [0xacda:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=76 command_id=0>
2021-07-15 20:27:33 DEBUG (MainThread) [zigpy.zcl] [0xacda:1:0x0006] ZCL request 0x0000: []
2021-07-15 20:27:33 DEBUG (MainThread) [zigpy.zcl] [0xacda:1:0x0006] No handler for cluster command 0
```
And the device automation looks the same only the "names" is different:
```
TS004F DA:
(SHORT_PRESS, BUTTON_1): {COMMAND: COMMAND_ON, CLUSTER_ID: 6, ENDPOINT_ID: 1},
(SHORT_PRESS, BUTTON_2): {COMMAND: COMMAND_OFF, CLUSTER_ID: 6, ENDPOINT_ID: 1},

E2002 DA:
(SHORT_PRESS, TURN_ON): {COMMAND: COMMAND_ON, CLUSTER_ID: 6, ENDPOINT_ID: 1},
(SHORT_PRESS, TURN_OFF): {COMMAND: COMMAND_OFF, CLUSTER_ID: 6, ENDPOINT_ID: 1},
```
And i was commenting out the on and off lines and HA was still receiving the zha_events with on and off.

I think for the moment is not the quirk that is not making the commands from cluster 0x0006 working, its somthing with the test system that is blocking it (old left quirks in the system that is not deleted or old out off sync ZHA components).

Dam that i cant getting the gray or black switch from EU to Austria !!!

MattWestb, 2021-07-17:
No zha_event is very strange then the device shall sending light commands to the coordinator if its OK paired.
HA debug logs from the user !!

MattWestb, 2021-07-18:
@koying May i doing one draft PR with your code so user can testing it easy ?
Perhaps we can getting more user testing the quirks and making it working OK.

koying, 2021-07-18:
Sure, no problem.
As a reminder, I don't have that device, so my involvement is just to help, here.

MattWestb, 2021-07-18:
I also dont have the device but like getting it for steering some tuya double mini dimmer in wall boxes (with or without external mechanical switches).
I have trying getting all the TS004X quirks being generic (only rigging on model) but all PR is not being accepted but hopefully they is going thru then its being easier for the user that is baying the devices is working out of the box also with new models from china.

You have the coding knowledge that i dont have and is great then fixing the problems the users is having with there devices.

MattWestb, 2021-07-18:
Thanks Chris !!!

PR #969 (2021-08-25): Adding support for tuya TS004F dimmer switches
Adding TS004F that is one no color controller or perhaps one no color scene controller and **not** one HA scene controller that the other TS004X is.

Fixes #914  
Fixes #897 
Fixes #917 
Plus [HA Community thread TS004X](https://community.home-assistant.io/t/help-adding-a-variation-of-tuya-ts0044-ts004f-to-zha/310579).

@koying have doing most coding in blind but the users have problem getting events from the on/off cluster that is little strange (i think its was problem in the local installation but its not verified) but the events from light lev [... cut]
closed 2021-08-25 by dmulcahey