zigpy/zha-device-handlers #918: [Device Support Request] Hama Smart Radiator Thermostat (TS0601 _TZE200_yw7cahqs)

marcis92, 2021-06-10:
**Is your feature request related to a problem? Please describe.**
Device pairs successfully, but no entities are created and the device is therefore not usable right now.

**Describe the solution you'd like**
I would like help getting this device paired in ZHA with controllable entities.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_yw7cahqs",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

marcis92, 2021-06-16:
@MattWestb Thank you very much for quick response.

It worked instantly and I could use all functions. I tested it in the past few days in different scenarios, everything works well.

closed 2021-06-16 by marcis92

MattWestb, 2021-06-16:
Great work !!

I have making on PR with the change.
You can trying downloading it (you only need the zhaquirks/tuya/thermostat_88teujp.py) and testing it but it shall being the same as you have testing or waiting util is merged.

MattWestb, 2021-06-19:
Its in the dev branch now :-))

So very likely its going in the next ZHA quirk release that hopefully is going in HA 2012.07.

DatGuyCoding, 2021-08-31:
I am not sure if anybody will read a comment in a closed ticket since I am fairly new to all this stuff. But I have the same problem as marcis92. The device is added and fully usable via the dashboard in home assisant. If I click on the device itself, no entitys beside battery status and power state is generated. Therefore I can't automate the heating in home assistant. Using a local quirk with modified parameters didn't work for me.

The zigbee device signature says the following, which is quite a bit different from marcis signature (even though its exactly the same device):

`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=0, maximum_buffer_size=82, maximum_incoming_transfer_size=255, server_mask=11264, maximum_outgoing_transfer_size=255, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0201",
        "0xef00"
      ],
      "out_clusters": [
        "0x0001",
        "0x0003",
        "0x0019",
        "0x0201"
      ]
    }
  },
  "manufacturer": "_TYST11_yw7cahqs",
  "model": "w7cahqs",
  "class": "zhaquirks.tuya.thermostat_88teujp.Thermostat_TYST11_c88teujp"
}`

I will wait a few days and if no one is responding here, I will start a new ticket. If you need further information, let me know. Thanks in advance!

MattWestb, 2021-08-31:
Your device ("_TYST11_yw7cahqs", "w7cahqs") was added in end of February #797 but i was not adding the second version that marcis92  is having so i was doing it with this linked PR.

Wot i can see have your device loaded the quirk: `"zhaquirks.tuya.thermostat_88teujp.Thermostat_TYST11_c88teujp"` so you shall not loading some local quirk for it then its over riding the committed one. 

Try deleting all files in your local quirk directory and restarting HA and see if you is getting the line that is saying the quirk is loaded.

If all is going well you is getting 2 entries on the device card one thermostat and one thermostat power (I think but not knowing then i have not the same version but my have battery % some other only battery low).

DatGuyCoding, 2021-09-01:
First of all: thank you for your quick response!
I did as you said and deleted the local quirk and renamed the folder. I reconnected the device and the signature gives the same output as before. I even checked another thermostate I added **before** I even tried to use the local quirks and it gives me the same device signature:

```{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=0, maximum_buffer_size=82, maximum_incoming_transfer_size=255, server_mask=11264, maximum_outgoing_transfer_size=255, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0201",
        "0xef00"
      ],
      "out_clusters": [
        "0x0001",
        "0x0003",
        "0x0019",
        "0x0201"
      ]
    }
  },
  "manufacturer": "_TYST11_yw7cahqs",
  "model": "w7cahqs",
  "class": "zhaquirks.tuya.thermostat_88teujp.Thermostat_TYST11_c88teujp"
}```

In my device card I get only **one entry** per thermostate **with 2 entities each**. And on/off and battery entity. Which leaves me with the ability to turn it on and off in automations but not setting a specific temperature.

I will try to re-install ZHA and add the devices again to make sure there is no remainder of the local quirks.

Is there anything else I can provide to you to find problem on this?

MattWestb, 2021-09-01:
The quirk is loaded OK but its looks like its not doing all things is shall doing.
For my Siterwell i getting this (its not using the same quirk but working more or less the same):
[image]
and with debug for tuya qiurk i is getting this one time i the hour:
```
2021-09-01 09:25:32 DEBUG (MainThread) [zhaquirks.tuya] [0x5ce2:1:0xef00] Received value [0, 0, 0, 241] for attribute 0x0203 (command 0x0001)
```
Its nicely reporting the local temperature.

I think doing one new install of HA is the best so no old crap is making bad things.

And no user that is using this quirk have reported problem but it can being  that  tuya have doing  one new version of the tuya MCU and not updated the manufacture and model IDs.

DatGuyCoding, 2021-09-01:
Okay, my Entities basicly look the same. Maybe I have a wrong understanding of what the thermostate should be able to do. Before I re-install my home assistant (I am pretty sure thats not the problem, because my thermostats were the first devices I added immediately after a new installation) could you please send me a screenshot of the available actions you can choose from when you try to setup an automation with the thermostat? For me the only option is to set the "HVAC status" from "off" to "heat". I can't adjust the temperature though.

MattWestb, 2021-09-01:
From the device card and add automation its looks like this:
[image]
And its working changing mode and temperature from the main screen by clicking on the 3 dots and i getting this:
[image]
or dragging / clicking on one new target temperature and the TRV reporting back the new set temperature set-point.

I think changing the set-point temperature you must sending the commands by cluster you have on the device in the automation.
From the device card you can finding then in "manage cluster" and its looks like this and you is getting cluster Nr and command / attribute number you need.
[image]
You can reading and setting attributes there like the set-pint temperature and all other parameters.
One not its only Moes TRV that have getting extra functions in ZHA then all other only have basic function support.

I dont have experience of sending cluster commands in HA with automatons then i using the static temperature and only changing if its too warm or cold from HA GUI but you can finding  how to do that in the HA forum.

DatGuyCoding, 2021-09-01:
Okay, as I already thought my expectations on the abilitys of the HVAC was wrong. By comparing the functions of your device and mine it looks like my HVAC is behaving correctly. I took it for granted that there would be an action to adjust the temperature with the automations without sending cluster commands.

Anyway: Thank your very much for all of your help an quick responses! I really appreciate this and it motivated me even more to dive into this topic! This seems to be a great community. Have a good one and stay healthy!

Adminiuga, 2021-09-01:
There might be no Device actions, but you always could use climate.* Service calls in your automations to control the entity

Adminiuga, 2021-09-01:
https://www.home-assistant.io/integrations/climate/

MattWestb, 2021-09-01:
Thanks Adminiuga that i have missed !!!

Its looks like this:
[image]
And then its working clicking on YAML mode and you can copy the code you was making and using it in the automation :-)))

MattWestb, 2021-09-01:
If making one new automation then the action par is looking the same so you can easy setting the target temperature and working mode with the GUI editor for automatons.

zewsan, 2021-12-20:
Hello, I'm having some issues with this one. I have 7 thermostats of this model and they will randomly change their target temperature, even if no automation has ever been created. 
The HA installation has never had any automations either.

In 1-3 days, the temp will be set down to 20, then to 16 and will stay on 16.
Any suggestions what I can look over?

Thanks!

Provox, 2021-12-23:
> 

Ensure yourself, the thermostats are in manual mode (clock led has to be off > button double click). I've experienced same behaviour on this thermostat while I have accidently turned on this mode.

cremor, 2022-02-26:
> I need to send it as 2 seperate steps though. So first on/off and then set the temperature.

I recently bought this thermostat and have the same problem. Has there been any progress to a "real" solution (so not having to send 2 commands)?

@marcis92 Tagging you, since you closed the issue. Does this work for you with a single command?

marcis92, 2022-03-04:
@cremor tbh I never used it within an automation or with seperate service calls.
I either use it manually via dashboard or with Scheduler component from HACS. That works perfectly.

Just did a quick test and I also have to turn the device on before set the temperature.

cremor, 2022-03-04:
@marcis92 Thanks for the confirmation.
@MattWestb Is this something that can be fixed in the quirk?

MattWestb, 2022-03-04:
@cremor This TRVs was added for around one year ago and has not being updated and no have interest in fixing its issues that have the knowledge to fixing it,

But one user have making one updated quirk that have more function implanted and also more function in ZHA GUI but we cant getting it merged in ZHA then we have no one that  can doing the test code. 
Feel free testing it as one local quirk in HA and using it if yu like the function that being added. 
https://github.com/zigpy/zha-device-handlers/issues/1189

And its the same as with my Siterw TRV need disabling the shuddering o the device or its doing its "own thing" or you must programming the times in the TRV local or with cluster commands then we have not (yet) implanted that in ZHA.

tomzi007, 2023-11-23:
> For getting the local quirk folder to working add this in HA config:
> 
> ```
> zha:
>   custom_quirks_path: /config/custom_zha_quirks/
> ```
> 
> Making one folder with the name `custom_zha_quirks` in you HA config directory. Putting the unzipped `thermostat_88teujp.py` in the folder and restarting HA.

Hello,
I have used this option and now it reads everything in the home assistant like a charm, e.g. when I rotate the knob I can immedatelly see it changing in HA (target temp.).  What doesn't work is when I want to change HVAC from on to off or vice versa. 
How to make it work?
Here is my device's signature:
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0301",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x000d",
        "0x0201",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_h4cgnbzg",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_trv_sas.Thermostat_TZE200_c88teujp"
}

BFipsen, 2024-10-25:
> Hello, I have used this option and now it reads everything in the home assistant like a charm, e.g. when I rotate the knob I can immedatelly see it changing in HA (target temp.). What doesn't work [...]

Did you make it work? I have the same problem now