zigpy/zha-device-handlers #921: [Device Support Request] Shenzhen HOMA HOMA001F

TPMunster, 2021-06-14:
**Is your feature request related to a problem? Please describe.**
Need support for this RGBW Zigbee Controller, as only ON/OFF currently works. It fully integrates with Philips Hue bridge and is the last device left on my Hue bridge before fully converted to Home Assistant. 

**Describe the solution you'd like**
Would like full support of both on/off, brightness, color picker. 

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
`{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=34457, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.Router: 1>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010d",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000",
        "0xffee"
      ],
      "out_clusters": [
        "0x0019",
        "0x1000",
        "0xffee"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0066",
      "in_clusters": [
        "0x0021"
      ],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "ShenZhen_Homa",
  "model": "HOMA001F",
  "class": "zigpy.device.Device"
}`

**Additional context**
Add any other context or screenshots about the feature request here.

MattWestb, 2021-08-03:
I think its one more or less basic Zigbee device but have one manufacture cluster `0xffee` that i not known wot is doing with the device.

Can you posting one copy of the "Zigbee device signature" (from the device card) and using ` ``` text ```  `so  its easier to reading ?

And also if its possible lone debug log (with `homeassistant.components.zha: debug` and `zigpy: debug`) of paring the device so can see wot color attribute the device is saying it have.

TPMunster, 2021-08-03:
Hi @MattWestb , thank you for reaching out. 
It's definitely just a simple Zigbee device, no fancy functions at all. 

Here are the requested formatted Device Signature and debug log attached:

```{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=34457, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.Router: 1>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010d",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000",
        "0xffee"
      ],
      "out_clusters": [
        "0x0019",
        "0x1000",
        "0xffee"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0066",
      "in_clusters": [
        "0x0021"
      ],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "ShenZhen_Homa",
  "model": "HOMA001F",
  "class": "zigpy.device.Device"
}

[ZHA debug.txt](https://github.com/zigpy/zha-device-handlers/files/6924276/ZHA.debug.txt)

MattWestb, 2021-08-03:
Looks very good but you can putting one ` ``` ` after the last log line for ending the block then the last link is working OK :-)

I can see that ZHA have configured attribute reporting on on/off, light level and color cluster but i think the device is not reporting any thing if not changing the devices state.
Try turning it on and off and waiting little and look wot the debug log is saying.

Debug setting in HA:
```
logger:
  default: info
  logs:
    homeassistant.core: info
    homeassistant.components.zha: debug
```
And restarting HA.
`home-assistant.log` is the name of the log file in the HA config directory.

I think its little strange then ZHA have finding all cluster needed so it shall also "giving them" to HA.

TPMunster, 2021-08-03:
Well, it's great to learn something every day, @MattWestb :) 

I have changed the logger setup, switched the device on/off/on, then downloaded the logfile.
Here it is: 
[home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/6925172/home-assistant.log)

MattWestb, 2021-08-03:
I have founding one on/off command and then the device is reporting OK attribute on cluster 0x0006 (on/off) and also cluster 0x0008 (light level) 254 = max brightness.

I can also see that ZHA is setting up the device with cluster 0x0300 = color control but its not using after that :-((

But you is have 2 or more color lights so you can reading the attributes on the color cluster (from the device card manage cluster and then color cluster (0x0300)) and reading attribute color capabilities  (0x400a) and clicking get zigbee attribute, also color mode and color temperature and being interesting and comparing with the HOMA. 
Some of the attributes is only being OK if the light is on (IKEA is reporting all ways OK = not standard) and comparing with the working lights.

I think the problems is the color capabilities is not being red OK or if false but i dont have experience of this things than all my lights is working OK.

TPMunster, 2021-08-03:
Hi again @MattWestb
I have another LED strip from LEDVANCE I just added. Here is its Device Signature:
```
{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4489, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.Router: 1>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010d",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x0b05",
        "0x1000",
        "0xfc0f"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LEDVANCE",
  "model": "FLEX RGBW Z3",
  "class": "zigpy.device.Device"
}
```
And here is the log file: 
[home-assistant_adding new LED strip.txt](https://github.com/zigpy/zha-device-handlers/files/6926318/home-assistant_adding.new.LED.strip.txt)

It contains only log from when i had added the new LEDVANCE LED Strip.

Are you able to get closer to a solution?

MattWestb, 2021-08-03:
Both looks like normal RGBCT lights and have GP proxy and one manufacture cluster.

The only interesting mast being wot the HOMA is saying of requesting the color capabilities (0x400a) on the color cluster.

TPMunster, 2021-08-03:
How can I investigate that? 🤔

closed 2022-02-06 by github-actions[bot]