zigpy/zha-device-handlers #922: [Device Support Request]

snoopfab, 2021-06-15:
**Is your feature request related to a problem? Please describe.**

Hi everybody, I just bought an heimann warding device (siren) and unfortunatly its recognized as a sensor device in HA.
[Model HS2WD manufactured by Heiman](https://zigbee.blakadder.com/Heiman_HS2WD.html)

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

I saw that the only quirks is for the heimann smoke sensor and alarm device so i guess this is the reason for HA to integrate the device as a sensor device.

**What i would like to be integrated :** 

- On/Off binary button for the sounding of the siren
- On/Off binary button for the lighning (flash) of the siren
- Battery Level sensor
- Sensor tempering
- Options for type of sounding if possible
- Options for type of flashing if possible

**Zigbee signature of the device :**

{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4619, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=0, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.Router: 1>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0403",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0009",
        "0x0500",
        "0x0502"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "Heiman",
  "model": "WarningDevice",
  "class": "zigpy.device.Device"
}

**zha logs of the pairing process:**

[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
Device 0x637f (00:0d:6f:00:15:3a:3b:f7) joined the network
[0x637f:zdo] ZDO request ZDOCmd.Device_annce: [0x637F, 00:0d:6f:00:15:3a:3b:f7, 142]
[0x637f] Requesting 'Node Descriptor'
Tries remaining: 2
[0x637f] Extending timeout for 0x47 request
[0x637f] Node Descriptor: NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4619, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=0, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.Router: 1>, *user_descriptor_available=False)
[0x637f] Discovering endpoints
Tries remaining: 3
[0x637f] Discovered endpoints: [1]
[0x637f:1] Discovering endpoint information
Tries remaining: 3
[0x637f:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=1027, device_version=0, input_clusters=[0, 1, 3, 4, 9, 1280, 1282], output_clusters=[3, 25])
[0x637f:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=77 command_id=Command.Read_Attributes_rsp>
[0x637f:1] Manufacturer: Heiman
[0x637f:1] Model: WarningDevice
Checking quirks for Heiman WarningDevice (00:0d:6f:00:15:3a:3b:f7)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
[... 75 lines cut ...]
[0x0000] Delivery error for seq # 0x17, on endpoint id 0 cluster 0x0031: Couldn't enqueue send data request: API is not running
[0x0000] Querying next starting at 1
Tries remaining: 3
[0x0000] request status: Status.SUCCESS. response: Neighbors(entries=5, start_index=1, neighbor_table_list=[Neighbor(extended_pan_id=XXXXXXXX, ieee=XXXXXXXX nwk=0x018E, packed=18, permit_joining=<PermitJoins.NotAccepting: 0>, depth=1, lqi=252, *device_type=<DeviceType.EndDevice: 2>, *relationship=<Relationship.Child: 1>, *rx_on_when_idle=<RxOnWhenIdle.Off: 0>)])
[0x0000] Querying next starting at 2
Tries remaining: 3
[0x0000] request status: Status.SUCCESS. response: Neighbors(entries=5, start_index=2, neighbor_table_list=[Neighbor(extended_pan_id=XXXXXXXX, ieee=XXXXXXXX, nwk=0x9686, packed=18, permit_joining=<PermitJoins.NotAccepting: 0>, depth=1, lqi=252, *device_type=<DeviceType.EndDevice: 2>, *relationship=<Relationship.Child: 1>, *rx_on_when_idle=<RxOnWhenIdle.Off: 0>)])
[0x0000] Querying next starting at 3
Tries remaining: 3
[0x0000] request status: Status.SUCCESS. response: Neighbors(entries=5, start_index=3, neighbor_table_list=[Neighbor(extended_pan_id=XXXXXXXX, ieee=XXXXXXXX, nwk=0xE2CA, packed=18, permit_joining=<PermitJoins.NotAccepting: 0>, depth=1, lqi=186, *device_type=<DeviceType.EndDevice: 2>, *relationship=<Relationship.Child: 1>, *rx_on_when_idle=<RxOnWhenIdle.Off: 0>)])
[0x0000] Querying next starting at 4
Tries remaining: 3
[0x0000] request status: Status.SUCCESS. response: Neighbors(entries=5, start_index=4, neighbor_table_list=[Neighbor(extended_pan_id=XXXXXXXX, ieee=XXXXXXXX, nwk=0x61AF, packed=37, permit_joining=<PermitJoins.Accepting: 1>, depth=1, lqi=253, *device_type=<DeviceType.Router: 1>, *relationship=<Relationship.Sibling: 2>, *rx_on_when_idle=<RxOnWhenIdle.On: 1>)])
[0x0000] Done scanning. Total 5 neighbours
[0x61af:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=97 command_id=Command.Report_Attributes>
[0x61af:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>)]]
[0x61af:1:0x0006] Attribute report received: on_off=0

**Additional context**

I know how to code in python so if i can help writing tests or the quirck i will do it gladly,  but i'm a total noobs in zigbee network programming so i need directions from the devs.
I saw that there is a siren.py in the tuya quircks which is probably a similar device so maybe it is a good starting point ?
Any direction or help will be greatly appreciated.

Thanks again for your time.

mbo18, 2021-08-28:
I've just received a siren from Heiman, model is HS2WD-Y and seems now to be a Tuya siren.

Signature is:
`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4619, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0403",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500",
        "0x0502",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0004",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TYZB01_8scntis1",
  "model": "TS0216",
  "class": "zigpy.device.Device"
}`

oliverbock, 2021-10-20:
I can control this device using a script like:

`service: zha.warning_device_warn
data:
  mode: 1
  level: 3
  duration: 1
  ieee: 80:4b:50:ff:fe:xx:xx:xx
`

The duration works.  The level is ignored.

mbo18, 2021-11-12:
@MattWestb Could you please help me in trying to create a quirk for this device? 
I don't know if it's a good idea but I would like to start with an already existing quirk. I just don't know which one ¯\_(ツ)_/¯

This is this device:
https://www.zigbee2mqtt.io/devices/HS2WD-E.html
https://github.com/Koenkk/zigbee-herdsman-converters/blob/19bb57b7bc2f686e9bf05e28a5b1b6219e8a7d46/devices/heiman.js#L263

Thank you

MattWestb, 2021-11-12:
Start with one copy of https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/heiman/smoke.py and name it `siren.py`.
Configure local quirks in HA and putting the new quirk in the folder.

Then updating the device signature with all information and cluster functions.
Then you must do import of all new clusters and deleting the you is not needing.
If not all imports is not OK ZHA is not starting but you is getting one error in the log that showing what is wrong.

Your device type is `IAS_WARNING_DEVICE` = 0x0403

Your new cluster is:
https://github.com/zigpy/zigpy/blob/7c168c1c2ca2a338676e5956d2277af7c40971d6/zigpy/zcl/clusters/security.py#L62
https://github.com/zigpy/zigpy/blob/7c168c1c2ca2a338676e5956d2277af7c40971d6/zigpy/zcl/clusters/security.py#L346

Then you can start fixing the replace part and hope all functions is OK and if not you must finding how to fixing it.

mbo18, 2021-11-12:
Thank you so much! I've started something and I currently have a quirk that match my device! \o/

Now how to get information from the embedded battery?

MattWestb, 2021-11-12:
From device card `manage cluster` and cluster 0x0001 and attribute 0x0021 click get zigbee attribute and waking the device so it can receiving the command.
Some device is not doing it OK and you must reading the voltage instead 0x0020.

mbo18, 2021-11-12:
This siren is powered by main but has a battery inside in order to still work for some time even if unplugged. So in ZHA, "Power Source" is equal to "Mains" but I would like ZHA to create a sensor for the battery.

0x0021 is 200 and 0x0020 is 41

MattWestb, 2021-11-12:
0x0012 is send with * 2 so in real its 100%
0x0020 is send in 0.1 volts so its 4.1 V in real.

In https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0201_zemismart.py its one sensor that is saying its main powered and the quirk is changing it to battery so ZHA is showing the battery indicator.

mbo18, 2021-11-12:
No way to create a sensor based on these value without rewriting Node descriptor?
Do you know where I can find Node descriptor values?
Thank you

MattWestb, 2021-11-12:
I think you is getting one then paring the device if you have debug logging.

MattWestb, 2021-11-12:
[class _NodeDescriptorEnums:](https://github.com/zigpy/zigpy/blob/a19a2a063d5edacb46722aeb56d538badd37fcc0/zigpy/zdo/types.py#L65) and under.

MattWestb, 2021-11-12:
https://github.com/zigpy/zigpy/blob/a19a2a063d5edacb46722aeb56d538badd37fcc0/zigpy/zdo/types.py#L14-L18

mbo18, 2021-11-12:
OK I have the battery working!!! (this is not trivial at all). Thank you again :)

Now to get entities for the siren sensor like done for this quirk: https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_siren.py

mbo18, 2021-11-12:
> I think you is getting one then paring the device if you have debug logging.
> 
> Sorry wrong answer :-((

That was true :)

MattWestb, 2021-11-12:
tuya ts0601 device is bad if you have one more normal devices then they is using one own protocol (DP = tuya MQTT 2 cloud) and have very strange functions "converted" to Zigbee standard in the quirk.
You can trying it but i think its better looking of some other siren that have getting quirks if you can using the function they is have done.

mbo18, 2021-11-13:
I would like to be able to set mode, level, strobe, etc. from ZHA like it’s possible in Z2M. Maybe I need to wait for ZHA to support siren entities?

MattWestb, 2021-11-13:
ZHA have getting one alarm panel for IAS but no Siren card.
If you can steering the functions with cluster commands and attributes you can making trample switches that is sending the commands in HA GUI.

If you good in wiring code you can also making one device card that have all your nice functions (i cant ;-(( ).

Hedda, 2021-12-06:
FYI, support for siren entities in ZHA integration component has now been added to the Home Assistant core:

https://github.com/home-assistant/core/pull/60920

(that will maybe not make it into HA 2021.12 release? https://rc.home-assistant.io/blog/2021/12/03/release-202112/ )

Are you also looking at adding device support for this new siren entity platform for Zigbee sirens (which also includes doorbells and chimes) to the ZHA integration component in Home Assistant core? 

As per the feature requested here:

https://community.home-assistant.io/t/zha-integration-to-use-siren-entity-platform-for-zigbee-sirens-doorbells-and-chimes/343075

_Request ZHA integration 1 use recently new siren entity platform introduced in Home Assistant 2021.8_

_" A siren entity is a device whose main purpose is to control siren devices like a doorbell or chime."_

[https://www.home-assistant.io/blog/2021/08/04/release-20218/#siren ](https://www.home-assistant.io/blog/2021/08/04/release-20218/#siren)

[https://www.home-assistant.io/integrations/siren/ ](https://www.home-assistant.io/integrations/siren/)

[https://developers.home-assistant.io/docs/core/entity/siren/ ](https://developers.home-assistant.io/docs/core/entity/siren/)

_So far Z-Wave JS, DeCONZ, and Tuya integrations (Devolco is WIP) has added siren entity platform:_

https://github.com/home-assistant/architecture/issues/375

https://github.com/home-assistant/core/pull/52948

https://github.com/home-assistant/core/pull/56397

https://github.com/home-assistant/core/pull/57780

https://github.com/home-assistant/core/pull/53400

mbo18, 2021-12-06:
It is in 2021.12. I’m currently running the beta and my Heiman siren is now recognized as a siren \o/

bsfaxi, 2021-12-17:
> It is in 2021.12. I’m currently running the beta and my Heiman siren is now recognized as a siren \o/

Hi mbo18,
I have acquired the same siren, and it's recognized by zha as you mentioned. But, I don't see the other info like battery, etc.
Could you please share your quirk code so I can test it too?
Thanks.

closed 2022-06-22 by github-actions[bot]

BabaIsYou, 2022-10-14:
> The cluster 0x0502 is having little different attribute then Zigbee standard i was finding them in some tuya dev docks.
> 
> 0x0502 IasWd Cluter
> 
>     * 0xF000 (int8)
>       
>       * 0 Disarm the alarm
>       * 1 Sound alarm
>       * 2 Light alarm
>       * 3 Sound and light alarm

@MattWestb what do you mean about this ? is 0xF000 a new command or a new attribute ?

MattWestb, 2022-10-14:
Its tuya version attributes that need being converted to Zigbee standard or ZHA need adding them as custom attributes for getting the right function in (Z)HA.

The standard Zigbee cluster you can see its not easy matching:
https://github.com/zigpy/zigpy/blob/16ec072eb5e1e94dac33e5d408f939cf6746d652/zigpy/zcl/clusters/security.py#L326

BabaIsYou, 2022-10-14:
Just to be sure. The only attribute in 0x502 cluster is 0x0000 max_duration. Do you mean that this item has another one that is  0xF000 ?