zigpy/zha-device-handlers #928: [BUG] IKEA Tradfri E1810 does not send zha events

fluffy88, 2021-06-19:
**Describe the bug**
The IKEA Tradfri E1810 five button remote does not send any events to HA after pairing.

I am using an XBee XStick V2 as my Zigbee coordinator.

I have two of the E1810 bought this week and neither are working. However the E1812 shortcut button I bought at the same time is working perfectly.

I noticed the Quirk being used for my 5 button remote is _zhaquirks.ikea.fivebtnremote.IkeaTradfriRemote_, should this be 
_zhaquirks.ikea.fivebtnremotezha.IkeaTradfriRemote_ instead?

Any help or pointers on how to further debug this would be greatly apprieciated! 

Thanks for all your great work!

**To Reproduce**
Steps to reproduce the behavior:
1. Go to ZHA integration and add new device
2. Put remote into pairing mode by pressing the pair button 4 times within 5 seconds
3. Device gets to "Interview complete, configuring" but never completes.

**Expected behavior**
ZHA should show a successfully added green box allowing the device to be renamed.

**Screenshots**
[image]

**Additional context**
The device will show up under the HA Devices tab even though the pairing doesn't ever complete, however when using Developer Tools to list to all events pressing buttons on the remote does not show any events being triggered.

Here is the device info as shown by the device card in HA.
>**Device info**
>TRADFRI remote control by IKEA of Sweden 
>Zigbee Coordinator 
>Firmware: 0x12223573 

>**Zigbee info**
>IEEE: 00:0d:6f:ff:fe:33:c5:89
>Nwk: 0xdf88
>Device Type: EndDevice
>LQI: Unknown 
>RSSI: Unknown 
>Last Seen: 2021-06-19T21:14:30 
>Power Source: Battery or Unknown 
>Quirk: zhaquirks.ikea.fivebtnremote.IkeaTradfriRemote 

Here are the debug logs during the pairing.
```
2021-06-19 21:01:10 WARNING (MainThread) [zigpy_deconz.api] No response to 'Command.device_state' command with seq id '0x02'
2021-06-19 21:01:13 WARNING (MainThread) [zigpy_zigate.api] No response to command 0x0002
2021-06-19 21:01:13 WARNING (MainThread) [zigpy_zigate.api] Retry command 0x0002
2021-06-19 21:01:16 WARNING (MainThread) [zigpy_xbee.api] at: No response to AP command
2021-06-19 21:01:17 WARNING (MainThread) [zigpy_cc.api] No response for: SRSP SYS version {}
2021-06-19 21:03:41 DEBUG (MainThread) [zigpy_xbee.zigbee.application] Broadcast request seq 1
2021-06-19 21:03:41 DEBUG (MainThread) [zigpy_xbee.api] Command tx_explicit (00:00:00:00:00:00:ff:fc, <BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>, 0, 0, 54, 0, 0, 0, b'\x01<\x00')
2021-06-19 21:03:41 DEBUG (MainThread) [zigpy_xbee.uart] Sending: b'\x11\x1e\x00\x00\x00\x00\x00\x00\xff\xfc\xff\xfc\x00\x00\x006\x00\x00\x00\x00\x01<\x00'
2021-06-19 21:03:42 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x8b\x1e\xff\xfe\x00\x00\x00'
2021-06-19 21:03:42 DEBUG (MainThread) [zigpy_xbee.api] Frame received: tx_status
2021-06-19 21:03:42 DEBUG (MainThread) [zigpy_xbee.api] tx_explicit to 0xfffe: TXStatus.SUCCESS after 0 tries. Discovery Status: DiscoveryStatus.SUCCESS, Frame #30
2021-06-19 21:03:42 DEBUG (MainThread) [zigpy_xbee.api] at command: NJ (60,)
2021-06-19 21:03:42 DEBUG (MainThread) [zigpy_xbee.api] Command at (b'NJ', b'<')
2021-06-19 21:03:42 DEBUG (MainThread) [zigpy_xbee.uart] Sending: b'\x08\x1fNJ<'
2021-06-19 21:03:42 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x88\x1fNJ\x00'
2021-06-19 21:03:42 DEBUG (MainThread) [zigpy_xbee.api] Frame received: at_response
2021-06-19 21:03:42 DEBUG (MainThread) [zigpy_xbee.api] at command: AC ()
2021-06-19 21:03:42 DEBUG (MainThread) [zigpy_xbee.api] Command at (b'AC', b'')
2021-06-19 21:03:42 DEBUG (MainThread) [zigpy_xbee.uart] Sending: b'\x08 AC'
2021-06-19 21:03:42 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x88 AC\x00'
2021-06-19 21:03:42 DEBUG (MainThread) [zigpy_xbee.api] Frame received: at_response
2021-06-19 21:03:42 DEBUG (MainThread) [zigpy_xbee.api] at command: CB (2,)
2021-06-19 21:03:42 DEBUG (MainThread) [zigpy_xbee.api] Command at (b'CB', b'\x02')
[... 289 lines cut ...]
2021-06-19 21:14:27 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x91\x00\ro\xff\xfe3\xc5\x89\xdf\x88\x00\x00\x00\x06\x00\x00B\x01\xfd\xff\x04\x01\x01\x19\x00\x00'
2021-06-19 21:14:27 DEBUG (MainThread) [zigpy_xbee.api] Frame received: explicit_rx_indicator
2021-06-19 21:14:27 DEBUG (MainThread) [zigpy_xbee.api] _handle_explicit_rx: (00:0d:6f:ff:fe:33:c5:89, 0xdf88, 0, 6, 66, b'01fdff040101190000')
2021-06-19 21:14:27 DEBUG (MainThread) [zigpy.zdo] [0xdf88:zdo] ZDO request ZDOCmd.Match_Desc_req: [0xFFFD, 260, [25], []]
2021-06-19 21:14:27 DEBUG (MainThread) [zigpy_xbee.zigbee.application] Zigbee request tsn #1: b'010000000101'
2021-06-19 21:14:27 DEBUG (MainThread) [zigpy_xbee.api] Command tx_explicit (00:0d:6f:ff:fe:33:c5:89, 0xDF88, 0, 0, <ZDOCmd.Match_Desc_rsp: 0x8006>, 0, 0, 0, b'\x01\x00\x00\x00\x01\x01')
2021-06-19 21:14:27 DEBUG (MainThread) [zigpy_xbee.uart] Sending: b'\x11?\x00\ro\xff\xfe3\xc5\x89\xdf\x88\x00\x00\x80\x06\x00\x00\x00\x00\x01\x00\x00\x00\x01\x01'
2021-06-19 21:14:28 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x8b?\xdf\x88\x00\x00@'
2021-06-19 21:14:28 DEBUG (MainThread) [zigpy_xbee.api] Frame received: tx_status
2021-06-19 21:14:28 DEBUG (MainThread) [zigpy_xbee.api] tx_explicit to 0xdf88: TXStatus.SUCCESS after 0 tries. Discovery Status: DiscoveryStatus.EXTENDED_TIMEOUT, Frame #63
2021-06-19 21:14:29 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x91\x00\ro\xff\xfe3\xc5\x89\xdf\x88\x01\x01\x00\x19\x01\x04A\x01\x03\x01\x01|\x11\xc1\x11s5"\x12\x01\x00'
2021-06-19 21:14:29 DEBUG (MainThread) [zigpy_xbee.api] Frame received: explicit_rx_indicator
2021-06-19 21:14:29 DEBUG (MainThread) [zigpy_xbee.api] _handle_explicit_rx: (00:0d:6f:ff:fe:33:c5:89, 0xdf88, 1, 25, 65, b'010301017c11c111733522120100')
2021-06-19 21:14:29 DEBUG (MainThread) [zigpy.zcl] [0xdf88:1:0x0019] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=3 command_id=1>
2021-06-19 21:14:29 DEBUG (MainThread) [zigpy.zcl] [0xdf88:1:0x0019] ZCL request 0x0001: [1, 4476, 4545, 304231795, 1]
2021-06-19 21:14:29 DEBUG (MainThread) [zigpy.zcl] [0xdf88:1:0x0019] OTA query_next_image handler for 'IKEA of Sweden TRADFRI remote control': field_control=1, manufacture_id=4476, image_type=4545, current_file_version=304231795, hardware_version=1
2021-06-19 21:14:29 DEBUG (MainThread) [zigpy.zcl] [0xdf88:1:0x0019] No OTA image is available
2021-06-19 21:14:29 DEBUG (MainThread) [zigpy_xbee.zigbee.application] Zigbee request tsn #3: b'19030298'
2021-06-19 21:14:29 DEBUG (MainThread) [zigpy_xbee.api] Command tx_explicit (00:0d:6f:ff:fe:33:c5:89, 0xDF88, 1, 1, 25, 260, 0, 0, b'\x19\x03\x02\x98')
2021-06-19 21:14:29 DEBUG (MainThread) [zigpy_xbee.uart] Sending: b'\x11@\x00\ro\xff\xfe3\xc5\x89\xdf\x88\x01\x01\x00\x19\x01\x04\x00\x00\x19\x03\x02\x98'
2021-06-19 21:14:30 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x8b@\xdf\x88\x00\x00@'
2021-06-19 21:14:30 DEBUG (MainThread) [zigpy_xbee.api] Frame received: tx_status
2021-06-19 21:14:30 DEBUG (MainThread) [zigpy_xbee.api] tx_explicit to 0xdf88: TXStatus.SUCCESS after 0 tries. Discovery Status: DiscoveryStatus.EXTENDED_TIMEOUT, Frame #64
2021-06-19 21:15:52 WARNING (MainThread) [homeassistant.components.zha.core.channels.base] [0xDF88:1:0x1000]: 'async_configure' stage failed: [0x0000:230:0x0006]: Message send failure
```

MattWestb, 2021-06-19:
I dont knowing how xbee is working but your E1810 is having on very old firmware 0x12223573 (= ZLL) and i dont knowing if the old version is working OK with the current quirks.

Eases is activating OTA updates for IKEA devices but for the moment not the test feed then its problems with the remotes for the moment.
https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates

fluffy88, 2021-06-19:
Hey Matt, thanks for the quick reply, I am going to test a firmware upgrade using the HA OTA feature you linked to. I do not have an IKEA gateway hub to use so hopefully the HA OTA works.

Thanks for the help.

fluffy88, 2021-06-20:
I ran the OTA update of the first remote last night using the OTA URL for the test feed which is the URL provided in the OTA guide you linked to and the button is now working properly, wohoo!!! Thanks for the suggestion @MattWestb.

I am now doing an update to the second remote using the non-test feed, http://fw.ota.homesmart.ikea.net/feed/version_info.json
Hopefully this one goes well too.

MattWestb, 2021-06-20:
Better not doing all with the test version then its not tested / is having bugs and the normal is very stable and then IKEA is putting the remote in the stable then do the last with it (very likely is some weeks).
The test feed is only good for some lights that is not working in the normal feed.

Great that you have getting the remote working in ZHA !!

I was baying one 10 W drive in the week (recycled = half price) and its was on one old ZLL (1.x) firmware but ZHA was updating it in some minutes and its working great for driving my LED stripes behind my TV and its following my Yamaha receiver is power on or off :-))

fluffy88, 2021-06-20:
Should I update the (OTA-Device-Firmware-Updates)[https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates] wiki page to remove the test feed and recommend the standard feed?

fluffy88, 2021-06-20:
Actually, nevermind, I just took a look at the [code](https://github.com/zigpy/zigpy/blob/350e626c1b57c91a4e831b1cb1cde412d5394855/zigpy/ota/provider.py#L141) and I see that the standard feed is the default URL.

MattWestb, 2021-06-20:
The test feed is normally working very well its only for the last week its being bad firmware there but its one private test feed for IKEA devs so its on own risk.

Its enough only comment the url or deleting the line and its is using the normal feed.
```
#      ikea_update_url: http://fw.test.ota.homesmart.ikea.net/feed/version_info.json
```
The recorded shall always being the standard for production systems and normal users and only using the then then need it.

fluffy88, 2021-06-20:
I updated the 2nd remote using the standard firmware feed but the remote was still not working using this firmware. For reference the remote reported the following details when querying the basic cluster as described in the OTA Device FIrmware Updates wiki page.
| Attribute     | Value    |
| ------------- | -------- |
| date_code     | 20190708 |
| app_version   | 33       |
| stack_version | 98       |
| sw_build_id   | 2.3.014  |

I then updated the firmware again but this time using the test feed and now both remotes are working correctly. The firmware details are as follows:
| Attribute     | Value    |
| ------------- | -------- |
| date_code     | 20210526 |
| app_version   | 33       |
| stack_version | 98       |
| sw_build_id   | 2.3.075  |

To summarise, updating the remotes firmware using the test feed worked, the remote is fully working on the latest test firmware.

Again, thank you for the help @MattWestb

MattWestb, 2021-06-20:
The "normal" firmware is normally needing doing one "reconfigure" from the device card after being bonded for getting the reporting all things to the coordinator.

Wot is not working with the 2.3.075 is group binding (sending direct commands to group of lights) but its very likely being fixed then its being released as stable.

fluffy88, 2021-06-20:
Ah, I did not try a reconfigure after the update to 2.3.014.

I don't plan to use group binding, since 2.3.075 is generating zha_events for HA to use for automations I am happy to stay on this version as the remote is working for my usage.

closed 2021-06-20 by fluffy88