zigpy/zha-device-handlers #929: [Device Support Request] Aqara Smart Knob H1 - lumi.remote.rkba01 - ZNXNKG02LM

matt2468, 2021-06-20:
I'm using a Sonoff ZHABridge flashed with Tasmota as my Zigbee Hub

**Describe the solution you'd like**
Support for: Battery, Dimmer, Press, Double Press, Press and Hold, Press and turn.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
`Device 0x3006 (54:ef:44:10:00:06:e9:eb) joined the network
Skip initialization for existing device 54:ef:44:10:00:06:e9:eb
[0x3006:zdo] ZDO request ZDOCmd.Device_annce: [0x3006, 54:ef:44:10:00:06:e9:eb, 132]
[0x3006:zdo] ZDO request ZDOCmd.Node_Desc_req: [0x0000]
[0x3006:zdo] Unsupported ZDO request:ZDOCmd.Node_Desc_req
[0x3006] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=132, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=True, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)
[0x3006] Discovering endpoints
Tries remaining: 3
[0x3006] Extending timeout for 0x07 request
[0x3006] Discovered endpoints: [1]
[0x3006:1] Discovering endpoint information
Tries remaining: 3
[0x3006] Extending timeout for 0x09 request
[0x3006:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=259, device_version=1, input_clusters=[0, 3, 1], output_clusters=[3, 25, 6])
[0x3006] Extending timeout for 0x0b request
[0x3006:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=11 command_id=Command.Read_Attributes_rsp>
[0x3006:1] Manufacturer: LUMI
[0x3006:1] Model: lumi.remote.rkba01
Checking quirks for LUMI lumi.remote.rkba01 (54:ef:44:10:00:06:e9:eb)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
device - 0x3006:54:ef:44:10:00:06:e9:eb entering async_device_initialized - is_new_join: True
device - 0x3006:54:ef:44:10:00:06:e9:eb has joined the ZHA zigbee network
[0x3006](lumi.remote.rkba01): started configuration
[0x3006:ZDO](lumi.remote.rkba01): 'async_configure' stage succeeded
[0x3006:1:0x0000]: finished channel configuration
[0x3006] Extending timeout for 0x0d request
[0x3006:1:0x0019]: finished channel configuration
[0x3006:1:0x0006]: bound 'on_off' cluster: Status.SUCCESS
[0x3006:1:0x0006]: finished channel configuration
[0x3006:1:0x0000]: 'async_configure' stage succeeded
[0x3006:1:0x0006]: 'async_configure' stage succeeded
[0x3006:1:0x0019]: 'async_configure' stage succeeded
[0x3006](lumi.remote.rkba01): completed configuration
[0x3006](lumi.remote.rkba01): stored in registry: ZhaDeviceEntry(name='LUMI lumi.remote.rkba01', ieee='54:ef:44:10:00:06:e9:eb', last_seen=1624184552.845776)
[0x3006] Extending timeout for 0x0f request
[0x3006:1:0x0003] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=15 command_id=Command.Default_Response>
[0x3006:1:0x0003]: executed 'trigger_effect' command with args: '(2, 0)' kwargs: '{}' result: [64, <Status.SUCCESS: 0>]
[0x3006](lumi.remote.rkba01): started initialization
[0x3006:ZDO](lumi.remote.rkba01): 'async_initialize' stage succeeded
[0x3006:1:0x0000]: initializing channel: from_cache: False
[0x3006:1:0x0000]: finished channel configuration
[0x3006:1:0x0006]: initializing channel: from_cache: False
[0x3006:1:0x0006]: finished channel configuration
[0x3006:1:0x0019]: initializing channel: from_cache: False
[0x3006:1:0x0019]: finished channel configuration
[0x3006:1:0x0000]: 'async_initialize' stage succeeded
[0x3006:1:0x0006]: 'async_initialize' stage succeeded
[0x3006:1:0x0019]: 'async_initialize' stage succeeded
[0x3006](lumi.remote.rkba01): power source: Mains
[0x3006](lumi.remote.rkba01): completed initialization`

**Additional context**
Pairing
To pair the device press and hold the middle button for about 10 seconds, continue to click the button through the pairing process.

Battery is not reported - Device power source is listed as Mains which is incorrect

Single Press, Dimming (both clockwise and counterclockwise) and Press and turn do not create zha_events - however they do create logs in the Sonoff Console logs

**Press and Hold**
`{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "54:ef:44:10:00:06:e9:eb",
        "unique_id": "54:ef:44:10:00:06:e9:eb:1:0x0006",
        "device_id": "90d78984b0f5a92616631c518d31d017",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "off",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2021-06-20T10:05:38.843651+00:00",
    "context": {
        "id": "0b9ed2ce803265a4a6de9256ee985476",
        "parent_id": null,
        "user_id": null
    }
}`

**Double Press**
`{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "54:ef:44:10:00:06:e9:eb",
        "unique_id": "54:ef:44:10:00:06:e9:eb:1:0x0006",
        "device_id": "90d78984b0f5a92616631c518d31d017",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "toggle",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2021-06-20T10:05:43.966304+00:00",
    "context": {
        "id": "c04256ef48be975713ba2fa08f722d78",
        "parent_id": null,
        "user_id": null
    }
}`

matt2468, 2021-10-16:
Looks like this has been integrated into zigbee2mqtt

`    {
        zigbeeModel: ['lumi.remote.rkba01'],
        model: 'ZNXNKG02LM',
        vendor: 'Xiaomi',
        description: 'Aqara knob H1 (wireless)',
        meta: {battery: {voltageToPercentage: '3V_2100'}},
        exposes: [e.battery(), e.battery_voltage(),
            e.action(['single', 'double', 'hold', 'release', 'start_rotating', 'rotation', 'stop_rotating']),
            exposes.enum('operation_mode', ea.ALL, ['event', 'command']).withDescription('Button mode'),
            exposes.numeric('action_rotation_angle', ea.STATE).withUnit('*').withDescription('Rotation angle'),
            exposes.numeric('action_rotation_angle_speed', ea.STATE).withUnit('*').withDescription('Rotation angle speed'),
            exposes.numeric('action_rotation_percent', ea.STATE).withUnit('%').withDescription('Rotation percent'),
            exposes.numeric('action_rotation_percent_speed', ea.STATE).withUnit('%').withDescription('Rotation percent speed'),
            exposes.numeric('action_rotation_time', ea.STATE).withUnit('ms').withDescription('Rotation time'),
        ],
        fromZigbee: [fz.xiaomi_on_off_action, fz.xiaomi_multistate_action, fz.xiaomi_battery, fz.aqara_opple_report,
            fz.aqara_knob_rotation],
        toZigbee: [tz.aqara_opple_operation_mode],
        onEvent: preventReset,
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint1 = device.getEndpoint(1);
            await endpoint1.write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f, disableResponse: true});
        },
    },`

https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/xiaomi.js

closed 2022-04-21 by github-actions[bot]

oneiropagides, 2023-03-12:
Can someone say if this is working with ZHA now?

oxc, 2023-03-17:
No, it doesn't, that's why I have opened #2266. I have a custom quirk nearly working, though. I recommend you subscribe the other issue for progress.