zigpy/zha-device-handlers #930: [Device Support Request] PTVO.info firmware quirk development

glcos, 2021-06-20:
I'm attemping to develop a quirk for a device based on PTVO firmware. This device has a DHT22 sensor attached and generates temperatore and humidity signals.
This is what I see in the logs when the device is paired (some not relevant lines removed):

```
2021-06-20 13:34:52 INFO (MainThread) [zigpy.device] [0x16c7] Discovered endpoints: [2, 1]
2021-06-20 13:34:53 INFO (MainThread) [zigpy.endpoint] [0x16c7:2] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=2, profile=260, device_type=256, device_version=0, input_clusters=[12], output_clusters=[])
2021-06-20 13:34:54 INFO (MainThread) [zigpy.endpoint] [0x16c7:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=256, device_version=0, input_clusters=[0, 2821, 6], output_clusters=[0, 6])
2021-06-20 13:34:54 DEBUG (MainThread) [zigpy.endpoint] [0x16c7:1] Manufacturer: ptvo.info
2021-06-20 13:34:54 DEBUG (MainThread) [zigpy.endpoint] [0x16c7:1] Model: ptvo.switch
2021-06-20 13:34:54 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for ptvo.info ptvo.switch (00:12:4b:00:21:6a:ce:fd)
2021-06-20 13:34:54 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'ptvo.ptvo.ptvoTemperature'>
2021-06-20 13:34:54 DEBUG (MainThread) [zigpy.quirks.registry] Found custom device replacement for 00:12:4b:00:21:6a:ce:fd: <class 'ptvo.ptvo.ptvoTemperature'>
```

The device sends both temperature and humidity on endpoint 2:

```
2021-06-17 21:41:48 DEBUG (MainThread) [zigpy.zcl] [0x3b49:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=60 command_id=Command.Report_Attributes>
2021-06-17 21:41:48 DEBUG (MainThread) [zigpy.zcl] [0x3b49:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>)]]
2021-06-17 21:41:48 DEBUG (MainThread) [zigpy.zcl] [0x3b49:1:0x0006] Attribute report received: on_off=0
2021-06-17 21:41:48 DEBUG (MainThread) [zigpy.zcl] [0x3b49:2:0x000c] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=61 command_id=Command.Report_Attributes>
2021-06-17 21:41:48 DEBUG (MainThread) [zigpy.zcl] [0x3b49:2:0x000c] ZCL request 0x000a: [[Attribute(attrid=85, value=<TypeValue type=Single, value=27.30000114440918>), Attribute(attrid=111, value=<TypeValue type=bitmap8, value=bitmap8.128|32|16|4|2|1>), Attribute(attrid=28, value=<TypeValue type=CharacterString, value=C>)]]
2021-06-17 21:41:48 DEBUG (MainThread) [zigpy.zcl] [0x3b49:2:0x000c] Attribute report received: present_value=27.30000114440918, status_flags=183, description=C
2021-06-17 21:41:48 DEBUG (MainThread) [zigpy.zcl] [0x3b49:2:0x000c] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=62 command_id=Command.Report_Attributes>
2021-06-17 21:41:48 DEBUG (MainThread) [zigpy.zcl] [0x3b49:2:0x000c] ZCL request 0x000a: [[Attribute(attrid=85, value=<TypeValue type=Single, value=53.400001525878906>), Attribute(attrid=111, value=<TypeValue type=bitmap8, value=bitmap8.128|32|8|4|2>), Attribute(attrid=28, value=<TypeValue type=CharacterString, value=%>)]]
2021-06-17 21:41:48 DEBUG (MainThread) [zigpy.zcl] [0x3b49:2:0x000c] Attribute report received: present_value=53.400001525878906, status_flags=174, description=%
```

My test quirk is here:

[https://github.com/glcos/PTVOquirk/blob/main/ptvo.py](url)

I see the new entities in HA but the value is "unknown"
If I have a starting point then I will further develop others quirks for this interesting platform.

Any help appreciated

glcos, 2021-06-20:
I eventually made it myself

https://github.com/glcos/PTVOquirk/blob/main/ptvo.py

MattWestb, 2021-08-03:
Great @glcos !!

Can you also including one guide in your git for user adding the quirk in local quirk folder or one link to one working guide ? 
 
One working guide https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274.

Thanks for contributing to the community !!

glcos, 2021-08-03:
PTVO.info is a configurable firmware for DIY enthusiasts and can support many sensors with countless configurations.
That said I can try to document what I have made, a small meteo station with Temperature Humidity and Rain detection using a CC2530+CC2591 module.
I will update my git with some more documentation as soon as I can, and post a notice here.

MattWestb, 2021-08-03:
Sounds nice !!

I was only thinking if user is installing your firmware they need also installing the quirk in ZHA if not having it in the HA release and you can also doing more version of the quirk with little different functions without making new PR with quirks for ZHA / HA and the user can easy making it working very easy. 
But best is that you is getting one quirk merged in ZHA so it working out of the box.

One link for how installing local quirks in ZHA can being enough and much better then nothing and the user is getting problem with your firmware.

glcos, 2021-08-03:
Just wanted to clarify that PTVO firmware is not mine, in the sense that I'm not the developer. I just used it in a couple of hobby projects. The firmware is developed and maintained by a russian guy. Unfortunately the documentation is not that detailed and the support group is in russian language only witch I do not understand.
However it's a great and versatile firmware for hobby DIY projects and it's proved stable and reliable in my tests.

closed 2022-02-06 by github-actions[bot]