zigpy/zha-device-handlers #933: [Device Support Request] Parkside Smart Watering Timer

Fenix999, 2021-06-22:
I would like to add this Parkside Smart Watering Timer

The actual bottom in home assistant works but I don't have any feedback if I press it on the device, also it seems to have a timer function that isn't available.

Device 0xeec8 (84:71:27:ff:fe:d3:3b:e3) joined the network
Device 84:71:27:ff:fe:d3:3b:e3 changed id (0x0b07 => 0xeec8)
[0xeec8:zdo] ZDO request ZDOCmd.Device_annce: [0xEEC8, 84:71:27:ff:fe:d3:3b:e3, 128]
[0xeec8] Extending timeout for 0x72 request
Checking quirks for _TZE200_htnnfasr TS0601 (84:71:27:ff:fe:d3:3b:e3)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
device - 61128:84:71:27:ff:fe:d3:3b:e3 entering async_device_initialized - is_new_join: True
device - 61128:84:71:27:ff:fe:d3:3b:e3 has joined the ZHA zigbee network
[61128](TS0601): started configuration
[61128:ZDO](TS0601): 'async_configure' stage succeeded
[0xeec8] Extending timeout for 0x74 request
[61128:1:0x0000]: finished channel configuration
[61128:1:0x0019]: finished channel configuration
[61128:1:0x0006]: bound 'on_off' cluster: Status.SUCCESS
[0xeec8] Extending timeout for 0x76 request
[0xeec8:1:0x0004] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=114 command_id=2>
Sending 'zdo_leave_req' failed: 
[61128:1:0x0006]: failed to set reporting for 'on_off' attr on 'on_off' cluster: 
[61128:1:0x0006]: finished channel configuration
[61128:1:0x0006]: 'async_configure' stage succeeded
[61128:1:0x0000]: 'async_configure' stage succeeded
[61128:1:0x0019]: 'async_configure' stage succeeded
[61128](TS0601): completed configuration
[61128](TS0601): stored in registry: ZhaDeviceEntry(name='_TZE200_htnnfasr TS0601', ieee='84:71:27:ff:fe:d3:3b:e3', last_seen=1624376423.8630354)
[0xeec8] Extending timeout for 0x78 request
[61128:1:0x0003]: command failed: 'trigger_effect' args: '(2, 0)' kwargs '{}' exception: ''
[61128](TS0601): started initialization
[61128:ZDO](TS0601): 'async_initialize' stage succeeded
[61128:1:0x0006]: initializing channel: from_cache: False
[0xeec8] Extending timeout for 0x7a request
[61128:1:0x0000]: initializing channel: from_cache: False
[61128:1:0x0000]: finished channel configuration
[61128:1:0x0019]: initializing channel: from_cache: False
[61128:1:0x0019]: finished channel configuration
[0x67e7:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=True> manufacturer=4447 tsn=221 command_id=Command.Report_Attributes>
[0x67e7:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=65281, value=<TypeValue type=CharacterString, value=!�(!�C!0>)]]
[0x67e7:1:0x0000] Attribute report received: 65281=!�(!�C!0
[0x67e7:1:0x0001] Voltage mV: [Min]:2820 < [RAW]:3055 < [Max]:3100, Battery Percent: 84.0
[61128:1:0x0006]: failed to get attributes '['on_off']' on 'on_off' cluster: 
[61128:1:0x0006]: finished channel configuration
[61128:1:0x0006]: 'async_initialize' stage succeeded
[61128:1:0x0000]: 'async_initialize' stage succeeded
[61128:1:0x0019]: 'async_initialize' stage succeeded
[61128](TS0601): power source: Battery or Unknown
[61128](TS0601): completed initialization
[61128:1:0x0006]: attempting to update onoff state - from cache: True

Efforts to try to implement this device are been made here:

https://github.com/Koenkk/zigbee2mqtt/issues/7695#

and here:

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4958

MattWestb, 2021-06-22:
If some one that understanding tuya zigbee magic one user have sniffing the setup and sending commands and getting the feedback in the Z2M thread.

Its looks it can being controlled with normal "on" command but is not reporting back normal zigbee attributes and its have one safety timer the using on commands so need sending new commands in interval for watering longer time.

Interesting product but my balcony is little to small for using it ;-))

pipiche38, 2021-06-22:
@MattWestb , usally the Tuya gateway do a kind of registration and write in some attribute at pairing phase. Do you do that ?

Fenix999, 2021-06-22:
I already have two hopping that someone smarter than me make it work, I will keep following the other threads to see if there is any development.

It's good to have that safety timer, it was one of my main concerns about the product since water is scarce and expensive.

Idea if our balcony doesn't have a tap you probably can use container with water and let gravity do it's thing.

MattWestb, 2021-06-22:
@pipiche38 The user have capture the paring and very early the ZBGW was writing one tuya attribute to the device that  i have posing in the Z2M thread and all after is tuya commands with normal zigbee default response.

MattWestb, 2021-06-22:
@Fenix999 I dont have any tap (but need it then the last level without sealing) but having french dropsystem then i going to Spain for holly days and its working OK only needing one good pump for refilling from the lower storage that i dont have finding.
And on very hot days all is dead if not watering extra in the afternoon :-((

MattWestb, 2021-07-16:
@Smanar some user  is getting DP from the device in Z2M https://github.com/Koenkk/zigbee-herdsman-converters/issues/2795#issuecomment-880947132.

Smanar, 2021-07-16:
```
Zigbee2MQTT:debug 2021-07-15 21:06:45: Received Zigbee message from '0xb4e3f9fffe0c6b70', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,60],"type":"Buffer"},"datatype":2,"dp":15,"fn":0,"status":0,"transid":2}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2021-07-15 21:06:45: zigbee-herdsman-converters:RTXZVG1Valve: NOT RECOGNIZED DP #15 with data {"status":0,"transid":2,"dp":15,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,60]}}
Zigbee2MQTT:debug 2021-07-15 21:06:45: Received Zigbee message from '0xb4e3f9fffe0c6b70', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":12,"fn":0,"status":0,"transid":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2021-07-15 21:06:45: zigbee-herdsman-converters:RTXZVG1Valve: NOT RECOGNIZED DP #12 with data {"status":0,"transid":3,"dp":12,"datatype":4,"fn":0,"data":{"type":"Buffer","data":[2]}}
Zigbee2MQTT:debug 2021-07-15 21:09:27: Received Zigbee message from '0xb4e3f9fffe0c6b70', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":7,"fn":0,"status":0,"transid":4}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2021-07-15 21:09:27: zigbee-herdsman-converters:RTXZVG1Valve: NOT RECOGNIZED DP #7 with data 
```

But we already have the on / off / timer / battery, what can be missing ?
And BTW reporting is now full working ?

MattWestb, 2021-07-16:
I dont knowing i was only see that they was getting more DP reports that is good but i dont knowing  them.

You still need very urgent one tuya  ZBGW and one bunch of tuya devices and also little python coding ;-)))

jvalst, 2021-07-20:
Should be nice if this one got embedded well. Now the switch is turning it on, and the auto power off is shutting it off. Same when I just click the button on the device, the auto power off kicks in after 60s. Seems like it's a pain to get this one working looking at multiple threads around deconz/z2m

jordyals, 2021-07-27:
Ok, for the ones interested in getting this thing running at least. I have it working now with an addition in Node-red (should be possible in Automations as well). Normally the default time-out of this device will be at 60 seconds, I'm giving this device an 'on' call each 50 seconds, so it keeps resetting the devices timeout. 

I'm using an irrigation page on my home-assistant with custom timing, and that one is giving the on/off state according the time I want. When the "on" is getting fired, my node-red will start an looptimer and keeps looping for each 50 seconds (with a max of x loops and a time of x minutes, so no infinite loop there if something is wrong). After he get's the off signal from my irrigation page, I stop the timer and the device goes off. 

Maybe someone can use this for the time being, hopefully in the future we can have some battery levels or also the "current state" of this device when triggering on the button itself.

closed 2022-01-30 by github-actions[bot]

MattWestb, 2022-03-26:
This device is needing tuya magic being casted on it for working OK.
Z2M have getting the DP commands working after implanting it.
https://github.com/Koenkk/zigbee2mqtt/issues/7695#issuecomment-1079660118

rforro, 2022-05-06:
Yet again I bought a Zigbee device which needs coding :D In the following weeks I'll do it.

MattWestb, 2022-05-06:
Servus @rforro so you have getting work until your summer holiday :-))
Then you start playing with the device and coding you need getting  the quirk casting "tuya magic spell" on it or its not behaving OK and you is not getting all function working OK.
Z2M have getting it working OK in the last mounts after implanting it.

rforro, 2022-05-16:
Very early version, less than alpha can be found here: https://gist.github.com/rforro/688a885bfdb09452ca31ffd795590a42

- It has a bug (tuya feature) which has to be solved: https://github.com/zigpy/zha-device-handlers/issues/1566

- You can use onoff button and see battery level directly (don't forget that the device has timer which turns it off afte XY minutes)
- if you want to change the timer duration or reset frost lock, currently you have to do it through `Manage Clusters`. (the after bug fix there will be entities in HA Frontend)

woll-knaeuel, 2024-04-13:
> Very early version, less than alpha can be found here: https://gist.github.com/rforro/688a885bfdb09452ca31ffd795590a42
> 
> * It has a bug (tuya feature) which has to be solved: [[BUG] Tuya data processing - ZCL frame multiple attributes #1566](https://github.com/zigpy/zha-device-handlers/issues/1566)
> * You can use onoff button and see battery level directly (don't forget that the device has timer which turns it off afte XY minutes)
> * if you want to change the timer duration (only set, currently cannot be read) or reset frost lock (set value 0), currently you have to do it through `Manage Clusters`. (when the bug will be fixed, I'll add entities into HA Frontend)
> * scheduler is not and will not be implemented as I don't see any reason to do so. HA is the scheduler and also there is no frontend UI to control the scheduler.
> 
> IMPORTANT After applying the quirk you have to pair the device again, because it stays enchanted (won't publish values).

Are there any plans or updates regarding adding the missing entities into HA Frontend? I have the on/off switch, battery level and remaining time but no easy way to access or set the timer. :-(