zigpy/zha-device-handlers #934: [Device Support Request] _TZE200_gubdgai2 TS0601

pex81, 2021-06-22:
**Is your feature request related to a problem? Please describe.**
I can connect it to ZHA, but it does not generate any sensors, switches or other.
It's a curtain motor.
Is it possible to integrate it?

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_gubdgai2",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

**Additional context**
Add any other context or screenshots about the feature request here.

pex81, 2021-06-24:
I was able to get the device to work.
However there are some oddities.
In the entity I can press the UP button, STOP button, but not the DOWN button.
But through the service: cover everything works.

ps: and the percentage shift does not work.

[image]

TheHolyFruitcake, 2021-06-25:
How did you get the upper and lower limits to work? The instructions I received do not work for me. How did you pair the device to home assistant?

pex81, 2021-06-25:
> How did you get the upper and lower limits to work? The instructions I received do not work for me. How did you pair the device to home assistant?

To associate it with the zigbee: 
- press the set button once and it will flash red 
- immediately afterwards hold down the set again until it flashes again and starts flashing red

TheHolyFruitcake, 2021-06-25:
This step I managed. I paired it with a the zigbee home automation integration with a zigbee dongle directly in my Raspberry pi 4. It links with the following device signature:

 "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_gubdgai2",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

But it does not give me options to go up, down or stop. I tried setting the upper and lower limit on the device itself as I thought this might be required, but I cannot get this to work. The instruction says to reset first by clicking on the set button and then the arrow up for 5 seconds. This does not always get a reset response. 
If i try to set the upper limit, I cannot bet a second light to blink. I both tried pressing the set button while the curtain is still going up, but this stops the curtain and pressing up then does not provide a response. And i tried setting it in the right position, pressing set and then up. Then the curtain just goes up. 

How did you get the lower and upper limit? Did you use the Tuya app for that?

MattWestb, 2021-06-25:
Its one tuya device that is using the tuya cluster for commands (0xef00).
Therefor the device need one quirk for "converting" tuya commands to real Zigbee commands for working.
And the quirk loaded for the device is `"class": "zigpy.device.Device"` = no quirk.

pex81, 2021-06-26:
> This step I managed. I paired it with a the zigbee home automation integration with a zigbee dongle directly in my Raspberry pi 4. It links with the following device signature:
>
> "node_descriptor": [...]

I changed __init__.py and ts0601.py to /usr/local/lib/python3.8/site-packages/zhaquirks/tuya

to line 87 of __init__.py I added:
```
     "_TZE200_gubdgai2": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
```
and to line 180 of ts0601.py I added:
```
             ("_TZE200_gubdgai2", "TS0601"),
```
i deleted 
```
rm -rf __pycache__
```
i restarted home assistant.

pex81, 2021-06-27:
what I don't know is how and if possible to carry out a calibration and how to manage the percentages.

MattWestb, 2021-07-28:
@pex81 if your device is working OK can you pleas making one PR so other user can getting there device working in ZHA  ?

If you have problem i can do it for you but i cant testing that the quirk is working OK then i dont have the device.

MattWestb, 2021-08-03:
I think some users have problem getting the calibration working from ZHA and have making on the device and getting it working (i dot have the device so i dont knowing only have reading it).

Look on the earlier commits for blinds for more information.

I have posted one PR for adding the device and i hope its better then no quirk at all for the moment so user can start using the devices in ZHA.

closed 2021-08-25 by dmulcahey

blakadder, 2021-09-05:
@pex81 can you post a link to the device or its model number?

pex81, 2021-09-06:
> @pex81 can you post a link to the device or its model number?

Mi model number is TM616EGBZTN

blakadder, 2021-09-07:
> > @pex81 can you post a link to the device or its model number?
> 
> My model number is TM616EGBZTN

any link, google has exactly 0 results on that model number

pex81, 2021-09-07:
> > > @pex81 can you post a link to the device or its model number?
> > 
> > 
> > My model number is TM616EGBZTN
> 
> any link, google has exactly 0 results on that model number

I don't have any links, this is my device.

[image]
[image]

cydia2020, 2023-01-24:
Ok I'm leaving this information out there in case anyone needs it. To set the limit on some of these device's variants, notably the M515EGBZTN by zemismart (version j.h.-v2.1-yb) (also reports as _TZE200_gubdgai2 in home assistant), you'll need to hold the up and down button simultaneously until the blue light begins flashing at roughly 2hz, you can then use the up/down/stop button to set your desired "upper" limit, once that's reached, press the stop button 5 times, then set the "lower" limit using the same method, press the stop button 5 times when you've reached your lower limit, the device then should confirm that the limits have been set with a blink of the red LED.

choportrend, 2023-07-04:
Hi,

I have this exact model. I'm very new to HA (less than a week in) and I'm hoping to integrate these motors locally into my set up. They're currently connected to a Tuya Zigbee gateway so I currently use the Tuya Cloud integration, but hoping to take this local. All the limits are set already through the app.

I tried linking one device to my zigbee stick connected to HA and it does connect, however none of the buttons on the entity work. I tried calling a service to it and nothing happens either. Under Zigbee Info it does say that this has a quirk (still learning how these work) and if I understand correctly this was deployed because of the guys here?

Is there a step I'm missing? Am I supposed to do something about the quirk, say edit something in my config file, or do these quirks make integrating these devices plug and play through ZHA? 

Would really appreciate any help on this issue, let me know if I need to provide more information

PR #1009 (2021-08-25): Adding _TZE200_gubdgai2 TS0601
Adding _TZE200_gubdgai2 TS0601 one more MOES cover / blind controller.

Fixes #934