zigpy/zha-device-handlers #935: [Device Support Request] TS0044 _TZ3000_dku2cfsc

GooG2e, 2021-06-23:
**Is your feature request related to a problem? Please describe.**
This is remote with 4 buttons but them shows as switches

**Describe the solution you'd like**
Change switches to buttons to use actions in automations

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
`{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0006"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    },
    "4": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_dku2cfsc",
  "model": "TS0044",
  "class": "zigpy.device.Device"
}`

fcamilleri, 2021-06-26:
I've got exactly the same issue.

MattWestb, 2021-07-28:
One Pr is posted to fixing the TS0044 switches but is not merged. #958

PR #958 (2021-08-25): Updating TS0042 using tuya general device name.
Changing name to general tuya device.

Fixes #935, fixes #952
closed 2021-08-25 by dmulcahey