zigpy/zha-device-handlers #937: [Device Support Request] TS0601 by _TZE200_bq5c8xfe (Temperature And Humidity Sensor)

brian-kehoe, 2021-06-24:
**Is your feature request related to a problem? Please describe.**
I can connect it to ZHA, it generates a device listing but no entities.
It is a Temperature And Humidity Sensor
Is it possible to integrate it?

[image]

**Describe the solution you'd like**
I would like to be able to see the Temperature And Humidity values

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
`{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_bq5c8xfe",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}`

**Additional context**
Version
supervisor-2021.06.3

rampachuflo, 2021-09-23:
Hello
I'm interested too
Did you solved it? thank you

brian-kehoe, 2021-10-18:
> Hello I'm interested too Did you solved it? thank you

No, there has been no update on this issue since I logged it. I have added the Debug logs to see if that will help resolve the issue.

jacekk015, 2021-12-02:
@brian-kehoe Still interested? We can try do make it supported device.

brian-kehoe, 2021-12-02:
Yes, definitely still interested in helping to get this working

jacekk015, 2021-12-02:
@MattWestb Where to we put it and how do we name it in your Matrix??
https://github.com/zigpy/zigpy/discussions/653#discussioncomment-314395

MattWestb, 2021-12-02:
The tuya TRV matrix is being very large and as discussion it only i that can ding changes.
I thinking doing one Wiki with tuya DP devices and subcategories and liking to smaller discussions for not getting them too large and not possible reading thm for user.
Also its possible changing  the link in the wiki to one new page if the old is not wanted.

MattWestb, 2021-12-02:
Ops i have making one more generell matrix for Zigbee and DP devices but its not up to date.
I can braking it in one Zigbee nad one DP for getting it working better.
https://github.com/zigpy/zigpy/wiki/tuya-Matrix

jacekk015, 2021-12-02:
@brian-kehoe 
Enable DEBUG logging in configuration.yaml
```
logger:
  default: info
  logs:
    homeassistant.components.zha: debug
    zigpy: debug
    zhaquirks: debug
```
Enable custom quirk option and put that file into that folder[create it if don't exists].
[ts0601_sensor.py.zip](https://github.com/zigpy/zha-device-handlers/files/7644731/ts0601_sensor.py.zip)
```
zha:
  custom_quirks_path: /config/zha_custom_quirks/
```

Restart HA!, remove device from HA, pair it, wait about 5 minutes and paste here ALL the logs from HA logs.
If it's big put it here as a file or use pastebin.com
Do NOT cut anything from the logs.
There will be no fireworks now, but device will present us in the logs all data it sends.

MattWestb, 2021-12-02:
Short answer for the original question:
i thing `ts0601_temperature.py`  or  `ts0601_temp.py` is OK and if its having extra humidity or light it can going in the same.

brian-kehoe, 2021-12-03:
> @brian-kehoe Enable DEBUG logging in configuration.yaml
> 
> ```
> logger:
>   default: info
>   logs:
>     homeassistant.components.zha: debug
>     zigpy: debug
>     zhaquirks: debug
> ```
> 
> Enable custom quirk option and put that file into that folder[create it if don't exists][unzip it ;-)] [ts0601_sensor.py.zip](https://github.com/zigpy/zha-device-handlers/files/7644731/ts0601_sensor.py.zip)
> 
> ```
> zha:
>   custom_quirks_path: /config/zha_custom_quirks/
> ```
> 
> Restart HA!, remove device from HA, pair it, wait about 5 minutes and paste here ALL the logs from HA logs. If it's big put it here as a file or use pastebin.com Do NOT cut anything from the logs. There will be no fireworks now, but device will present us in the logs all data it sends.
[Log file 03122021.txt](https://github.com/zigpy/zha-device-handlers/files/7647817/Log.file.03122021.txt)

I have attached the log file here

jacekk015, 2021-12-03:
@brian-kehoe Here's first dev version.
Check what's working, what's not.
After you put the file into the "zha_custom_quirks" folder restart is needed!
[ts0601_temperature.py.zip](https://github.com/zigpy/zha-device-handlers/files/7649357/ts0601_temperature.py.zip)

Your device shows battery level under attribute 0x0204, and log says it's value is 40.
Original Tuya code makes from it 80% probably.
If you could put a new pack of batteries and work a bit with that 5-10 minutes, and put here logs.
We will see which value is sent for 100% new batteries. I'm not even sure if that remaining percentage or level.

 
I've seen in Z2M and in your logs one parameter that we don't know what it is.
It's under attribute 0x0403 - and your logs shows a value [1] there.
Do you have some settings option on your device?? Maybe battery low alarm or something??
There's need to be some action that changes 0x0403 from [1] to [0] - we could code it, but we need to know what it is.
Maybe you used it with original Tuya app and know what it is?
```
2021-12-03 08:05:10 DEBUG (MainThread) [zigpy.zcl] [0x2be0:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=232, command_id=1027, function=0, data=[1, 1])]
2021-12-03 08:05:10 DEBUG (MainThread) [zhaquirks.tuya] [0x2be0:1:0xef00] Received value [1] for attribute 0x0403 (command 0x0001)
```

brian-kehoe, 2021-12-08:
When I place ts0601_temperature.py into the \config\custom_zha_quirks folder it seems to do something to the ZHA integration

Logger: homeassistant.config_entries
Source: deps/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py:665
First occurred: 14:03:18 (1 occurrences)
Last logged: 14:03:18

Error setting up entry TI CC2531 USB CDC, s/n: __0X00124B001CCC7BB0 - Texas Instruments for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 304, in async_setup
    result = await component.async_setup_entry(hass, self)  # type: ignore
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 102, in async_setup_entry
    await zha_gateway.async_initialize()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 152, in async_initialize
    self.application_controller = await app_controller_cls.new(
  File "/usr/local/lib/python3.9/site-packages/zigpy/application.py", line 61, in new
    await app._load_db()
  File "/usr/local/lib/python3.9/site-packages/zigpy/application.py", line 53, in _load_db
    await self._dblistener.load()
  File "/usr/local/lib/python3.9/site-packages/zigpy/appdb.py", line 383, in load
    device = zigpy.quirks.get_device(device)
  File "/usr/local/lib/python3.9/site-packages/zigpy/quirks/__init__.py", line 44, in get_device
    return _DEVICE_REGISTRY.get_device(device)
  File "/usr/local/lib/python3.9/site-packages/zigpy/quirks/registry.py", line 148, in get_device
    device = candidate(device._application, device.ieee, device.nwk, device)
  File "/usr/local/lib/python3.9/site-packages/zigpy/quirks/__init__.py", line 98, in __init__
    self.add_endpoint(endpoint_id, replace_device=replaces)
  File "/usr/local/lib/python3.9/site-packages/zigpy/quirks/__init__.py", line 113, in add_endpoint
    ep = custom_ep_type(self, endpoint_id, replacement_data, replace_device)
  File "/usr/local/lib/python3.9/site-packages/zigpy/quirks/__init__.py", line 137, in __init__
    cluster = c(self, is_server=True)
  File "/config/deps/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py", line 665, in __init__
    self.endpoint.device.battery_bus.add_listener(self)
AttributeError: 'TuyaSensor' object has no attribute 'battery_bus'

brian-kehoe, 2021-12-08:
Thanks, I have added two new log files
1) New .py file and old batteries
2) Remove device, install new batteries and re-add

[Log file 08122021 - Old Batteries.txt](https://github.com/zigpy/zha-device-handlers/files/7677094/Log.file.08122021.-.Old.Batteries.txt)
[Log file 08122021 - Remove Readd and New Batteries.txt](https://github.com/zigpy/zha-device-handlers/files/7677095/Log.file.08122021.-.Remove.Readd.and.New.Batteries.txt)

jacekk015, 2021-12-08:
@brian-kehoe You didn't wrote most important thing.
Does it work???
I see 19,1 degrees, 48% humidity and 100% battery level

I also see still unknown attribute 0x0403 with value [2]
```
2021-12-08 15:32:46 DEBUG (MainThread) [zhaquirks.tuya] [0x19e1:1:0xef00] Received value [2] for attribute 0x0403 (command 0x0001)
```

brian-kehoe, 2021-12-08:
Oh, I didn't know it was returning data to the logs because on the front end it is just displaying "Unknown" for the Temperature, Humidity and Power values

[image]

jacekk015, 2021-12-08:
@brian-kehoe 
Used new approach for this code, made some correction now.
But I'm surprised that battery level didn't work properly - it should with this code.
Replace file and restart HA
[ts0601_temperature.py.zip](https://github.com/zigpy/zha-device-handlers/files/7677887/ts0601_temperature.py.zip)

brian-kehoe, 2021-12-08:
Seeing some odd behaviour. I removed, restarted and re-added a few times

Attempt 1
[image]

Attempt 2
[image]

[Log file 08122021 - New File.txt](https://github.com/zigpy/zha-device-handlers/files/7678134/Log.file.08122021.-.New.File.txt)

jacekk015, 2021-12-09:
@brian-kehoe You have some strange problems in communication:
```
2021-12-08 18:33:31 DEBUG (MainThread) [zigpy.zcl] [0x6752:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=116 command_id=1>
2021-12-08 18:33:31 DEBUG (MainThread) [zigpy.zcl] [0x6752:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=91, command_id=516, function=0, data=[4, 0, 0, 0, 100])]
2021-12-08 18:33:31 DEBUG (MainThread) [zhaquirks.tuya] [0x6752:1:0xef00] Received value [0, 0, 0, 100] for attribute 0x0204 (command 0x0001)
2021-12-08 18:33:31 DEBUG (MainThread) [zigpy_znp.zigbee.application] Max concurrency reached, delaying requests (10 enqueued)
```
**Max concurrency reached, delaying requests (10 enqueued)**
It started to show when you've put new batteries.

Here corrected file - try it now:
[ts0601_temperature.py.zip](https://github.com/zigpy/zha-device-handlers/files/7684613/ts0601_temperature.py.zip)

brian-kehoe, 2021-12-09:
New file applied, restarted, removed and readded device

The device does not appear to have any entities

[image]

[Log file 09122021.txt](https://github.com/zigpy/zha-device-handlers/files/7685237/Log.file.09122021.txt)

jacekk015, 2021-12-09:
@brian-kehoe You have some Zigbee problems with communication. That device wasn't properly initialized.
You've got plenty of errors in the log.
Other devices are also dropping on communication:
```
2021-12-09 12:54:34 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7C47:1:0x0008]: failed to get attributes '['current_level']' on 'level' cluster: Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>
```

closed 2022-06-14 by github-actions[bot]